libusb/libusb #901: Which interface can I use to set the baud rate？

zj1838277795, 2021-03-29:
When I connected my USB device to Linux, I could not receive data, timeout, but I connected the USB device to Windows, opened the serial port with the serial port debugging tool, and unplugged the USB device without closing the serial port, and then connected to Linux, I could receive the data. Sending data can be done.
`int main(int argc, char *argv[])
{

    libusb_context *usb_context = nullptr;
    libusb_device **dev_list;
    libusb_device *dev = NULL;
    libusb_device *dev_temp;
    libusb_device_handle *dev_handle = NULL;
    int ret = -1;
    ssize_t cnt;
    if (ret = libusb_init(&usb_context) < 0)
    {
        printf("init error:%d\n", ret);
        return ret;
    }
    libusb_set_debug(usb_context, 1);
    if (cnt = libusb_get_device_list(usb_context, &dev_list) < 0)
    {
        printf("get device list error:%d\n", cnt);
        return cnt;
    }
    int i = 0;
    while ((dev_temp = dev_list[i++]) != NULL)
    {
        struct libusb_device_descriptor desc;
        int r = libusb_get_device_descriptor(dev_temp, &desc);
        if (r < 0)
        {
            printf("get device descriptor error:%d\n", r);
            return NULL;
        }
        printf("%x:%x (bus %d, device %d),bNumberInterfaces:%d\n", desc.idVendor, desc.idProduct,
               libusb_get_bus_number(dev_temp), libusb_get_device_address(dev_temp), desc.bNumConfigurations);
        if ((desc.idProduct == 60000) && (desc.idVendor == 4292))
        {
            dev = dev_temp;
            printf("bNumConfigurations: %d\n", desc.bNumConfigurations);
            break;
        }
    }
    if (dev == NULL)
    {
        printf("find dev error\n");
        return -1;
    }
    printf("find dev success\n");
    ret = libusb_open(dev, &dev_handle);
    if (dev_handle == NULL)
    {
        printf("open device error:%d\n", ret);
        return ret;
    }
    else
        printf("open device success:%d\n", ret);
    ret = libusb_kernel_driver_active(dev_handle, 0);
    if (ret == 1)
    {
        printf("driver is active, kernel_active ret:%d,%s,\n", ret, libusb_strerror((libusb_error)ret));
        ret = libusb_detach_kernel_driver(dev_handle, 0);
        if (ret != 0)
            printf("detach driver error, detach_driver ret:%d,%s,\n", ret, libusb_strerror((libusb_error)ret));
    }
    else if (ret == 0)
        printf("driver is not active, kernel_active ret:%d,%s,\n", ret, libusb_strerror((libusb_error)ret));
    int config = 0;
    libusb_get_configuration(dev_handle, &config);
    printf("libusb_get_configuration config:%d.\n", config);
    if (config != 1)
    {
        ret = libusb_set_configuration(dev_handle, 1);
        if (ret != 0)
            printf("libusb_set_configuration failed\n");
        else
            printf("libusb_set_configuration success\n");
    }
    if (ret = libusb_claim_interface(dev_handle, 0) < 0)
    {
        printf("claim interface error:%d\n", ret);
        libusb_free_device_list(dev_list, 1);
        libusb_exit(usb_context);
        return ret;
    }
    else if (ret == 0)
        printf("claim interface success:%d\n", ret);
    struct libusb_config_descriptor *conf = nullptr;
    ret = libusb_get_active_config_descriptor(dev, &conf);
    if (ret == 0)
    {
        printf("get active_config_descriptor success\n");
        if (conf)
        {
            printf("bConfigurationValue:%d\n", conf->bConfigurationValue);
            unsigned char data_rec[64] = "\0";
            int length = 0;
            unsigned char data[1] = {90};
            for (int i = 0; i < 20; i++)
            {
                ret = libusb_bulk_transfer(dev_handle, 0x01 /*LIBUSB_ENDPOINT_OUT*/, data, 1, &length, 0); 
                if (ret == 0)
                    printf("success\n");
                ret = libusb_bulk_transfer(dev_handle, 0x81, data_rec, 64, &length, 4000);
                if (length > 0)
                    printf("recv %d bytes, data: %s\n", length, data_rec);
                else
                    printf("ret = %d,%s\n", ret, libusb_strerror((libusb_error)ret));
                memset(data_rec, 0, 64);
            }
        }
    }
    else if (ret != 0)
    {
        printf("get config_descriptor error\n");
    }
    libusb_release_interface(dev_handle, 0); 
    libusb_close(dev_handle);
    libusb_free_device_list(dev_list, 1);
    libusb_exit(usb_context);
    return 0;
}
`

mcuee, 2021-03-31:
You need to provide more details about the device by providing the output of "lsusb -vvv".

And if your device is a USB serial port, then it is better you use the Serial API under Linux. Just wondering why you want to use libusb? Is the driver not existing under Windows? What is the Windows driver used?

Apparently you are missing some initiation codes to get the device in proper state. If you do not know the communication protocol of the device, then you have to carry out reverse engineering.

mcuee, 2021-03-31:
If this happens to be a USB CDC-ACM device, here are two examples for your reference. But you should really use the Serial API and not libusb.

https://github.com/tytouf/libusb-cdc-example/blob/master/cdc_example.c
https://github.com/NordicPlayground/node-usb-cdc-acm

mcuee, 2021-04-01:
Close for now. Please use libusb-devel for further question. 

Github is used for issue reporting and not technical support.

closed 2021-04-01 by mcuee

TheClogger666, 2023-12-20:
> Close for now. Please use libusb-devel for further question.
> 
> Github is used for issue reporting and not technical support.

You're an idiot.