libusb/libusb #902: Pyusb with new version of libusb 1.0 on HID mangling one byte of data on Windows 10

EnochMakenp, 2021-03-30:
Using the latest PyUSB, in conjunction with the latest libusb1.0 on Windows 10 (2020-12-11) with an access card reader that behave as a HID device is missing the last byte of data in the ctrl_transfer communication. I'm asking for 8 bytes of data and only 7 are read, which is annoying as I want to read the card UID, coded on 8 bytes...

The code I use in PyUSB is:
```
    device = usb.core.find(idVendor=vendorID, idProduct=productID)
    ret = device.ctrl_transfer(0x21, 0x09, 0x0300, 0, (0x8f,0x00,0x00,0x00,0x00,0x00,0x00,0x00),8) #makes the request
    response = device.ctrl_transfer(0xa1, 1, 0x0300, 0, 8) # read the response, ask for 8 bytes
```
and after this the content of response is :
array('B', [186, 137, 24, 128, 0, 0, 229])
Note that I have only 7 bytes instead of the 8 requested 
(if I ask for 7 bytes, the command fail, but if I request 9 or more bytes , I still get the 7 bytes...)

I used Wireshark (with USBPcap) to spy on the USB transaction and the 8 bytes of data are available on the USB packet, as can be see in the following transcript:
```
Frame 8: 36 bytes on wire (288 bits), 36 bytes captured (288 bits) on interface wireshark_extcap2040, id 0
USB URB
    [Source: 5.11.0]
    [Destination: host]
    USBPcap pseudoheader length: 28
    IRP ID: 0xffff960ef084e060
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CONTROL_TRANSFER (0x0008)
    IRP information: 0x01, Direction: PDO -> FDO
    URB bus id: 5
    Device address: 11
    Endpoint: 0x80, Direction: IN
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 8
    [Request in: 7]
    [Time from request: 0.000164000 seconds]
    Control transfer stage: Complete (3)
    [bInterfaceClass: Unknown (0xffff)]
    CONTROL response data: ba8918800000e5c9
```
The control data (ba8918800000e5c9) has 8 bytes and in the response above I'm missing the last byte "c9", and the collected bytes in python stop at "e5" (=229).

This Wireshark test is the reason why I suspect PyUSB or libusb1.0 messing around - with a preference for the second one, as there is a new version that is only 3 months old (note that I have made the same report on PyUSB).

Note that I can not revert to the previous version of libusb1.0 (previous to 2020-12-11) as it does not work at all anymore ending systematically with "[Errno 19] No such device (it may have been disconnected). " when I try to use ctrl_transfer.

jonasmalacofilho, 2021-03-31:
I think there might be a bug in the WinUSB backend.

Specifically, I suspect the handling of the id == 0 case in [`hid_copy_transfer_data`](https://github.com/libusb/libusb/blob/c6a35c56016ea2ab2f19115d2ea1e85e0edae155/libusb/os/windows_winusb.c#L4092).  Before libusb 1.0.24 [`_hid_get_report`](https://github.com/libusb/libusb/blob/c6a35c56016ea2ab2f19115d2ea1e85e0edae155/libusb/os/windows_winusb.c#L3473) handled copying the data, but now `hid_copy_transfer_data` does it, and I don't think that the `length--;` statement should be there, at least in the case of a control transfer:

https://github.com/libusb/libusb/blob/c6a35c56016ea2ab2f19115d2ea1e85e0edae155/libusb/os/windows_winusb.c#L4111-L4117

If the id is zero, we should indeed skip the first byte of `transfer_priv->hid_buffer`, since it is not part of the actual report.  But the report size should not change.

But that statement has been there (in some form) for years, so I could very well be wrong.  Or maybe it's correct for some other case.

mcuee, 2021-03-31:
Just want to clarify, we call this HID backend, not WinUSB.

jonasmalacofilho, 2021-03-31:
@mcuee

Thanks, I'll try to remember that in the future.

mcuee, 2021-03-31:
@EnochMakenp 
One general comments:
We recommend using HIDAPI for HID device. 
https://github.com/libusb/libusb/wiki/FAQ#Does_libusb_support_USB_HID_devices
You can use HIDAPI python API here.
https://github.com/trezor/cython-hidapi

If your python code is only for Windows, you can also try to use this.
https://github.com/rene-aguirre/pywinusb

mcuee, 2021-03-31:
Going back to this paticular issue, just wondering if you can try asking for 9 bytes of data. The handling of report ID 0 (no report ID) is rather troublesome.

jonasmalacofilho, 2021-03-31:
The user reported having already tried that.

> (if I ask for 7 bytes, the command fail, but if I request 9 or more bytes , I still get the 7 bytes...)

That probably happens because `hid_copy_transfer_data` gets called with a `length` value that does not comes from the user, but rather the number of actually transferred bytes.

mcuee, 2021-03-31:
Not so sure if this HIDAPI issue is somewhat related.
https://github.com/libusb/hidapi/issues/229

EnochMakenp, 2021-03-31:
I need to highlight what may be an important point: this code did work before (the original code was developed in 2018) with the same reader. I restarted development 2 weeks ago because we need to incorporate a new type of card reader. but I restarted the program with the old reader for testing (and remembering what I did 3 years ago :-) before going to the new reader and, as I mentioned earlier, the program managed to `find` the device but failed on the `ctrl_transfer` command with "[Errno 19] No such device (it may have been disconnected). ".
Upgrading the libusb1.0.dll to the latest version 1.0.24 did fix this issue... but after closer inspection I noticed the missing byte.
It is not completely clear to me why the older code did not run anymore on my computer, but I suspected either some Windows update, or some PyUSB update.

(Note that the code need to run on  a Windows PC (for enrolment) and on a RaspberryPI (for access) , that is the reason why I choose in 2018 the multiplatform PyUSB libusb1.0 solution but I did not know about the HIDAPI library, I may give it a try...)

mcuee, 2021-03-31:
@EnochMakenp 
Thanks for the clarifications. 
Just to prove it is an Windows HID backend problem, you can try to use Zadig to change the driver to WinUSB to see if the issue is still there or not. Thanks.

Zadig: https://zadig.akeo.ie/

mcuee, 2021-03-31:
FYI we have a few issues and limitations of the Windows HID backend.
https://github.com/libusb/libusb/issues?q=is%3Aopen+is%3Aissue+label%3AHID

And actually I have raised an RFC to remove the Windows HID backend.
https://github.com/libusb/libusb/issues/758

EnochMakenp, 2021-03-31:
I used Zadig and changed the HidUSB v10.0.18362.175 to the WinUSB driver for the HID compliant card reader... and bingo, my code now return the 8 bytes of data from the card !

It seems thus that the problem arises from a conjunction of a change in HidUSB (the version used now seems to appear in 2019, after my initial development knowing the computer where the code runs for enrolment is on WIndows7 and probably did not see that change) and a change in libusb as pointed out by @jonasmalacofilho.

I probably have now an acceptable procedure for my particular case (use Zadig on the Windows machine where I deploy it to change the driver to WinUSB) because from what I understand, on the RaspberryPI the code should work without issue if it is just a Windows backend issue.

Thanks a lot for your help and do not hesitate to ask me more question/test that you could need to propose a fix... or a "won't fix"...

diabolo38, 2021-03-31:
i had same kind of issue with a custom hid device transfert over hid back that were not correct (desc report id prepend to rd etc..  )
i end up with same solution replaced native backend  with libusbk  a far better option for anyone willing to do more than just getting hid report

mcuee, 2021-04-01:
Thanks for the updates. From the findings the issue is related to the HID backend.

commit f7624bf0 (2021-09-23): Fix Windows HID backend missing byte

When the report id is zero, we should indeed skip the first byte of
transfer_priv->hid_buffer, since it is not part of the actual report.
But the report size should not change.

Thanks to Jonas Malaco who has done the investigation in this issue.
Reference: Jonas Malaco's comments
https://github.com/libusb/libusb/issues/902#issuecomment-810726897

Fixes #902
Fixes #977
Closes #986
closed 2021-09-23 by tormodvolden