libusb/libusb #903: Deadlock calling libusb_close() within a transfer callback

achan1989, 2021-03-31:
The documentation for async completion handling says "The user-specified callback is called from an event handling context. It is therefore important that no calls are made into libusb that will attempt to perform any event handling. Examples of such functions are any listed in the synchronous API and any of the blocking functions that retrieve USB descriptors."

As I understand it, `libusb_close()` does not perform event handling and its documentation says "This is a non-blocking function; no requests are sent over the bus."

Yet when I call `libusb_close()` from within my transfer callback it deadlocks.  Single thread using libusb v1.0.23.11397.  Is this the expected behaviour (if so, the documentation could be made clearer), or is this a bug?  Logs attached.

[debug-log.txt](https://github.com/libusb/libusb/files/6238184/debug-log.txt)
[stack.txt](https://github.com/libusb/libusb/files/6238185/stack.txt)

jwrdegoede, 2021-03-31:
> As I understand it, |libusb_close()| does not perform event handling and its documentation says "This is a non-blocking function; no requests are sent over the bus."

So the issue here is that libusb_close() does:

```
	free(dev_handle);
```

But the transfer holds a reference to dev_handle->dev device and it unref-s that reference
after calling the callback:

```
        if (transfer->callback)
                transfer->callback(transfer);
        /* transfer might have been freed by the above call, do not use from
         * this point. */
        if (flags & LIBUSB_TRANSFER_FREE_TRANSFER)
                libusb_free_transfer(transfer);
        libusb_unref_device(dev_handle->dev);
```

So now you have `transfer->callback()` (your code) first doing `free(dev_handle)` and then the: `libusb_unref_device(dev_handle->dev)` call happens dereferencing the dev_handle pointer which now points to free-ed memory. Nothing good can come of this.

I guess you could try modifying the usbi_handle_transfer_completion() function inside libusb and cache dev_handle->dev before calling the callback. That might fix the problem because I don't see any obvious locking issue with doing what you do at a quick glance.

BTW under which OS is this? The OS specific code is calling `usbi_handle_transfer_completion()` and it also gets called through `usbi_backend.close(dev_handle)` from `libusb_close()` I did not check the backend locking, so maybe the backend is holding a lock here causing the deadlock.

> Yet when I call |libusb_close()| from within my transfer callback it deadlocks. Single thread using libusb v1.0.23.11397. Is this the expected behaviour (if so, the documentation could be made clearer), or is this a bug?

Calling libusb_close() from within a transfer callback is (I believe) a novel way of using libusb. If we can easily get it to work, then I guess we could allow it. But if turns out that this is hard to fix, then IMHO it is fair to call the deadlock expected behavior and then you will need to rework your code to close the device from outside the transfer-callback.

achan1989, 2021-03-31:
This is Ubuntu 20.04

> But the transfer holds a reference to dev_handle->dev device and [usbi_handle_transfer_completion] unref-s that reference
after calling the callback.

Oh, that's troubling.  "luckily" it deadlocked before reaching that point.  I think it's the `ctx->open_devs_lock` in `do_close()`.  The backend code holds `ctx->open_devs_lock` during `op_handle_events()` --> `reap_for_handle()` --> `handle_control_completion()` --> `usbi_handle_transfer_completion()` --> user callback --> `libusb_close()` --> `do_close()`

I don't think I'm capable of messing with libusb code right now.  I'm trying to write a safe wrapper around libusb, so I think I can prevent people from closing the device in the callback.

jwrdegoede, 2021-03-31:
Thank you for the deadlock analysis, so yes this sounds like a case of: "just don't do that".

If you can provide a patch / pull-req to get the documentation updated to explicitly document this then that would be great.