libusb/libusb #905: lisbusb.h: properly pack all USB descriptor definitions taken from the USB spec

lerdmann, 2021-04-05:
Likely, everyone (libusb implementers and also users) will expect the various USB descriptor definitions to have the original byte packing as defined by the USB spec.

For those descriptor definitions that extend USB definitions, I suggest to align pointers to 8-byte boundaries by adding padding bytes.

I stumbled over this problem when I was using "struct libusb_device_descriptor" in structure "struct libusb_device" on a new backend implementation. My assumption was that I could just copy the whole 18-byte block to that structure but that is not true if the structure is not byte packed.

LudovicRousseau, 2021-04-05:
`struct libusb_device_descriptor` is part of the libusb API so can't be changes without breaking the ABI (application binary interface).
I do not see any benefit from a packed structure for an application.

It should be easy to copy the structure in portable C. No need to expect (or suppose) the structure is 18-bytes long.
What are your motivations exactly?

lerdmann, 2021-04-05:
Initial motivation is that my backend reads the device descriptor from the USB device and copies it unchanged to the device descriptor structure which cannot be done if the target structure is not going to be packed.Other motivation is that users might request a raw descriptor from the USB device (like config descriptor) and might attempt to use the descriptor structures to access a field directly which will also fail. The descriptor structures are misleading in that they make you believe that you can do so where in fact you cannot.

mcuee, 2021-04-06:
I do not see the benefits of this idea which breaks the ABI.

mcuee, 2021-04-07:
Close for now. If others have strong opinion for this as well, please re-open.

closed 2021-04-07 by mcuee