libusb/libusb #908: libusb_kernel_driver_active() fails on Mac OS 11.1

programmingkidx, 2021-04-14:
I am trying to use an USB sound card in a program. To use it I have to detach the operating system's driver. Before I can do this I need to detect the interface the operating system is using. The libusb_kernel_driver_active() function fails to find the active interface every time. I did a full search from 0 to 255 as the interface number.

mcuee, 2021-04-15:
Kernel driver detaching is not supported under macOS.

mcuee, 2021-04-15:
On the other hand, libusb_kernel_driver_active() should still work. You need to specify the right interface and you are supposed to know the interface number. 

You can use the 'lsusb" utility to dump the descriptor under Linux, USBView under Windows.

There is a fork of "lsusb" under macOS as well. Or you can use the "testlibusb -v" from libusb example.

programmingkidx, 2021-04-15:
I tried ever interface number available (from 0 to 255). The USB sound card uses interface 0 to 3. Here is a picture of its information from IORegistryExplorer:

[image]

hjelmn, 2021-04-16:
Can you provide output with ```LIBUSB_DEBUG=100``. Can't say this feature has been tested in a long long time so it could easily be broken.

programmingkidx, 2021-04-16:
Here is the output you requested:

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000032] [0000650a] libusb: debug [libusb_init] created default context
[ 0.000048] [0000650a] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000233] [0000650a] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xbaff96d
[ 0.000252] [0000650a] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xbaff96d
[ 0.001420] [0000650a] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.001424] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.001425] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0310
[ 0.001426] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.001427] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.001428] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.001429] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.001430] [0000650a] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x2109
[ 0.001430] [0000650a] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0817
[ 0.001431] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x03c4
[ 0.001432] [0000650a] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.001438] [0000650a] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.001439] [0000650a] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x03
[ 0.001440] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.001451] [0000650a] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.001453] [0000650a] libusb: debug [process_new_device] allocating new device in context 0x127e94760 for with session 0xbaff96d
[ 0.001457] [0000650a] libusb: debug [process_new_device] found device with address 1 port = 2 parent = 0x0 at 0x127e9722a
[ 0.001494] [0000650a] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xbca6ccc
[ 0.001517] [0000650a] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xbca6ccc/0x100000 against cached device with sessionID/locationID 0xbaff96d/0x200000
[ 0.001520] [0000650a] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xbca6ccc
[ 0.001755] [0000650a] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.001761] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.001763] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.001764] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.001766] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.001767] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.001768] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.001770] [0000650a] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x2109
[ 0.001771] [0000650a] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x2817
[ 0.001773] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x03c4
[ 0.001774] [0000650a] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.001776] [0000650a] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.001777] [0000650a] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x03
[ 0.001778] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.001789] [0000650a] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.001791] [0000650a] libusb: debug [process_new_device] allocating new device in context 0x127e94760 for with session 0xbca6ccc
[ 0.001794] [0000650a] libusb: debug [process_new_device] found device with address 2 port = 1 parent = 0x0 at 0x127e96daa
[ 0.001836] [0000650a] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xd67fcd4
[ 0.001856] [0000650a] libusb: debug [darwin_get_cached_device] parent sessionID: 0xbaff96d
[ 0.001858] [0000650a] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xd67fcd4/0x220000 against cached device with sessionID/locationID 0xbca6ccc/0x100000
[ 0.001860] [0000650a] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xd67fcd4/0x220000 against cached device with sessionID/locationID 0xbaff96d/0x200000
[ 0.001862] [0000650a] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xd67fcd4
[ 0.002019] [0000650a] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002023] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002024] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0310
[ 0.002026] [0000650a] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[... cut ...]
[17.604788] [0000650a] libusb: debug [libusb_attach_kernel_driver] interface 10
[17.604790] [0000650a] libusb: debug [libusb_attach_kernel_driver] interface 11
[17.604801] [0000650a] libusb: debug [libusb_attach_kernel_driver] interface 12
[17.604805] [0000650a] libusb: debug [libusb_attach_kernel_driver] interface 13
[17.604807] [0000650a] libusb: debug [libusb_attach_kernel_driver] interface 14
[17.604809] [0000650a] libusb: debug [libusb_attach_kernel_driver] interface 15
[17.604816] [0000650a] libusb: debug [libusb_close]

hjelmn, 2021-04-16:
And of course I never bothered with any logging in that function (except on error). Let me take a look.

programmingkidx, 2021-04-17:
I thought of one way to reimplement the libusb_kernel_driver_active() function. We could take the output of "ioreg -Src IOUSBDevice" and parse it. We could search for the name and other properties of a USB device to make sure we have the right one. Then search for IOUSBHostInterface. If there is a child object to it when we know the  interface has an attached driver. Here is what the output of my USB sound card looks like:

  | | +-o C-Media USB Audio Device   @00110000  <class IOUSBHostDevice, id 0x100000e2d, registered, matched, active, busy 0 (88 ms), retain 29>
  | |   | {
  | |   |   "sessionID" = 45872715564
  | |   |   "USBSpeed" = 1
  | |   |   "idProduct" = 8
  | |   |   "iManufacturer" = 0
  | |   |   "bDeviceClass" = 0
  | |   |   "IOPowerManagement" = {"PowerOverrideOn"=Yes,"CapabilityFlags"=32768,"MaxPowerState"=2,"DevicePowerState"=2,"DriverPowerState"=0,"ChildrenPowerState"=2,"CurrentPowerState"=2}
  | |   |   "bcdDevice" = 256
  | |   |   "bMaxPacketSize0" = 64
  | |   |   "iProduct" = 1
  | |   |   "iSerialNumber" = 0
  | |   |   "bNumConfigurations" = 1
  | |   |   "USB Product Name" = "C-Media USB Audio Device   "
  | |   |   "USB Address" = 3
  | |   |   "locationID" = 1114112
  | |   |   "bDeviceSubClass" = 0
  | |   |   "bcdUSB" = 272
  | |   |   "kUSBCurrentConfiguration" = 1
  | |   |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  | |   |   "bDeviceProtocol" = 0
  | |   |   "USBPortType" = 0
  | |   |   "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
  | |   |   "Device Speed" = 1
  | |   |   "idVendor" = 3468
  | |   |   "kUSBProductString" = "C-Media USB Audio Device   "
  | |   |   "IOGeneralInterest" = "IOCommand is not serializable"
  | |   |   "kUSBAddress" = 3
  | |   | }
  | |   | 
  | |   +-o AppleUSBHostCompositeDevice  <class AppleUSBHostCompositeDevice, id 0x100000e31, !registered, !matched, active, busy 0, retain 4>
  | |   +-o IOUSBHostInterface@0  <class IOUSBHostInterface, id 0x100000e32, registered, matched, active, busy 0 (19 ms), retain 8>
  | |   | +-o AppleUSBAudioDevice  <class AppleUSBAudioDevice, id 0x100000e3a, registered, matched, active, busy 0 (3 ms), retain 21>
  | |   |   +-o AppleUSBAudioEngine  <class AppleUSBAudioEngine, id 0x100000e3c, registered, matched, active, busy 0 (1 ms), retain 15>
  | |   |   | +-o AppleUSBAudioStream  <class AppleUSBAudioStream, id 0x100000e3d, registered, matched, active, busy 0 (1 ms), retain 9>
  | |   |   | +-o IOAudioLevelControl  <class IOAudioLevelControl, id 0x100000e51, !registered, !matched, active, busy 0, retain 8>
  | |   |   | | +-o IOAudioControlUserClient  <class IOAudioControlUserClient, id 0x100000e5e, !registered, !matched, active, busy 0, retain 6>
  | |   |   | +-o IOAudioToggleControl  <class IOAudioToggleControl, id 0x100000e52, !registered, !matched, active, busy 0, retain 8>
  | |   |   | | +-o IOAudioControlUserClient  <class IOAudioControlUserClient, id 0x100000e5f, !registered, !matched, active, busy 0, retain 6>
  | |   |   | +-o IOAudioLevelControl  <class IOAudioLevelControl, id 0x100000e53, !registered, !matched, active, busy 0, retain 8>
  | |   |   | | +-o IOAudioControlUserClient  <class IOAudioControlUserClient, id 0x100000e60, !registered, !matched, active, busy 0, retain 6>
  | |   |   | +-o IOAudioToggleControl  <class IOAudioToggleControl, id 0x100000e54, !registered, !matched, active, busy 0, retain 8>
  | |   |   | | +-o IOAudioControlUserClient  <class IOAudioControlUserClient, id 0x100000e61, !registered, !matched, active, busy 0, retain 6>
  | |   |   | +-o IOAudioEngineUserClient  <class IOAudioEngineUserClient, id 0x100000e62, !registered, !matched, active, busy 0, retain 6>
  | |   |   +-o AppleUSBAudioEngine  <class AppleUSBAudioEngine, id 0x100000e55, registered, matched, active, busy 0 (1 ms), retain 14>
  | |   |     +-o AppleUSBAudioStream  <class AppleUSBAudioStream, id 0x100000e56, registered, matched, active, busy 0 (1 ms), retain 9>
  | |   |     +-o IOAudioLevelControl  <class IOAudioLevelControl, id 0x100000e5b, !registered, !matched, active, busy 0, retain 8>
  | |   |     | +-o IOAudioControlUserClient  <class IOAudioControlUserClient, id 0x100000e63, !registered, !matched, active, busy 0, retain 6>
  | |   |     +-o IOAudioLevelControl  <class IOAudioLevelControl, id 0x100000e5c, !registered, !matched, active, busy 0, retain 8>
  | |   |     | +-o IOAudioControlUserClient  <class IOAudioControlUserClient, id 0x100000e64, !registered, !matched, active, busy 0, retain 6>
  | |   |     +-o IOAudioToggleControl  <class IOAudioToggleControl, id 0x100000e5d, !registered, !matched, active, busy 0, retain 8>
  | |   |     | +-o IOAudioControlUserClient  <class IOAudioControlUserClient, id 0x100000e65, !registered, !matched, active, busy 0, retain 6>
  | |   |     +-o IOAudioEngineUserClient  <class IOAudioEngineUserClient, id 0x100000e66, !registered, !matched, active, busy 0, retain 6>
  | |   +-o IOUSBHostInterface@1  <class IOUSBHostInterface, id 0x100000e33, registered, matched, active, busy 0 (16 ms), retain 10>
  | |   | +-o AppleUSBAudioStream  <class AppleUSBAudioStream, id 0x100000e56, registered, matched, active, busy 0 (1 ms), retain 9>
  | |   +-o IOUSBHostInterface@2  <class IOUSBHostInterface, id 0x100000e34, registered, matched, active, busy 0 (15 ms), retain 10>
[... cut ...]
  | |       +-o IOHIDInterface  <class IOHIDInterface, id 0x100000e4a, registered, matched, active, busy 0 (18 ms), retain 10>
  | |         +-o AppleUserHIDEventDriver  <class AppleUserHIDEventService, id 0x100000e4b, registered, matched, active, busy 0 (0 ms), retain 12>
  | |           +-o IOHIDEventServiceUserClient  <class IOHIDEventServiceUserClient, id 0x100000e50, !registered, !matched, active, busy 0, retain 6>

Finding the right interface is as simple as looking for "IOUSBHostInterface@"<interface number>.

Would this be an acceptable solution?

hjelmn, 2021-04-20:
I see the issue. Will try to fix it this week.

hjelmn, 2021-04-20:
At some point Apple changed the hierarchy. Essentially we just need to get the IORegistry child entry of the USB interface entry and check for the property there.

programmingkidx, 2021-04-30:
Any progress yet?

hjelmn, 2021-07-11:
Can you try again with the current git master? The implementation of this function has changed.

programmingkidx, 2021-07-13:
I tried libusb at commit 339d24c8235476ceee43f7d704415ccab85e2583.
My USB sound card did not work. Sorry. 
Please let me know if you need more debugging information.

programmingkidx, 2021-07-13:
I set LIBUSB_DEBUG=10 and recorded this output:

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000038] [000b9527] libusb: debug [libusb_init] libusb v1.0.24.11636
[ 0.000110] [000b9527] libusb: debug [usbi_add_event_source] add fd 11 events 1
[ 0.000418] [000b9527] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xbb65d34
[ 0.000450] [000b9527] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xbb65d34
[ 0.002622] [000b9527] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002629] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002631] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0310
[ 0.002632] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002633] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002635] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002636] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002637] [000b9527] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x2109
[ 0.002638] [000b9527] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0817
[ 0.002640] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x03c4
[ 0.002641] [000b9527] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002643] [000b9527] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002644] [000b9527] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x03
[ 0.002645] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002660] [000b9527] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002662] [000b9527] libusb: debug [process_new_device] allocating new device in context 0x158e75390 for with session 0xbb65d34
[ 0.002669] [000b9527] libusb: debug [process_new_device] found device with address 1 port = 2 parent = 0x0 at 0x158e77b5a
[ 0.002737] [000b9527] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xbf2cdf6
[ 0.002758] [000b9527] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xbf2cdf6/0x100000 against cached device with sessionID/locationID 0xbb65d34/0x200000
[ 0.002760] [000b9527] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xbf2cdf6
[ 0.003014] [000b9527] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003018] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003020] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.003021] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003022] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003023] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.003025] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.003026] [000b9527] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x2109
[ 0.003027] [000b9527] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x2817
[ 0.003028] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x03c4
[ 0.003030] [000b9527] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003031] [000b9527] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003032] [000b9527] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x03
[ 0.003034] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003044] [000b9527] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003046] [000b9527] libusb: debug [process_new_device] allocating new device in context 0x158e75390 for with session 0xbf2cdf6
[ 0.003049] [000b9527] libusb: debug [process_new_device] found device with address 2 port = 1 parent = 0x0 at 0x158e7622a
[ 0.003117] [000b9527] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xd5d0e34
[ 0.003142] [000b9527] libusb: debug [darwin_get_cached_device] parent sessionID: 0xbb65d34
[ 0.003144] [000b9527] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xd5d0e34/0x220000 against cached device with sessionID/locationID 0xbf2cdf6/0x100000
[ 0.003146] [000b9527] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xd5d0e34/0x220000 against cached device with sessionID/locationID 0xbb65d34/0x200000
[ 0.003148] [000b9527] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xd5d0e34
[ 0.003305] [000b9527] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003309] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003310] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0310
[ 0.003311] [000b9527] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[... cut ...]
[16.078166] [000b9527] libusb: info [darwin_claim_interface] no interface found; setting configuration: 1
[16.078321] [000b9527] libusb: error [darwin_claim_interface] interface not found
[18.013877] [000b9527] libusb: debug [libusb_get_device_list]  
[18.013916] [000b9527] libusb: debug [libusb_get_device_descriptor]  
[18.013918] [000b9527] libusb: debug [libusb_get_device_descriptor]  
[18.013919] [000b9527] libusb: debug [libusb_get_device_descriptor]  
[18.013919] [000b9527] libusb: debug [libusb_get_device_descriptor]

mcuee, 2021-07-18:
@programmingkidx Could you post the codes for the above debug log? Take note there are a few things.
1)if libusb_kernel_driver_active() does not work under macOS, that is a libusb bug
2) not able to detach kernel driver -- that is possible and it is not a libusb bug.

programmingkidx, 2021-07-18:
I'm sorry but I don't know what you mean by post codes. Did you want me to send you the code I use to test libusb? I use libusb with a computer emulator called QEMU. libusb allows for real USB devices to be used in emulated environment.

hjelmn, 2021-07-18:
Well. The only question is does driver_active now work as expected? Your log shows that it can't detach the kernel driver due to missing entitlements. So the detach failure is expected.

hjelmn, 2021-07-18:
You can Google macOS entitlements to find out how to use then with your app.

mcuee, 2021-07-19:
Just did a quick hack of xusb example to test and libusb_kernel_driver_active() is working under my Mac Mini M1 (Big Sur).

```
libusb on  master [!] ❯ git diff
diff --git a/examples/xusb.c b/examples/xusb.c
index bf328fe..e60f7b8 100644
--- a/examples/xusb.c
+++ b/examples/xusb.c
@@ -867,6 +867,30 @@ static int test_device(uint16_t vid, uint16_t pid)
        printf("             nb interfaces: %d\n", nb_ifaces);
        if (nb_ifaces > 0)
                first_iface = conf_desc->usb_interface[0].altsetting[0].bInterfaceNumber;
+
+    r = libusb_kernel_driver_active(handle, 1);
+
+       for (iface = 0; iface < nb_ifaces; iface++)
+       {
+               printf("\nCheck kernel driver active for interface %d... \n", iface);
+               r = libusb_kernel_driver_active(handle, iface);
+               if (r == LIBUSB_SUCCESS) {
+                       perr("   No kernel driver attached.\n");
+               }
+               else if (r == 1) {
+                       perr("   There is a kernel driver attached.\n");
+               }
+               else if (r == LIBUSB_ERROR_NOT_SUPPORTED) {
+                       perr("   Not supported for this platform.\n");
+               }
+               else if (r == LIBUSB_ERROR_NO_DEVICE) {
+                       perr("   Device not present.\n");
+               }
+               else {
+                       perr("   Other errors.\n");
+               }
+       }
+
        for (i=0; i<nb_ifaces; i++) {
                printf("              interface[%d]: id = %d\n", i,
                        conf_desc->usb_interface[i].altsetting[0].bInterfaceNumber);
```

Run log 1:
```

libusb on  master [!] ❯ ./examples/xusb 046d:c52b
Using libusb v1.0.24.11636

Opening device 046D:C52B...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 046D:C52B
         bcdDevice: 2410
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 3
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access

Check kernel driver active for interface 0... 
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   There is a kernel driver attached.

Check kernel driver active for interface 1... 
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   There is a kernel driver attached.

Check kernel driver active for interface 2... 
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   There is a kernel driver attached.
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.01
       endpoint[0].address: 81
           max packet size: 0008
          polling interval: 08
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.02
       endpoint[0].address: 82
           max packet size: 0008
          polling interval: 02
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 83
           max packet size: 0020
          polling interval: 02

Claiming interface 0...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Claiming interface 1...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Claiming interface 2...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Reading string descriptors:
   String (0x01): "Logitech"
   String (0x02): "USB Receiver"

Releasing interface 0...
Releasing interface 1...
Releasing interface 2...
Closing device...```

mcuee, 2021-07-19:
Run log 2: FTDI USB Serial Device is no longer openning the device in exclusive mode in Big Sur and therefore it shows as no kernel driver attached.

```

libusb on  master [!] ❯ ./examples/xusb 0403:6001
Using libusb v1.0.24.11636

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
libusb: info [darwin_claim_interface] no interface found; setting configuration: 1
libusb: error [darwin_claim_interface] interface not found

Check kernel driver active for interface 0... 
   No kernel driver attached.
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Releasing interface 0...
Closing device...
```

mcuee, 2021-07-19:
@hjelmn The warning and error message are not related to this issue. I will create another issue to check if there is a need to reduce the level of the message or not.

mcuee, 2021-07-19:
Run log 3 for a USB headset. Again libusb_kernel_driver_active() is correct.

```

libusb on  master [!] ❯ ./examples/xusb 047f:c025
Using libusb v1.0.24.11636

Opening device 047F:C025...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 047F:C025
         bcdDevice: 0135
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 4
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access

Check kernel driver active for interface 0... 
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   There is a kernel driver attached.

Check kernel driver active for interface 1... 
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   There is a kernel driver attached.

Check kernel driver active for interface 2... 
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   There is a kernel driver attached.

Check kernel driver active for interface 3... 
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   There is a kernel driver attached.
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.01.00
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.02.00
interface[1].altsetting[1]: num endpoints = 1
   Class.SubClass.Protocol: 01.02.00
       endpoint[0].address: 81
           max packet size: 00C8
          polling interval: 01
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.02.00
interface[2].altsetting[1]: num endpoints = 1
   Class.SubClass.Protocol: 01.02.00
       endpoint[0].address: 01
           max packet size: 00C0
          polling interval: 01
              interface[3]: id = 3
interface[3].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 84
           max packet size: 0025
          polling interval: 01

Claiming interface 0...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Claiming interface 1...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Claiming interface 2...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Claiming interface 3...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Reading string descriptors:
   String (0x01): "Plantronics"
   String (0x02): "Plantronics C320-M"
   String (0x03): "CB13A3E40E8E47D6A40769C27E90A38E"

Releasing interface 0...
Releasing interface 1...
Releasing interface 2...
Releasing interface 3...
Closing device...
```

mcuee, 2021-07-19:
> I'm sorry but I don't know what you mean by post codes. Did you want me to send you the code I use to test libusb? I use libusb with a computer emulator called QEMU. libusb allows for real USB devices to be used in emulated environment.

I want you to post the code which shows that libusb_kernel_driver_active() is not working for you with the latest libusb git head. From my own testing, it works fine.

For your second question with using QEMU and USB device under the virtual environment, that is a seperate issue. And you may check out Nathan's suggestion, or ask the QEMU community. I have tried to use UTM (based on QEMU) and Debian VMs under my Mac Mini M1, it seems to work with my USB devices last time I tested it.

mcuee, 2021-07-19:
@programmingkidx Your original issue report is correct. libusb-1.0.24 has the issue in libusb_kernel_driver_active and I can reproduce the issue. But I belive it has been fixed in the latest git. Please check again.

libusb-1.0.24 issue: PICKit 2 is an HID device and it has the HID driver attached. So the libusb_kernel_driver_active() output is not correct.

```
libusb/libusb-1.0.24/buildarm64 ❯ ./examples/xusb 04d8:0033
Using libusb v1.0.24.11584

Opening device 04D8:0033...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 04D8:0033
         bcdDevice: 0002
   iMan:iProd:iSer: 1:2:3
          nb confs: 2

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1

Check kernel driver active for interface 0... 
   No kernel driver attached.
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 01
           max packet size: 0040
          polling interval: 01

Claiming interface 0...
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Reading string descriptors:
   String (0x01): "Microchip Technology Inc."
   String (0x02): "PICkit 2 Microcontroller Programmer"
   String (0x03): "pk2new"

Releasing interface 0...
Closing device...

```

mcuee, 2021-07-22:
@programmingkidx Maybe you want to try out https://github.com/libusb/libusb/pull/959 to see if works better for you. Thanks.

mcuee, 2021-07-23:
@programmingkidx #959 has been merged. I will close this issue now. Please reopen if you still encounter issues.

programmingkidx, 2021-07-24:
I am using libusb at this commit:

commit d2522b3a4a88f42b7719db68260700c215bc8148 (HEAD -> master, origin/master, origin/HEAD)
Author: Nathan Hjelm <hjelmn@google.com>
Date:   Wed Jul 21 23:12:19 2021 -0600

    darwin: only attempt to auto-detach driver is one is active

I tried to use a USB audio device and did see it correctly listed by QEMU console using the 'info usb' command, but it didn't work. Here is the output:

qemu-system-ppc: libusb_detach_kernel_driver: -12 [NOT_SUPPORTED]
qemu-system-ppc: libusb_detach_kernel_driver: -12 [NOT_SUPPORTED]
qemu-system-ppc: libusb_detach_kernel_driver: -12 [NOT_SUPPORTED]
qemu-system-ppc: libusb_detach_kernel_driver: -12 [NOT_SUPPORTED]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_detach_kernel_driver: -12 [NOT_SUPPORTED]
qemu-system-ppc: libusb_detach_kernel_driver: -12 [NOT_SUPPORTED]
qemu-system-ppc: libusb_detach_kernel_driver: -12 [NOT_SUPPORTED]
qemu-system-ppc: libusb_detach_kernel_driver: -12 [NOT_SUPPORTED]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]
qemu-system-ppc: libusb_kernel_driver_active: -5 [NOT_FOUND]

I am sure I am using the correct libusb. I used the install_name_tool command to point QEMU to this library. I even added a printf() message to libusb_init() just to be safe. I did see the message I made. So I think something is still not right with libusb.

mcuee, 2021-07-24:
@programmingkidx It seems to me this is a different issue related to qemu. Can you run libusb xusb example against your USB soundcard to see if libusb_kernel_driver_active() is working or not?

What is this qemu-system-ppc? What is your macOS system? Are you using a PPC Mac?

mcuee, 2021-07-24:
@programmingkidx I did a simple google search and it seems to me qemu does not work well with USB passthrough. So you may want to ask in the QEMU side.

Ref:
https://stackoverflow.com/questions/65816169/how-to-use-usb-with-qemu-on-a-mac-host

closed 2021-07-24 by mcuee

mcuee, 2021-07-24:
BTW, you may want to give UTM a try, it seems to work reasonably well under my Mac Mini M1 with regard to USB device passthrough.