libusb/libusb #909: WinUSB ISO packet backend is broken in multiple ways

dontech, 2021-04-16:
Hi Libusb,

As always thanks for your great library.

I have been porting some old applications from libusb-0.1 to libusb-1.x. This works great for e.g. Linux, where int, bulk and iso streams work fantastic in latest libusb.

However, on Windows, the WinUSB backend for ISO packets is more or less unusable in the current state. There seem to be allot of issues:

1) If you keep writing libusb_submit_transfer (async), the data stream is severely corrupted. To make matters worse, if you keep going, the kernel driver (WinUSB) even locks up, requireing a system hard shutdown. This is not libusbs fault, but libusb seems to do something to trigger this fairly easily.

2) Reading using the async interface for iso is also broken. After submitting (and resubmitting) packets for 300-400 times, the callbacks stop occurring. Only queuing one at a time makes it slightly more stable, but still not usable.

3) WinUSB seems to loop the iso data that has been sent it it. Not sure if this is something libusb can stop. This means that when you stop sending data to WinUSB, old data is being sent in a loop in some cases.

Notes 1: The same program works flawlessly on Linux using LibUSB.
Notes 2: The same device works flawlessly in Windows using older library (libusb-0.1 with libusb0.sys).
Notes 3: Control, Bulk and Int transfers work great on all systems.

Setup: Windows 10 version 1909, X64.

Any ideas? Need some traces/logs?

Thanks,

/pedro

mcuee, 2021-04-16:
There are two more different backend to try for isoc transfer. Please try libusbk.sys driver first to see if that works or not.

There was a past report to say that libusbk.sys works but not winusb for isoc transfer.
https://github.com/libusb/libusb/issues/759

If that still does not work, try usbdk.

mcuee, 2021-04-16:
Another reference and potential improvement for WinUSB.

Issue:
https://github.com/libusb/libusb/issues/747

Pull request:
https://github.com/libusb/libusb/issues/759

dontech, 2021-04-16:
Hi there mcuee!

Yes, i am finally starting to move to 1.0! I am probably the last man standing on 0.1... :-)
I still use the 0.1 subsystem allot however, because is is just so damn reliable, and the code is easy to change/fix.

For your inquires: I would rather not start using libusbk, but i guess i could do some testing with it, if you think it helps.

I am however fairly sure its the WinUSB backend that's broken. I think there are some issues with queuing of packets that goes haywire at some point.

I think using time on WinUSB is a good investment, as Microsoft is going to be supporting WinUSB for a long foreseeable future.

dontech, 2021-04-16:
#759 seems like a workaround more than a fix, and it seems like the user found a way around some of the issues. It should probably be fixed in the library instead.

mcuee, 2021-04-19:
Yes I agree that WinUSB is the way to go as it is supported by Microsoft.

On the other hand, please still test libusbk to see if the issue is generic or only with WinUSB. Thanks. 

I think ISOC transfer is more difficult to test. In your case, it is good that you have a known good FW for testing.

dontech, 2021-05-04:
OK, i will try when i find the time.

mcuee, 2021-07-05:
@dontech Not exactly to solve your issue, but just a pointer.

Travis has created a development branch for libusbk here. It adds WinUSB isochronous transfer support. New benchmark FW for Cypress EZ-USB FX3 super speed usb device controller has been added. libusbk library and kbench now have better support of isochronous transfer for both libusbk driver and WinUSB driver.
https://github.com/mcuee/libusbk/tree/WinUSB-ISO

You can see the impressive results from kbench from Travis using the Cypress CYUSB3KIT-003 EZ-USB® FX3™ SuperSpeed Explorer Kit.
https://github.com/mcuee/libusbk/tree/WinUSB-ISO/BmFW/CYPRESS_FX3
https://www.cypress.com/documentation/development-kitsboards/cyusb3kit-003-ez-usb-fx3-superspeed-explorer-kit

I have also got the same Cypress board and I can verify the results from Travis.

mcuee, 2021-07-05:
Maybe libusb isoc codes can now be unified (WinUSB/libusbk) to use the latest libusbk dll.

mcuee, 2021-11-09:
> Maybe libusb isoc codes can now be unified (WinUSB/libusbk) to use the latest libusbk dll.

This may not be a good idea, people can use WinUSB without libusbk.dll.

dontech, 2021-11-09:
On Tue, 2021-11-09 at 01:08 -0800, mcuee wrote:
> > Maybe libusb isoc codes can now be unified (WinUSB/libusbk) to use
> > the latest libusbk dll.
> This may not be a good idea, people can use WinUSB without
> libusbk.dll.

Which is fairly broken currently...

mcuee, 2021-11-09:
@dontech BTW, are you only using full speed isoc transfer? For that purpose libusb0.sys will probably work well enough. For high speed or super speed USB isoc transfer, you will have better lucks with libusbk.sys or WinUSB.

tormodvolden, 2022-03-21:
I don't know but maybe the now merged #1095 fixed the third issue in the original post.

dontech, 2022-03-21:
I will try to retest soon with these new ISO fixes, as soon as there is a new test-build to give a spin.

tormodvolden, 2022-03-21:
Great. An RC will come within a few days. But you can also get Windows binaries built from CI at any time. You can access it through the green tick marker in the git log on GitHub or directly at https://ci.appveyor.com/project/LudovicRousseau/libusb/history
Latest build today: https://ci.appveyor.com/project/LudovicRousseau/libusb/builds/42970249/job/qu1lx89ah2frlbvw/artifacts

tormodvolden, 2022-03-29:
I have tested isochronous streaming over both WinUSB and libusbK with 1.0.26-rc1 successfully, using libfreenect and an Xbox 360 Kinect camera. I think the first and second issue in the original post may be improved by the ContinueStream work in #749.

mcuee, 2022-04-01:
@dontech libusb-1.0.26rc1 has been out. I think the isoc support should be better now. Maybe you want to give it a try.

I am still facing issues with High Speed and SuperSpeed isoc transfer. I have not tried Full Speed USB device.

#749 and #1069 (partial fix to use RAW_IO policy) may help WinUSB isoc transfer stability.
Ref: https://github.com/libusb/libusb/issues/1104

mcuee, 2023-02-17:
@dontech 

Just wondering if you have done any tests with libusb 1.0.26 or better libusb git HEAD. Thanks.