libusb/libusb #910: Enable TLS on all subdomains of libusb.info

KOLANICH, 2021-04-20:
, including api.libusb.info  and log.libusb.info

mcuee, 2021-04-20:
api.libusb.info will redirect to http://libusb.sourceforge.net/api-1.0/index.html and I think we have no control over Sourceforge.

mcuee, 2021-04-20:
log.libusb.info will redirect to https://github.com/libusb/libusb/blob/master/ChangeLog and it is already using https.

KOLANICH, 2021-04-20:
>log.libusb.info will redirect to https://github.com/libusb/libusb/blob/master/ChangeLog and it is already using https.

The redirect itself is insecure and adversary can use it to deliver JS-based exploits.

>api.libusb.info will redirect to http://libusb.sourceforge.net/api-1.0/index.html and I think we have no control over Sourceforge.

How about using GitHub Pages then?

mcuee, 2021-04-20:
Personally I think this is not an important issue. But I will just leave this open.

LudovicRousseau, 2021-04-20:
I can access the DNS configuration for libusb.info at https://cp.blacknight.com/
But I am surprised to NOT find any record for log.libusb.info. and api.libusb.info.
[image]

I don't understand how these 2 names can be resolved. What am I missing?

mcuee, 2021-04-23:
Interesting. 
api.libusb.info and log.libusb.info point to the same IPv4 address: 78.153.212.172.

LudovicRousseau, 2021-04-23:
And 78.153.212.172 is pemsslpr01.blacknight.com.

I had a look again and I found how it works.
[image]

We can't use https: for sourceforge.
@KOLANICH what change do you propose exactly?

tormodvolden, 2021-04-23:
> api.libusb.info will redirect to http://libusb.sourceforge.net/api-1.0/index.html and I think we have no control over Sourceforge.

You can switch the SF site to https (however, \*facepalm\* it means domain name change .net -> .io):
https://sourceforge.net/p/libusb/admin/ext/https/

tormodvolden, 2021-04-23:
@KOLANICH here's for you: http://neil.franklin.ch/Save_HTTP/

KOLANICH, 2021-04-23:
>@KOLANICH what change do you propose exactly?

You can move the website to sourceforge.io, or you can move it to GitHub pages. I feel like moving to GH pages is slightly more preferred, since you are already using GitHub, and pages can be updated using Actions pipelines.

LudovicRousseau, 2021-04-24:
- I upgraded sourceforge to use https. The SF web site is now at https://libusb.sourceforge.io and api.libusb.info redirect to the https page.
- I removed the redirection for nabble.libusb.info since this service is no more available.
- I also updated the web site https://libusb.info/ to use https when available.

@KOLANICH Can we  consider the issue to be fixed?

KOLANICH, 2021-04-24:
>I upgraded sourceforge to use https. The SF web site is now at https://libusb.sourceforge.io 

Works, redirects to https://libusb.info/. :+1:

>and api.libusb.info redirect to the https page.

The redirect (to https://libusb.sourceforge.io/api-1.0/index.html) itself is still going over an insecure connection, over https it still is not available. Can anything be done with this?  If nothing can be done to make the server on that domain to support TLS, it can probably be possible to make the server to return the redirect to just `https://libusb.sourceforge.io/api`, then to have `https://libusb.sourceforge.io/api` be just an alias to the actual version of the api docs. Then I can write a HTTPSEverywhere rule rewriting `api.libusb.info` to `https://libusb.sourceforge.io/api`

>I also updated the web site https://libusb.info/ to use https when available.

Nice, thank you. May also make sense to add it into https://github.com/chromium/hstspreload.org .

LudovicRousseau, 2021-04-24:
What do you mean by "The redirect (...) itself is still going over an insecure connection"?

I have:
```
$ curl http://api.libusb.info
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>302 Found</title>
</head><body>
<h1>Found</h1>
<p>The document has moved <a href="https://libusb.sourceforge.io/api-1.0/index.html">here</a>.</p>
</body></html>
```
So the redirection to https works.

And
```
$ curl https://api.libusb.info
curl: (7) Failed to connect to api.libusb.info port 443: Connection refused
```
I don't know if we can change that since the redirection is handled by blacknight.com web server.

mcuee, 2021-05-04:
Close for now as this is the best we can do for now.

closed 2021-05-04 by mcuee