libusb/libusb #912: Segfault under MinGW-w64

kulkin-fish, 2021-04-29:
analysis-dbg.exe caused an Access Violation at location 000000006B608500 in module libusb-1.0.dll Reading from location 000000000000054C.

AddrPC           Params
000000006B608500 000000002568F930 00007FFD00000004 000000006B62FA50  libusb-1.0.dll!usbi_signal_transfer_completion  [D:/msys64/tmp/libusb-1.0.24/libusb/io.c @ 1734]
  1732: 
  1733: if (dev_handle) {
> 1734: struct libusb_context *ctx = HANDLE_CTX(dev_handle);
  1735: unsigned int event_flags;
  1736: 
000000006B60BD99 00000000222EDC80 0000000000000000 00000000222CC7D0  libusb-1.0.dll!windows_iocp_thread  [D:/msys64/tmp/libusb-1.0.24/libusb/os/windows_common.c @ 442]
   440: usbi_dbg("transfer %p completed, length %lu",
   441:  USBI_TRANSFER_TO_LIBUSB_TRANSFER(itransfer), ULONG_CAST(num_bytes));
>  442: usbi_signal_transfer_completion(itransfer);
   443: }
   444: 
00007FFDD06BAF5A 00007FFDD07106D0 00000000222CC7D0 0000000000000000  msvcrt.dll!_beginthreadex
00007FFDD06BB02C 0000000000000000 0000000000000000 0000000000000000  msvcrt.dll!_endthreadex
00007FFDD0737034 0000000000000000 0000000000000000 0000000000000000  KERNEL32.DLL!BaseThreadInitThunk
00007FFDD0922651 0000000000000000 0000000000000000 0000000000000000  ntdll.dll!RtlUserThreadStart

kulkin-fish, 2021-04-29:
[libusb-bug.txt](https://github.com/libusb/libusb/files/6397880/libusb-bug.txt)

mcuee, 2021-04-29:
Why do you think this is a libusb bug and not the application "analysis-dbg.exe"? 

Are you the author of the application? If yes please post the relevant codes.

kulkin-fish, 2021-04-29:
#include <libusb-1.0/libusb.h>

int main(int argc, char* argv[]){
  libusb_context* libusbContext = nullptr;
  if(libusb_init(&libusbContext) != LIBUSB_SUCCESS)
    return -1;

  libusb_device** devList = nullptr;
  const int devCount = libusb_get_device_list(libusbContext, &devList);
  if(devCount <= 0){
    libusb_exit(libusbContext);
    return 0;
  }

  for(int i = 0; i < devCount; ++i){
    libusb_device_descriptor desc;
    libusb_get_device_descriptor(devList[i], &desc);

    if(desc.idVendor != 0x0525)
      continue;

    libusb_device_handle* handle = nullptr;
    if(libusb_open(devList[i], &handle) != LIBUSB_SUCCESS)
      continue;

    libusb_claim_interface(handle, 0);
    libusb_close(handle);
  }
  libusb_free_device_list(devList, 1);
  libusb_exit(libusbContext);
  return 0;
}

Minimal compilable sample to reproduce the error. Compilation was done with 'gcc.exe main.cpp -lusb-1.0 -omain' command.
I am using gcc-9.3.0 and libusb-1.0.24. There was no error with libusb-1.0.23.
If comment out line 'libusb_claim_interface(handle, 0);' the error disappears. As you can see from debugger output the error occurs in a separate thread created by libusb_init(...). It does not looks like a bug in my program.
[image]

mcuee, 2021-05-01:
Did you install the supported driver?
https://github.com/libusb/libusb/wiki/Windows#Driver_Installation

kulkin-fish, 2021-05-01:
Driver libusbK is installed. As I mentioned above there is no problem with libusb-1.0.23.
[libusb-error-2.txt](https://github.com/libusb/libusb/files/6409272/libusb-error-2.txt)
[image]

mcuee, 2021-05-04:
Thanks for the updates. Please help to post the USB Descriptors. That may play a part.

You can use USBView utility or other utilities.
Microsoft USBView: 
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview
https://ftdichip.com/utilities/ (small download, FTDI version)

mcuee, 2021-05-04:
The other possibility, please switch to WinUSB driver and delete libusbk.dll from the system to see if this issue is applicable only to libusbk or it also applies to WinUSB. Thanks.

mcuee, 2021-05-04:
BTW, which MinGW compiler you are using? 

Do you use the prebuilt binaries, in that case, please take note that we do not support MinGW.org 32bit compilers. You have to use MinGW-w64 compilers (32bit and 64bit), and you may have to build the binary from source due to compatibilities between different compiler versions.

mcuee, 2021-05-18:
@kulkin-fish Please help to test git master to see if the issue goes away. Thanks.

kulkin-fish, 2021-05-18:
Indeed master from git has no such bug. But has another one :)
Looks like some deadlock.
[image]

mcuee, 2021-05-19:
Thanks for the updates. Did this issue happen with WinUSB driver?

kulkin-fish, 2021-05-19:
[image]

photron, 2021-05-19:
I tested your minimal example (but changed the VID to match my device) and cannot reproduce the deadlock. But I'm using WinUSB instead of libusbK.

According to your backtrace thread 1 is waiting for the IOCP thread (thread 5 in the backtrace) in the Windows backend to exit. But thread 5 is trying to lock a dev_handle and is blocked there. The question is now who is holding the dev_handle lock, or who has not unlocked the dev_handle lock.

Can you make your example more minimal, or does it have to open the device and claim the interface for the deadlock to occur?

Does the deadlock occur every time?

Could you provide a debug log again?

kulkin-fish, 2021-05-19:
Situation is stable. I can reproduce it every time running application. If comment 'libusb_claim_interface' the problem is not occured. Log is:
[image]

One more thing. With LIBUSB_DEBUG=4 my DrMinGW popuped with text:
[image]

I made some modifications to my source:
```
#include <libusb-1.0/libusb.h>

int main(int argc, char* argv[]){
  libusb_context* libusbContext = nullptr;
  if(libusb_init(&libusbContext) != LIBUSB_SUCCESS)
    return -1;

  libusb_device** devList = nullptr;
  const int devCount = libusb_get_device_list(libusbContext, &devList);
  if(devCount <= 0){
    libusb_exit(libusbContext);
    return 0;
  }

  for(int i = 0; i < devCount; ++i){
    libusb_device_descriptor desc;
    libusb_get_device_descriptor(devList[i], &desc);

    if(desc.idVendor != 0x0525)
      continue;

    libusb_device_handle* handle = nullptr;
    if(libusb_open(devList[i], &handle) != LIBUSB_SUCCESS)
      continue;

    if(argc > 1)
      libusb_claim_interface(handle, 0);
    libusb_close(handle);
  }
  libusb_free_device_list(devList, 1);
  libusb_exit(libusbContext);
  return 0;
}

```
Running with any parameter in command line causes libusb_claim_interface to be called. Without any command line parameter libusb_claim_interface not called. Running without parameters is OK every time (e.g. without caliming interface). Running with any parameter is deadlocked everytime.

kulkin-fish, 2021-05-19:
Even without LIBUSB_DEBUG i've got the message:
[image]

mcuee, 2021-05-20:
@kulkin-fish Please switch to WinUSB driver using Zadig, and see if the issue is still there or not.

kulkin-fish, 2021-05-22:
> @kulkin-fish Please switch from libusbk driver to WinUSB driver using Zadig, and see if the issue is still there or not.
> 
> Unless you hit into WinUSB driver limitations, you should use WinUSB driver and not libusbk.

Confirmed. Switch to WinUSB driver eliminates the problem. Does that mean libusb has no bugs with libusbK?

mcuee, 2021-05-24:
Thanks for the updates. So looks like there is an issue with libusbk backend.

Please help to post the debug log with WinUSB as well for comparison. Thanks.

kulkin-fish, 2021-05-28:
> Thanks for the updates. So looks like there is an issue with libusbk backend.
> 
> Please help to post the debug log with WinUSB as well for comparison. Thanks.

[image]

mcuee, 2021-06-04:
I can reproduce your issue under Windows 10 and MSYS2 MinGW-w64. So I can confirm there is an issue with libusbk.sys backend.

I have a USB device with firmware compatible with WinUSB device. 
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/automatic-installation-of-winusb

When I switch the driver from WinUSB to libusbk using Zadig.  And I run your program (just change the VID to 0x04d8 as it is from Microchip), it crashes just as what you reported. 

When I switch the driver back to WinUSB using Windows Device Manager, it will be okay.

mcuee, 2021-06-05:
@dickens Please take a look when you got the time. Thanks.

mcuee, 2021-06-08:
I am not so sure if it is the code or not, but this also happens under libusb-1.0.24 and current git. So this should not be a show stopper for 1.0.25.

mcuee, 2021-06-08:
@kulkin-fish Please help to try out 1.0.23 as well and this may help to narrow down the potential issue. Thanks.

mcuee, 2021-06-08:
Initially I was worried it is related to https://github.com/libusb/libusb/commit/b51c743e4210756a98f4f60c69a34745e4b27a55 because of #863 (fixed in https://github.com/libusb/libusb/commit/a108bb8e185cc5c41b296d35dfa83e5e5b13b2c0) but apparently not since 1.0.24 release already got the issue with libusbk.

mcuee, 2021-06-08:
This may be related to #844 as well which is fixed by commit https://github.com/libusb/libusb/commit/b51c743e4210756a98f4f60c69a34745e4b27a55

mcuee, 2021-06-08:
#844

mcuee, 2021-06-08:
I did some tests and it seems to me libusb git head already fixed the issue. Initially I was not testing correctly because I was still using the dll from MSYS2's libusb-1.0.dll which was 1.0.24. Now I just use libusb-1.0.dll built from the latest git and it seems me the issue is no longer there.

@kulkin-fish Please test the latest git.

mcuee, 2021-06-09:
On the other hand, xusb example runs fine with 1.0.24 and latest git. So maybe the code has some issues as well.

mcuee, 2021-06-09:
@kulkin-fish Maybe you want to take some references from the xusb example and other official libusb examples as well.

tormodvolden, 2021-06-09:
Does it crash if you release all claimed interfaces before libusb_exit() ?

mcuee, 2021-06-09:
BTW, I tested with an HID device and the test code does not crash. Device using WinUSB driver also does not crash. So it is very specific to libusbk driver (through libusbK.dll).

mcuee, 2021-06-09:
> Does it crash if you release all claimed interfaces before libusb_exit() ?

It seems to me libusbk backend may have an issue. I changed to the following code and it still crashes with the latest git head.

```
#include <libusb-1.0/libusb.h>

int main(int argc, char* argv[]){
  libusb_context* libusbContext = nullptr;
  if(libusb_init(&libusbContext) != LIBUSB_SUCCESS)
    return -1;

  libusb_device** devList = nullptr;
  const int devCount = libusb_get_device_list(libusbContext, &devList);
  if(devCount <= 0){
    libusb_exit(libusbContext);
    return 0;
  }

  for(int i = 0; i < devCount; ++i){
    libusb_device_descriptor desc;
    libusb_get_device_descriptor(devList[i], &desc);

    if(desc.idVendor != 0x0403)
      continue;

    libusb_device_handle* handle = nullptr;
    if(libusb_open(devList[i], &handle) != LIBUSB_SUCCESS)
      continue;

    if(argc > 1){
          libusb_claim_interface(handle, 0);
	  libusb_release_interface(handle, 0);
	  }
    libusb_close(handle);
  }
  libusb_free_device_list(devList, 1);
  libusb_exit(libusbContext);
  return 0;
}
```

mcuee, 2021-06-24:
Narrowing down the commits.

This is okay.
https://github.com/libusb/libusb/commit/9c28ad219b654011783a42ec888ca87dbda704a6#diff-bd7e9ed3e7d8a4a69d0664da0f6b8dbbff0b2c608d8dd63de02c379c98cf88b4

This is not okay.
https://github.com/libusb/libusb/commit/ba6b8bcb7ea204e65a3deec3be81aacc9f4b6d5a#diff-bd7e9ed3e7d8a4a69d0664da0f6b8dbbff0b2c608d8dd63de02c379c98cf88b4

mcuee, 2021-06-24:
Further testing:
This is okay: https://github.com/libusb/libusb/commit/4261cbefc716e49d459426593cef0104482ec43b#diff-bd7e9ed3e7d8a4a69d0664da0f6b8dbbff0b2c608d8dd63de02c379c98cf88b4

So the issue comes from https://github.com/libusb/libusb/commit/ba6b8bcb7ea204e65a3deec3be81aacc9f4b6d5a

mcuee, 2021-06-24:
History: https://github.com/libusb/libusb/commits/master/libusb/os/windows_winusb.c

```
https://github.com/libusb/libusb/commit/ba6b8bcb7ea204e65a3deec3be81aacc9f4b6d5a
Commits on Aug 11, 2020
Windows: Use I/O completion ports for transfers …

Windows: Use I/O completion ports for transfers
As a first step in removing the Windows poll() emulation, switch the
transfers to use an I/O completion port. A dedicated per-context thread
will wait on the I/O completion port and report transfer completions
using usbi_signal_transfer_completion(). This enables the complete
removal of the handle_events() function for the Windows backend and
removes the notion of one "file descriptor" per transfer.

```

mcuee, 2021-06-24:
It seems to commits https://github.com/libusb/libusb/commit/b51c743e4210756a98f4f60c69a34745e4b27a55 fixed some issues from https://github.com/libusb/libusb/commit/ba6b8bcb7ea204e65a3deec3be81aacc9f4b6d5a but may not be enough for libusbk yet.

mcuee, 2021-06-24:
I just apply a simple patch to testlibusb example and it seems to re-create the issue.

```

C:\work\libusb\libusb_test [master ≡ +0 ~1 -0 !]> git diff
diff --git a/examples/testlibusb.c b/examples/testlibusb.c
index ba00f90..f3e411f 100644
--- a/examples/testlibusb.c
+++ b/examples/testlibusb.c
@@ -203,6 +203,14 @@ static void print_device(libusb_device *dev, libusb_device_handle *handle)
                                printf("  Product:                   %s\n", (char *)string);
                }

+               if (desc.idVendor == 0x04d8 ) {
+                       ret = libusb_claim_interface(handle, 0);
+                       if (ret > 0)
+                               printf("  Successfully claimed interface for Microchip USB device \n");
+                       else
+                               printf("  Failed to claim interface for Microchip USB device \n");
+               }
+
                if (desc.iSerialNumber && verbose) {
                        ret = libusb_get_string_descriptor_ascii(handle, desc.iSerialNumber, string, sizeof(string));
                        if (ret > 0)
```

Run log:
```

C:\work\libusb\libusb_test [master ≡ +0 ~1 -0 !]> .\x64\Release\examples\testlibusb.exe
Dev (bus 3, device 5): 0424 - 5744 speed: 5G
Dev (bus 3, device 13): 046D - C077 speed: 1.5M
  Manufacturer:              Logitech
  Product:                   USB Optical Mouse
Dev (bus 3, device 6): 0BDA - 5413 speed: 480M
Dev (bus 3, device 3): 0BDA - 0413 speed: 5G
Dev (bus 3, device 7): 413C - B06E speed: 480M
Dev (bus 1, device 1): 046D - C534 speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 1, device 4): 8087 - 0AAA speed: 12M
Dev (bus 3, device 8): 0424 - 2744 speed: 480M
Dev (bus 3, device 2): 0BDA - 5487 speed: 480M
Dev (bus 3, device 1): 0BDA - 0487 speed: 10G
Dev (bus 3, device 24): 0424 - 2740 speed: 480M
  Manufacturer:              Microchip Tech
  Product:                   Hub Controller
Dev (bus 3, device 26): 047F - C056 speed: 12M
  Manufacturer:              Plantronics
  Product:                   Plantronics Blackwire 3220 Series
Dev (bus 3, device 25): 0BDA - 402E speed: 480M
Dev (bus 1, device 0): 8086 - A36D speed: 5G
Dev (bus 1, device 2): 0A5C - 5842 speed: 480M
Dev (bus 3, device 0): 8086 - 15F0 speed: 10G
Dev (bus 2, device 0): 8086 - 15DB speed: 5G
Dev (bus 3, device 12): 413C - 2107 speed: 1.5M
  Manufacturer:              DELL
  Product:                   Dell USB Entry Keyboard
Dev (bus 3, device 4): 0BDA - 8153 speed: 5G
Dev (bus 3, device 10): 413C - B06F speed: 480M
Dev (bus 3, device 23): 04D8 - FA2E speed: 12M
  Manufacturer:              Travis Robinson
  Product:                   Benchmark Device
  Failed to claim interface for Microchip USB device
Dev (bus 1, device 3): 0BDA - 58FD speed: 480M
```

mcuee, 2021-06-24:
Same thing for the above test code.

```
#include <stdio.h>
#include <libusb-1.0/libusb.h>

int main(int argc, char* argv[]){
  libusb_context* libusbContext = nullptr;
  if(libusb_init(&libusbContext) != LIBUSB_SUCCESS)
    return -1;

  libusb_device** devList = nullptr;
  const int devCount = libusb_get_device_list(libusbContext, &devList);
  if(devCount <= 0){
    libusb_exit(libusbContext);
    return 0;
  }

  for(int i = 0; i < devCount; ++i){
    libusb_device_descriptor desc;
    libusb_get_device_descriptor(devList[i], &desc);

    if(desc.idVendor != 0x04d8)
      continue;

    libusb_device_handle* handle = nullptr;
    if(libusb_open(devList[i], &handle) != LIBUSB_SUCCESS)
      continue;
	
	printf("Found device \n");
	
    if(argc > 1){
	  printf("Claiming interface \n");
      if (libusb_claim_interface(handle, 0)) {
		  printf("Successfully claimed interface \n");
	      printf("Releasing interface 0 \n");
	      if (libusb_release_interface(handle, 0)) 
		      printf("Successfully released interface \n");
		  else
			  printf("Failed to release interface \n");
	      }
	    printf("Failed to claim interface \n");
		}
    libusb_close(handle);
  }
  libusb_free_device_list(devList, 1);
  libusb_exit(libusbContext);
  return 0;
}
```
```

mcuee, 2021-06-25:
Interestingly I can not reproduce the anomaly with latest git any more.

```
$ ./main1.exe 1
Found device
Claiming interface
Successfully claimed interface
Releasing interface 0
Successfully released interface
```

Full debug log:
```

$ ./main1.exe 1
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000325] [00003ac8] libusb: debug [libusb_init] libusb v1.0.24.11630
[ 0.000502] [00003ac8] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000090 events 0
[ 0.000657] [00003ac8] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000816] [00003ac8] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000A0 events 0
[ 0.001025] [00003ac8] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.001190] [00003ac8] libusb: debug [htab_create] using 1021 entries hash table
[ 0.006000] [00003ac8] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.010621] [00003ac8] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.8.0
[ 0.014672] [00003ac8] libusb: info [windows_init] UsbDk backend is not available
[ 0.015016] [00003ac8] libusb: debug [libusb_get_device_list]
[ 0.016125] [000037c4] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.030006] [00003ac8] libusb: debug [winusb_get_device_list] allocating new device for session [87]
[ 0.030270] [00003ac8] libusb: debug [winusb_get_device_list] allocating new device for session [88]
[ 0.030596] [00003ac8] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.030813] [00003ac8] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.030978] [00003ac8] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.031307] [00003ac8] libusb: debug [get_api_type] driver(s): libusbK
[ 0.031508] [00003ac8] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.031675] [00003ac8] libusb: debug [winusb_get_device_list] allocating new device for session [13]
[ 0.032210] [00003ac8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.032365] [00003ac8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.032529] [00003ac8] libusb: debug [winusb_get_device_list] allocating new device for session [65]
[ 0.032829] [00003ac8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.033036] [00003ac8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.033194] [00003ac8] libusb: debug [winusb_get_device_list] allocating new device for session [66]
[ 0.033491] [00003ac8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.033684] [00003ac8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.033854] [00003ac8] libusb: debug [winusb_get_device_list] allocating new device for session [67]
[ 0.034154] [00003ac8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.034354] [00003ac8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.034504] [00003ac8] libusb: debug [winusb_get_device_list] allocating new device for session [68]
[ 0.034851] [00003ac8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A36D&SUBSYS_091A1028&REV_10\3&11583659&0&A0' bus number 1
[ 0.035009] [00003ac8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15DB&SUBSYS_091A1028&REV_02\71D9F51854B3020000' bus number 2
[ 0.036285] [00003ac8] libusb: debug [winusb_get_device_list] found existing device for session [67]
[ 0.036593] [00003ac8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.036747] [00003ac8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.036887] [00003ac8] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 1): 'USB\VID_046D&PID_C534\5&E9F3E45&0&1'
[ 0.037145] [00003ac8] libusb: debug [winusb_get_device_list] found existing device for session [5]
[ 0.037325] [00003ac8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.037446] [00003ac8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.037542] [00003ac8] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 14): 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
[ 0.038175] [00003ac8] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&36020D6F&0&0' reports 26 ports
[ 0.038369] [00003ac8] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&36020D6F&0&0'
[ 0.038668] [00003ac8] libusb: debug [winusb_get_device_list] found existing device for session [66]
[ 0.038883] [00003ac8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.038999] [00003ac8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 269 bytes)
[ 0.039126] [00003ac8] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 10): 'USB\VID_0A5C&PID_5842\0123456789ABCD'
[ 0.039471] [00003ac8] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\7&2452366F&0&0' reports 4 ports
[ 0.039603] [00003ac8] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\7&2452366F&0&0'
[ 0.039969] [00003ac8] libusb: debug [winusb_get_device_list] extra GUID: {26BEE760-0B93-8B57-199C-7BDA395646D1}
[ 0.040093] [00003ac8] libusb: debug [winusb_get_device_list] found existing device for session [13]
[... cut ...]
l[ 0.051342] [000037c4] libusb: debug [windows_iocp_thread] ignoring overlapped 000000A435DFF8B0 for handle 0000027F08E4A050 (device 1.10)
a[ 0.051429] [000037c4] libusb: debug [windows_iocp_thread] ignoring overlapped 000000A435DFF8B0 for handle 0000027F08E4A050 (device 1.10)
imed interface
Releasing interface 0
[ 0.053067] [00003ac8] libusb: debug [libusb_release_interface] interface 0
Successfully released interface
[ 0.054241] [00003ac8] libusb: debug [libusb_close]
[ 0.054310] [00003ac8] libusb: debug [libusb_get_device_descriptor]
[ 0.054365] [00003ac8] libusb: debug [libusb_get_device_descriptor]
```

mcuee, 2021-06-25:
However, if I go back to OP's test program, I will still be able to recreate the crash under MinGW.

```
#include <libusb-1.0/libusb.h>

int main(int argc, char* argv[]){
  libusb_context* libusbContext = nullptr;
  if(libusb_init(&libusbContext) != LIBUSB_SUCCESS)
    return -1;

  libusb_device** devList = nullptr;
  const int devCount = libusb_get_device_list(libusbContext, &devList);
  if(devCount <= 0){
    libusb_exit(libusbContext);
    return 0;
  }

  for(int i = 0; i < devCount; ++i){
    libusb_device_descriptor desc;
    libusb_get_device_descriptor(devList[i], &desc);

    if(desc.idVendor != 0x04d8)
      continue;

    libusb_device_handle* handle = nullptr;
    if(libusb_open(devList[i], &handle) != LIBUSB_SUCCESS)
      continue;

    if(argc > 1)
      libusb_claim_interface(handle, 0);
    libusb_close(handle);
  }
  libusb_free_device_list(devList, 1);
  libusb_exit(libusbContext);
  return 0;
}

$ ./main.exe 1
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000325] [000000d0] libusb: debug [libusb_init] libusb v1.0.24.11630
[ 0.000497] [000000d0] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000124 events 0
[ 0.000679] [000000d0] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000846] [000000d0] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000130 events 0
[ 0.001049] [000000d0] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.001223] [000000d0] libusb: debug [htab_create] using 1021 entries hash table
[ 0.004558] [000000d0] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.005563] [000000d0] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.8.0
[ 0.010269] [000000d0] libusb: info [windows_init] UsbDk backend is not available
[ 0.010576] [000000d0] libusb: debug [libusb_get_device_list]
[ 0.011719] [00004408] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.028619] [000000d0] libusb: debug [winusb_get_device_list] allocating new device for session [87]
[ 0.028930] [000000d0] libusb: debug [winusb_get_device_list] allocating new device for session [88]
[ 0.029307] [000000d0] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.029550] [000000d0] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.029735] [000000d0] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.030120] [000000d0] libusb: debug [get_api_type] driver(s): libusbK
[ 0.030340] [000000d0] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.030502] [000000d0] libusb: debug [winusb_get_device_list] allocating new device for session [13]
[ 0.031025] [000000d0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.031174] [000000d0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.031284] [000000d0] libusb: debug [winusb_get_device_list] allocating new device for session [65]
[ 0.031500] [000000d0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.031651] [000000d0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.031758] [000000d0] libusb: debug [winusb_get_device_list] allocating new device for session [66]
[ 0.031971] [000000d0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.032116] [000000d0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.032223] [000000d0] libusb: debug [winusb_get_device_list] allocating new device for session [67]
[ 0.032440] [000000d0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.032589] [000000d0] libusb: debug [get_api_type] matched driver name against Composite API
[... 61 lines cut ...]
[ 0.044189] [000000d0] libusb: debug [set_composite_interface] interface[3] already set - ignoring HID collection: HID\VID_047F&PID_C025&MI_03&COL02\7&147046C1&0&0001
[ 0.044712] [000000d0] libusb: debug [get_api_type] driver(s): libusbK
[ 0.044811] [000000d0] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.046698] [000000d0] libusb: debug [libusb_get_device_descriptor]
[ 0.046771] [000000d0] libusb: debug [libusb_get_device_descriptor]
[ 0.046827] [000000d0] libusb: debug [libusb_get_device_descriptor]
[ 0.046918] [000000d0] libusb: debug [libusb_get_device_descriptor]
[ 0.047013] [000000d0] libusb: debug [libusb_get_device_descriptor]
[ 0.047109] [000000d0] libusb: debug [libusb_get_device_descriptor]
[ 0.047168] [000000d0] libusb: debug [libusb_open] open 1.11
[ 0.047252] [000000d0] libusb: debug [libusb_claim_interface] interface 0
[ 0.047350] [000000d0] libusb: debug [winusbx_claim_interface] claimed interface 0
[ 0.047422] [000000d0] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 01 to interface 0
[ 0.047518] [000000d0] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[ 0.047686] [000000d0] libusb: debug [libusb_close]
[ 0.047723] [00004408] libusb: debug [windows_iocp_thread] ignoring overlapped 000000597FBFF880 for handle 00000172FBECE160 (device 1.11)
[ 0.047782] [000000d0] libusb: debug [libusb_get_device_descriptor]
[ 0.047889] [000000d0] libusb: debug [libusb_get_device_descriptor]
[ 0.047941] [000000d0] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.048010] [000000d0] libusb: debug [libusb_unref_device] destroy device 1.4
[ 0.048077] [000000d0] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.048140] [000000d0] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.048202] [000000d0] libusb: debug [libusb_unref_device] destroy device 1.11
Segmentation fault
```

dmitry-zakablukov, 2021-06-30:
@mcuee , @kulkin-fish , fixed: #943

mcuee, 2021-07-01:
@dmitry-zakablukov Great. I will test this fix.

mcuee, 2021-07-01:
@dmitry-zakablukov Yes I can confirm #943 fixed this issue. Thanks.

mcuee, 2021-07-09:
#943 has been merged, close this issue.

closed 2021-07-09 by mcuee