libusb/libusb #913: libusb 1.0.24 uses a lot of cpu in libusb_bulk_transfer under Linux

r4nd0m6uy, 2021-05-11:
I am maintaining a BSP with Yocto on a ARM Linux device. After upgrading Yocto that uses libusb 1.0.24, I noticed that one application uses a lot of CPU.

I'm not a maintainer of the given application (xow) but after some investigation, the issue seems to lie in the _libusb_bulk_transfer_ call:
https://github.com/medusalix/xow/blob/master/dongle/usb.cpp#L111

Downgrading to libusb 1.0.23 fixed the CPU usage problem.

It seems a lot of work was done regarding file descriptor handling, I'm wondering if something went wrong here:
https://github.com/libusb/libusb/commit/d67eb5beaa44c17c09f089a522ff483f4978a978

* Is that a known issue?
* Is there _autoconf_ flags I am missing?
* Is there a workaround?

Best regards,

Guy

mcuee, 2021-05-17:
Not a known issue. Please post the debug log by running the application with environment variable LIBUSB_DEBUG set as 4.

mcuee, 2021-05-17:
You may also want to try the latest git which fixes some critical issues for Linux.

r4nd0m6uy, 2021-05-17:
Alright, thanks for the inputs, I will let you know

r4nd0m6uy, 2021-05-17:
So I did some tests

> You may also want to try the latest git which fixes some critical issues for Linux.

The latest git (https://github.com/libusb/libusb/commit/35e8ee98a36143da4d226fc99e95673cea188bb0) shows the same issue.

> You may also want to try the latest git which fixes some critical issues for Linux.

Version 1.0.23  (working) has only few outputs, most interesting part is the bulk_transfer:

```
[ 3.785735] [000002c3] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 512
[ 3.785775] [000002c3] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 3.885697] [000002c2] libusb: debug [handle_events] poll() returned 1
[ 3.885793] [000002c2] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=96
[ 3.885822] [000002c2] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 3.885845] [000002c2] libusb: debug [handle_bulk_completion] all URBs in transfer reaped --> complete!
[ 3.885870] [000002c2] libusb: debug [usbi_handle_transfer_completion] transfer 0x75100b60 has callback 0x76f00605
[ 3.885892] [000002c2] libusb: debug [sync_transfer_cb] actual_length=96
[ 3.885937] [000002c2] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.885962] [000002c2] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.885963] [000002c3] libusb: debug [libusb_free_transfer] transfer 0x75100b60
[ 3.886045] [000002c3] libusb: debug [libusb_alloc_transfer] transfer 0x75100b60
[ 3.886070] [000002c3] libusb: debug [libusb_submit_transfer] transfer 0x75100b60
```

On version 1.0.24 and git, it explodes the log (log file is 10x bigger). I have the feeling it is doing polling intensively:
```
[ 4.386872] [000002c9] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 512
[ 4.386933] [000002c9] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 4.386698] [000002c8] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 4.386999] [000002c8] libusb: debug [handle_event_trigger] event triggered
[ 4.387056] [000002c9] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 4.387079] [000002c9] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 4.387106] [000002c9] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 4.387123] [000002c9] libusb: debug [handle_event_trigger] event triggered
[ 4.387141] [000002c9] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 4.387157] [000002c9] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 4.387177] [000002c9] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 4.387194] [000002c9] libusb: debug [handle_event_trigger] event triggered
[ 4.387211] [000002c9] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 4.387228] [000002c9] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 4.387248] [000002c9] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 4.387265] [000002c9] libusb: debug [handle_event_trigger] event triggered
/* Same repeats a lot */
```

On git, logs are pretty similar.

[Complete logs are available here ](https://r4nd0m6uy.ch/documents/libusb-logs.tar.gz). I would ignore startup code and concentrate on normal operation (from 3 seconds).

If you need any more information, I would be glad to provide them!

r4nd0m6uy, 2021-05-17:
Not sure if that is relevant but the application handles some UNIX signals (SIGINT, SIGTERM, SIGUSR1)

mcuee, 2021-05-20:
Are you able to reproduce the issue at an x86/Amd64 Linux machine? Just want to know if the issue is specific to your Arm Linux build or it is a generic issue.

r4nd0m6uy, 2021-05-20:
> Are you able to reproduce the issue at an x86/Amd64 Linux machine? Just want to know if the issue is specific to your Arm Linux build or it is a generic issue.

It might take me some time but I will let you know

r4nd0m6uy, 2021-05-20:
(Un)fortunately, the behavior is the same on X86_64 Linux machine :disappointed: ...

[You can find the logs here](https://r4nd0m6uy.ch/documents/libusb-logs-x86_64.tar.gz). Since my machine is much faster,  be aware that the logs of the git version exploded to 700MB non compressed

mcuee, 2021-05-21:
https://github.com/libusb/libusb/pull/856

Please try out pull-request #856. It should fix this issue. It is pending to be merged for 1.0.25 release.

r4nd0m6uy, 2021-05-21:
Wonderful, this fixes the problem for me! Thanks a lot, waiting for 1.0.25 release, closing the issue now.

closed 2021-05-21 by r4nd0m6uy