libusb/libusb #917: `libusb_wait_for_event` does not wake after transfer completion, only after `libusb_unlock_events`

kevinmehall, 2021-05-23:
I was trying to implement something resembling `libusb_handle_events_completed` using the low-level APIs following the [example in the documentation](https://libusb.sourceforge.io/api-1.0/libusb_mtasync.html#threadwait).

Simplified and translated from Rust back to C:
```c
while (!atomic_load(completed)) {
    if (libusb_try_lock_events(ctx) == 0) {
        while (!atomic_load(completed) && libusb_event_handling_ok(ctx)) {
            libusb_handle_events_locked(ctx, null);
        }
        libusb_unlock_events(ctx);
    } else {
        libusb_lock_event_waiters(ctx);
        while (!atomic_load(completed) && libusb_event_handler_active(ctx)) {
            libusb_wait_for_event(ctx, null);
        }
        libusb_unlock_event_waiters(ctx);
    }
}
```

When this runs on two threads waiting for two different transfers (with two different `completed` flags), I see one thread stuck in `libusb_wait_for_event`, even when the transfer completes and its callback is called on the other thread. It only wakes when the other thread eventually exits the loop.

Looking at the [implementation of `libusb_wait_for_event`](https://github.com/libusb/libusb/blob/35e8ee98a36143da4d226fc99e95673cea188bb0/libusb/io.c#L1996), I see that it waits on a condvar `event_waiters_cond`. I see that [`libusb_unlock_events` calls `usbi_cond_broadcast(&ctx->event_waiters_cond)`](https://github.com/libusb/libusb/blob/35e8ee98a36143da4d226fc99e95673cea188bb0/libusb/io.c#L1834), but this appears to be the only other reference to `event_waiters_cond` outside of initialization and finalization, and even the only call to `usbi_cond_broadcast` in the entire codebase.

I noticed that unlike the example and my code, the implementation of `libusb_handle_events_timeout_completed` does not loop with the event lock held. My code works properly after changing the second `while` to an `if` so the lock is released after every call to `libusb_handle_events_locked`.

It seems that either:

1. The documentation is wrong: `libusb_wait_for_event` is only intended to wait for `libusb_unlock_events` and not transfer completions, and it should be documented that you are not allowed to hold the events lock across multiple calls to `libusb_handle_events_locked` The example code should be updated accordingly.
2. The implementation is wrong: there should be another call to `usbi_cond_broadcast(&ctx->event_waiters_cond)` somewhere when a transfer completes.

Am I missing something?

kevinmehall, 2021-05-25:
It seems the missing call to `usbi_cond_broadcast(&ctx->event_waiters_cond)` was removed in a3a4806d494b8ff3dd213d5eb408b657bcdd47f0:
> usbi_handle_transfer_completion() is only called by backends when
    handling transfer completion. The backend can only make this call
    whilst holding the events lock, therefore it is pointless to send
    this signal. Any threads waiting to be signaled will be woken up
    and will either find that an event handler is still active or will
    try to obtain the events lock and be blocked.

This doesn't account for the third possibility that the woken thread notices that the transfer it's waiting for has completed, and goes on to do something else with that transfer instead of continuing to wait for events.

mcuee, 2023-03-31:
@tormodvolden and @hjelmn 

Please take a look at this to see if this is just a documentation issue or there is a real bug here. Thanks.