libusb/libusb #918: Problems with libusb and USB Ethernet extenders

hperaza, 2021-05-27:
libusb is unable to list and open devices connected via a USB-Ethernet extender. Specifically, I'm having problems with an EXSYS EX-6002 extender: Windows recognizes the extender, as well as the remote device, and so do other programs (e.g. UsbDeview, printer control programs, etc.). This is what libusb gives:

libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'USB\ROOT_HUB_GENERIC&REV_0201\C21B1C1100000000'
libusb: warning [init_device] ancestor for device 'USB\VID_04D8&PID_F85D\01202020' not found at depth 1
libusb: warning [winusb_get_device_list] failed to initialize device 'USB\VID_04D8&PID_F85D\01202020'

The device I'm trying to access is the 04D8/F85D one. Note that is gets detected, and that the hub also gets detected but not included in the enumeration list because libusb is expecting a path with a fixed format (PCI\\VEN_%04hx&DEV_%04hx%*s according to the code). Since the hub is not included in the list, libusb cannot include the remote device either even when it detects it.

I believe a more robust scheme should be used by libusb (e.g. saving the path string instead of just the VID/PID pair and test children nodes against it). Moreover, since the hub is merely virtual and not a real USB device, libusb cannot expect it to have a valid VID/PID (USB-IF may even deny registering a VID for the same reason).

mcuee, 2021-05-27:
Please post the detail debug log and I am not so sure if the above warning is the real reason.

Please also refer to the past merged patch for another virtual USB device, not so sure if that patch helps you or not. Basically you need to add the list of the virtual hub driver in libusb Windows backend codes first, and then see if there are any things else which needs to be done.
https://github.com/libusb/libusb/pull/257

hperaza, 2021-05-28:
Please find the log below. Line @[0.186648] complains about the ancestor not being found for the hub. Windows lists the ancestor as \HTREE\ROOT\0

[ 0.000593] [00007b70] libusb: debug [libusb_init] created default context
[ 0.000810] [00007b70] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.002082] [00007b70] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
[ 0.003032] [00007b70] libusb: debug [htab_create] using 1021 entries hash table

[ 0.016154] [00007b70] libusb: info [winusbx_init] WinUSB DLL available (without isoch support)
[ 0.016943] [00007b70] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.019196] [00007b70] libusb: info [windows_init] UsbDk backend is not available
[ 0.019656] [00007b70] libusb: debug [usbi_add_event_source] add HANDLE 00000034 events 0
[ 0.020978] [00007b70] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.021087] [00009030] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.022554] [00007b70] libusb: debug [usbi_add_event_source] add HANDLE 0000012C events 0
[ 0.023271] [00007b70] libusb: debug [libusb_get_device_list]
[ 0.153888] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [DA8]
[ 0.154545] [00007b70] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'NUSB3'
[ 0.155396] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [2E70]
[ 0.156552] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [2FB4]
[ 0.157481] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [2FF0]
[ 0.158490] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [302C]
[ 0.159442] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [307C]
[ 0.161194] [00007b70] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.161912] [00007b70] libusb: debug [get_api_type] lower filter driver(s): bcbtums;btwampfl
[ 0.162610] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [828]
[ 0.163416] [00007b70] libusb: debug [get_api_type] driver(s): DisplayLinkUsbIo_x64
[ 0.163623] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [1540]
[ 0.164142] [00007b70] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.164447] [00007b70] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.164792] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [1868]
[ 0.165503] [00007b70] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.165650] [00007b70] libusb: debug [get_api_type] matched driver name against HID API
[ 0.166080] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [2680]
[ 0.166679] [00007b70] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.167001] [00007b70] libusb: debug [get_api_type] matched driver name against HID API
[ 0.167369] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [26CC]
[ 0.168073] [00007b70] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.168378] [00007b70] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.169895] [00007b70] libusb: debug [winusb_get_device_list] allocating new device for session [2718]
[ 0.170547] [00007b70] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_1033&DEV_0194&SUBSYS_04A31028&REV_04\4&869802D&0&00E3' bus number 1
[ 0.170935] [00007b70] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1C26&SUBSYS_04A31028&REV_04\3&11583659&0&E8' bus number 2
[ 0.171331] [00007b70] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1C2D&SUBSYS_04A31028&REV_04\3&11583659&0&D0' bus number 3
[ 0.171825] [00007b70] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'USB\ROOT_HUB_GENERIC&REV_0201\C21B1C1100000000' bus number 4
[ 0.172167] [00007b70] libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'USB\ROOT_HUB_GENERIC&REV_0201\C21B1C1100000000'
[ 0.185146] [00007b70] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&33127676&0' reports 3 ports
[ 0.185363] [00007b70] libusb: debug [init_device] (bus: 3, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&33127676&0'
[ 0.186116] [00007b70] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&D94D0D7&0' reports 3 ports
[ 0.186316] [00007b70] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&D94D0D7&0'
[ 0.186648] [00007b70] libusb: debug [winusb_get_device_list] unlisted ancestor for 'USB\ROOT_HUB_GENERIC&REV_0201\C21B1C1100000000' (non USB HID, newly connected, etc.) - ignoring
[ 0.187165] [00007b70] libusb: debug [winusb_get_device_list] found existing device for session [2680]
[ 0.187589] [00007b70] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.187971] [00007b70] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.188428] [00007b70] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 4): 'USB\VID_046D&PID_C062\6&DD7E971&0&4'
[ 0.189003] [00007b70] libusb: debug [winusb_get_device_list] extra GUID: {6FE858A9-E07D-4FC2-A1F5-F25D33748EE3}
[ 0.189389] [00007b70] libusb: debug [winusb_get_device_list] found existing device for session [1868]
[... cut ...]
[ 1.314028] [00007b70] libusb: debug [libusb_unref_device] destroy device 1.0
[ 1.314448] [00007b70] libusb: debug [libusb_exit]
[ 1.314921] [00007b70] libusb: debug [libusb_exit] destroying default context
[ 1.314998] [00007b70] libusb: warning [libusb_exit] some libusb_devices were leaked
[ 1.315436] [00007b70] libusb: debug [usbi_remove_event_source] remove HANDLE 0000012C
[ 1.315943] [00007b70] libusb: debug [usbi_remove_event_source] remove HANDLE 00000034
[ 1.316366] [00009030] libusb: debug [windows_iocp_thread] I/O completion thread exiting

mcuee, 2021-05-31:
What is the name of the driver for the virtual USB hub? That may help to sort out the issue.

hperaza, 2021-05-31:
I believe is "vpcusb".

mcuee, 2021-06-05:
@hperaza 
Thanks for the info. The codes for WinUSB backends have been changed so that approach in #257 will no longer work (by adding "vpcusb" to the hub_driver_names() list).

I think you have provided enough info. Hopefully when Chris got the time to look at this he will come out with a solution.

@dickens Please take a look when you got the time. Thanks.

mcuee, 2021-06-07:
@hperaza You are welcome to create a pull request based on your idea of the robust scheme or just ignore the error on VID/PID of the parent of the root hub.

dimgit, 2021-07-29:
A better naming scheme will not fix the issue, for all extenders.
I am using Sylex DS-600 and it does not work with libusb (1.0.24). I was hoping it is just a naming issue - yes, the Sylex driver uses some non-standard path, and libusb complained that it cannot determine the VID/PID, but that was not the reason why it removed my device from the list of detected devices (libusb_get_device_list).
After debugging the code, I discovered that the Sylex driver does not respond correctly to the IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX that the WinUSB backend sends, as part of the device initialization, during enumeration. Sylex does nothing when that IO req is sent - no error is returned, and the connection info structure, passed to the IO request, is simply not touched at all.

I need libusb to work with Sylex, at the moment, and the only way I am able to resolve this right now is to detect if the hub, the device is connected to, is Sylex and if Yes, then retrieve the device descriptor using IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION (which thankfully works with Sylex) and initialize the device with that.
Maybe the Windows backend should not give up right away if IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX returns garbage (or nothing), or fails, but instead try one last resort with IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION?

mcuee, 2021-07-30:
@dimgit Thanks for the analysis. Have you contacted Silex and ask them to fix their driver to support IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX? It seems that is the best solution.

dimgit, 2021-07-30:
@mcuee Yes, I contacted Silex. Waiting for response.

mcuee, 2022-04-01:
SInce this is more an issue with Silex, I will close this. Please reopen if you have new info. Thanks.

closed 2022-04-01 by mcuee