libusb/libusb #919: SANE scanning issues on Big Sur

skelband, 2021-05-27:
At the SANE project we have been having a number of reports of MacOS users unable to scan using USB devices.
This seems to be primarily related to Big Sur but we cannot be sure that it is isolated to it.

We requested a diag log which is attached.

The log indicates that 1.0.24 is in use.
The salient lines seem to be here:
```
...
[ 0.562806] [00055a80] libusb: debug [libusb_alloc_transfer] transfer 0x116604258
[ 0.562820] [00055a80] libusb: debug [libusb_submit_transfer] transfer 0x116604258
[ 0.562833] [00055a80] libusb: debug [ep_to_pipeRef] converting ep address 0x88 to pipeRef and interface
[ 0.562837] [00055a80] libusb: debug [ep_to_pipeRef] pipe 2 on interface 0 matches
[ 0.562856] [00055a80] libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): no connection to an IOService (code = 0xe00002c0)
[ 0.562861] [00055a80] libusb: debug [libusb_free_transfer] transfer 0x116604258
[ 0.562871] [00055a80] libusb: debug [libusb_clear_halt] endpoint 88
...
```
Related issue at the SANE project is here: https://gitlab.com/sane-project/backends/-/issues/473

[scanimage-macOS-11.3-arm64.log](https://github.com/libusb/libusb/files/6555655/scanimage-macOS-11.3-arm64.log)

Any assistance would be really appreciated. Further information can be supplied.

mcuee, 2021-05-31:
Have you tried the latest git head?

onedrawingperday, 2021-05-31:
Hello @mcuee 

I am one of the users who reported the problem at the SANE project and contributed the above log.

Unfortunately installing libusb with the latest git HEAD does not solve the problem on my end.

For reference I executed the following commands and got the same I/O error:
```
brew uninstall --ignore-dependencies libusb
brew install --HEAD -s libusb 
...
==> ./configure --prefix=/opt/homebrew/Cellar/libusb/HEAD-35e8ee9
==> make install
...
scanimage -d 'pixma:04A91901' --verbose --resolution 150 --output-file output.png
scanimage: scanning image of size 1276x1754 pixels at 24 bits/pixel
scanimage: acquiring RGB frame
scanimage: min/max graylevel value = 255/0
scanimage: sane_read: Error during device I/O
```

onedrawingperday, 2021-05-31:
Furthermore I would like to add that I am seeing this problem not only on Big Sur with a MacBook Pro 2020 arm64 but also on Catalina with a mid-2012 MacBook Pro.

In both cases the scanner is recognised by SANE but when trying to scan, libusb throws the aforementioned error.

Also I tried scanning with both the original Canoscan 8800F USB 2.0 cable along with the Apple USB-C adapter, as well as another cable (without the Apple adapter) USB-C to USB 2.0.

In both cases the error persists.

The scanner also works fine with SANE, simplescan in a box running Xubuntu 20.04 LTS.

The scanner also works fine on MacOS with the proprietary VueScan software on both Big Sur and Catalina. 

However as I still able to scan on Linux, I am unwilling to pay the hefty price tag for VueScan.

mcuee, 2021-06-02:
Looks like bulk transfer failed based on the debug log. 

```

[ 0.562806] [00055a80] libusb: debug [libusb_alloc_transfer] transfer 0x116604258
[ 0.562820] [00055a80] libusb: debug [libusb_submit_transfer] transfer 0x116604258
[ 0.562833] [00055a80] libusb: debug [ep_to_pipeRef] converting ep address 0x88 to pipeRef and interface
[ 0.562837] [00055a80] libusb: debug [ep_to_pipeRef] pipe 2 on interface 0 matches
[ 0.562856] [00055a80] libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): no connection to an IOService (code = 0xe00002c0)
[ 0.562861] [00055a80] libusb: debug [libusb_free_transfer] transfer 0x116604258
[ 0.562871] [00055a80] libusb: debug [libusb_clear_halt] endpoint 88
[ 0.562875] [00055a80] libusb: debug [ep_to_pipeRef] converting ep address 0x88 to pipeRef and interface
[ 0.562878] [00055a80] libusb: debug [ep_to_pipeRef] pipe 2 on interface 0 matches
[ 0.562882] [00055a80] libusb: warning [darwin_clear_halt] ClearPipeStall: no connection to an IOService
[ 0.562893] [00055a80] libusb: debug [libusb_alloc_transfer] transfer 0x116604208
[ 0.562896] [00055a80] libusb: debug [libusb_submit_transfer] transfer 0x116604208
[ 0.562899] [00055a80] libusb: debug [ep_to_pipeRef] converting ep address 0x07 to pipeRef and interface
[ 0.562902] [00055a80] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 0.562904] [00055a80] libusb: error [submit_bulk_transfer] bulk transfer failed (dir = Out): no connection to an IOService (code = 0xe00002c0)
[ 0.562908] [00055a80] libusb: debug [libusb_free_transfer] transfer 0x116604208
[ 0.562912] [00055a80] libusb: debug [libusb_clear_halt] endpoint 7
[ 0.562915] [00055a80] libusb: debug [ep_to_pipeRef] converting ep address 0x07 to pipeRef and interface
[ 0.562918] [00055a80] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 0.562920] [00055a80] libusb: warning [darwin_clear_halt] ClearPipeStall: no connection to an IOService
[ 0.562925] [00055a80] libusb: debug [libusb_alloc_transfer] transfer 0x116604208
[ 0.562927] [00055a80] libusb: debug [libusb_submit_transfer] transfer 0x116604208
[ 0.562930] [00055a80] libusb: debug [ep_to_pipeRef] converting ep address 0x07 to pipeRef and interface
[ 0.562933] [00055a80] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 0.562936] [00055a80] libusb: error [submit_bulk_transfer] bulk transfer failed (dir = Out): no connection to an IOService (code = 0xe00002c0)
[ 0.562939] [00055a80] libusb: debug [libusb_free_transfer] transfer 0x116604208
[ 0.562942] [00055a80] libusb: debug [libusb_clear_halt] endpoint 7
[ 0.562945] [00055a80] libusb: debug [ep_to_pipeRef] converting ep address 0x07 to pipeRef and interface
[ 0.562947] [00055a80] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 0.562950] [00055a80] libusb: warning [darwin_clear_halt] ClearPipeStall: no connection to an IOService
scanimage: min/max graylevel value = 255/0
scanimage: sane_read: Error during device I/O
```

mcuee, 2021-06-02:
Just want to check if it is the changes of libusb or the changes of Catalina / Big Sur which caused the problem, please help to try out the following.
1) Does the issue happen on pre-Catalina release?
2) Does libusb 1.0.23 or 1.0.22 work under Catalina / Big Sur?

mcuee, 2021-06-02:
A more brutal force method will be using the USB Analyzer to compare the communication log when using VueScan and Sane.

onedrawingperday, 2021-06-02:
@mcuee 

Capturing USB traffic on macOS would [require disabling system integrity](https://developer.apple.com/forums/thread/124875). I am afraid that I am unwilling to do that. 

Regarding your other questions:

> 1. Does the issue happen on pre-Catalina release?
> 2. Does libusb 1.0.23 or 1.0.22 work under Catalina / Big Sur?

1. I cannot find out the answer to this question as it would entail downgrading the mid-2012 MacBook and that is just not possible.
2.  I am afraid that I cannot test the previous libusb releases. As per [this comment](https://github.com/mvp/uhubctl/issues/282#issuecomment-730130425) v.1.0.23 has a bug on Apple Silicon MacOS, v.1.0.22 is for Intel Macs.

Perhaps @dvkch can also look into your questions (he is the other user who submitted the original issue at the SANE backends repo on GitLab).

mcuee, 2021-06-03:
Actually I was thinking that some user may have access to USB HW Analyzer. So I was saying "brutal force" method. I do not expect many users to have that. But who knows, maybe one of the users have it.

One popular HW USB analyzer which works on macOS (and Linux and Windows).
https://www.totalphase.com/products/beagle-usb480/

mcuee, 2021-06-03:
And yes libusb 1.0.23 has a bug for macOS.
As for  libusb1.0.22, I have not tested it but 1.0.22 may work for Apple Silicon Macs.

mcuee, 2021-06-03:
The error codes seems to indicate the Device is lost during the transfer.

https://github.com/libusb/libusb/blob/master/libusb/os/darwin_usb.c#L102
```
  case kIOReturnNoDevice:
    return "no connection to an IOService";
```

mcuee, 2021-06-03:
For this type of low level error, the only way I can think of is to capture the USB traffic, compare the USB traffic between Sane under macOS and Sane under Linux, or Sane under macOS with VueScan under macOS.

dvkch, 2021-06-04:
I already have SIP disabled on macOS, here is the packets I have been able to capture using Wireshark. Systems : 

- macOS BigSur 11.3, sane 1.0.32, libusb 1.0.24
- Ubuntu 21.04, kernel 5.11.0-16-generic running in VMware Fusion in BigSur 11.3, sane 1.0.32

The command executed was `scanimage -d pixma:04A9173B_610308 > test.jpg`

[sane captures.zip](https://github.com/libusb/libusb/files/6600876/sane.captures.zip)

onedrawingperday, 2021-06-04:
I am posting the following workaround just in case someone else needs to scan on an M1 Mac with an old scanner *today*.

Install [UTM](https://github.com/utmapp/UTM/) version [v2.1.1](https://github.com/utmapp/UTM/releases/tag/v2.1.1) (that comes with USB support) and create a Ubuntu VM.  It runs fine. If anyone needs further instructions let me know.

Of course resolving this issue in `libusb` will make scanning on Big Sur easier.

mcuee, 2021-06-05:
Nice tip to use UTM 2.1.1. 
Now I can use a Linux VM easily with my Mac Mini M1 to carry out tests for libusb and other USB related open source projects.

mcuee, 2021-06-05:
@dvkch Thanks for the capture. At lease from the macOS capture, it is much shorter but the USB transfer status retured are all successful and nothing has a failed return status (libusb log shows kIOReturnNoDevice). I am not so sure if it is because the limit of the USB capture under macOS or not.

mcuee, 2021-06-05:
[image]

macOS capture.

mcuee, 2021-06-05:
Linux capture sequences are a bit different, no CONTROL IN transfer but got INTERUPT IN transfer.
[image]

mcuee, 2021-06-05:
I do have a Brother scanner but I under macOS I use the Brother iPrint&Sacn app and I do not use USB but rather wireless.

Anyway, I just connect through USB but I can not get Sane to recognize my Brother DCP-L2540DW. It seems to me my scanner is not supported by Sane.

```
~ ❯ scanimage -V
scanimage (sane-backends) 1.0.32; backend version 1.0.32

~ ❯ scanimage -L

No scanners were identified. If you were expecting something different,
check that the scanner is plugged in, turned on and detected by the
sane-find-scanner tool (if appropriate). Please read the documentation
which came with this software (README, FAQ, manpages).

~ took 20s ❯ sane-find-scanner

  # sane-find-scanner will now attempt to detect your scanner. If the
  # result is different from what you expected, first make sure your
  # scanner is powered up and properly connected to your computer.

  # No SCSI scanners found. If you expected something different, make sure that
  # you have loaded a kernel SCSI driver for your SCSI adapter.

found USB scanner (vendor=0x04f9 [Brother], product=0x0328 [DCP-L2540DW]) at libusb:002:016

  # Your USB scanner was (probably) detected. It may or may not be supported by
  # SANE. Try scanimage -L and read the backend's manpage.

  # Not checking for parallel port scanners.

  # Most Scanners connected to the parallel port or other proprietary ports
  # can't be detected by this program.

  # You may want to run this program as root to find all devices. Once you
  # found the scanner devices, be sure to adjust access permissions as
  # necessary.

```

mcuee, 2021-06-05:
BTW, I just took a very quick glance of the Sane usb code here.
https://gitlab.com/sane-project/backends/-/blob/master/sanei/sanei_usb.c

I am not so sure if the code here is a potential problem for macOS or not. They are potential dangerous for macOS, like clear_halt and set_alt_interace().
https://gitlab.com/sane-project/backends/-/blob/master/sanei/sanei_usb.c#L3131

mcuee, 2021-07-14:
Just an update from libusb side, #852 has also been merged.

From SANE side, they are looking at the concerns I raised.
https://gitlab.com/sane-project/backends/-/issues/473

tormodvolden, 2022-02-01:
Please retest with 1.0.25. It is already in Homebrew, and also macports have the equivalent in libusb-devel:
https://ports.macports.org/port/libusb-devel/details/

tenzap, 2022-02-01:
I already tested with 1.0.25 rc1. Same issue

mcuee, 2022-02-02:
> I already tested with 1.0.25 rc1 macports libusb-devel. Same issue

Comment out the following workaround to see it helps a bit or not.

https://gitlab.com/sane-project/backends/-/blob/master/sanei/sanei_usb.c#L3131
```
#elif defined(HAVE_LIBUSB)
  int ret;

  /* This call seems to be required by Linux xhci driver
   * even though it should be a no-op. Without it, the
   * host or driver does not send the clear to the device.
   * We intentionally ignore the return val */
  if (workaround)
    {
      sanei_usb_set_altinterface (dn, devices[dn].alt_setting);
    }

  ret = libusb_clear_halt (devices[dn].lu_handle, devices[dn].bulk_in_ep);
  if (ret){
    DBG (1, "sanei_usb_clear_halt: BULK_IN ret=%d\n", ret);
    return SANE_STATUS_INVAL;
  }

  ret = libusb_clear_halt (devices[dn].lu_handle, devices[dn].bulk_out_ep);
  if (ret){
    DBG (1, "sanei_usb_clear_halt: BULK_OUT ret=%d\n", ret);
    return SANE_STATUS_INVAL;
  }
#else /* not HAVE_LIBUSB_LEGACY && not HAVE_LIBUSB */
  DBG (1, "sanei_usb_clear_halt: libusb support missing\n");
#endif /* HAVE_LIBUSB_LEGACY || HAVE_LIBUSB */

```

diabolo38, 2022-02-02:
from the log it's seam an ep  pipe operation to a non existing or opened  pipe not a "stall" that would result from halt not cleared ?
The workaround is actuality doing a set  interface,  so could it be a set interface somewhere missing or active ep list refresh/broken in the back end ?

tormodvolden, 2022-02-02:
The variable `workaround` is set if the environment variable `SANE_USB_WORKAROUND` is set to `1`. Has this been tried?

tormodvolden, 2022-02-02:
From where in `usbDev_Prepare()` (the log says usbDev_PrepareScan but that is a typo) are the bulk transfers coming?

Looking at our Darwin backend, the error comes from GetPipePropertiesV3() before the bulk transfer is actually submitted. I am not sure if it is relevant here, but we are not using GetPipePropertiesV3() consistently, but still the now deprecated GetPipeProperties() in some places, in particular in get_endpoints(). Maybe the deprecated one has become broken in recent macOS.

mcuee, 2022-02-03:
> The variable `workaround` is set if the environment variable `SANE_USB_WORKAROUND` is set to `1`. Has this been tried? @mcuee Did you mean commenting out the libusb_clear_halt() calls as well?

Yes that is what I mean. To comment out libusb_clear_halt() as well.

mcuee, 2022-02-03:
> from the log it's seam an ep pipe operation to a non existing or opened pipe not a "stall" that would result from halt not cleared ? The workaround is actuality doing a set interface, so could it be a set interface somewhere missing or active ep list refresh/broken in the back end ?

Good point.

@tenzap Please post the USB descriptors. You can use USB Porber or the real lsusb under macOS.
https://github.com/Homebrew/homebrew-core/pull/89341

tenzap, 2022-02-03:
@mcuee here is the output of `lsusb -v` (+ I attatched output with all devices). This is made with macports' lsusb version (v007)

[lsusb.out.txt.gz](https://github.com/libusb/libusb/files/7992515/lsusb.out.txt.gz)

```
Bus 006 Device 002: ID 04a9:2220 Canon, Inc. CanoScan LIDE 25
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol       255 
  bMaxPacketSize0         8
  idVendor           0x04a9 Canon, Inc.
  idProduct          0x2220 CanoScan LIDE 25
  bcdDevice            1.00
  iManufacturer           1 Canon
  iProduct                2 CanoScan
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol    255 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              16
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

Use of SANE_USB_WORKAROUND=1 doesn't change the issue
[out.txt.gz](https://github.com/libusb/libusb/files/7992605/out.txt.gz)

I didn't test with removal of the piece of code in sane backends you mention because it is already in sane-backends 1.0.27, and 1.0.27 works fine.
https://gitlab.com/sane-project/backends/-/blob/1.0.27/sanei/sanei_usb.c#L2230

tormodvolden, 2022-02-03:
It might be good to include `SANE_DEBUG_SANEI_USB=50` in the command, it can give some context to the libusb calls.

I notice "reader_process started (forked)", I guess on Linux it is using threads instead?

Anyway, this bug was opened with the suspicion that there was something with Big Sur and libusb. Now it seems clear that it boils down to changes in sane between 1.0.27 and 1.0.28. I think we must wait for more investigation on the sane side before we can do more on the libusb side.

mcuee, 2022-02-04:
> Anyway, this bug was opened with the suspicion that there was something with Big Sur and libusb. Now it seems clear that it boils down to changes in sane between 1.0.27 and 1.0.28. I think we must wait for more investigation on the sane side before we can do more on the libusb side.

I suspect there are multiple issues here. But yes I agree we can not do much without the investigations on the Sane side.

Ref: the following issue is more concerned about Big Sur: sane-backends fails to scan on Big Sur 11.3; the reports there refer to sane-backends version 1.0.32
https://gitlab.com/sane-project/backends/-/issues/473

tenzap's issue may be different: CanoScan LiDE25 doesn’t scan since v1.0.28
https://gitlab.com/sane-project/backends/-/issues/565

cyanfish, 2022-08-27:
For the difference between SANE 1.0.27 and 1.0.28, I might have found what changed (using pthreads in 1.0.27 vs forked processes in 1.0.28). I put a comment in the relevant [SANE issue](https://gitlab.com/sane-project/backends/-/issues/565#note_1079913177).

tormodvolden, 2022-08-27:
> using pthreads in 1.0.27 vs forked processes in 1.0.28

So you were able to build libsane with pthreads even if pthread_t is not integer. Does that mean that sanei_thread implementation is fixed and it should default to use_pthread=yes again?

The use of fork seems to be the issue here. All USB transfers fail after the fork. It can as you say be something with macOS and accessing the USB device from different processes. I wonder if the forking path would work on Linux.

cyanfish, 2022-08-27:
> Does that mean that sanei_thread implementation is fixed and it should default to use_pthread=yes again?

Excellent question. According to [this issue](https://gitlab.com/sane-project/backends/-/issues/153) the check is still needed, but I wonder if it's working as intended on macOS. The concern seems to be that pthread_t might be a struct, while on macOS it's a pointer, which to me seems like it should work as the only integer-like requirements are that it can be compared and cast to/from -1. I added a comment to the linked issue as well.

tormodvolden, 2022-08-28:
This is a relevant quote from SANE backends README.darwin, indicating that the fork path never worked on macOS:

> Backends that use the function "fork" may not work at least with USB scanners.
> That's a limitation of MacOS X (doesn't use file descriptors for USB access).
> Most backends have been modified to use threads on MacOS X and should
> work. Others don't use fork at all. Both types of backends should work with
> MacOS X. Changing the backends to use threads is an ongoing effort.
>

cyanfish, 2022-08-28:
Ah. Well, that explains it.

I'm now trying to track down the second macOS issue I'm having with my Canon MP495. I've done some testing with VMs, and it's pretty strange:
- Windows host / Ubuntu VM / SANE: succeeds
- macOS host / Ubuntu VM / SANE: fails, USB device gets into an unresponsive state (until disconnected+reconnected)
- macOS host / Ubuntu VM / Vuescan: succeeds (unless SANE has previously put it in the unresponsive state)

The fact that it varies depending on the host OS even with the same VM indicates it's some interaction with the USB communication done by SANE/libusb and the macOS USB implementation, and not something specific to the macOS build like the fork issue.

I'm going to continue going over the Wireshark USB captures I have and see if I can find any clues.

cyanfish, 2022-08-28:
I was able to get scanning working for my MP495, and have a partial understanding of what the cause is.

The flow that my scanner's backend (pixma) does is something like this:
1. Send some bulk out/in requests
2. Read from the interrupt endpoint with 100ms timeout until it has no data remaining and times out (to clear the buffers)
3. Send some more bulk out/in requests
4. Read from the interrupt endpoint with 1000ms timeout (with the expectation that the scanner will send data when it's ready)
5. Continue with actual scan via bulk in/out requests

I noticed that the failure was happening after the interrupt requests (which time out). So bypassing those (skipping 2 and replacing 4 with a sleep) makes it work. So potentially we could patch the pixma backend to behave more like Vuescan and not read from the interrupt endpoint during operation, but ideally we can figure out why this problem is happening as I'm sure it's affecting multiple backends.

After some more experimentation I found that doing the interrupt request with a short timeout (10ms) works fine. There's an inconsistent threshold somewhere in the 20ms-80ms range where anything about that causes failures. Also 10x 10ms requests fails. But it's not just a pure timing issue as a 1000ms sleep works fine. Moving the interrupts before step 1 regardless of timeout also works fine.

So I guess libusb has some kind of issue handling requests that timeout on macOS? I'm not sure if it's specific to interrupts.

cyanfish, 2022-08-30:
After a lot of hair-pulling trying to figure out why the libusb interrupt code wasn't being called, I found this:
https://gitlab.com/sane-project/backends/-/blob/master/backend/pixma/pixma_io_sanei.c#L69
```
/* MAC OS X does not support timeouts in darwin/libusb interrupt reads
 * This is a very basic turnaround for MAC OS X
 * Button scan will not work with this wrapper */
#ifdef __APPLE__
# define sanei_usb_read_int sanei_usb_read_bulk
#endif
```

Removing that still fails, but also changing this line in libusb to use `ClearPipeStall` instead of `ClearPipeStallBothEnds` fixes it:
https://github.com/libusb/libusb/blob/c060e9ce30ac2e3ffb49d94209c4dae77b6642f7/libusb/os/darwin_usb.c#L2271
```
/* newer versions of darwin support clearing additional bits on the device's endpoint */
// kresult = (*(cInterface->interface))->ClearPipeStallBothEnds(cInterface->interface, pipeRef);
kresult = (*(cInterface->interface))->ClearPipeStall(cInterface->interface, pipeRef);
```

Note that comment is specifically referring to the "both ends" part, as an older [revision](https://github.com/libusb/libusb/blob/b49f6bf5c910d0fd694ecf165d7927673707bff9/libusb/os/darwin_usb.c#L898) has this code:
```
#if (InterfaceVersion < 190)
  kresult = (*(cInterface->interface))->ClearPipeStall(cInterface->interface, pipeRef);
#else
  /* newer versions of darwin support clearing additional bits on the device's endpoint */
  kresult = (*(cInterface->interface))->ClearPipeStallBothEnds(cInterface->interface, pipeRef);
#endif
```

Documentation for ClearPipeStallBothEnds:
> This function is equivalent to ClearPipeStall except that it also attempts to clear the halt and toggle bits on the device's endpoint for the pipe by sending a ClearFeature(ENDPOINT_HALT) to the default control pipe in the device, specifying the endpoint for the pipe represented by pipeRef. For most devices, this resynchronizes the data toggle between the two endpoints to ensure that there is no loss of data.
https://developer.apple.com/documentation/iokit/iousbinterfaceinterface190/1559241-clearpipestallbothends?language=objc

I'm not familiar enough with usb to understand the implications of skipping that, but given the libusb comment, it doesn't sound like it's critical.

That function is also called in [one more spot](https://github.com/libusb/libusb/blob/c060e9ce30ac2e3ffb49d94209c4dae77b6642f7/libusb/os/darwin_usb.c#L1682), but since that's the clear_halt function and the point of BothEnds is to clear the halt bit, I would guess that call should remain, but I'm not sure.

None of this would directly affect SANE/libusb running in the Ubuntu VM, but as the VM software (UTM/qemu) is itself using libusb, I suspect a libusb fix with that change would fix the VM too.

cyanfish, 2022-08-31:
And it looks like there's already a pending PR that does an equivalent change:
https://github.com/libusb/libusb/pull/1117

So all we need is for that to be merged.

tormodvolden, 2022-08-31:
Did you test your scanner with #1117?

cyanfish, 2022-08-31:
Yes, after fixing the uninitialized variable so it compiles, it works.

tormodvolden, 2022-08-31:
Thanks. So the PR was apparently not compile-tested for recent macOS... I see now that the CI build for the PR had failed. I have updated the PR to fix the uninitialized variable.

r-oldenburg, 2023-01-02:
Hey guys, 
I do have the exact same error with my Mustek BearPaw 2400 CU and Mac M1 Monterey.
Installed sane-backends with brew (1.1.1) and using backend `gt68xx' version 1.0.84.
The scan process always stops when the first "bulk transfer" starts.

I tried with SANE_USB_WORKAROUND=1 to no avail.
I have also built libusb locally from https://github.com/libusb/libusb/pull/1117 and used it. Same error.

It seems that connection gets lost before the bulk transfer starts and that internally fails when calling `GetPipePropertiesV3`.

```scanimage: scanning image of size 1281x1772 pixels at 8 bits/pixel
scanimage: acquiring gray frame
[09:59:56.493752] [sanei_usb] sanei_usb_read_bulk: trying to read 1344 bytes
[25.831893] [00ea81d4] libusb: debug [libusb_submit_transfer] transfer 0x145705a60
[25.831905] [00ea81d4] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[25.831907] [00ea81d4] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[25.831929] [00ea81d4] libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): no connection to an IOService (code = 0xe00002c0)
[25.831934] [00ea81d4] libusb: debug [libusb_free_transfer] transfer 0x145705a60
[09:59:56.494702] [sanei_usb] sanei_usb_read_bulk: read failed (still got 0 bytes): No such device (it may have been disconnected)
[25.832314] [00ea81d4] libusb: debug [libusb_clear_halt] endpoint 0x81
[25.832316] [00ea81d4] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[25.832318] [00ea81d4] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[25.832321] [00ea81d4] libusb: warning [darwin_clear_halt] ClearPipeStall: no connection to an IOService
[09:59:56.495884] [dll] sane_read(handle=0x600003c540a0,data=0x14580dc00,maxlen=32768,lenp=0x16d23e49c)
scanimage: min/max graylevel value = 255/0
libpng error: No IDATs written into file```

Also using UTM does not seem to help. There other problems appear (no working connection or strange errors like "invalid argument").

Anybody any ideas?

r-oldenburg, 2023-01-02:
Update: Running a UTM VM with ubuntu-22.10-live-server-arm64 works!
Interestingly using USB 2.0 **and** 3.0 (setup was done with 2.0 then after successful test switched to 3.0 -> also ok)

Now trying with VMWare Fusion...

r-oldenburg, 2023-01-02:
Update 2: Funny. Also working with VMWare Fusion / Ubuntu 22.10
Previous tries with VMWare / Fedora failed...

And sill Mac OS locally fails the whole time with the described failure above.

cyanfish, 2023-01-06:
In case it's helpful you can try with [NAPS2 for Mac](https://github.com/cyanfish/naps2/discussions/35) which includes [patches](https://github.com/cyanfish/naps2-sane/tree/main/patches) for libusb and SANE. Or try those patches yourself.

tormodvolden, 2023-01-06:
The libusb fix in #1117 has been merged to git master. I believe that is all needed from libusb and the rest must be fixed in sane-backends.

r-oldenburg, 2023-01-09:
> In case it's helpful you can try with [NAPS2 for Mac](https://github.com/cyanfish/naps2/discussions/35) which includes [patches](https://github.com/cyanfish/naps2-sane/tree/main/patches) for libusb and SANE. Or try those patches yourself.

What a great tool!!! I did not know about it.
Unfortunately also NAPS2 fails to scan with my BearPaw. It does recognize it and allows me to select it but then it fails when "Aquiring data". It sais "The selected scanne is offline".

When I have the time I will try the patches locally.

mcuee, 2023-01-24:
I will close this for now. Please re-open if anyone find SANE issue with libusb again. Thanks.

closed 2023-01-24 by mcuee

tormodvolden, 2025-03-19:
They finally fixed USB scanning on macOS a year ago https://gitlab.com/sane-project/backends/-/commit/dcf4241470fa9998514d3d2c3744817e1c866120 included in sane 1.3.1 (released 2024-05-22).