libusb/libusb #920: Is it possible to claim interfaces on macOS while using kernel extensions for other interfaces?

NorbiPeti, 2021-06-01:
I'm developing an application that uses a custom serial driver to communicate with a device. The device has other interfaces as well (HID, audio) which I don't want to access but want the system to pick up. I tried the latest HEAD (35e8ee9) and used set_auto_detach_kernel_driver() to detach the interface from the driver and it works without any issues, however, it also prevents the other interfaces from being detected.

Is there some way to be able to claim the needed interfaces while letting the system pick up the rest? Or do I have to resort to unloading the kext while the application is running?

Tested on macOS 11.3.1

mcuee, 2021-06-18:
@osy Just wondering if you can shed some light on this. Thanks.

osy, 2021-06-18:
No it’s not possible. The capture apis only works on the whole device.

mcuee, 2021-06-19:
Thanks. So I will create a FAQ item in the will and then close this issue.

mcuee, 2021-06-25:
FAQ updated.
https://github.com/libusb/libusb/wiki/FAQ#How_can_I_run_libusb_applications_under_Mac_OS_X_if_there_is_already_a_kernel_extension_installed_for_the_device_and_claim_exclusive_access

Please also refer to the details mentioned in #911 .

closed 2021-06-25 by mcuee