libusb/libusb #921: Include Windows ARM64 binaries in releases

prayaas-a, 2021-06-03:
As of #423, support for the Windows ARM platforms was added (as far as I'm aware, only ARM64 is supported). While this change made it to the 1.0.24 release, it wasn't included as a binary in the package. Including the ARM64 build might help discoverability and adoption on those platforms. Further, the documentation could be updated to reflect that these platforms are supported.

Side note: is ARM64 Windows support being tested actively?

mcuee, 2021-06-03:
Arm Windows will be a niche and we may not have the resources to provide the binaries. Take note we provide binaries for both GCC and VS. I have no idea the support of GCC ARM compiler for Windows.

mcuee, 2021-06-03:
In any case, even the Windows x86/x64 binaries are provided as is and may not work for different compiler versions.

mcuee, 2021-06-05:
So I will say do not expect any time soon for a Windows ARM binary to be available.

@dickens Please chime in here as well if you got the time.

sidd-kishan, 2021-06-06:
Please pursue it we at Windows on Rasberry Pi community will be glad to extend support in testing your drivers and tools for ARM64. We have Windows on ARM running on Rasberry Pi 4 B+ which will be a great place to test out the ARM64 Binaries.

sidd-kishan, 2021-06-06:
Cant, we just do `vcpkg install libusb` ? As libusb is already part of vcpkg.

mcuee, 2021-06-07:
@sidd-kishan You can try to vcpkg to see if it works for Raspberry Pi Windows on ARM or not. If it works, that will be good. I do not use vcpkg myself under Windows, I use MSYS2 MinGW-w64.

I have the Raspeberry Pi 400 myself but I have no interests in running Windows on ARM myself. Linux is pretty solid on my Raspberry Pi 400 (Raspberry Pi OS, Ubuntu, or Manjaro).

mcuee, 2021-06-07:
If vcpkg does not support the ARM architecture, you can always create a ticket for vcpkg.

mcuee, 2021-06-14:
BTW, if you use VS2019 and install the necessary tools, it is pretty easy to build libusb for ARM64. I seldom use VS but just as a test and it seems to be pretty smooth to build for ARM64 and ARM.

```
Rebuild started...
1>------ Rebuild All started: Project: libusb-1.0 (static), Configuration: Release ARM64 ------
2>------ Rebuild All started: Project: getopt, Configuration: Release ARM64 ------
3>------ Rebuild All started: Project: libusb-1.0 (dll), Configuration: Release ARM64 ------
3>core.c
2>getopt.c
2>getopt1.c
1>core.c
2>getopt_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\lib\getopt.lib
3>descriptor.c
1>descriptor.c
3>events_windows.c
1>events_windows.c
3>hotplug.c
1>hotplug.c
3>io.c
1>io.c
3>strerror.c
1>strerror.c
3>sync.c
1>sync.c
3>threads_windows.c
1>threads_windows.c
3>windows_common.c
1>windows_common.c
3>windows_usbdk.c
1>windows_usbdk.c
3>windows_winusb.c
1>windows_winusb.c
3>   Creating library C:\work\libusb\libusb\msvc\..\ARM64\Release\dll\libusb-1.0.lib and object C:\work\libusb\libusb\msvc\..\ARM64\Release\dll\libusb-1.0.exp
1>libusb_static_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\lib\libusb-1.0.lib
3>Generating code
4>------ Rebuild All started: Project: dpfp, Configuration: Release ARM64 ------
5>------ Rebuild All started: Project: dpfp_threaded, Configuration: Release ARM64 ------
6>------ Rebuild All started: Project: fxload, Configuration: Release ARM64 ------
7>------ Rebuild All started: Project: hotplugtest, Configuration: Release ARM64 ------
8>------ Rebuild All started: Project: listdevs, Configuration: Release ARM64 ------
9>------ Rebuild All started: Project: sam3u_benchmark, Configuration: Release ARM64 ------
10>------ Rebuild All started: Project: stress, Configuration: Release ARM64 ------
11>------ Rebuild All started: Project: testlibusb, Configuration: Release ARM64 ------
12>------ Rebuild All started: Project: xusb, Configuration: Release ARM64 ------
4>dpfp.c
5>dpfp.c
3>Finished generating code
7>hotplugtest.c
8>listdevs.c
9>sam3u_benchmark.c
10>stress.c
6>ezusb.c
12>xusb.c
11>testlibusb.c
3>libusb_dll_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\dll\libusb-1.0.dll
6>fxload.c
10>testlib.c
4>Generating code
7>Generating code
5>Generating code
8>Generating code
9>Generating code
12>Generating code
10>Generating code
11>Generating code
4>Finished generating code
7>Finished generating code
6>Generating code
9>Finished generating code
8>Finished generating code
5>Finished generating code
10>Finished generating code
12>Finished generating code
11>Finished generating code
6>Finished generating code
4>dpfp_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\examples\dpfp.exe
7>hotplugtest_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\examples\hotplugtest.exe
8>listdevs_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\examples\listdevs.exe
9>sam3u_benchmark_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\examples\sam3u_benchmark.exe
5>dpfp_threaded_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\examples\dpfp_threaded.exe
11>testlibusb_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\examples\testlibusb.exe
10>stress_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\tests\stress.exe
12>xusb_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\examples\xusb.exe
6>fxload_2019.vcxproj -> C:\work\libusb\libusb\ARM64\Release\examples\fxload.exe
========== Rebuild All: 12 succeeded, 0 failed, 0 skipped ==========

```

prayaas-a, 2021-06-14:
@mcuee yup, this was my experience too. That's why I thought including the binaries would be a trivial task. But it looks like I'd forgotten about the MinGW version - and there's not exactly an easy and reliable way to generate Windows ARM64 binaries using that yet. I'm hoping that side of the ecosystem catches up eventually (especially with ARM computing being much bigger now, OEMs jumping on it, and the availability of low cost Windows ARM devices) - it would be nice to see.

sidd-kishan, 2021-06-16:
@mcuee Please share your project compilation configuration, so that we can try to compile the same on Windows 10 ARM64 device.

mcuee, 2021-06-16:
@sidd-kishan No change needed for the solution files. I just install the language tools with latest VS2019 community edition and it can be built out of the box. The ARM and ARM64 configurations are already in the solution files.

mcuee, 2021-06-18:
In this case, I will close the issue for now.

closed 2021-06-18 by mcuee

mcuee, 2021-07-08:
Ref: vcpkg already supports ARM binary for VS.
https://github.com/a1ien/rusb/issues/68#issuecomment-873561030
```
./vcpkg install libusb:arm64-windows
```