libusb/libusb #922: libusb stress test failed

mcuee, 2021-06-04:
I just tried the latest git and then the stress test failed. No debug logs were given so I am not so sure what happened. I tested with both ARM64 M1 build and x86_64 build under my Mac Mini M1 fully upgraded (using Homebrew).

Early exit with latest git.
```
libusb/buildarm64 on master > ./tests/stress 
Starting test run: init_and_exit...
[1]    45358 abort      ./tests/stress
```

Proper out with 1.0.24 release.
```
libusb/libusb-1.0.24/build took 3s ❯ ./tests/stress 
Starting test run: init_and_exit...
Success (0)
Starting test run: get_device_list...
Success (0)
Starting test run: many_device_lists...
Success (0)
Starting test run: default_context_change...
Success (0)
---
Ran 4 tests
Passed 4 tests
Failed 0 tests
Error in 0 tests
Skipped 0 tests
```

mcuee, 2021-06-04:
Under my Raspberry Pi running Ubuntu 64bit Linux, it just hangs.

```
libusb on master > ./tests/stress
Starting test run: init_and_exit...
^C

```

mcuee, 2021-06-04:
Same segfault under Windows 10 x86_64.

```
/c/work/libusb/libusb
$ ./tests/stress.exe
Starting test run: init_and_exit...
Success (0)
Starting test run: get_device_list...
Success (0)
Starting test run: many_device_lists...
Success (0)
Starting test run: default_context_change...
Segmentation fault

```

mcuee, 2021-06-04:
Looks like it is with the default context change test. Maybe it has something to do with the test code or the latest commit https://github.com/libusb/libusb/commit/32a22069428cda9d63aa666e92fb8882a83d4515

https://github.com/libusb/libusb/blob/master/tests/stress.c#L117

```
/** Tests that the default context (used for various things including
 * logging) works correctly when the first context created in a
 * process is destroyed. */
```

mcuee, 2021-06-05:
@hjelmn Maybe this needs to be fixed before the RC.

hjelmn, 2021-06-05:
Ok. can take a look.

mcuee, 2021-06-06:
Just a relevant info, if I run xusb under my Mac Mini M1, it will have warning in the end. Maybe it is related to macOS stress test failure.

```

libUsb/buildarm64 on  master [?] ❯ ./examples/xusb 04d8:fa2e
Using libusb v1.0.24.11617

Opening device 04D8:FA2E...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 04D8:FA2E
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 00.00.00
       endpoint[0].address: 01
           max packet size: 0020
          polling interval: 00
       endpoint[1].address: 81
           max packet size: 0020
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Travis Robinson"
   String (0x02): "Benchmark Device"
   String (0x03): "LUSBW1"

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  92 00 00 00 00 01 05 00 01 00 84 00 00 00 07 00  ................
  00000010  00 00 2a 00 44 00 65 00 76 00 69 00 63 00 65 00  ..*.D.e.v.i.c.e.
  00000020  49 00 6e 00 74 00 65 00 72 00 66 00 61 00 63 00  I.n.t.e.r.f.a.c.
  00000030  65 00 47 00 55 00 49 00 44 00 73 00 00 00 50 00  e.G.U.I.D.s...P.
  00000040  00 00 7b 00 46 00 37 00 30 00 32 00 34 00 32 00  ..{.F.7.0.2.4.2.
  00000050  43 00 37 00 2d 00 46 00 42 00 32 00 35 00 2d 00  C.7.-.F.B.2.5.-.
  00000060  34 00 34 00 33 00 42 00 2d 00 39 00 45 00 37 00  4.4.3.B.-.9.E.7.
  00000070  45 00 2d 00 41 00 34 00 32 00 36 00 30 00 46 00  E.-.A.4.2.6.0.F.
  00000080  33 00 37 00 33 00 39 00 38 00 32 00 7d 00 00 00  3.7.3.9.8.2.}...
  00000090  00 00                                            ..

Releasing interface 0...
Closing device...
libusb: warning [libusb_exit] device 2.6 still referenced
libusb: warning [libusb_exit] device 2.2 still referenced
```

mcuee, 2021-06-07:
Maybe Stress test code is not correct.

Nathan mentioned the following in #777. A potential fix is in #923.
> Interesting question. I have always assumed this is true but we don't document it. We can call:
> 
> ```
> libusb_set_option(NULL, ...);
> ```
> 
> before the default context is created, right? If not the stress test is wrong. Otherwise I have a fix that will allow this. Will hold off of this change until I get a consensus.

mcuee, 2021-06-07:
From https://github.com/libusb/libusb/issues/777

> I identifed why the Darwin backend is failing the stress test. Will have a fix today.
> 
> The problem is in the ordering in libusb_exit caused by destroying the cached device list when the init count goes to 0. Still not sure I agree with this behavior change. I used a library destructor function for a reason... I can fix the ordering issue but this code is somewhat more fragile than I would like.

mcuee, 2021-06-07:
Commit https://github.com/libusb/libusb/commit/676076ce6cbed8331a22c33ca84b9cae326cff5b helps macOS stress test to have the same result as Linux and Windows. Then #923 should be able to fix the remaining segfault issues with Stress test.
```
libusb/buildarm64 on master [?] > ./tests/stress 
Starting test run: init_and_exit...
Success (0)
Starting test run: get_device_list...
Success (0)
Starting test run: many_device_lists...
Success (0)
Starting test run: default_context_change...
[1]    6732 segmentation fault  ./tests/stress
```

mcuee, 2021-06-08:
Commit #923 fixed the issue under Windows 10. I belive it will fix the issue under macOS and Linux as well. I will test later.

```
MINGW64 /c/work/libusb/libusb
$ ./tests/stress.exe
Starting test run: init_and_exit...
Success (0)
Starting test run: get_device_list...
Success (0)
Starting test run: many_device_lists...
Success (0)
Starting test run: default_context_change...
Success (0)
---
Ran 4 tests
Passed 4 tests
Failed 0 tests
Error in 0 tests
Skipped 0 tests
```

mcuee, 2021-06-08:
No issues under macOS and Linux as well.

closed 2021-06-08 by mcuee