libusb/libusb #924: Device still referenced warning when running xusb example under macOS

mcuee, 2021-06-07:
Latest git still have the following issue.

```
./examples/xusb 04d8:fa2e
Using libusb v1.0.24.11620

Opening device 04D8:FA2E...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 04D8:FA2E
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 00.00.00
       endpoint[0].address: 01
           max packet size: 0020
          polling interval: 00
       endpoint[1].address: 81
           max packet size: 0020
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Travis Robinson"
   String (0x02): "Benchmark Device"
   String (0x03): "LUSBW1"

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  92 00 00 00 00 01 05 00 01 00 84 00 00 00 07 00  ................
  00000010  00 00 2a 00 44 00 65 00 76 00 69 00 63 00 65 00  ..*.D.e.v.i.c.e.
  00000020  49 00 6e 00 74 00 65 00 72 00 66 00 61 00 63 00  I.n.t.e.r.f.a.c.
  00000030  65 00 47 00 55 00 49 00 44 00 73 00 00 00 50 00  e.G.U.I.D.s...P.
  00000040  00 00 7b 00 46 00 37 00 30 00 32 00 34 00 32 00  ..{.F.7.0.2.4.2.
  00000050  43 00 37 00 2d 00 46 00 42 00 32 00 35 00 2d 00  C.7.-.F.B.2.5.-.
  00000060  34 00 34 00 33 00 42 00 2d 00 39 00 45 00 37 00  4.4.3.B.-.9.E.7.
  00000070  45 00 2d 00 41 00 34 00 32 00 36 00 30 00 46 00  E.-.A.4.2.6.0.F.
  00000080  33 00 37 00 33 00 39 00 38 00 32 00 7d 00 00 00  3.7.3.9.8.2.}...
  00000090  00 00                                            ..

Releasing interface 0...
Closing device...
libusb: warning [libusb_exit] device 2.6 still referenced
libusb: warning [libusb_exit] device 2.2 still referenced

```

mcuee, 2021-06-07:
From https://github.com/libusb/libusb/issues/777 with regard to macOS specific issue with Stress Test. Maybe this is related.

> I identifed why the Darwin backend is failing the stress test. Will have a fix today.
> 
> The problem is in the ordering in libusb_exit caused by destroying the cached device list when the init count goes to 0. Still not sure I agree with this behavior change. I used a library destructor function for a reason... I can fix the ordering issue but this code is somewhat more fragile than I would like.

mcuee, 2021-06-07:
Full debug log here.

```

libUsb/buildarm64 on  master [?] took 26s ❯ ./examples/xusb -d 04d8:fa2e
Using libusb v1.0.24.11620

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000034] [0000987a] libusb: debug [libusb_init] created default context
[ 0.000071] [0000987a] libusb: debug [libusb_init] libusb v1.0.24.11620
[ 0.000132] [0000987a] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000439] [0000987a] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb4ae3ed
[ 0.000476] [0000987a] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb4ae3ed
[ 0.002259] [0000987a] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002264] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002266] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002267] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002269] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002270] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002271] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002272] [0000987a] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002273] [0000987a] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002274] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002275] [0000987a] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002281] [0000987a] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002283] [0000987a] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002284] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002298] [0000987a] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002300] [0000987a] libusb: debug [process_new_device] allocating new device in context 0x11de0dc70 for with session 0xb4ae3ed
[ 0.002303] [0000987a] libusb: debug [process_new_device] found device with address 1 port = 5 parent = 0x0 at 0x11de110aa
[ 0.002382] [0000987a] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb6be948
[ 0.002405] [0000987a] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb6be948/0x2100000 against cached device with sessionID/locationID 0xb4ae3ed/0x2500000
[ 0.002407] [0000987a] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb6be948
[ 0.002792] [0000987a] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002796] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002797] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002798] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002799] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002800] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002801] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002803] [0000987a] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002804] [0000987a] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002805] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002806] [0000987a] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002808] [0000987a] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002809] [0000987a] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002810] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002822] [0000987a] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002824] [0000987a] libusb: debug [process_new_device] allocating new device in context 0x11de0dc70 for with session 0xb6be948
[ 0.002826] [0000987a] libusb: debug [process_new_device] found device with address 2 port = 1 parent = 0x0 at 0x11de111aa
[ 0.002901] [0000987a] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb6dec56
[ 0.002929] [0000987a] libusb: debug [darwin_get_cached_device] parent sessionID: 0xb4ae3ed
[ 0.002931] [0000987a] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb6dec56/0x2510000 against cached device with sessionID/locationID 0xb6be948/0x2100000
[ 0.002933] [0000987a] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb6dec56/0x2510000 against cached device with sessionID/locationID 0xb4ae3ed/0x2500000
[ 0.002935] [0000987a] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb6dec56
[ 0.003126] [0000987a] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003130] [0000987a] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...]
[ 0.089325] [0000987a] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.089372] [0000987a] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.089417] [0000987a] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.089462] [0000987a] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.089503] [0000987a] libusb: debug [usbi_remove_event_source] remove fd 3
[ 0.089529] [0000987a] libusb: warning [libusb_exit] device 2.6 still referenced
[ 0.089531] [0000987a] libusb: warning [libusb_exit] device 2.2 still referenced```

hjelmn, 2021-06-07:
Will take a look.

mcuee, 2021-06-07:
BTW, I did some other tests under my Mac Mini M1 and libusb works fine. I actually dig out the old libusbdotnet 2.2.8 binaries released in 2010 and it still works.
https://github.com/LibUsbDotNet/LibUsbDotNet/issues/137

hjelmn, 2021-06-07:
@mcuee I don't own any Microsoft products so I can't run xusb. Will look at the logs.

hjelmn, 2021-06-07:
Ah, I see. It can do more than xbox controllers and you are running with winusb. Ok. Can't help much here :)

hjelmn, 2021-06-07:
Ignore that. I see. It is a string returned by the device. Throwing me off here

hjelmn, 2021-06-07:
Looking for a device I can use with xusb now.

hjelmn, 2021-06-07:
Don't see the error with a Rio S10.

mcuee, 2021-06-08:
The issue may not be with the particular benchmark test device I have (which is a good test device for libusb, libusb-win32, libusbk and libusbdotnet, FW developed by Travis Robinson), rather it could be with my other devices connected to my Mac Mini M1 (two powered hub and quite some USB devices connected). I will try with my other devices to see if the issue is still there or not.

mcuee, 2021-06-08:
Okay, I find out the issues are with external USB Hub. I have two external powered hub in previous tests, after removing one of them, then I can see the result has only one device still referenced. If I totally removed the two USB hubs, then no more devices unreferenced after the program.

Eg: one hub
```
libusb/buildarm64 on master > ./examples/xusb 04d8:fa2e
Using libusb v1.0.24.11623

Opening device 04D8:FA2E...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 04D8:FA2E
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 00.00.00
       endpoint[0].address: 01
           max packet size: 0020
          polling interval: 00
       endpoint[1].address: 81
           max packet size: 0020
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Travis Robinson"
   String (0x02): "Benchmark Device"
   String (0x03): "LUSBW1"

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  92 00 00 00 00 01 05 00 01 00 84 00 00 00 07 00  ................
  00000010  00 00 2a 00 44 00 65 00 76 00 69 00 63 00 65 00  ..*.D.e.v.i.c.e.
  00000020  49 00 6e 00 74 00 65 00 72 00 66 00 61 00 63 00  I.n.t.e.r.f.a.c.
  00000030  65 00 47 00 55 00 49 00 44 00 73 00 00 00 50 00  e.G.U.I.D.s...P.
  00000040  00 00 7b 00 46 00 37 00 30 00 32 00 34 00 32 00  ..{.F.7.0.2.4.2.
  00000050  43 00 37 00 2d 00 46 00 42 00 32 00 35 00 2d 00  C.7.-.F.B.2.5.-.
  00000060  34 00 34 00 33 00 42 00 2d 00 39 00 45 00 37 00  4.4.3.B.-.9.E.7.
  00000070  45 00 2d 00 41 00 34 00 32 00 36 00 30 00 46 00  E.-.A.4.2.6.0.F.
  00000080  33 00 37 00 33 00 39 00 38 00 32 00 7d 00 00 00  3.7.3.9.8.2.}...
  00000090  00 00                                            ..

Releasing interface 0...
Closing device...
libusb: warning [libusb_exit] device 2.2 still referenced

libusb/buildarm64 on master > ./examples/listdevs
04d8:fa2e (bus 2, device 17) path: 1.4
046d:c52b (bus 2, device 15) path: 1.1.3
0bda:5411 (bus 2, device 4) path: 1.1
05e3:0749 (bus 2, device 5) path: 5.1.1
0bda:0411 (bus 2, device 3) path: 5.1
0bda:5411 (bus 2, device 2) path: 1
0bda:0411 (bus 2, device 1) path: 5

libusb/buildarm64 on master > ./examples/xusb 0bda:5411
Using libusb v1.0.24.11623

Opening device 0BDA:5411...
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access

Reading device descriptor:
            length: 18
      device class: 9
               S/N: 0
           VID:PID: 0BDA:5411
         bcdDevice: 0117
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: 3 caps
    USB 2.0 extension:
      attributes             : F41E
    USB 3.0 capabilities:
      attributes             : 00
      supported speeds       : 000E
      supported functionality: 01
    Container ID:
      {e5cdb920-3970-11e0-a935-0002a5d5c51b}

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 09.00.01
       endpoint[0].address: 81
           max packet size: 0001
          polling interval: 0C
interface[0].altsetting[1]: num endpoints = 1
   Class.SubClass.Protocol: 09.00.02
       endpoint[0].address: 81
           max packet size: 0001
          polling interval: 0C

Claiming interface 0...
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Reading string descriptors:
   String (0x01): "Generic"
   String (0x02): "4-Port USB 2.0 Hub"

Releasing interface 0...
Closing device...

```

mcuee, 2021-06-08:
1.0.24 release does not have this issue.

```
libusb/libusb-1.0.24/buildarm64 ❯ ./examples/xusb -d 04d8:fa2e
Using libusb v1.0.24.11584

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000021] [00018d1c] libusb: debug [libusb_init] created default context
[ 0.000041] [00018d1c] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000236] [00018d1c] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1089908cf4
[ 0.000274] [00018d1c] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1089908cf4
[ 0.001997] [00018d1c] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002005] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002007] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002009] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002011] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002012] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002013] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002014] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002016] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002017] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002019] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002020] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002022] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002023] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002044] [00018d1c] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002046] [00018d1c] libusb: debug [process_new_device] allocating new device in context 0x13c704ea0 for with session 0x1089908cf4
[ 0.002050] [00018d1c] libusb: debug [process_new_device] found device with address 21 port = 1 parent = 0x0 at 0x13c60897a
[ 0.002114] [00018d1c] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1089d4af07
[ 0.002135] [00018d1c] libusb: debug [darwin_get_cached_device] parent sessionID: 0x1089908cf4
[ 0.002137] [00018d1c] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x1089d4af07/0x2110000 against cached device with sessionID/locationID 0x1089908cf4/0x2100000
[ 0.002139] [00018d1c] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1089d4af07
[ 0.002401] [00018d1c] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002406] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002407] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002409] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002410] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002411] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002413] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002414] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002415] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002417] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002418] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002420] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002421] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002422] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002438] [00018d1c] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002440] [00018d1c] libusb: debug [process_new_device] allocating new device in context 0x13c704ea0 for with session 0x1089d4af07
[ 0.002443] [00018d1c] libusb: debug [process_new_device] found device with address 22 port = 1 parent = 0x13c608b30 at 0x13c608c0a
[ 0.002499] [00018d1c] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x108a1632c4
[ 0.002519] [00018d1c] libusb: debug [darwin_get_cached_device] parent sessionID: 0x1089d4af07
[ 0.002520] [00018d1c] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x108a1632c4/0x2111000 against cached device with sessionID/locationID 0x1089d4af07/0x2110000
[ 0.002522] [00018d1c] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x108a1632c4/0x2111000 against cached device with sessionID/locationID 0x1089908cf4/0x2100000
[ 0.002539] [00018d1c] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x108a1632c4
[ 0.002729] [00018d1c] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002733] [00018d1c] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...]
[ 0.012603] [00018d1c] libusb: debug [libusb_unref_device] destroy device 2.25
[ 0.012604] [00018d1c] libusb: debug [libusb_unref_device] destroy device 2.24
[ 0.012606] [00018d1c] libusb: debug [libusb_unref_device] destroy device 2.23
[ 0.012607] [00018d1c] libusb: debug [libusb_unref_device] destroy device 2.22
[ 0.012608] [00018d1c] libusb: debug [libusb_unref_device] destroy device 2.21
[ 0.012610] [00018d1c] libusb: debug [usbi_remove_event_source] remove fd 3
[ 0.012653] [00018d20] libusb: debug [darwin_event_thread_main] darwin event thread exiting

```

tormodvolden, 2021-06-08:
Please not that the "still referenced" warning was added after 1.0.24, in commit v1.0.24-3-g1a08aa8, so at least you won't see that message on the 1.0.24 release.

I can also see a list of devices still referenced on Linux. I see this with dfu-util, but only if I do a firmware upload from a device. When just listing the connected DFU devices it doesn't happen. It doesn't go away if I plug the device directly into the laptop, but there is maybe an internal USB hub in any case.

tormodvolden, 2021-06-08:
I can reproduce with this program:
```
#include "libusb.h"

void main(void)
{
	libusb_context *ctx;
	libusb_device **list;
	ssize_t num_devs;
	ssize_t i;

	libusb_init(&ctx);
	num_devs = libusb_get_device_list(ctx, &list);
	for (i = 0; i < num_devs; ++i) {
		struct libusb_device_descriptor desc;
		struct libusb_device *dev = list[i];
	}
	libusb_free_device_list(list, 0);
	libusb_exit(ctx);
}
```

hjelmn, 2021-06-08:
So the bug is in the core and could have existed in 1.0.24.

tormodvolden, 2021-06-08:
Ho hum, my test program should probably have libusb_free_device_list(list, 1) so I guess I was not hitting mcuee's issue.

mcuee, 2021-06-09:
> So the bug is in the core and could have existed in 1.0.24.

Yes based on what Tormod Volden mentioned. And it does not only happen under macOS but also Linux. I believe Windows may also be the same. So this will not be a blocker for 1.0.25RC1. We can still continue the RC1 process.

hjelmn, 2021-06-11:
I can reproduce this. Woo! Looking.

hjelmn, 2021-06-11:
I see the issue.... Think about how best to fix this.

hjelmn, 2021-06-11:
Ack, had one typo. Fixed.

mcuee, 2021-06-11:
Now the issue becomes a bit different.

```
/examples/xusb 04d8:fa2e
Using libusb v1.0.24.11627

Opening device 04D8:FA2E...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 04D8:FA2E
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 00.00.00
       endpoint[0].address: 01
           max packet size: 0020
          polling interval: 00
       endpoint[1].address: 81
           max packet size: 0020
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Travis Robinson"
   String (0x02): "Benchmark Device"
   String (0x03): "LUSBW1"

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  92 00 00 00 00 01 05 00 01 00 84 00 00 00 07 00  ................
  00000010  00 00 2a 00 44 00 65 00 76 00 69 00 63 00 65 00  ..*.D.e.v.i.c.e.
  00000020  49 00 6e 00 74 00 65 00 72 00 66 00 61 00 63 00  I.n.t.e.r.f.a.c.
  00000030  65 00 47 00 55 00 49 00 44 00 73 00 00 00 50 00  e.G.U.I.D.s...P.
  00000040  00 00 7b 00 46 00 37 00 30 00 32 00 34 00 32 00  ..{.F.7.0.2.4.2.
  00000050  43 00 37 00 2d 00 46 00 42 00 32 00 35 00 2d 00  C.7.-.F.B.2.5.-.
  00000060  34 00 34 00 33 00 42 00 2d 00 39 00 45 00 37 00  4.4.3.B.-.9.E.7.
  00000070  45 00 2d 00 41 00 34 00 32 00 36 00 30 00 46 00  E.-.A.4.2.6.0.F.
  00000080  33 00 37 00 33 00 39 00 38 00 32 00 7d 00 00 00  3.7.3.9.8.2.}...
  00000090  00 00                                            ..

Releasing interface 0...
Closing device...
Assertion failed: (refcnt >= 0), function libusb_unref_device, file ../../libusb/core.c, line 1184.
[1]    6933 abort      ./examples/xusb 04d8:fa2e

```

mcuee, 2021-06-11:
Debug log:

```

libusb/buildarm64 on  master [?] ❯ export LIBUSB_DEBUG=4
libusb/buildarm64 on  master [?] ❯ ./examples/xusb 04d8:fa2e
Using libusb v1.0.24.11627

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000021] [0000bb5a] libusb: debug [libusb_init] created default context
[ 0.000043] [0000bb5a] libusb: debug [libusb_init] libusb v1.0.24.11627
[ 0.000098] [0000bb5a] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000361] [0000bb5a] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb306a6d
[ 0.000399] [0000bb5a] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb306a6d
[ 0.002637] [0000bb5a] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002647] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002648] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002650] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002651] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002652] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002654] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002655] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002656] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002657] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002658] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002660] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002661] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002662] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002682] [0000bb5a] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002684] [0000bb5a] libusb: debug [process_new_device] allocating new device in context 0x11c706780 for with session 0xb306a6d
[ 0.002687] [0000bb5a] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x11c709c1a
[ 0.002752] [0000bb5a] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb5716e7
[ 0.002775] [0000bb5a] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb5716e7/0x2500000 against cached device with sessionID/locationID 0xb306a6d/0x2100000
[ 0.002778] [0000bb5a] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb5716e7
[ 0.003030] [0000bb5a] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003034] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003035] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.003036] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003037] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003039] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.003040] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.003041] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.003042] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.003043] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.003044] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003050] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003051] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003052] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003063] [0000bb5a] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003065] [0000bb5a] libusb: debug [process_new_device] allocating new device in context 0x11c706780 for with session 0xb5716e7
[ 0.003067] [0000bb5a] libusb: debug [process_new_device] found device with address 2 port = 5 parent = 0x0 at 0x11c709c9a
[ 0.003125] [0000bb5a] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb70cc74
[ 0.003145] [0000bb5a] libusb: debug [darwin_get_cached_device] parent sessionID: 0xb306a6d
[ 0.003147] [0000bb5a] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb70cc74/0x2110000 against cached device with sessionID/locationID 0xb5716e7/0x2500000
[ 0.003148] [0000bb5a] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb70cc74/0x2110000 against cached device with sessionID/locationID 0xb306a6d/0x2100000
[ 0.003150] [0000bb5a] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb70cc74
[ 0.003314] [0000bb5a] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003318] [0000bb5a] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...]
[ 0.085101] [0000bb5a] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.085147] [0000bb5a] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.085192] [0000bb5a] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.085235] [0000bb5a] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.085238] [0000bb5a] libusb: debug [libusb_unref_device] destroy device 2.1
Assertion failed: (refcnt >= 0), function libusb_unref_device, file ../../libusb/core.c, line 1184.
[1]    7029 abort      ./examples/xusb 04d8:fa2e
```

mcuee, 2021-06-11:
I got two USB Hubs connected to the Mac Mini M1.
```
libusb/buildarm64 on  master [?] ❯ export LIBUSB_DEBUG=3    
libusb/buildarm64 on  master [?] ❯ ./examples/testlibusb
Dev (bus 2, device 14): 04D8 - FA2E speed: 12M
  Manufacturer:              Travis Robinson
  Product:                   Benchmark Device
Dev (bus 2, device 11): 1915 - 1025 speed: 12M
  Manufacturer:              ZY.Ltd
  Product:                   ZY Control Mic
Dev (bus 2, device 10): 1366 - 0101 speed: 12M
  Manufacturer:              SEGGER
  Product:                   J-Link
Dev (bus 2, device 9): 1A40 - 0201 speed: 480M
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
  Product:                   USB 2.0 Hub [MTT]
Dev (bus 2, device 8): 046D - C52B speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 2, device 7): 0781 - 5595 speed: 5G
  Manufacturer:              SanDisk
  Product:                   Ultra USB 3.0
Dev (bus 2, device 6): 05E3 - 0749 speed: 5G
  Manufacturer:              Generic
  Product:                   USB3.0 Card Reader
Dev (bus 2, device 5): 1A40 - 0201 speed: 480M
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
  Product:                   USB 2.0 Hub [MTT]
Dev (bus 2, device 4): 0BDA - 0411 speed: 5G
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub
Dev (bus 2, device 3): 0BDA - 5411 speed: 480M
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
Dev (bus 2, device 2): 0BDA - 0411 speed: 5G
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub
Dev (bus 2, device 1): 0BDA - 5411 speed: 480M
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
```

mcuee, 2021-06-11:
Same thing for a generic FTDI USB Serial Adapter.

```

libusb on  master [?] ❯ ./buildarm64/examples/xusb 0403:6001
Using libusb v1.0.24.11627

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Releasing interface 0...
Closing device...
Assertion failed: (refcnt >= 0), function libusb_unref_device, file ../../libusb/core.c, line 1184.
[1]    31357 abort      ./buildarm64/examples/xusb 0403:6001

```

hjelmn, 2021-06-11:
Ok. Will take a look shortly.

hjelmn, 2021-06-11:
Thinking about it my fix was too simplistic. I know what went wrong. fix coming shortly.

hjelmn, 2021-06-11:
Should be really fixed now.

mcuee, 2021-06-11:
It goes back to previous behavior.

```

libUsb on  master [?] took 4s ❯ ./examples/xusb 0403:6001
Using libusb v1.0.24.11628

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Releasing interface 0...
Closing device...
libusb: warning [libusb_exit] device 2.6 still referenced
libusb: warning [libusb_exit] device 2.2 still referenced
```

hjelmn, 2021-06-11:
Ugh. It went away for me. Maybe I need more devices on the hub or a second hub. We seem to have a reference counting issue.

hjelmn, 2021-06-11:
Ok, have nested hubs. Still no luck reproducing the issue with head.

mcuee, 2021-06-12:
> Ok, have nested hubs. Still no luck reproducing the issue with head.

Interesting that the issue goes away for me for single HUB but not nested hub yet.

Here is the debug log with nested hub.

<details>
<summary> Click to expand </summary>

```

libusb/buildarm64 on  master [?] ❯ ./examples/xusb 0403:6001
Using libusb v1.0.24.11627

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000021] [00004fde] libusb: debug [libusb_init] created default context
[ 0.000041] [00004fde] libusb: debug [libusb_init] libusb v1.0.24.11627
[ 0.000093] [00004fde] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000306] [00004fde] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb58f629
[ 0.000342] [00004fde] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb58f629
[ 0.002010] [00004fde] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002016] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002018] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002019] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002021] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002022] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002023] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002025] [00004fde] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002026] [00004fde] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002027] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002029] [00004fde] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002031] [00004fde] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002032] [00004fde] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002033] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002050] [00004fde] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002053] [00004fde] libusb: debug [process_new_device] allocating new device in context 0x123005fe0 for with session 0xb58f629
[ 0.002057] [00004fde] libusb: debug [process_new_device] found device with address 1 port = 5 parent = 0x0 at 0x12300640a
[ 0.002108] [00004fde] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb797f8f
[ 0.002135] [00004fde] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb797f8f/0x2100000 against cached device with sessionID/locationID 0xb58f629/0x2500000
[ 0.002137] [00004fde] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb797f8f
[ 0.002487] [00004fde] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002492] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002493] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002495] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002496] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002497] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002498] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002500] [00004fde] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002501] [00004fde] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002503] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002504] [00004fde] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002516] [00004fde] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002517] [00004fde] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002518] [00004fde] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002533] [00004fde] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002535] [00004fde] libusb: debug [process_new_device] allocating new device in context 0x123005fe0 for with session 0xb797f8f
[ 0.002537] [00004fde] libusb: debug [process_new_device] found device with address 2 port = 1 parent = 0x0 at 0x12300952a
[ 0.002592] [00004fde] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb7b94c8
[ 0.002613] [00004fde] libusb: debug [darwin_get_cached_device] parent sessionID: 0xb58f629
[ 0.002616] [00004fde] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb7b94c8/0x2510000 against cached device with sessionID/locationID 0xb797f8f/0x2100000
[ 0.002618] [00004fde] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb7b94c8/0x2510000 against cached device with sessionID/locationID 0xb58f629/0x2500000
[ 0.002619] [00004fde] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb7b94c8
[... cut ...]
[ 0.106260] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.106294] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.106332] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.106374] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.30
[ 0.106377] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.2
Assertion failed: (refcnt >= 0), function libusb_unref_device, file ../../libusb/core.c, line 1184.
[1]    2829 abort      ./examples/xusb 0403:6001
```

</details>

mcuee, 2021-06-12:
Topology on the USB connections

```

libusb/buildarm64 on  master [?] ❯ system_profiler SPUSBDataType
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleT8103USBXHCI

    USB 3.1 Bus:

      Host Controller Driver: AppleT8103USBXHCI

    USB 3.0 Bus:

      Host Controller Driver: AppleT8103USBXHCIFL1100
      PCI Device ID: 0x1100 
      PCI Revision ID: 0x0010 
      PCI Vendor ID: 0x1b73 

        4-Port USB 3.0 Hub:

          Product ID: 0x0411
          Vendor ID: 0x0bda  (Realtek Semiconductor Corp.)
          Version: 1.17
          Speed: Up to 5 Gb/s
          Manufacturer: Generic
          Location ID: 0x02500000 / 1
          Current Available (mA): 900
          Current Required (mA): 0
          Extra Operating Current (mA): 0

            4-Port USB 3.0 Hub:

              Product ID: 0x0411
              Vendor ID: 0x0bda  (Realtek Semiconductor Corp.)
              Version: 1.17
              Speed: Up to 5 Gb/s
              Manufacturer: Generic
              Location ID: 0x02510000 / 3
              Current Available (mA): 900
              Current Required (mA): 0
              Extra Operating Current (mA): 0

        4-Port USB 2.0 Hub:

          Product ID: 0x5411
          Vendor ID: 0x0bda  (Realtek Semiconductor Corp.)
          Version: 1.17
          Speed: Up to 480 Mb/s
          Manufacturer: Generic
          Location ID: 0x02100000 / 2
          Current Available (mA): 500
          Current Required (mA): 0
          Extra Operating Current (mA): 0

            USB 2.0 Hub [MTT]:

              Product ID: 0x0201
              Vendor ID: 0x1a40  (TERMINUS TECHNOLOGY INC.)
              Version: 1.00
              Speed: Up to 480 Mb/s
              Location ID: 0x02140000 / 30
              Current Available (mA): 500
              Current Required (mA): 100
              Extra Operating Current (mA): 0

                USB 2.0 Hub [MTT]:

                  Product ID: 0x0201
                  Vendor ID: 0x1a40  (TERMINUS TECHNOLOGY INC.)
                  Version: 1.00
                  Speed: Up to 480 Mb/s
                  Location ID: 0x02145000 / 8
                  Current Available (mA): 500
                  Current Required (mA): 100
                  Extra Operating Current (mA): 0

                USB Receiver:

                  Product ID: 0xc52b
                  Vendor ID: 0x046d  (Logitech Inc.)
                  Version: 24.10
                  Speed: Up to 12 Mb/s
                  Manufacturer: Logitech
                  Location ID: 0x02144000 / 32
                  Current Available (mA): 500
                  Current Required (mA): 98
                  Extra Operating Current (mA): 0

                FT232R USB UART:

                  Product ID: 0x6001
                  Vendor ID: 0x0403  (Future Technology Devices International Limited)
                  Version: 6.00
                  Serial Number: A8007Ub5
                  Speed: Up to 12 Mb/s
                  Manufacturer: FTDI
                  Location ID: 0x02143000 / 13
                  Current Available (mA): 500
                  Current Required (mA): 90
                  Extra Operating Current (mA): 0

            4-Port USB 2.0 Hub:

              Product ID: 0x5411
              Vendor ID: 0x0bda  (Realtek Semiconductor Corp.)
              Version: 1.17
              Speed: Up to 480 Mb/s
              Manufacturer: Generic
              Location ID: 0x02110000 / 4
              Current Available (mA): 500
              Current Required (mA): 0
              Extra Operating Current (mA): 0
```

mcuee, 2021-06-12:
USB topology

libusb/buildarm64 on  master [?] ❯ ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 27>
  +-o AppleT8103USBXHCI@01000000  <class AppleT8103USBXHCI, id 0x100000299, registered, matched, active, busy 0 (18 ms)$
  +-o AppleT8103USBXHCI@00000000  <class AppleT8103USBXHCI, id 0x100000295, registered, matched, active, busy 0 (22 ms)$
  +-o AppleT8103USBXHCIFL1100@02000000  <class AppleT8103USBXHCIFL1100, id 0x1000006f4, registered, matched, active, bu$
    +-o 4-Port USB 3.0 Hub@02500000  <class IOUSBHostDevice, id 0x100000741, registered, matched, active, busy 0 (149 m$
    | +-o 4-Port USB 3.0 Hub@02510000  <class IOUSBHostDevice, id 0x100000750, registered, matched, active, busy 0 (135$
    +-o 4-Port USB 2.0 Hub@02100000  <class IOUSBHostDevice, id 0x10000074e, registered, matched, active, busy 0 (1443 $
      +-o 4-Port USB 2.0 Hub@02110000  <class IOUSBHostDevice, id 0x100000767, registered, matched, active, busy 0 (17 $
      +-o USB 2.0 Hub [MTT]@02140000  <class IOUSBHostDevice, id 0x100000d4c, registered, matched, active, busy 0 (458 $
        +-o USB Receiver@02144000  <class IOUSBHostDevice, id 0x100000d71, registered, matched, active, busy 0 (75 ms),$
        +-o USB 2.0 Hub [MTT]@02145000  <class IOUSBHostDevice, id 0x100000d8a, registered, matched, active, busy 0 (97$
        +-o FT232R USB UART@02143000  <class IOUSBHostDevice, id 0x100000e98, registered, matched, active, busy 0 (58 m$

mcuee, 2021-06-12:
No issues under Linux (tested with Raspberry Pi 400 running 64bit Ubuntu Linux), with all the USB devices connected.

```
libusb on  master took 3s ❯ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
                |__ Port 1: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M
            |__ Port 4: Dev 6, If 0, Class=Hub, Driver=hub/7p, 480M
                |__ Port 4: Dev 9, If 2, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 4: Dev 9, If 0, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 4: Dev 9, If 1, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 5: Dev 10, If 0, Class=Hub, Driver=hub/7p, 480M
                    |__ Port 5: Dev 11, If 2, Class=Human Interface Device, Driver=usbhid, 12M
                    |__ Port 5: Dev 11, If 0, Class=Audio, Driver=snd-usb-audio, 12M
                    |__ Port 5: Dev 11, If 3, Class=Human Interface Device, Driver=usbhid, 12M
                    |__ Port 5: Dev 11, If 1, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 3: Dev 8, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 4: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
libusb on  master ❯ ./examples/xusb 0403:6001
Using libusb v1.0.24.11629

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Releasing interface 0...
Closing device...

```

mcuee, 2021-06-12:
Debug Log under Linux.

<details>
<summary> Click to expand </summary>

libusb on  master ❯ export LIBUSB_DEBUG=4
libusb on  master ❯ ./examples/xusb 0403:6001
Using libusb v1.0.24.11629

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000051] [00008601] libusb: debug [libusb_init] created default context
[ 0.000091] [00008601] libusb: debug [libusb_init] libusb v1.0.24.11629
[ 0.000176] [00008601] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000233] [00008601] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000246] [00008601] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000287] [00008601] libusb: debug [get_kernel_version] reported kernel version is 5.11.0-1009-raspi
[ 0.000543] [00008601] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000584] [00008601] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000626] [00008601] libusb: debug [op_init] sysfs is available
[ 0.002073] [0000860f] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.030389] [00008601] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.031063] [00008601] libusb: debug [linux_get_device_address] scan usb1
[ 0.031516] [00008601] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.031560] [00008601] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.031590] [00008601] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.032336] [00008601] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.032378] [00008601] libusb: debug [linux_get_device_address] scan 1-1
[ 0.032495] [00008601] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.032513] [00008601] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.032529] [00008601] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.032664] [00008601] libusb: debug [linux_get_parent_info] dev 0xaaaae72cb500 (1-1) has parent 0xaaaae729fe70 (usb1) port 1
[ 0.033098] [00008601] libusb: debug [linux_get_device_address] getting address for device: 1-1.3 detached: 0
[ 0.033120] [00008601] libusb: debug [linux_get_device_address] scan 1-1.3
[ 0.033241] [00008601] libusb: debug [linux_get_device_address] bus=1 dev=3
[ 0.033258] [00008601] libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
[ 0.033273] [00008601] libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
[ 0.033384] [00008601] libusb: debug [linux_get_parent_info] dev 0xaaaae72cb380 (1-1.3) has parent 0xaaaae72cb500 (1-1) port 3
[ 0.033872] [00008601] libusb: debug [linux_get_device_address] getting address for device: 1-1.3.1 detached: 0
[ 0.033889] [00008601] libusb: debug [linux_get_device_address] scan 1-1.3.1
[ 0.034001] [00008601] libusb: debug [linux_get_device_address] bus=1 dev=5
[ 0.034019] [00008601] libusb: debug [linux_enumerate_device] busnum 1 devaddr 5 session_id 261
[ 0.034033] [00008601] libusb: debug [linux_enumerate_device] allocating new device for 1/5 (session 261)
[ 0.034146] [00008601] libusb: debug [linux_get_parent_info] dev 0xaaaae72a10d0 (1-1.3.1) has parent 0xaaaae72cb380 (1-1.3) port 1
[ 0.034641] [00008601] libusb: debug [linux_get_device_address] getting address for device: 1-1.3.1.1 detached: 0
[ 0.034662] [00008601] libusb: debug [linux_get_device_address] scan 1-1.3.1.1
[ 0.034799] [00008601] libusb: debug [linux_get_device_address] bus=1 dev=7
[ 0.034817] [00008601] libusb: debug [linux_enumerate_device] busnum 1 devaddr 7 session_id 263
[ 0.034832] [00008601] libusb: debug [linux_enumerate_device] allocating new device for 1/7 (session 263)
[ 0.034948] [00008601] libusb: debug [linux_get_parent_info] dev 0xaaaae72a14d0 (1-1.3.1.1) has parent 0xaaaae72a10d0 (1-1.3.1) port 1
[ 0.035383] [00008601] libusb: debug [linux_get_device_address] getting address for device: 1-1.3.4 detached: 0
[ 0.035401] [00008601] libusb: debug [linux_get_device_address] scan 1-1.3.4
[ 0.035517] [00008601] libusb: debug [linux_get_device_address] bus=1 dev=6
[ 0.035532] [00008601] libusb: debug [linux_enumerate_device] busnum 1 devaddr 6 session_id 262
[ 0.035547] [00008601] libusb: debug [linux_enumerate_device] allocating new device for 1/6 (session 262)
[ 0.035661] [00008601] libusb: debug [linux_get_parent_info] dev 0xaaaae72a0790 (1-1.3.4) has parent 0xaaaae72cb380 (1-1.3) port 4
[ 0.036105] [00008601] libusb: debug [linux_get_device_address] getting address for device: 1-1.3.4.3 detached: 0
[ 0.036123] [00008601] libusb: debug [linux_get_device_address] scan 1-1.3.4.3
[ 0.036236] [00008601] libusb: debug [linux_get_device_address] bus=1 dev=8
[ 0.036252] [00008601] libusb: debug [linux_enumerate_device] busnum 1 devaddr 8 session_id 264
[ 0.036267] [00008601] libusb: debug [linux_enumerate_device] allocating new device for 1/8 (session 264)
[ 0.036375] [00008601] libusb: debug [linux_get_parent_info] dev 0xaaaae72a0d60 (1-1.3.4.3) has parent 0xaaaae72a0790 (1-1.3.4) port 3
[ 0.036819] [00008601] libusb: debug [linux_get_device_address] getting address for device: 1-1.3.4.4 detached: 0
[ 0.036839] [00008601] libusb: debug [linux_get_device_address] scan 1-1.3.4.4
[ 0.036957] [00008601] libusb: debug [linux_get_device_address] bus=1 dev=9
[... cut ...]
[ 0.081370] [00008601] libusb: debug [libusb_unref_device] destroy device 1.7
[ 0.081386] [00008601] libusb: debug [libusb_unref_device] destroy device 1.5
[ 0.081402] [00008601] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.081420] [00008601] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.081438] [00008601] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.081455] [00008601] libusb: debug [usbi_remove_event_source] remove fd 4
[ 0.081497] [00008601] libusb: debug [usbi_remove_event_source] remove fd 3
```

```

</details>

mcuee, 2021-06-12:
If I look at the debug log for macOS, the strange part is probbaly the following.
[ 0.105895] [00004fde] libusb: debug [darwin_claim_interface] interface opened
[ 0.105896] [00004fde] libusb: debug [darwin_restore_state] darwin/restore_state: device state restored

```
[ 0.103885] [00004fde] libusb: debug [darwin_restore_state] darwin/restore_state: reclaiming interfaces
[ 0.103886] [00004fde] libusb: debug [darwin_restore_state] darwin/restore_state: re-claiming interface 0
[ 0.104966] [00004fde] libusb: debug [get_endpoints] building table of endpoints.
[ 0.105430] [00004fde] libusb: debug [get_endpoints] interface: 0 pipe 1: dir: 1 number: 1
[ 0.105881] [00004fde] libusb: debug [get_endpoints] interface: 0 pipe 2: dir: 0 number: 2
[ 0.105895] [00004fde] libusb: debug [darwin_claim_interface] interface opened
[ 0.105896] [00004fde] libusb: debug [darwin_restore_state] darwin/restore_state: device state restored
Closing device...
[ 0.105903] [00004fde] libusb: debug [libusb_close]  
[ 0.105992] [00004fde] libusb: debug [libusb_exit] destroying default context
[ 0.106006] [00004fe1] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 0.106079] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.10
[ 0.106169] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.8
[ 0.106217] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.32
[ 0.106260] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.106294] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.106332] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.106374] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.30
[ 0.106377] [00004fde] libusb: debug [libusb_unref_device] destroy device 2.2
Assertion failed: (refcnt >= 0), function libusb_unref_device, file ../../libusb/core.c, line 1184.
[1]    2829 abort      ./examples/xusb 0403:6001

```

mcuee, 2021-06-12:
I also tested under Windows 10 with the same two USB hubs and there are also no issues. So looks like the issue is very specific to macOS and to my setup. I would say it is not a big issue though and this can be post 1.0.25 release if we can not find a solution so soon.

hjelmn, 2021-06-12:
Wait. Now it is back to aborting. Thats odd. Let me look through the log and see if there is enough info to trace this down.

hjelmn, 2021-06-12:
Ok, and now it is back to left a reference for me. Must have been a merge from earlier that re-broke mine.

hjelmn, 2021-06-12:
One more error in reference counting. Double (or worse) counted the parent device on re-enumeration in the darwin backend.

closed 2021-06-12 by hjelmn

mcuee, 2021-06-12:
Great. There is no issue now. Now this is really closed. 👍

mcuee, 2021-06-12:
Just attach the latest debug log for reference.

<details>
<summary> Click to expand </summary>

```
libusb on  master [?] ❯ export LIBUSB_DEBUG=4                                                                                   
libusb on  master [?] ❯ ./examples/xusb 0403:6001
Using libusb v1.0.24.11630

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000035] [0000d2a7] libusb: debug [libusb_init] created default context
[ 0.000071] [0000d2a7] libusb: debug [libusb_init] libusb v1.0.24.11630
[ 0.000135] [0000d2a7] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000309] [0000d2a7] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb37ff2c
[ 0.000342] [0000d2a7] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb37ff2c
[ 0.002779] [0000d2a7] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002784] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002785] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002787] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002788] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002790] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002791] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002792] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002793] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002794] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002795] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002797] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002799] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002800] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002816] [0000d2a7] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002818] [0000d2a7] libusb: debug [process_new_device] allocating new device in context 0x132706720 for with session 0xb37ff2c
[ 0.002821] [0000d2a7] libusb: debug [process_new_device] found device with address 1 port = 2 parent = 0x0 at 0x132709b0a
[ 0.002871] [0000d2a7] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb39bc94
[ 0.002894] [0000d2a7] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb39bc94/0x2600000 against cached device with sessionID/locationID 0xb37ff2c/0x2200000
[ 0.002896] [0000d2a7] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb39bc94
[ 0.003122] [0000d2a7] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003126] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003128] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.003129] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003130] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003131] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.003133] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.003134] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.003135] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.003136] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.003137] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003138] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003139] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003141] [0000d2a7] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003153] [0000d2a7] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003155] [0000d2a7] libusb: debug [process_new_device] allocating new device in context 0x132706720 for with session 0xb39bc94
[ 0.003157] [0000d2a7] libusb: debug [process_new_device] found device with address 2 port = 6 parent = 0x0 at 0x132709c0a
[ 0.003197] [0000d2a7] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb53f9c7
[ 0.003212] [0000d2a7] libusb: debug [darwin_get_cached_device] parent sessionID: 0xb39bc94
[ 0.003214] [0000d2a7] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb53f9c7/0x2610000 against cached device with sessionID/locationID 0xb39bc94/0x2600000
[ 0.003216] [0000d2a7] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb53f9c7/0x2610000 against cached device with sessionID/locationID 0xb37ff2c/0x2200000
[ 0.003217] [0000d2a7] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb53f9c7
[... cut ...]
[ 0.115128] [0000d2a7] libusb: debug [libusb_unref_device] destroy device 2.6
[ 0.115187] [0000d2a7] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.115228] [0000d2a7] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.115273] [0000d2a7] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.115317] [0000d2a7] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.115353] [0000d2a7] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.115392] [0000d2a7] libusb: debug [usbi_remove_event_source] remove fd 3
```

</details>

commit a0e77381 (2021-06-11): core: ensure that all devices are properly cleaned up on libusb_exit

When cleaning up the context on libusb_exit the last step is to to call
hotplug_exit. This function took one pass over the devices and released any
devices where the reference count had reached 0. All remaining devices
were assumed to have leaked references and a warning message was printed
out. Unfortunately, this cleanup was too simplistic. It ignored the
references created when a device was the parent of another device (which
takes a reference). This reference is released when the child device is
released.

To ensure that [... cut]
commit 87a8578b (2021-06-11): core: really fix dangling devices

The prior fix made an incorrect assumption about the reference counting of
parent devices. This CL fixes the issue by checking if the parent device
should be removed from the list as well. This is needed as when the child
device is released the parent may also be released.

Closes #924
commit e3dda19b (2021-06-12): darwin: release device parent reference when re-enumerating device

This commit fixes a dangling device reference that can occur when a device is
being re-enumerated. The problem is the code was unconditionally re-caching the
parent device which then takes a reference on the parent device. If the field
was already populated this reference was not needed. To ensure the code works
even if the parent device also changes during the re-enumeration the code
release the parent reference and grabs a fresh reference.

Fixes #924