libusb/libusb #926: Uwp (Windows.Devices.Usb) backend for libusb

ankurverma85, 2021-06-11:
The following documentation illustrates the apis available for interfacing with usb devices from uwp processes 

https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/talking-to-usb-devices-start-to-finish#step6

Its should theoretically be possible to use cppwinrt projections of these and bridge a c++ library as a libusb backend

This would not only modernize the windows support for libusb but also enable support for using the library on non desktop based windows hardware like Xbox and hololens etc.

mcuee, 2021-06-11:
Thanks. If you can implement a prototype and create a pull-request, that will be very welcome.

ankurverma85, 2021-06-11:
I'm a complete novice when it comes to the USB internals but I'd be happy to setup the scaffolding around Creating C++ static/library that uses cppwinrt projections to interact with the apis and serve as the initial seed.
We dont actually need any dot-net .
Should be simple enough to do a purely c++ <-> c libusb-backend interfacing. 

if someone can help out with the actual code to bridge the two together I can definitely setup a branch to get things started.

ankurverma85, 2021-06-11:
I did a basic cmake based scaffolding that tries to follow the examples and query all usb devices by implemeting the usbi backend call here:

https://github.com/ankurverma85/libusb/tree/uwp

To build:
If you have VS2019 you can open the folder cmake/ in there and run the test.exe to 
The CMakeLists.txt included is configured to only use the uwp backed (in uwp.cpp) on windows, so you wont get the other windows backends.

mcuee, 2021-06-14:
Thanks for the quick start. This will probably not be a simple effort.

I tried your branch, initially cmake-gui complains missing config.h.in, so I copy config.h from the msvc directory to the root directory and also the CMakeLists.txt file. Then cmake process is okay for VS2019.

However I can not build the solution.

```
Build started...
1>------ Build started: Project: usb-1.0, Configuration: Debug x64 ------
1>core.c
1>C:\work\libusb\libusb_uwp\libusb\libusbi.h(35,10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
1>descriptor.c
1>C:\work\libusb\libusb_uwp\libusb\libusbi.h(35,10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
1>hotplug.c
1>C:\work\libusb\libusb_uwp\libusb\libusbi.h(35,10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
1>io.c
1>C:\work\libusb\libusb_uwp\libusb\libusbi.h(35,10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
1>sync.c
1>C:\work\libusb\libusb_uwp\libusb\libusbi.h(35,10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
1>strerror.c
1>C:\work\libusb\libusb_uwp\libusb\libusbi.h(35,10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
1>events_windows.c
1>C:\work\libusb\libusb_uwp\libusb\os\events_windows.c(23,10): fatal error C1083: Cannot open include file: 'libusbi.h': No such file or directory
1>threads_windows.c
1>C:\work\libusb\libusb_uwp\libusb\os\threads_windows.c(22,10): fatal error C1083: Cannot open include file: 'libusbi.h': No such file or directory
1>Generating Code...
1>Done building project "usb-1.0.vcxproj" -- FAILED.
2>------ Build started: Project: test, Configuration: Debug x64 ------
2>testlib.c
2>C:\work\libusb\libusb_uwp\tests\libusb_testlib.h(41,48): error C2061: syntax error: identifier '__attribute__'
2>C:\work\libusb\libusb_uwp\tests\libusb_testlib.h(41,48): error C2059: syntax error: ';'
2>C:\work\libusb\libusb_uwp\tests\libusb_testlib.h(41,48): error C2059: syntax error: 'constant'
2>C:\work\libusb\libusb_uwp\tests\testlib.c(132,8): warning C4996: 'freopen': This function or variable may be unsafe. Consider using freopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
2>stress.c
2>C:\work\libusb\libusb_uwp\tests\stress.c(24,10): fatal error C1083: Cannot open include file: 'libusb.h': No such file or directory
2>Generating Code...
2>Done building project "test.vcxproj" -- FAILED.
3>------ Build started: Project: ALL_BUILD, Configuration: Debug x64 ------
3>Building Custom Rule C:/work/libusb/libusb_uwp/CMakeLists.txt
========== Build: 1 succeeded, 2 failed, 1 up-to-date, 0 skipped ==========

```

ankurverma85, 2021-06-14:
oops that was my bad. 
Forgot to check-in the config.h.in
Will push shortly

The config.h from msvc wont cleanly work. 

The config.h.in is a more reliable way as it does not rely on hard-coded values but is a cmake-auto-detect equivalent to configure

photron, 2021-06-17:
I did a narrow focused incomplete libusb API implementation for UWP using CX for my specific use case. You can find it here:

https://github.com/Tinkerforge/brickd/blob/master/src/build_data/windows/libusb_uwp/libusb_uwp.cpp

The thing that took me the longest to figure out: Why do the read transfers seem to complete out-of-order. This happens because the transfer completion notification is done by a thread pool that might reorder its tasks. The rest was straight forward.

Be aware that there is a limitation to the Windows.Devices.Usb API. It only deals with USB devices that have a DeviceInterfaceGUID set in the Windows registry.

mcuee, 2021-06-30:
Indeed Opening Cmake with VS2019 works.

```
1> CMake generation started for default configuration: 'x64-Debug'.
1> Command line: "C:\WINDOWS\system32\cmd.exe" /c "%SYSTEMROOT%\System32\chcp.com 65001 >NUL && "C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2019\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe"  -G "Ninja"  -DCMAKE_BUILD_TYPE:STRING="Debug" -DCMAKE_INSTALL_PREFIX:PATH="C:\work\libusb\libusb_uwp\cmake\out\install\x64-Debug" -DCMAKE_C_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30037/bin/Hostx64/x64/cl.exe" -DCMAKE_CXX_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30037/bin/Hostx64/x64/cl.exe"  -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2019\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" "C:\work\libusb\libusb_uwp\cmake" 2>&1"
1> Working directory: C:\work\libusb\libusb_uwp\cmake\out\build\x64-Debug
1> [CMake] -- The C compiler identification is MSVC 19.29.30038.1
1> [CMake] -- The CXX compiler identification is MSVC 19.29.30038.1
1> [CMake] -- Detecting C compiler ABI info
1> [CMake] -- Detecting C compiler ABI info - done
1> [CMake] -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30037/bin/Hostx64/x64/cl.exe - skipped
1> [CMake] -- Detecting C compile features
1> [CMake] -- Detecting C compile features - done
1> [CMake] -- Detecting CXX compiler ABI info
1> [CMake] -- Detecting CXX compiler ABI info - done
1> [CMake] -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30037/bin/Hostx64/x64/cl.exe - skipped
1> [CMake] -- Detecting CXX compile features
1> [CMake] -- Detecting CXX compile features - done
1> [CMake] -- Looking for clock_gettime
1> [CMake] -- Looking for clock_gettime - not found
1> [CMake] -- Looking for pthread_condattr_setclock
1> [CMake] -- Looking for pthread_condattr_setclock - not found
1> [CMake] -- Looking for pthread_setname_np
1> [CMake] -- Looking for pthread_setname_np - not found
1> [CMake] -- Looking for pthread_threadid_np
1> [CMake] -- Looking for pthread_threadid_np - not found
1> [CMake] -- Looking for eventfd
1> [CMake] -- Looking for eventfd - not found
1> [CMake] -- Looking for pipe2
1> [CMake] -- Looking for pipe2 - not found
1> [CMake] -- Looking for syslog
1> [CMake] -- Looking for syslog - not found
1> [CMake] -- Looking for include file asm/types.h
1> [CMake] -- Looking for include file asm/types.h - not found
1> [CMake] -- Looking for include file dlfcn.h
1> [CMake] -- Looking for include file dlfcn.h - not found
1> [CMake] -- Looking for include file inttypes.h
1> [CMake] -- Looking for include file inttypes.h - found
1> [CMake] -- Looking for include file stdint.h
1> [CMake] -- Looking for include file stdint.h - found
1> [CMake] -- Looking for include file stdlib.h
1> [CMake] -- Looking for include file stdlib.h - found
1> [CMake] -- Looking for include file string.h
1> [CMake] -- Looking for include file string.h - found
1> [CMake] -- Looking for include file strings.h
1> [CMake] -- Looking for include file strings.h - not found
1> [CMake] -- Looking for include file sys/stat.h
1> [CMake] -- Looking for include file sys/stat.h - found
1> [CMake] -- Looking for include file sys/time.h
1> [CMake] -- Looking for include file sys/time.h - not found
1> [CMake] -- Looking for include file sys/types.h
1> [CMake] -- Looking for include file sys/types.h - found
1> [CMake] -- Looking for include file unistd.h
1> [CMake] -- Looking for include file unistd.h - not found
1> [CMake] -- Looking for 4 include files stdlib.h, ..., float.h
1> [CMake] -- Looking for 4 include files stdlib.h, ..., float.h - found
1> [CMake] -- Looking for EFD_CLOEXEC
1> [CMake] -- Looking for EFD_CLOEXEC - not found
1> [CMake] -- Looking for EFD_NONBLOCK
1> [CMake] -- Looking for EFD_NONBLOCK - not found
1> [CMake] -- Looking for TFD_CLOEXEC
1> [CMake] -- Looking for TFD_CLOEXEC - not found
1> [CMake] -- Looking for timerfd_create
1> [CMake] -- Looking for timerfd_create - not found
1> [CMake] -- Performing Test HAVE_STRUCT_TIMESPEC
1> [CMake] -- Performing Test HAVE_STRUCT_TIMESPEC - Success
1> [CMake] -- Configuring done
1> [CMake] -- Generating done
1> [CMake] -- Build files have been written to: C:/work/libusb/libusb_uwp/cmake/out/build/x64-Debug
1> Extracted CMake variables.
[... 17 lines cut ...]
  [7/13] Building C object CMakeFiles\usb-1.0.dir\C_\work\libusb\libusb_uwp\libusb\io.c.obj
  [8/13] Building C object CMakeFiles\usb-1.0.dir\C_\work\libusb\libusb_uwp\libusb\descriptor.c.obj
  [9/13] Building C object CMakeFiles\usb-1.0.dir\C_\work\libusb\libusb_uwp\libusb\core.c.obj
  [10/13] Building C object CMakeFiles\usb-1.0.dir\C_\work\libusb\libusb_uwp\libusb\os\events_windows.c.obj
  [11/13] Building CXX object CMakeFiles\usb-1.0.dir\C_\work\libusb\libusb_uwp\libusb\os\uwp.cpp.obj
C:\work\libusb\libusb_uwp\libusb\libusb.h(783): warning C4200: nonstandard extension used: zero-sized array in struct/union
  C:\work\libusb\libusb_uwp\libusb\libusb.h(783): note: This member will be ignored by a defaulted constructor or copy/move assignment operator
C:\work\libusb\libusb_uwp\libusb\libusb.h(808): warning C4200: nonstandard extension used: zero-sized array in struct/union
[... cut ...]
C:\work\libusb\libusb_uwp\libusb\libusbi.h(876): warning C4200: nonstandard extension used: zero-sized array in struct/union
  C:\work\libusb\libusb_uwp\libusb\libusbi.h(876): note: This member will be ignored by a defaulted constructor or copy/move assignment operator
C:\work\libusb\libusb_uwp\libusb\os\uwp.cpp(42): warning C4244: '=': conversion from '_Rep' to 'long', possible loss of data
          with
          [
              _Rep=__int64
          ]
  [12/13] Linking CXX static library usb-1.0.lib
  [13/13] Linking CXX executable test.exe

Build All succeeded.
```

mcuee, 2021-06-30:
@photron Nice work. I understand this is not complete implementation of libusb but it seems to be pretty good. Maybe you want to create a pull request and labeled it WIP. Thanks.

ankurverma85, 2021-06-30:
Thats awesome @photron . 
However looks like its not really a straight forward port from your repository into the libusb repository as a PR. 
From what I gather its going to need some major refactoring before it can be done as PR here.

Meanwhile that gives me a lot more context 
I added opening and closing of usb-device-handles and querying of device_list to the branch.
@mcuee , The tests seem to be doing only that much for now unfortunately :(

Tails86, 2025-10-01:
I've been looking into the provided sample and how winrt handles USB devices. Here are some things I have noticed while playing around with it:
- Calling ` UsbDevice::GetDeviceSelector(0xFFFF, 0xFFFF);` results in the filter string: `System.Devices.InterfaceClassGuid:="{DEE824EF-729B-4A0E-9C14-B7117D33A817}" AND System.Devices.InterfaceEnabled:=System.StructuredQueryType.Boolean#True AND System.DeviceInterface.WinUsb.UsbVendorId:=FFFF AND System.DeviceInterface.WinUsb.UsbProductId:=FFFF` 
    - This of course only finds DeviceInterfaces which conform to WinUsb
- Using the following filter string will retrieve ALL connected DeviceInterfaces for a particular VID/PID: `System.Devices.InterfaceEnabled:=System.StructuredQueryType.Boolean#True AND System.Devices.DeviceInstanceId:~~"USB\VID_FFFF&PID_FFFF"`
- There doesn't seem to be any way to get the `iManufacturer`, `iProduct`, or `iSerialNumber` from the device descriptor, but all other data is available through many convoluted calls
- String descriptors don't seem to be randomly accessible, but I haven't gone far enough down this rabbit hole yet - it may be possible to get the [raw descriptors](https://learn.microsoft.com/en-us/uwp/api/windows.devices.usb.usbdescriptor?view=winrt-26100) from the root DeviceInstance and just look for any descriptors marked as string (the above `i` values may be retrievable this way too)
- The serial may be retrieved from the end of the `DeviceInstanceId` (or Id()) of the root DeviceInstance (the one that doesn't have `MI_*` in its ID)

It doesn't seem straightforward at all, but it looks like most of the necessary data is available and could be mapped to libusb structures.

Tails86, 2025-10-25:
I'm pretty motivated to get this completed. This is in a proof-of-concept stage, so please forgive the messiness of it:
https://github.com/Tails86/libusb-cmake/pull/1/files

Any tips as far as libusb structure is concerned would be greatly appreciated! I've been hacking my way through things to try and learn how the guts of libusb are supposed to come together.

Notes:
- I used the cmake variant of libusb just because I'm more familiar with how cmake works
- This compiles for both UWP and WIN32 desktop (`WIN32_WINRT` is provided to force the winrt implementation for Windows when not WINDOWS_STORE system name)
- Using unmanaged C++ to get the greatest compatibility out of this
- winrt requires C++17, so I figured that I may as well just use C++ standard library for threading and event handling to avoid Windows.h
    - Event handling isn't quite working correctly yet - I'm getting `LIBUSB_ERROR_BUSY` errors on `do_sync_bulk_transfer()` when trying to send many transfers in succession
    - Everything works fine if I switch over to the windows platform code (this code won't compile for UWP, so this is only an option for testing purposes on Win32 desktop)
- All of the data not provided by winrt can simply be retrieved through control requests
- For some odd reason, winrt forces you to open an interface in order to do control transfers, so at least 1 interface must be available to properly list the device
- Interrupt endpoints aren't implemented yet, but that should be similar to bulk
- It doesn't look like winrt supports isochronous communication as far as I can tell