libusb/libusb #928: Openocd ST-Link connect issue using latest libusb git

mcuee, 2021-06-11:
Issue reported here.
https://sourceforge.net/p/openocd/tickets/308/

This issue is for latest libusb-1.0.dll version 1.0.24.11627 (and 1.0.24.11623 as I checked).
For older version 1.0.24.11608 all works fine.

mcuee, 2021-06-11:
Reports details:

For years I compile openocd and libusb-1.0 by myself from the newest sources (cross-compilation under ubuntu on Windows 10 for windows 10 x64). I noticed problem with the newest libusb-1.0.dll - I can't connect to st-link (ver. 2.1 and also 3). Log ends with:
```
Debug: 46 239 stlink_usb.c:3420 stlink_open(): stlink_open
Debug: 47 243 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3744 serial:
Debug: 48 248 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3748 serial:
Debug: 49 255 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374b serial:
Debug: 50 261 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374d serial:
Debug: 51 269 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374e serial:
Debug: 52 275 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374f serial:
Debug: 53 280 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3752 serial:
Debug: 54 289 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3753 serial:
```
and openocd is exited without any information like "open failed". Connecting to DAP-Link v2 with bulk works. 
This issue is for latest libusb-1.0.dll version 1.0.24.11627 (and 1.0.24.11623 as I checked).
For older version 1.0.24.11608 all works fine:
```
Debug: 46 217 stlink_usb.c:3420 stlink_open(): stlink_open
Debug: 47 221 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3744 serial:
Debug: 48 226 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3748 serial:
Debug: 49 234 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374b serial:
Debug: 50 240 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374d serial:
Debug: 51 246 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374e serial:
Debug: 52 254 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374f serial:
Debug: 53 260 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3752 serial:
Debug: 54 269 stlink_usb.c:3432 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3753 serial:
Info : 55 1722 stlink_usb.c:1318 stlink_usb_version(): STLINK V2J37M27 (API v2) VID:PID 0483:3752
Debug: 56 1729 stlink_usb.c:1543 stlink_usb_exit_mode(): MODE: 0x01
```

mcuee, 2021-06-11:
I can reproduce the issue under my Mac Mini M1. 1.0.24 release version works and latest 1.0.24.11617 does not.

1.0.24 release:
```
libusb/buildarm64 on  master [?] ❯ openocd -s /opt/homebrew/share/openocd/scripts -f board/stm32f4discovery.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J30S0 (API v2) VID:PID 0483:3748
Info : Target voltage: 2.874370
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
```

mcuee, 2021-06-11:
Debug log.

```

libusb/buildarm64 on  master [?] ❯ export LIBUSB_DEBUG=4
libusb/buildarm64 on  master [?] ❯ openocd -s /opt/homebrew/share/openocd/scripts -f board/stm32f4discovery.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000041] [0001fc88] libusb: debug [libusb_init] libusb v1.0.24.11627
[ 0.000132] [0001fc88] libusb: debug [usbi_add_event_source] add fd 5 events 1
[ 0.000349] [0001fc88] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb306a6d
[ 0.000390] [0001fc88] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb306a6d
[ 0.002216] [0001fc88] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002223] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002225] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002227] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002228] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002230] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002231] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002233] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002234] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002236] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002238] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002239] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002240] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002242] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002260] [0001fc88] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002263] [0001fc88] libusb: debug [process_new_device] allocating new device in context 0x12f804080 for with session 0xb306a6d
[ 0.002266] [0001fc88] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x12e743ada
[ 0.002321] [0001fc88] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb5716e7
[ 0.002348] [0001fc88] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb5716e7/0x2500000 against cached device with sessionID/locationID 0xb306a6d/0x2100000
[ 0.002351] [0001fc88] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb5716e7
[ 0.002619] [0001fc88] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002623] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002624] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002626] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002627] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002629] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002630] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002632] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002633] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002635] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002637] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002638] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002640] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002641] [0001fc88] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002653] [0001fc88] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002655] [0001fc88] libusb: debug [process_new_device] allocating new device in context 0x12f804080 for with session 0xb5716e7
[ 0.002658] [0001fc88] libusb: debug [process_new_device] found device with address 2 port = 5 parent = 0x0 at 0x12e743cfa
[ 0.002702] [0001fc88] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb70cc74
[ 0.002721] [0001fc88] libusb: debug [darwin_get_cached_device] parent sessionID: 0xb306a6d
[ 0.002723] [0001fc88] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb70cc74/0x2110000 against cached device with sessionID/locationID 0xb5716e7/0x2500000
[... cut ...]
[ 0.010114] [0001fc88] libusb: debug [ep_to_pipeRef] converting ep address 0x02 to pipeRef and interface
[ 0.010116] [0001fc88] libusb: debug [ep_to_pipeRef] pipe 2 on interface 0 matches
[ 0.010190] [0001fc88] libusb: debug [libusb_submit_transfer] transfer 0x12e744f68
[ 0.010193] [0001fc88] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 0.010194] [0001fc88] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[1]    28925 segmentation fault  openocd -s /opt/homebrew/share/openocd/scripts -f board/stm32f4discovery.cfg
```

mcuee, 2021-06-11:
Another report for macOS which I am not so sure if related or not.
https://sourceforge.net/p/openocd/tickets/305/
Freeze on Mac OS Catalina 10.15.7 (19H1030)

mcuee, 2021-06-11:
No issue under my Raspberry Pi 400 running Ubuntu Linux 64bit with the latest git version of openocd.

```
openocd/build on  master [?] took 21s ❯ openocd -s /usr/local/share/openocd/scripts -f board/stm32f4discovery.cfg
Open On-Chip Debugger 0.11.0+dev-00210-gbb81ec8bf (2021-06-11-22:04)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J30S0 (API v2) VID:PID 0483:3748
Info : Target voltage: 2.872962
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections

```

mcuee, 2021-06-11:
No issues with openocd 0.11 release with latest git libusb either. So Linux is probably not affected.

```
libusb/openocd-0.11.0/build took 17s ❯ openocd -s /usr/local/share/openocd/scripts -f board/stm32f4discovery.cfg 
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J30S0 (API v2) VID:PID 0483:3748
Info : Target voltage: 2.872962
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections

```

mcuee, 2021-06-12:
Debug log using lldb.

```
openocd on  master [!?] ❯ lldb ./buildarm64/src/openocd 
(lldb) target create "./buildarm64/src/openocd"
Current executable set to '/Users/mcuee/build/openocd/openocd/buildarm64/src/openocd' (arm64).
(lldb) run -s ./tcl -f board/stm32f4discovery.cfg
Process 16560 launched: '/Users/mcuee/build/openocd/openocd/buildarm64/src/openocd' (arm64)
Open On-Chip Debugger 0.11.0+dev-00210-gbb81ec8bf (2021-06-11-22:35)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Process 16560 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x128)
    frame #0: 0x00000001a14b5d9c libsystem_pthread.dylib`pthread_mutex_lock
libsystem_pthread.dylib`pthread_mutex_lock:
->  0x1a14b5d9c <+0>:  ldr    x8, [x0]
    0x1a14b5da0 <+4>:  mov    w9, #0x545a
    0x1a14b5da4 <+8>:  movk   w9, #0x4d55, lsl #16
    0x1a14b5da8 <+12>: cmp    x8, x9
Target 0: (openocd) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x128)
  * frame #0: 0x00000001a14b5d9c libsystem_pthread.dylib`pthread_mutex_lock
    frame #1: 0x00000001005d8908 libusb-1.0.0.dylib`libusb_get_next_timeout + 56
    frame #2: 0x00000001005d81e0 libusb-1.0.0.dylib`get_next_timeout + 32
    frame #3: 0x00000001005d8090 libusb-1.0.0.dylib`libusb_handle_events_timeout_completed + 84
    frame #4: 0x00000001005d8838 libusb-1.0.0.dylib`libusb_handle_events_completed + 36
    frame #5: 0x000000010004820c openocd`stlink_usb_xfer_rw [inlined] sync_transfer_wait_for_completion(transfer=0x0000000100c558e8) at stlink_usb.c:506:7 [opt]
    frame #6: 0x00000001000481e8 openocd`stlink_usb_xfer_rw at stlink_usb.c:614 [opt]
    frame #7: 0x000000010004812c openocd`stlink_usb_xfer_rw(handle=<unavailable>, cmdsize=<unavailable>, buf=<unavailable>, size=<unavailable>) at stlink_usb.c:705 [opt]
    frame #8: 0x00000001000479c0 openocd`stlink_usb_usb_xfer_noerrcheck(handle=0x0000000100c517b0, buf=<unavailable>, size=<unavailable>) at stlink_usb.c:806:8 [opt]
    frame #9: 0x00000001000470cc openocd`stlink_usb_version [inlined] stlink_usb_xfer_noerrcheck(handle=0x0000000100c517b0, buf=<unavailable>, size=6) at stlink_usb.c:231:9 [opt]
    frame #10: 0x00000001000470b8 openocd`stlink_usb_version(handle=0x0000000100c517b0) at stlink_usb.c:1155 [opt]
    frame #11: 0x00000001000476fc openocd`stlink_usb_usb_open(handle=0x0000000100c517b0, param=0x00000001002fc8e8) at stlink_usb.c:3157:9 [opt]
    frame #12: 0x0000000100045574 openocd`stlink_open [inlined] stlink_usb_open(handle=0x0000000100c517b0, param=0x00000001002fc8e8) at stlink_usb.c:218:9 [opt]
    frame #13: 0x0000000100045564 openocd`stlink_open(param=0x00000001002fc8e8, mode=<unavailable>, fd=0x00000001002fc938) at stlink_usb.c:3442 [opt]
    frame #14: 0x000000010000cffc openocd`hl_layout_open(adapter=0x00000001002fc8e8) at hla_layout.c:44:8 [opt]
    frame #15: 0x000000010000c508 openocd`hl_interface_open(tr=HL_TRANSPORT_SWD) at hla_interface.c:56:15 [opt]
    frame #16: 0x000000010000d578 openocd`hl_transport_init(cmd_ctx=<unavailable>) at hla_transport.c:183:15 [opt]
    frame #17: 0x00000001001d172c openocd`jim_command_dispatch [inlined] run_command(context=0x0000000100c07230, c=0x0000000100c37410, words=0x0000000100c3f570, num_words=1) at command.c:582:15 [opt]
    frame #18: 0x00000001001d16e0 openocd`jim_command_dispatch [inlined] exec_command(interp=0x0000000100c09600, cmd_ctx=0x0000000100c07230, c=0x0000000100c37410, argc=1, argv=0x0000000100c3f750) at command.c:950 [opt]
    frame #19: 0x00000001001d1698 openocd`jim_command_dispatch(interp=0x0000000100c09600, argc=1, argv=0x0000000100c3f750) at command.c:1001 [opt]
    frame #20: 0x00000001001e5940 openocd`JimInvokeCommand(interp=0x0000000100c09600, objc=1, objv=0x0000000100c3f750) at jim.c:10514:19 [opt]
    frame #21: 0x00000001001de530 openocd`Jim_EvalObjVector(interp=0x0000000100c09600, objc=1, objv=0x0000000100c3f750) at jim.c:10576:15 [opt]
    frame #22: 0x00000001001e5b60 openocd`Jim_EvalObjPrefix(interp=<unavailable>, prefix=0x0000000100c2af70, objc=0, objv=<unavailable>) at jim.c:10595:11 [opt]
    frame #23: 0x00000001001d15bc openocd`jim_command_dispatch(interp=0x0000000100c09600, argc=2, argv=0x000000016fdff0a8) at command.c:967:17 [opt]
    frame #24: 0x00000001001e5940 openocd`JimInvokeCommand(interp=0x0000000100c09600, objc=2, objv=0x000000016fdff0a8) at jim.c:10514:19 [opt]
    frame #25: 0x00000001001e6124 openocd`Jim_EvalObj(interp=0x0000000100c09600, scriptObjPtr=<unavailable>) at jim.c:11006:23 [opt]
    frame #26: 0x00000001001e7970 openocd`Jim_EvalSource(interp=0x0000000100c09600, filename=0x0000000000000000, lineno=0, script=<unavailable>) at jim.c:11298:18 [opt]
    frame #27: 0x00000001001d2168 openocd`command_run_line(context=0x0000000100c07230, line="transport init") at command.c:626:14 [opt]
    frame #28: 0x0000000100001044 openocd`handle_init_command(cmd=0x000000016fdff210) at openocd.c:152:11 [opt]
[... cut ...]
    frame #34: 0x00000001001e7970 openocd`Jim_EvalSource(interp=0x0000000100c09600, filename=0x0000000000000000, lineno=0, script=<unavailable>) at jim.c:11298:18 [opt]
    frame #35: 0x00000001001d2168 openocd`command_run_line(context=0x0000000100c07230, line="init") at command.c:626:14 [opt]
    frame #36: 0x0000000100000e14 openocd`openocd_main [inlined] openocd_thread(argc=5, argv=0x000000016fdff4a8, cmd_ctx=0x0000000100c07230) at openocd.c:310:9 [opt]
    frame #37: 0x0000000100000dcc openocd`openocd_main(argc=5, argv=0x000000016fdff4a8) at openocd.c:356 [opt]
    frame #38: 0x00000001a14d9450 libdyld.dylib`start + 4
(lldb) 

```

mcuee, 2021-06-12:
It seems the issue is somewhere in openocd as the other JTAG debuggers seem to work fine.

Eg: using JTAGKey 2 with another STM32F103 board.

```
openocd/test/stm32 ❯ openocd -s /opt/homebrew/share/openocd/scripts -f ./stm32_jtagkey2.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Info : JTAG tap: stm32f1x.cpu tap/device found: 0x3ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x3)
Info : JTAG tap: stm32f1x.bs tap/device found: 0x16410041 (mfg: 0x020 (STMicroelectronics), part: 0x6410, ver: 0x1)
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: stm32f1x.cpu tap/device found: 0x3ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x3)
Info : JTAG tap: stm32f1x.bs tap/device found: 0x16410041 (mfg: 0x020 (STMicroelectronics), part: 0x6410, ver: 0x1)
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800272c msp: 0x20000810
Info : device id = 0x20036410
Info : flash size = 128kbytes
Info : JTAG tap: stm32f1x.cpu tap/device found: 0x3ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x3)
Info : JTAG tap: stm32f1x.bs tap/device found: 0x16410041 (mfg: 0x020 (STMicroelectronics), part: 0x6410, ver: 0x1)
shutdown command invoked
```

mcuee, 2021-06-13:
openocd st-link driver:
https://github.com/ntfreak/openocd/blob/master/src/jtag/drivers/stlink_usb.c

mcuee, 2021-06-13:
Debug log from openocd side.

```
                                                  
openocd/test/stm32 ❯ openocd -d -s /usr/local/share/openocd/scripts -f board/stm32f4discovery.cfg
Open On-Chip Debugger 0.11.0+dev-00210-gbb81ec8bf (2021-06-13-10:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 1 options.c:63 configuration_output_handler(): debug_level: 3
User : 4 1 options.c:63 configuration_output_handler(): 
Debug: 5 1 configuration.c:44 add_script_search_dir(): adding /usr/local/share/openocd/scripts
Debug: 6 1 options.c:244 add_default_dirs(): bindir=/opt/homebrew/Cellar/open-ocd/HEAD-bb81ec8/bin
Debug: 7 1 options.c:245 add_default_dirs(): pkgdatadir=/opt/homebrew/Cellar/open-ocd/HEAD-bb81ec8/share/openocd
Debug: 8 1 options.c:246 add_default_dirs(): exepath=/opt/homebrew/Cellar/open-ocd/HEAD-bb81ec8/bin
Debug: 9 1 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 10 1 configuration.c:44 add_script_search_dir(): adding /Users/mcuee/Library/Preferences/org.openocd
Debug: 11 1 configuration.c:44 add_script_search_dir(): adding /Users/mcuee/.config/openocd
Debug: 12 1 configuration.c:44 add_script_search_dir(): adding /Users/mcuee/.openocd
Debug: 13 1 configuration.c:44 add_script_search_dir(): adding /opt/homebrew/Cellar/open-ocd/HEAD-bb81ec8/bin/../share/openocd/site
Debug: 14 1 configuration.c:44 add_script_search_dir(): adding /opt/homebrew/Cellar/open-ocd/HEAD-bb81ec8/bin/../share/openocd/scripts
Debug: 15 1 command.c:201 script_debug(): command - ocd_find board/stm32f4discovery.cfg
Debug: 16 1 configuration.c:99 find_file(): found /usr/local/share/openocd/scripts/board/stm32f4discovery.cfg
Debug: 17 1 command.c:201 script_debug(): command - ocd_find interface/stlink.cfg
Debug: 18 1 configuration.c:99 find_file(): found /usr/local/share/openocd/scripts/interface/stlink.cfg
Debug: 19 1 command.c:201 script_debug(): command - adapter driver hla
Debug: 20 1 command.c:201 script_debug(): command - adapter driver hla
Debug: 21 1 command.c:201 script_debug(): command - hla_layout stlink
Debug: 22 1 hla_interface.c:242 hl_interface_handle_layout_command(): hl_interface_handle_layout_command
Debug: 23 1 command.c:201 script_debug(): command - hla_device_desc ST-LINK
Debug: 24 1 hla_interface.c:216 hl_interface_handle_device_desc_command(): hl_interface_handle_device_desc_command
Debug: 25 1 command.c:201 script_debug(): command - hla_vid_pid 0x0483 0x3744 0x0483 0x3748 0x0483 0x374b 0x0483 0x374d 0x0483 0x374e 0x0483 0x374f 0x0483 0x3752 0x0483 0x3753
Debug: 26 1 command.c:201 script_debug(): command - transport select hla_swd
Debug: 27 1 command.c:201 script_debug(): command - transport select hla_swd
Debug: 28 1 hla_transport.c:205 hl_swd_transport_select(): hl_swd_transport_select
Debug: 29 1 command.c:201 script_debug(): command - ocd_find target/stm32f4x.cfg
Debug: 30 1 configuration.c:99 find_file(): found /usr/local/share/openocd/scripts/target/stm32f4x.cfg
Debug: 31 1 command.c:201 script_debug(): command - ocd_find target/swj-dp.tcl
Debug: 32 1 configuration.c:99 find_file(): found /usr/local/share/openocd/scripts/target/swj-dp.tcl
Debug: 33 1 command.c:201 script_debug(): command - transport select
Debug: 34 1 command.c:201 script_debug(): command - transport select
Debug: 35 1 command.c:201 script_debug(): command - ocd_find mem_helper.tcl
Debug: 36 1 configuration.c:99 find_file(): found /usr/local/share/openocd/scripts/mem_helper.tcl
Debug: 37 2 command.c:201 script_debug(): command - add_usage_text mrw address
Debug: 38 2 command.c:201 script_debug(): command - add_help_text mrw Returns value of word in memory.
Debug: 39 2 command.c:201 script_debug(): command - add_usage_text mrh address
Debug: 40 2 command.c:201 script_debug(): command - add_help_text mrh Returns value of halfword in memory.
Debug: 41 2 command.c:201 script_debug(): command - add_usage_text mrb address
Debug: 42 2 command.c:201 script_debug(): command - add_help_text mrb Returns value of byte in memory.
Debug: 43 2 command.c:201 script_debug(): command - add_usage_text mmw address setbits clearbits
Debug: 44 2 command.c:201 script_debug(): command - add_help_text mmw Modify word in memory. new_val = (old_val & ~clearbits) | setbits;
Debug: 45 2 command.c:201 script_debug(): command - transport select
Debug: 46 2 command.c:201 script_debug(): command - transport select
Debug: 47 2 command.c:201 script_debug(): command - transport select
Debug: 48 2 command.c:201 script_debug(): command - transport select
Debug: 49 2 command.c:201 script_debug(): command - transport select
Debug: 50 2 command.c:201 script_debug(): command - transport select
Debug: 51 2 command.c:201 script_debug(): command - swd newdap stm32f4x cpu -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id 0x2ba01477
Debug: 52 2 command.c:201 script_debug(): command - swd newdap stm32f4x cpu -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id 0x2ba01477
Debug: 53 2 hla_tcl.c:111 jim_hl_newtap_cmd(): Creating New Tap, Chip: stm32f4x, Tap: cpu, Dotted: stm32f4x.cpu, 8 params
Debug: 54 2 hla_tcl.c:121 jim_hl_newtap_cmd(): Processing option: -irlen
Debug: 55 2 hla_tcl.c:121 jim_hl_newtap_cmd(): Processing option: -ircapture
Debug: 56 2 hla_tcl.c:121 jim_hl_newtap_cmd(): Processing option: -irmask
Debug: 57 2 hla_tcl.c:121 jim_hl_newtap_cmd(): Processing option: -expected-id
Debug: 58 2 core.c:1484 jtag_tap_init(): Created Tap: stm32f4x.cpu @ abs position 0, irlen 0, capture: 0x0 mask: 0x0
[... cut ...]
Debug: 155 3 stlink_usb.c:3434 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374d serial: 
Debug: 156 3 stlink_usb.c:3434 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374e serial: 
Debug: 157 3 stlink_usb.c:3434 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374f serial: 
Debug: 158 3 stlink_usb.c:3434 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3752 serial: 
Debug: 159 3 stlink_usb.c:3434 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3753 serial: 
[1]    72547 segmentation fault  openocd -d -s /usr/local/share/openocd/scripts -f board/stm32f4discovery.cfg
```

mcuee, 2021-06-13:
It seems to me that the application can not find the ST-Link V2 device. Strange.

mcuee, 2021-06-13:
I was trying to build st-link with latest libusb-git to see if st-link is okay or not, but it failed to build.

mcuee, 2021-06-13:
After fixing the cmake build issue of stlink project, it has no issues connecting to ST-Link. So the issue seems to be with openocd stlink codes.

mcuee, 2021-06-19:
I can reproduce on Windows side as well.

1.0.24 release 1.0.24.11584
```
$ ./bin/openocd.exe -f board/stm32f4discovery.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J30S0 (API v2) VID:PID 0483:3748
Info : Target voltage: 2.869596
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
shutdown command invoked
```

Latest git 1.0.24.11630
```
$ ./bin/openocd.exe -f board/stm32f4discovery.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Segmentation fault
```

mcuee, 2021-06-23:
Looks like the big patch in libusb (https://github.com/libusb/libusb/commit/32a22069428cda9d63aa666e92fb8882a83d4515) is causing openocd to fail.

```

openocd/test/stm32 ❯ brew info libusb
libusb: stable 1.0.24 (bottled), HEAD
Library for USB device access
https://libusb.info/
/opt/homebrew/Cellar/libusb/HEAD-32a2206 (6 files, 767.5KB) *
  Built from source
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/libusb.rb
License: LGPL-2.1-or-later
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 60,921 (30 days), 194,606 (90 days), 818,103 (365 days)
install-on-request: 6,327 (30 days), 18,921 (90 days), 78,245 (365 days)
build-error: 0 (30 days)
openocd/test/stm32 ❯ openocd -s /opt/homebrew/share/openocd/scripts -f board/stm32f4discovery.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
[1]    19689 segmentation fault  openocd -s /opt/homebrew/share/openocd/scripts -f board/stm32f4discovery.cfg
```

mcuee, 2021-06-23:
But I am not so sure if libusb change is the real issue or not. It could also be openocd st-link code has some underlying issues which cause the libusb changes to trigger the fault.

mcuee, 2021-06-23:
From https://sourceforge.net/p/openocd/tickets/308/

Looks like openocde code is the issue.
https://github.com/ntfreak/openocd/blob/master/src/jtag/drivers/stlink_usb.c#L496
```
static void sync_transfer_wait_for_completion(struct libusb_transfer *transfer)
{
	int r, *completed = transfer->user_data;

	/* Assuming a single libusb context exists.  There no existing interface into this
	 * module to pass a libusb context.
	 */
	struct libusb_context *ctx = NULL;

	while (!*completed) {
		r = libusb_handle_events_completed(ctx, completed);
		if (r < 0) {
			if (r == LIBUSB_ERROR_INTERRUPTED)
				continue;
			libusb_cancel_transfer(transfer);
			continue;
		}
	}
}
```

But it uses the libusb_helper.c function here.
https://github.com/ntfreak/openocd/blob/master/src/jtag/drivers/libusb_helper.c#L156
```
int jtag_libusb_open(const uint16_t vids[], const uint16_t pids[],
		const char *serial,
		struct libusb_device_handle **out,
		adapter_get_alternate_serial_fn adapter_get_alternate_serial)
{
	int cnt, idx, errCode;
	int retval = ERROR_FAIL;
	bool serial_mismatch = false;
	struct libusb_device_handle *libusb_handle = NULL;

	if (libusb_init(&jtag_libusb_context) < 0)
		return ERROR_FAIL;

	cnt = libusb_get_device_list(jtag_libusb_context, &devs);
```

mcuee, 2021-06-23:
So based on the above conclusion, looks like the big patch in libusb (32a2206) changed a bit of the behavior of libusb code but openocd stlink_usb.c does not handle the context right. 

Why FTDI adapter does not have such issue, because it does not use libusb_helper.h and uses properly libusb by itself with the mpsse backend.
https://github.com/ntfreak/openocd/blob/master/src/jtag/drivers/mpsse.c

Why does other libusb_helper based adapter not have such an issue, because they do not use async transfer and not use the context.

mcuee, 2021-06-23:
To proof my analysis, I use the following dirty fix of libusb_helper.c.

```
diff --git a/src/jtag/drivers/libusb_helper.c b/src/jtag/drivers/libusb_helper.c
index f0122d534..05c6e5574 100644
--- a/src/jtag/drivers/libusb_helper.c
+++ b/src/jtag/drivers/libusb_helper.c
@@ -163,7 +163,7 @@ int jtag_libusb_open(const uint16_t vids[], const uint16_t pids[],
        bool serial_mismatch = false;
        struct libusb_device_handle *libusb_handle = NULL;
 
-       if (libusb_init(&jtag_libusb_context) < 0)
+       if (libusb_init(NULL) < 0)
                return ERROR_FAIL;
 
        cnt = libusb_get_device_list(jtag_libusb_context, &devs);
```

Then openocd git latest HEAD has no issues with latest libsub git HEAD.

```
openocd on  master [!] took 1m36s ❯ brew info libusb
libusb: stable 1.0.24 (bottled), HEAD
Library for USB device access
https://libusb.info/
/opt/homebrew/Cellar/libusb/HEAD-e3dda19 (22 files, 565.0KB) *
  Built from source on 2021-06-23 at 22:08:47
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/libusb.rb
License: LGPL-2.1-or-later
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 60,921 (30 days), 194,606 (90 days), 818,103 (365 days)
install-on-request: 6,327 (30 days), 18,921 (90 days), 78,245 (365 days)
build-error: 0 (30 days)

openocd on  master [!] ❯ ./src/openocd -s ./tcl -f board/stm32f4discovery.cfg
Open On-Chip Debugger 0.11.0+dev-00225-g42a0bf3c3-dirty (2021-06-23-22:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J30S0 (API v2) VID:PID 0483:3748
Info : Target voltage: 2.876336
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections

```

mcuee, 2021-06-24:
Antonio Borneo has proposed a fix and it is under review in openocd gerrit. I will close this as it is not an issue with libusb.

http://openocd.zylin.com/#/c/6331/1
```
stlink: fix SIGSEGV with libusb v1.0.24-33-g32a2206 (11618)

The stlink driver incorrectly uses a NULL pointer for libusb's
struct libusb_context.
The correct value to be used is local in libusb_helper.c.

Move in the helper file, in a wrapper function, the only call that
requires the above value, and let stlink driver to use this
wrapper.

This issue has not triggered any visible problem until a code
refactoring [1] in libusb has made OpenOCD crashing on Windows and
on MacOS.
```

closed 2021-06-24 by mcuee

mcuee, 2022-02-11:
"brew install openocd --HEAD" should fix the issue as well.

mcuee, 2022-02-11:
The issue is also reported here now that libusb-1.0.25 has been released.
https://github.com/Homebrew/homebrew-core/pull/94207#issuecomment-1029206795

dot4qu, 2022-02-12:
I tried to install the newest of openocd but I assumed the patch hadn't been released yet. That's a better solution, thanks!