libusb/libusb #930: fxload example under macOS

mcuee, 2021-06-12:
The following error message may not be a real error.

```
~/build/libusb ❯ ./libusb/examples/fxload -i ./CYStream.hex
found device 'Cypress EZ-USB FX2LP (68013A/68014A/68015A/68016A)' [04b4:8613] (2,24)
microcontroller type: fx2lp
./CYStream.hex: type Intel HEX
stop CPU
... WROTE: 4571 bytes, 47 segments, avg 97
reset CPU
libusb: error [darwin_reenumerate_device] USBDeviceReEnumerate: device not opened for exclusive access
```

mcuee, 2021-06-13:
```

~/build/libusb ❯ ./libusb/examples/fxload -i ./CYStream.hex 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000047] [00029fb6] libusb: debug [libusb_init] created default context
[ 0.000084] [00029fb6] libusb: debug [libusb_init] libusb v1.0.24.11630
[ 0.000148] [00029fb6] libusb: debug [usbi_add_event_source] add fd 3 events 1
... skipped ...
found device 'Cypress EZ-USB FX2LP (68013A/68014A/68015A/68016A)' [04b4:8613] (2,7)
[ 0.008208] [00029fb6] libusb: debug [libusb_open] open 2.7
[ 0.008288] [00029fb6] libusb: debug [darwin_open] device open for access
[ 0.008296] [00029fb6] libusb: debug [libusb_claim_interface] interface 0
[ 0.008332] [00029fb6] libusb: debug [darwin_reenumerate_device] darwin/reenumerate_device: restoring state...
[ 0.008345] [00029fb6] libusb: debug [darwin_open] device open for access
[ 0.008347] [00029fb6] libusb: debug [darwin_restore_state] darwin/restore_state: reclaiming interfaces
[ 0.008349] [00029fb6] libusb: debug [darwin_restore_state] darwin/restore_state: device state restored
[ 0.008364] [00029fb6] libusb: info [darwin_claim_interface] no interface found; setting configuration: 1
[ 0.009172] [00029fb6] libusb: debug [get_endpoints] building table of endpoints.
[ 0.009193] [00029fb6] libusb: debug [darwin_claim_interface] interface opened
microcontroller type: fx2lp
./CYStream.hex: type Intel HEX
stop CPU
[ 0.009250] [00029fb6] libusb: debug [libusb_alloc_transfer] transfer 0x127610108
[ 0.009255] [00029fb6] libusb: debug [libusb_submit_transfer] transfer 0x127610108
[ 0.009285] [00029fb6] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.009296] [00029fb6] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.009298] [00029fb6] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.009316] [00029fb6] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 0.009340] [00029fb9] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.009351] [00029fb6] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.009355] [00029fb6] libusb: debug [handle_event_trigger] event triggered
[ 0.009356] [00029fb6] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 0.009358] [00029fb6] libusb: debug [usbi_handle_transfer_completion] transfer 0x127610108 has callback 0x102d8fb54
[ 0.009360] [00029fb6] libusb: debug [sync_transfer_cb] actual_length=1
[ 0.009377] [00029fb6] libusb: debug [libusb_free_transfer] transfer 0x127610108
[ 0.009596] [00029fb6] libusb: debug [libusb_alloc_transfer] transfer 0x127610108
[ 0.009600] [00029fb6] libusb: debug [libusb_submit_transfer] transfer 0x127610108
[ 0.009621] [00029fb6] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.009623] [00029fb6] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.009625] [00029fb6] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 0.009669] [00029fb9] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.009679] [00029fb6] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.009681] [00029fb6] libusb: debug [handle_event_trigger] event triggered
[ 0.009683] [00029fb6] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 0.009685] [00029fb6] libusb: debug [usbi_handle_transfer_completion] transfer 0x127610108 has callback 0x102d8fb54
[ 0.009687] [00029fb6] libusb: debug [sync_transfer_cb] actual_length=18
[ 0.009690] [00029fb6] libusb: debug [libusb_free_transfer] transfer 0x127610108
[ 0.009693] [00029fb6] libusb: debug [libusb_alloc_transfer] transfer 0x127610108
[ 0.009694] [00029fb6] libusb: debug [libusb_submit_transfer] transfer 0x127610108
[ 0.009714] [00029fb6] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.009716] [00029fb6] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.009718] [00029fb6] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 0.009760] [00029fb9] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.009772] [00029fb6] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.009774] [00029fb6] libusb: debug [handle_event_trigger] event triggered
[ 0.009775] [00029fb6] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 0.009778] [00029fb6] libusb: debug [usbi_handle_transfer_completion] transfer 0x127610108 has callback 0x102d8fb54
[ 0.009779] [00029fb6] libusb: debug [sync_transfer_cb] actual_length=10
[ 0.009782] [00029fb6] libusb: debug [libusb_free_transfer] transfer 0x127610108
[ 0.009789] [00029fb6] libusb: debug [libusb_alloc_transfer] transfer 0x127706378
[ 0.009791] [00029fb6] libusb: debug [libusb_submit_transfer] transfer 0x127706378
[ 0.009821] [00029fb6] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.009823] [00029fb6] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.009825] [00029fb6] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[... cut ...]
[ 1.831328] [00029fb6] libusb: debug [libusb_unref_device] destroy device 2.6
[ 1.831358] [00029fb6] libusb: debug [libusb_unref_device] destroy device 2.4
[ 1.831534] [00029fb6] libusb: debug [libusb_unref_device] destroy device 2.5
[ 1.831564] [00029fb6] libusb: debug [libusb_unref_device] destroy device 2.3
[ 1.831686] [00029fb6] libusb: debug [libusb_unref_device] destroy device 2.2
[ 1.831724] [00029fb6] libusb: debug [libusb_unref_device] destroy device 2.1
[ 1.831754] [00029fb6] libusb: debug [usbi_remove_event_source] remove fd 3
```

mcuee, 2021-06-13:
Basically the device actually changed to a different PID after reset.

mcuee, 2021-06-13:
The question whether the following should be a warning or a fault.
https://github.com/libusb/libusb/blob/master/libusb/os/darwin_usb.c#L1730

```
  /* from macOS 10.11 ResetDevice no longer does anything so just use USBDeviceReEnumerate */
  kresult = (*(dpriv->device))->USBDeviceReEnumerate (dpriv->device, options);
  if (kresult != kIOReturnSuccess) {
    usbi_err (HANDLE_CTX (dev_handle), "USBDeviceReEnumerate: %s", darwin_error_str (kresult));
    dpriv->in_reenumerate = false;
    return darwin_to_libusb (kresult);
  }
```

hjelmn, 2021-06-14:
Yeah. This doesn't seem right given that debug log. Will take a look.

mcuee, 2021-07-10:
Latest git commit (https://github.com/libusb/libusb/commit/339d24c8235476ceee43f7d704415ccab85e2583) which fixed #945 also fixes this one.

closed 2021-07-10 by mcuee