libusb/libusb #931: Failed to build stlink with latest libusb git under macOS

mcuee, 2021-06-13:
https://github.com/stlink-org/stlink/issues/1152
I am not so sure if this is a libusb (eg: pkg-config script) issue or stlink issue.

libusb git has introduced a new function under macOS and this seems to break st-link build.

```
stlink/buildm1 on  develop [?] via △ v3.20.3 took 2s ❯ sw_vers
ProductName:	macOS
ProductVersion:	11.4
BuildVersion:	20F71

stlink/buildm1 on  develop [?] via △ v3.20.3 ❯ make -i
Consolidate compiler generated dependencies of target stlink-static
[ 22%] Built target stlink-static
Consolidate compiler generated dependencies of target st-trace
[ 25%] Linking C executable bin/st-trace
Undefined symbols for architecture arm64:
  "_SecTaskCopyValueForEntitlement", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_SecTaskCreateFromSelf", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[ 27%] Built target st-trace
[ 30%] Building C object CMakeFiles/st-flash.dir/src/st-flash/flash.c.o
[ 32%] Building C object CMakeFiles/st-flash.dir/src/st-flash/flash_opts.c.o
[ 35%] Linking C executable bin/st-flash
Undefined symbols for architecture arm64:
  "_SecTaskCopyValueForEntitlement", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_SecTaskCreateFromSelf", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[ 35%] Built target st-flash
[ 37%] Building C object CMakeFiles/stlink-shared.dir/src/common.c.o
[ 40%] Building C object CMakeFiles/stlink-shared.dir/src/stlink-lib/chipid.c.o
[ 42%] Building C object CMakeFiles/stlink-shared.dir/src/stlink-lib/flash_loader.c.o
[ 45%] Building C object CMakeFiles/stlink-shared.dir/src/stlink-lib/logging.c.o
[ 47%] Building C object CMakeFiles/stlink-shared.dir/src/stlink-lib/md5.c.o
[ 50%] Building C object CMakeFiles/stlink-shared.dir/src/stlink-lib/sg.c.o
[ 52%] Building C object CMakeFiles/stlink-shared.dir/src/stlink-lib/usb.c.o
[ 55%] Building C object CMakeFiles/stlink-shared.dir/src/stlink-lib/helper.c.o
[ 57%] Linking C shared library lib/libstlink.dylib
Undefined symbols for architecture arm64:
  "_SecTaskCopyValueForEntitlement", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_SecTaskCreateFromSelf", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[ 57%] Built target stlink-shared
[ 60%] Building C object CMakeFiles/st-util.dir/src/st-util/gdb-remote.c.o
[ 62%] Building C object CMakeFiles/st-util.dir/src/st-util/gdb-server.c.o
[ 65%] Building C object CMakeFiles/st-util.dir/src/st-util/semihosting.c.o
[ 67%] Linking C executable bin/st-util
Undefined symbols for architecture arm64:
  "_SecTaskCopyValueForEntitlement", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_SecTaskCreateFromSelf", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[ 67%] Built target st-util
[ 70%] Building C object CMakeFiles/st-info.dir/src/st-info/info.c.o
[ 72%] Linking C executable bin/st-info
Undefined symbols for architecture arm64:
  "_SecTaskCopyValueForEntitlement", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_SecTaskCreateFromSelf", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[ 72%] Built target st-info
[ 75%] Building C object src/stlink-gui/CMakeFiles/stlink-gui.dir/gui.c.o
make[2]: *** No rule to make target `lib/libstlink.1.7.0.dylib', needed by `bin/stlink-gui'.  Stop.
[ 77%] Built target stlink-gui
[ 80%] Building C object src/stlink-gui/CMakeFiles/stlink-gui-local.dir/gui.c.o
make[2]: *** No rule to make target `lib/libstlink.1.7.0.dylib', needed by `bin/stlink-gui-local'.  Stop.
[ 82%] Built target stlink-gui-local
[ 85%] Building C object tests/CMakeFiles/test-flash.dir/flash.c.o
[ 87%] Building C object tests/CMakeFiles/test-flash.dir/__/src/st-flash/flash_opts.c.o
[ 90%] Linking C executable ../bin/test-flash
[ 90%] Built target test-flash
[ 92%] Building C object tests/CMakeFiles/test-usb.dir/usb.c.o
[ 95%] Linking C executable ../bin/test-usb
Undefined symbols for architecture arm64:
  "_SecTaskCopyValueForEntitlement", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_SecTaskCreateFromSelf", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[ 95%] Built target test-usb
[ 97%] Building C object tests/CMakeFiles/test-sg.dir/sg.c.o
[100%] Linking C executable ../bin/test-sg
Undefined symbols for architecture arm64:
  "_SecTaskCopyValueForEntitlement", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_SecTaskCreateFromSelf", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[100%] Built target test-sg

```

mcuee, 2021-06-13:
Relevant commit is here: https://github.com/libusb/libusb/pull/911/commits/9f8df764171abd870c680908bb6d37e857558090

Looks like application needs to add the Security framework if they use libusb-1.0.a.
 	
LIBS="${LIBS} -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -Wl,-framework,Security"

mcuee, 2021-06-13:
So it seems to me applications using CMake and libusb needs to be updated. I do not know much about CMake and my simple fix does not work.

```
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -187,7 +187,8 @@ if (APPLE)     # ... with Apple macOS libraries
     find_library(ObjC objc)
     find_library(CoreFoundation CoreFoundation)
     find_library(IOKit IOKit)
-    target_link_libraries(${STLINK_LIB_SHARED} ${LIBUSB_LIBRARY} ${SSP_LIB} ${ObjC} ${CoreFoundation} ${IOKit})
+    find_library(Security Security)
+    target_link_libraries(${STLINK_LIB_SHARED} ${LIBUSB_LIBRARY} ${SSP_LIB} ${ObjC} ${CoreFoundation} ${IOKit} ${Security})
 elseif (WIN32) # ... with Windows libraries
     target_link_libraries(${STLINK_LIB_SHARED} ${LIBUSB_LIBRARY} ${SSP_LIB} wsock32 ws2_32)
 else ()
@@ -229,7 +230,8 @@ if (APPLE)     # ... with Apple macOS libraries
     find_library(ObjC objc)
     find_library(CoreFoundation CoreFoundation)
     find_library(IOKit IOKit)
-    target_link_libraries(${STLINK_LIB_STATIC} ${LIBUSB_LIBRARY} ${SSP_LIB} ${ObjC} ${CoreFoundation} ${IOKit})
+    find_library(Security Security)
+    target_link_libraries(${STLINK_LIB_SHARED} ${LIBUSB_LIBRARY} ${SSP_LIB} ${ObjC} ${CoreFoundation} ${IOKit} ${Security})
 elseif (WIN32) # ... with Windows libraries
     target_link_libraries(${STLINK_LIB_STATIC} ${LIBUSB_LIBRARY} ${SSP_LIB} wsock32 ws2_32)
 else ()

```

mcuee, 2021-06-13:
Anyway, I think this is not a libusb issue.

mcuee, 2021-06-13:
Somehow stlink wants to use libusb-1.0.a for static link. If using libusb-1.0.dylib maybe things will be easier.
The followng simple patch fixed the issue if stlink is willing to use dynamic link with libusb under macOS.

```
diff --git a/cmake/modules/Findlibusb.cmake b/cmake/modules/Findlibusb.cmake
index d861bdd..330dbb0 100644
--- a/cmake/modules/Findlibusb.cmake
+++ b/cmake/modules/Findlibusb.cmake
@@ -16,7 +16,7 @@ if (APPLE)                                                 # macOS
         HINTS /usr /usr/local /opt
         PATH_SUFFIXES libusb-1.0
         )
-    set(LIBUSB_NAME libusb-1.0.a)
+    set(LIBUSB_NAME libusb-1.0.dylib)
     find_library(
         LIBUSB_LIBRARY NAMES ${LIBUSB_NAME}
         HINTS /usr /usr/local /opt
```

hjelmn, 2021-06-14:
Yeah. Don't think this is a libusb issues but I will check. If a project wants to static link libusb they need to supply all the linker dependencies. In the past this has been mainly IOKit but now security is needed. Best to dynamic link is possible.

mcuee, 2021-06-14:
I checked and I do not see anything wrong from the libusb side. So I will close this one. Anyway I have the issue raised in stlink github page. If they find something wrong in libusb, I can reopen this issue.

closed 2021-06-14 by mcuee

LudovicRousseau, 2021-06-14:
The libusb pkg-config file (generated on macOS Mojave) `libusb-1.0.pc` contains:
```
Libs: -L${libdir} -lusb-1.0
Libs.private:  -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -Wl,-framework,Security
```
The Security framefork is already present.

I am not a pkg-config expert. But maybe stlink is not using the correct information.
Something like `pkg-config --libs --static libusb-1.0` should return the correct information.

mcuee, 2021-06-14:
That is the whole problem, they do not use pkg-config under macOS.
https://github.com/stlink-org/stlink/blob/develop/cmake/modules/Findlibusb.cmake

mcuee, 2021-06-14:
And not so sure why they even delete my issue.