libusb/libusb #934: LibUSB - Streaming Video to a Microcontroller Stutters (LibUSBK Bulk Mode)

Cybdyn-Systems, 2021-06-15:
I am trying to stream a video file to a MCU, but whenever it plays the video on the device it's stuttering pretty moderately. If I use my PC whilst streaming, it gets considerably worse.

For example, if I am streaming and then I scroll through chat history in say, Discord, the stuttering becomes much, much worse which is incredibly strange, but may be a key factor in finding the root of the problem.

NOTE: We must stream 2352 bytes in 512 byte chunks (this is for the ISO9660 disc video format I am streaming and is critical to the system that is playing it so do not think that this is the problem or error).

Here is my current code implementation.

```
// main streaming loop
// we send the data by 512 bytes, so 2352 fits into 5 x 512
// [0] code
// [1] sync_id
// [2] readsync_id
// [3, 4, 5, 6] cd_lba
int VideoStream(char *image_path)
{
    #define BYTES_PER_SECTOR 2352

    int error = 0; // the error returned from the LibUSB transfer functions
    int received = 0;
    int length = 0;

    unsigned char curr_req_id = 0, req_id = 0;
    unsigned char curr_rdsync_id = 0, rdsync_id = 0;
    unsigned char wr_lock = 0;
    unsigned char *pbuf;
    int fread_result; // contains the result from 'fread'

    unsigned char code;
    unsigned int cd_lba, offset;

    memset(cd_buffer, 0, CD_BUFFER_MAX_LEN); // clear the 'cd_buffer' array before we use it

    fp = fopen(image_path, "rb"); // open the video file (read mode)

    if (fp == NULL)
    {
        #ifdef DEBUG
            printf("Failed to open the video file \"%s\"\n", image_path);
        #endif
        
        return 0;
    }

    fseek(fp, 0L, SEEK_END);

    // calculate the size of the file
    size_t fsize = ftell(fp);
    if (fsize <= 0) return 0;
    fseek(fp, 0, SEEK_SET);

    // clear the 'cd_buffer' array again
    memset(cd_buffer, 0, CD_BUFFER_MAX_LEN);

    // main streaming loop
    while (1)
    {
        if(libusb_interrupt_transfer(handle, BULK_EP_IN, cd_buffer, 16, &transferred, 3000) < 0)
        {
            StopUSBStream();
            
            // these crash the program? why??
            //libusb_close(handle);
            //libusb_exit(NULL);
            
            return 0;
        }
        
        code = cd_buffer[0];
        curr_req_id = cd_buffer[1];
        curr_rdsync_id = cd_buffer[2];
        
        if (req_id == curr_req_id) continue;
        
        req_id = curr_req_id;
        
        if (code == 0x99)
        {
            // idle
        }
        else if (code == 0x01) // data request
        {
            if (wr_lock)
            {
                if (rdsync_id != curr_rdsync_id)
                {
                    wr_lock = 0;
                }
            }
            
            if (!wr_lock)
            {
                cd_lba = cd_buffer[3] + 0x0000100 * cd_buffer[4] + 0x0010000 * cd_buffer[5] + 0x1000000 * cd_buffer[6];
                
                offset = (cd_lba - 150) * 0x930;
                fseek(fp, 0, SEEK_SET);
                fseek(fp, offset, SEEK_SET);
                fread_result = fread(cd_buffer + 4, 1, BYTES_PER_SECTOR, fp);
                
                if (fread_result < BYTES_PER_SECTOR)
                {
                    int rd_try = 10;
                    int rd_cmplt = 0;
                    int rd_more = BYTES_PER_SECTOR;
                    int rd_total = fread_result;
                    
                    while (rd_try)
                    {
                        rd_more -= fread_result;
                        offset += fread_result;
                        fseek(fp, offset, SEEK_SET);
                        fread_result = fread(cd_buffer + 4, 1, rd_more, fp);
                        rd_total += fread_result;
                        
                        if (rd_total == BYTES_PER_SECTOR) rd_cmplt = 1; // if we have read all the bytes, flag that the reading is completed
                        
                        rd_try--; // negate 1 byte
                    }
                    
                    // if reading is still not complete, flag that we have an error
                    if (!rd_cmplt) rd_cmplt = 0;
                }
                
                pbuf = &cd_buffer[2];
                
                int n = CD_BUFFER_MAX_LEN;
                int c = 0;
                int wrlen = 512;
                
                while (n)
                {
                    pbuf[0] = 2; 
                    pbuf[1] = c;
                    error = libusb_bulk_transfer(handle, BULK_EP_OUT, pbuf, wrlen, &transferred, 0);
                    pbuf += (wrlen-2);
                    n -= wrlen;
                    c++;
                }
                
                if (error == 0)
                {
                    rdsync_id = curr_rdsync_id;
                    wr_lock = 1;
                }
            }
        }
    }
}
```

Is this a bug in LibUSB? I cannot figure out why this is happening. The stream used to work fine when we were streaming via MSC/MSD, but when we switched to HID we noticed this stuttering issue.

Thanks.

mcuee, 2021-06-15:
> The stream used to work fine when we were streaming via MSC/MSD, but when we switched to HID we noticed this stuttering issue.

What do you mean by MSC/MSD and HID here?

mcuee, 2021-06-15:
Also it is better to ask question like this in libusb-devel mailing list.

mcuee, 2021-06-15:
Also please check which driver are you using? You may want to try out WinUSB instead of libusbk to see things perform better or not. And you are using the Synchronous API which may be subject to the issue you are seeing. You may want to change to async API.
libusb documentation: https://libusb.sourceforge.io/api-1.0/

Cybdyn-Systems, 2021-06-15:
> > The stream used to work fine when we were streaming via MSC/MSD, but when we switched to HID we noticed this stuttering issue.
> 
> What do you mean by MSC/MSD and HID here? And what do you mean by MCU (Microcontrolller) in the first sentence?

We used to use the generic Windows Mass Storage Class/Device driver.
We have a MCU which sets up the USB endpoints and protocol to communicate to the host (PC).

I will try changing drivers and then if that doesn't help, try changing to the async method.

mcuee, 2021-06-16:
HID is slow. If you have packet size of 64Bytest (full speed), it will be limited to 64 Bytes/1ms = 64KB /s.

mcuee, 2021-06-16:
Please post the device descriptor.

Cybdyn-Systems, 2021-06-16:
> HID is slow using interrupt transfer. If you have packet size of 64Bytest (full speed), it will be limited to 64 Bytes/1ms = 64KB /s. On the other hand, MSC user bulk transfer, the speed will be up to 64Bytes x 19 /1ms = 1216 Bytes/ms = 1,216,000 Bytes/s
> 
> For High Speed, interrupt transfer can run faster but Windows HID driver may be the limiting factor.

We are using a bulk endpoint so there should be no interrupt based slowdowns. Also, we are using the `libusbk` driver.
I will post the device descriptor soon.

Cybdyn-Systems, 2021-06-16:
> Please post the device descriptor.

[image]

mcuee, 2021-06-16:
Thanks for the clarifications. I was confused by your post "but when we switched to HID we noticed this stuttering issue". 

I suggest you try the following three things.
1) Try WinUSB driver instead of libusbk driver to see if there is any change
2) Try async API
3) Please post your question to libusb-devel mailing list.

This github is for issue reporting and it is better to ask questions like this in libusb-devel .

Cybdyn-Systems, 2021-06-17:
Yes, the max packet size is 0x200 (512).

1) I tried using WinUSB, but the problem persists.
2) We'll have a look at it trying the sync route.
3) I was hesitant to use that mailing list because who the hell actually uses mailing lists these days.

mcuee, 2021-06-17:
You will be surprised that many people actually prefer to use mailing list than forum or github issues.

Anyway, I will tell you that libusb-devel mailing list has expert users like Tim Roberts who is an leading Windows Driver expert in the world and also an USB expert,

closed 2021-06-17 by mcuee