libusb/libusb #936: libusb: error [submit_iso_transfer] submiturb failed, errno=11

xiayibing, 2021-06-16:
cpu model               : ICT Loongson-3 V0.4  FPU V0.1   MIPS

OS1:Linux version 4.19.0-loongson-3-desktop
OS2:Linux version 5.4.18-23-generic (buildd@ubuntu)

In function submit_iso_transfer this below operation of ioctl returns error with EAGAIN.
     ioctl(dpriv->fd, IOCTL_USBFS_SUBMITURB, urbs[i]);
I repeated this issue again and again on the above OS1  and OS2 system.

mcuee, 2021-06-16:
You need to provide more details. Like the details of your device (lsusb -vvv) and your codes.

Isochronous transfer is not easy to get right. It can be a problem with your device, your code or the host Linux side, especially when device is a not common Loongson MIPS platform.

mcuee, 2021-06-16:
https://stackoverflow.com/questions/4058368/what-does-eagain-mean
EAGAIN is often raised when performing non-blocking I/O. It means "there is no data available right now, try again later".

mcuee, 2021-06-17:
Please use libusb-devel for further discussions. Thanks.

closed 2021-06-17 by mcuee

xiayibing, 2021-06-17:
The same code is OK under X86 on Linux version 5[MIPS.txt](https://github.com/libusb/libusb/files/6666936/MIPS.txt)3.0-28-generic (buildd@lcy01-amd64-009). 
I guess usbfs is different .  I noticed there are three different places: 
In MIPS.txt 
           1. Couldnot open device, some information will be missing.
           2.  guidFormat is different but minor.
           3.  bNumCompressionPatterns 	1

    at /home/ht706/usb/repoX/HSRUSB/Server/atusbd/atusbd/src/atusbd_export_thread.c:1271
1271        ret = libusb_submit_transfer(priv->transfer);
(gdb) p priv->transfer
$19 = (struct libusb_transfer *) 0x7fffe8003e70
(gdb) p * (struct libusb_transfer *)priv->transfer
$20 = {dev_handle = 0x7ffff00b1a40, flags = 0 '\000', endpoint = 131 '\203', type = 1 '\001', timeout = 0, status = LIBUSB_TRANSFER_COMPLETED, length = 384000, actual_length = 0,
  callback = 0x12003cd24 <libusb_callback>, user_data = 0x7fffe8002e80, buffer = 0x7fffe80044e0 "\200", num_iso_packets = 128, iso_packet_desc = 0x7fffe8003eac}
[MIPS.txt](https://github.com/libusb/libusb/files/6667004/MIPS.txt)
[x86.txt](https://github.com/libusb/libusb/files/6667005/x86.txt)

mcuee, 2021-06-17:
In that case, you have to ask in the  Loongson MIPS related forum or other support channels. We will not be able to help.