libusb/libusb #937: libusb-1.0.24 returns fewer devices than libusb-1.0.23

marriusco, 2021-06-16:
Hello,

I have this device:
Zorloo Zorloo Ztella USB Audio as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/0003:2454:1292.0047/input/input72 
(my Ubuntu 20 has libusb-1.0.23)  

I have a small program (see at the end) whic with the USB inserted prints. It links with -lusb-1.0 (1.0.23) 
It detects 6 devices.

DEVICE: VENDOR: 7531  PRODUCT: 3 
DEVICE: VENDOR: 32903  PRODUCT: 2602 
DEVICE: VENDOR: 3034  PRODUCT: 22758 
DEVICE: VENDOR: 9300  PRODUCT: 4754  <------  My USB MQA device
DEVICE: VENDOR: 1204  PRODUCT: 96 
DEVICE: VENDOR: 7531  PRODUCT: 2 

If I run the same program on 1.0.24 the USB DAC MQA device is not shown

DEVICE: VENDOR: 3034  PRODUCT: 22758 
DEVICE: VENDOR: 32903  PRODUCT: 2602 
DEVICE: VENDOR: 7531  PRODUCT: 3 
DEVICE: VENDOR: 1204  PRODUCT: 96 
DEVICE: VENDOR: 7531  PRODUCT: 2 
: 2 

- small program

```cpp
	num_devs = libusb_get_device_list(usb_context, &devs);int i=0;
	while ((dev = devs[i++]) != NULL) {
		struct libusb_device_descriptor desc;
		struct libusb_config_descriptor *conf_desc = NULL;
		int res = libusb_get_device_descriptor(dev, &desc);
		unsigned short dev_vid = desc.idVendor;
		unsigned short dev_pid = desc.idProduct;

		printf("DEVICE: VENDOR: %d  PRODUCT: %d \n", dev_vid, dev_pid);
      }
```

mcuee, 2021-06-17:
I believe this is the same as https://github.com/libusb/libusb/issues/831

Please use latest git and it should have been fixed.

mcuee, 2021-06-18:
I will close this for now. If this is indeed a different issue, please reopen.

closed 2021-06-18 by mcuee