libusb/libusb #938: device_event_log_impl.cc(214)] [08:28:15.375] USB: usb_device_handle_win.cc:1058 Failed to read descriptor from node connection: A device attached to the system is not functioning

ravipratap8, 2021-06-21:
Edition	Windows 10 Enterprise
Version	20H2
Installed on	‎18-‎03-‎2021
OS build	19042.1052
Serial number	PF2PPKVV
Experience	Windows Feature Experience Pack 120.2212.2020.0

Running Protractor Tool for web application automation

getting below error.

DevTools listening on ws://127.0.0.1:53985/devtools/browser/f085b501-5c4e-43b7-9717-11dc0d9267ac
[17904:15248:0621/082815.349:ERROR:device_event_log_impl.cc(214)] [08:28:15.349] USB: usb_service_win.cc:389 Could not read device interface GUIDs: The system cannot find the file specified. (0x2)
[17904:15248:0621/082815.375:ERROR:device_event_log_impl.cc(214)] [08:28:15.375] USB: usb_device_handle_win.cc:1058 Failed to read descriptor from node connection: A device attached to the system is not functioning. (0x1F)
[17904:15248:0621/082815.377:ERROR:device_event_log_impl.cc(214)] [08:28:15.377] USB: usb_device_handle_win.cc:1058 Failed to read descriptor from node connection: A device attached to the system is not functioning. (0x1F)

mcuee, 2021-06-21:
Read the WiKi, FAQ, etc, before asking questions. Do provide details of device, the version of OS, the code, the debug log, etc, when you raise an issue.

Please also read here.
https://github.com/libusb/libusb/wiki/Troubleshooting

mcuee, 2021-06-21:
Did you install WinUSB driver with Zadig?

ravipratap8, 2021-06-21:
No i did not do anything regarding to USB

Just installed

Node js 
Visual studio code
Added 
"devDependencies": {
    "cucumber": "^7.0.0-rc.0",
    "protractor": "^7.0.0",
    "protractor-cucumber-framework": "^8.2.0"

and getting below error

PS C:\e2eframework> protractor C:\e2eframework\config.js
[10:54:52] I/launcher - Running 1 instances of WebDriver
[10:54:52] I/direct - Using ChromeDriver directly...

DevTools listening on ws://127.0.0.1:56349/devtools/browser/79b561a0-3e8e-4b88-9b2c-5c4d585ddf3a
[18900:13784:0621/105453.151:ERROR:device_event_log_impl.cc(214)] [10:54:53.151] USB: usb_service_win.cc:389 Could not read device interface GUIDs: The system cannot find the file specified. (0x2)
[18900:13784:0621/105453.161:ERROR:device_event_log_impl.cc(214)] [10:54:53.161] USB: usb_device_handle_win.cc:1058 Failed to read descriptor from node connection: A device attached to the system is not functioning. (0x1F)
[18900:13784:0621/105453.163:ERROR:device_event_log_impl.cc(214)] [10:54:53.163] USB: usb_device_handle_win.cc:1058 Failed to read descriptor from node connection: A device attached to the system is not functioning. (0x1F)

0 scenarios
0 steps
0m00.001s (executing steps: 0m00.000s)
[10:54:54] I/launcher - 0 instance(s) of WebDriver still running
[10:54:54] I/launcher - chrome #01 passed
PS C:\e2eframework>

mcuee, 2021-06-21:
Read the Wiki and FAQ first. I will close this for now.

closed 2021-06-21 by mcuee