libusb/libusb #941: libusb_init() return LIBUSB_ERROR_IO

DarylXiang, 2021-06-25:
In Android, when call libusb_init()  will return LIBUSB_ERROR_IO, and I traced the libusb code show that in sysfs_get_device_list() try to open dir "/sys/bus/usb/devices" and failed because of permission denied. Then I try to "chmod 777 /sys/bus/usb/devices", however problem still exist.

`
static int sysfs_get_device_list(struct libusb_context *ctx) {
    DIR * devices = **opendir(SYSFS_DEVICE_PATH)**;
    struct dirent *entry;
    int num_devices = 0;
    int num_enumerated = 0;

    if (!devices) {
        ALOGD("opendir devices failed (path:%s), errno=%d, error msg=%s", SYSFS_DEVICE_PATH, errno, strerror(errno));
        usbi_err(ctx, "opendir devices failed, errno=%d", errno);
        return LIBUSB_ERROR_IO;
    }
}
`

mcuee, 2021-06-25:
Read here for a proper way to use libusb under Android.
https://github.com/libusb/libusb/pull/830

DarylXiang, 2021-06-25:
I have seen that issue, but libusb_wrap_sys_device() need a fileDescriptor from Java UsbDeviceConnection. It isn't a application apk project, I can't get fileDescriptor.

mcuee, 2021-06-25:
You can continue the discussion here if you think #830 is not good enough.
https://github.com/libusb/libusb/issues/717

Or you can try #874.

mcuee, 2021-06-26:
@DarylXiang Please reopen this issue if you think your issue is different from #717.

mcuee, 2021-06-26:
Similar reports:
https://github.com/libusb/libusb/issues/864

mcuee, 2021-06-26:
Many older issues of Android has been fixed by https://github.com/libusb/libusb/pull/760 which is included in 1.0.24 release, if you use rooted Android. If you use unrooted Android, you have to use the method mentioned by #830 as of now.

mcuee, 2021-06-26:
@yshteinm I know I closed #864 and refer you to #830. Still can you comment here that if your Android is rooted or not? Thanks.

mcuee, 2021-07-26:
@DarylXiang What do you mean by the following? Sorry I do not know much about Android.
> I have seen that issue, but libusb_wrap_sys_device() need a fileDescriptor from Java UsbDeviceConnection. It isn't a application apk project, I can't get fileDescriptor.

Will #874 help you better?

Or will the approach mentioned by @normanr here is better for you?
https://github.com/libusb/libusb/pull/874#issuecomment-873322802

mcuee, 2021-07-26:
Relevant discussion: termux pyusb enhancement
https://github.com/pyusb/pyusb/pull/287