libusb/libusb #944: C++ symbol for libusb_set_option not exported correctly

tckillian, 2021-07-02:
The libusb_set_option() function is not being properly exported for use with C++ in MSVC.

I believe the export was briefly specified correctly in @dickens original commit, here:
https://github.com/libusb/libusb/blob/v1.0.22-rc1/libusb/libusb-1.0.def#L156

It looks like it's changed several times since then, but the export style of this function does not match other exports in the master branch.

mcuee, 2021-07-02:
Please help to create a pull request if you can. Thanks.

mcuee, 2021-07-11:
@tckillian @tormodvolden  Please check if #949 is correct or not. Thanks.

mcuee, 2021-07-11:
First time learning to create a pull request. :-)

tormodvolden, 2021-07-11:
From going through the git log, it was only changed once, in commit 0aac390, maybe inadvertently, @dickens?

BTW shouldn't the ordinal numbers be unique for each function?
https://docs.microsoft.com/en-us/cpp/build/exporting-from-a-dll-using-def-files?view=msvc-160

It looks like people regenerate this file from a DLL instead of maintaining it by hand.

tormodvolden, 2021-07-11:
Note the original commit was commit 539f22e2, not in the RC that you both are referring to. It was actually modified twice:
- commit 46874013 removed the "@8" ordinal number.
- commit 0aac390 added the underscore

tormodvolden, 2021-07-27:
@hjelmn what do you think about the ordinal numbers?

mcuee, 2021-07-27:
@tormodvolden It seems to me the ordinal number is not unique. I believe it is there to support legacy code (32bit only). But now we need to see if 8 is correct or not based on the following.

Ref: https://docs.microsoft.com/en-us/cpp/build/reference/exports?view=msvc-160
>Because the MSVC compiler uses name decoration for C++ functions, you must either use the decorated name internal_name or define the exported functions by using extern "C" in the source code. The compiler also decorates C functions that use the __stdcall calling convention with an underscore (_) prefix and a suffix composed of the at sign (@) followed by the number of bytes (in decimal) in the argument list.

mcuee, 2021-07-27:
@tormodvolden From libusb-1.0.def history file, it seems that this should be manually generated.
https://github.com/libusb/libusb/commit/b88ce7d2450a2d77b84cca5ffe6bcb449a5ae05b
```
* Generated from a MinGW32 generated DLL using the command:
  echo -e "LIBRARY\nEXPORTS" > libusb-1.0.def; strings libusb-1.0.dll | grep libusb | grep @ | sort | sed -e "s/\(.*\)@\([0-9]*\)/  \1\n  \1@\2 = \1/" >> libusb-1.0.def
```

mcuee, 2021-07-27:
Chris is also generating the def file from the DLL.
https://github.com/libusb/libusb/commit/cec3514dd4c6a56b543559874f2b40f3e0c67fe2
```Windows: Regenerate libusb-1.0.def file from latest DLL```

But many times it will be manually created.

mcuee, 2021-07-27:
The usage of the def file is to make the DLL interchangeable between MinGW and MSVC. There was a long discussion here. 

https://marc.info/?l=libusb-devel&m=128076361626828&w=4
List:       libusb-devel
Subject:    Re: [Libusb-devel] interchangeability of MinGW and MS DLLs
From:       Pete Batard 
Date:       2010-08-02 15:39:52

All things considered, I think the solution I'd prefer would be to ditch 
__declspec(dllexport) altogether and revert to using a def file (while 
leaving WINAPI as the default calling convention).

The problem of keeping the .def in sync with what we have in our source 
could be solved with a sed script, called at the end of autogen.sh, that 
would update the .def by picking up all API_EXPORTED calls we have in 
core (and of course, as current .def would be provided as well).

This would keep libusb.h free from the redundant LIBUSB_EXP, as Daniel 
wanted, keep the use of WINAPI, and might ease up the life of 
developers, as having a .def allows for easy creation of the DLL library 
wrapper, in case they don't have access to a prebuilt one for whatever 
reason.

Any other views?

Regards,

/Pete

mcuee, 2021-07-30:
Closing this for now. If the users encounter issues with libusb-1.0.def file, then we can reopen.

I will ask in the libusb-devel mailing list as well. I understand that it is meant to make the DLL generated by MSVC to be compatible with MinGW compilers. So far I can see they are compatible but I am using 64bit Windows and I think it does not matter for 64bit Windows.

mcuee, 2021-09-05:
Actually I do not know if libusb_set_option() can be exported correctly for 32bit Windows (it is not needed for 64bit Windows as far as I know) because it takes variadic.

```
/c/work/libusb/libusb-1.0.24_bin/MinGW32/dll
$ echo -e "LIBRARY\nEXPORTS" > libusb-1.0.def; strings libusb-1.0.dll | grep libusb | grep @ | sort | sed -e "s/\(.*\)@\([0-9]*\)/  \1\n  \1@\2 = \1/" >> libusb
-1.0.def

LIBRARY
EXPORTS
  _libusb_alloc_streams
  _libusb_alloc_streams@16 = _libusb_alloc_streams
  _libusb_alloc_transfer
  _libusb_alloc_transfer@4 = _libusb_alloc_transfer
  _libusb_attach_kernel_driver
  _libusb_attach_kernel_driver@8 = _libusb_attach_kernel_driver
  _libusb_bulk_transfer
  _libusb_bulk_transfer@24 = _libusb_bulk_transfer
  _libusb_cancel_transfer
  _libusb_cancel_transfer@4 = _libusb_cancel_transfer
  _libusb_claim_interface
  _libusb_claim_interface@8 = _libusb_claim_interface
  _libusb_clear_halt
  _libusb_clear_halt@8 = _libusb_clear_halt
  _libusb_close
  _libusb_close@4 = _libusb_close
  _libusb_control_transfer
  _libusb_control_transfer@32 = _libusb_control_transfer
  _libusb_detach_kernel_driver
  _libusb_detach_kernel_driver@8 = _libusb_detach_kernel_driver
  _libusb_dev_mem_alloc
  _libusb_dev_mem_alloc@8 = _libusb_dev_mem_alloc
  _libusb_dev_mem_free
  _libusb_dev_mem_free@12 = _libusb_dev_mem_free
  _libusb_error_name
  _libusb_error_name@4 = _libusb_error_name
  _libusb_event_handler_active
  _libusb_event_handler_active@4 = _libusb_event_handler_active
  _libusb_event_handling_ok
  _libusb_event_handling_ok@4 = _libusb_event_handling_ok
  _libusb_exit
  _libusb_exit@4 = _libusb_exit
  _libusb_free_bos_descriptor
  _libusb_free_bos_descriptor@4 = _libusb_free_bos_descriptor
  _libusb_free_config_descriptor
  _libusb_free_config_descriptor@4 = _libusb_free_config_descriptor
  _libusb_free_container_id_descriptor
  _libusb_free_container_id_descriptor@4 = _libusb_free_container_id_descriptor
  _libusb_free_device_list
  _libusb_free_device_list@8 = _libusb_free_device_list
  _libusb_free_pollfds
  _libusb_free_pollfds@4 = _libusb_free_pollfds
  _libusb_free_ss_endpoint_companion_descriptor
  _libusb_free_ss_endpoint_companion_descriptor@4 = _libusb_free_ss_endpoint_companion_descriptor
  _libusb_free_ss_usb_device_capability_descriptor
  _libusb_free_ss_usb_device_capability_descriptor@4 = _libusb_free_ss_usb_device_capability_descriptor
  _libusb_free_streams
  _libusb_free_streams@12 = _libusb_free_streams
  _libusb_free_transfer
  _libusb_free_transfer@4 = _libusb_free_transfer
  _libusb_free_usb_2_0_extension_descriptor
  _libusb_free_usb_2_0_extension_descriptor@4 = _libusb_free_usb_2_0_extension_descriptor
  _libusb_get_active_config_descriptor
  _libusb_get_active_config_descriptor@8 = _libusb_get_active_config_descriptor
  _libusb_get_bos_descriptor
  _libusb_get_bos_descriptor@8 = _libusb_get_bos_descriptor
  _libusb_get_bus_number
  _libusb_get_bus_number@4 = _libusb_get_bus_number
  _libusb_get_config_descriptor
  _libusb_get_config_descriptor@12 = _libusb_get_config_descriptor
  _libusb_get_config_descriptor_by_value
[... 271 lines cut ...]
  libusb_set_pollfd_notifiers
  libusb_set_pollfd_notifiers@16 = libusb_set_pollfd_notifiers
  libusb_setlocale
  libusb_setlocale@4 = libusb_setlocale
  libusb_strerror
  libusb_strerror@4 = libusb_strerror
  libusb_submit_transfer
  libusb_submit_transfer@4 = libusb_submit_transfer
  libusb_transfer_get_stream_id
  libusb_transfer_get_stream_id@4 = libusb_transfer_get_stream_id
  libusb_transfer_set_stream_id
  libusb_transfer_set_stream_id@8 = libusb_transfer_set_stream_id
  libusb_try_lock_events
  libusb_try_lock_events@4 = libusb_try_lock_events
  libusb_unlock_event_waiters
  libusb_unlock_event_waiters@4 = libusb_unlock_event_waiters
  libusb_unlock_events
  libusb_unlock_events@4 = libusb_unlock_events
  libusb_unref_device
  libusb_unref_device@4 = libusb_unref_device
  libusb_wait_for_event
  libusb_wait_for_event@8 = libusb_wait_for_event
  libusb_wrap_sys_device
  libusb_wrap_sys_device@12 = libusb_wrap_sys_device
```

mcuee, 2021-09-05:
MinGW.org gendef utility (I just found that MinGW.org is still alive from OSDN).

```
;
; Definition file of libusb-1.0.dll
; Automatic generated by gendef
; written by Kai Tietz 2008
;
LIBRARY "libusb-1.0.dll"
EXPORTS
GUID_DEVINTERFACE_LIBUSB0_FILTER DATA
GUID_DEVINTERFACE_USB_DEVICE DATA
GUID_DEVINTERFACE_USB_HOST_CONTROLLER DATA
GUID_DEVINTERFACE_USB_HUB DATA
active_contexts_list DATA
active_contexts_lock DATA
discovered_devs_append
htab_hash
libusb_alloc_streams@16
libusb_alloc_streams@16@16
libusb_alloc_transfer@4
libusb_alloc_transfer@4@4
libusb_attach_kernel_driver@8
libusb_attach_kernel_driver@8@8
libusb_bulk_transfer@24
libusb_bulk_transfer@24@24
libusb_cancel_transfer@4
libusb_cancel_transfer@4@4
libusb_claim_interface@8
libusb_claim_interface@8@8
libusb_clear_halt@8
libusb_clear_halt@8@8
libusb_close@4
libusb_close@4@4
libusb_control_transfer@32
libusb_control_transfer@32@32
libusb_detach_kernel_driver@8
libusb_detach_kernel_driver@8@8
libusb_dev_mem_alloc@8
libusb_dev_mem_alloc@8@8
libusb_dev_mem_free@12
libusb_dev_mem_free@12@12
libusb_error_name@4
libusb_error_name@4@4
libusb_event_handler_active@4
libusb_event_handler_active@4@4
libusb_event_handling_ok@4
libusb_event_handling_ok@4@4
libusb_exit@4
libusb_exit@4@4
libusb_free_bos_descriptor@4
libusb_free_bos_descriptor@4@4
libusb_free_config_descriptor@4
libusb_free_config_descriptor@4@4
libusb_free_container_id_descriptor@4
libusb_free_container_id_descriptor@4@4
libusb_free_device_list@8
libusb_free_device_list@8@8
libusb_free_pollfds@4
libusb_free_pollfds@4@4
libusb_free_ss_endpoint_companion_descriptor@4
libusb_free_ss_endpoint_companion_descriptor@4@4
libusb_free_ss_usb_device_capability_descriptor@4
libusb_free_ss_usb_device_capability_descriptor@4@4
libusb_free_streams@12
libusb_free_streams@12@12
libusb_free_transfer@4
libusb_free_transfer@4@4
libusb_free_usb_2_0_extension_descriptor@4
libusb_free_usb_2_0_extension_descriptor@4@4
[... 143 lines cut ...]
usbi_destroy_timer
usbi_disarm_timer
usbi_disconnect_device
usbi_get_device_by_session_id
usbi_get_monotonic_time
usbi_handle_disconnect@4
usbi_handle_transfer_cancellation
usbi_handle_transfer_completion
usbi_hotplug_deregister
usbi_hotplug_match
usbi_hotplug_notification
usbi_io_exit
usbi_io_init
usbi_log
usbi_remove_event_source
usbi_sanitize_device
usbi_signal_event
usbi_signal_transfer_completion
usbi_wait_for_events
windows_error_str
windows_force_sync_completion
windows_version DATA
winusb_backend DATA

```

mcuee, 2021-09-05:
@pbatard Sorry Pete, need to disturb you to see what is the right way to export libusb_set_option(). Thanks.

mcuee, 2021-09-05:
Ref: https://github.com/vpelletier/python-libusb1/issues/75#issuecomment-909733308

pbatard, 2021-09-05:
> what is the right way to export libusb_set_option()

Don't know if that helps but when compiling the libusb DLL using VS2019, you get a `libusb-1.0.exp` generated.
Then you can issue something like ` strings libusb-1.0.exp  | grep _libusb | sort | uniq` and you'll get:
```
_libusb_alloc_streams@16
_libusb_alloc_transfer@4
_libusb_attach_kernel_driver@8
_libusb_bulk_transfer@24
_libusb_cancel_transfer@4
_libusb_claim_interface@8
_libusb_clear_halt@8
_libusb_close@4
_libusb_control_transfer@32
_libusb_detach_kernel_driver@8
_libusb_dev_mem_alloc@8
_libusb_dev_mem_free@12
_libusb_error_name@4
_libusb_event_handler_active@4
_libusb_event_handling_ok@4
_libusb_exit@4
_libusb_free_bos_descriptor@4
_libusb_free_config_descriptor@4
_libusb_free_container_id_descriptor@4
_libusb_free_device_list@8
_libusb_free_pollfds@4
_libusb_free_ss_endpoint_companion_descriptor@4
_libusb_free_ss_usb_device_capability_descriptor@4
_libusb_free_streams@12
_libusb_free_transfer@4
_libusb_free_usb_2_0_extension_descriptor@4
_libusb_get_active_config_descriptor@8
_libusb_get_bos_descriptor@8
_libusb_get_bus_number@4
_libusb_get_config_descriptor@12
_libusb_get_config_descriptor_by_value@12
_libusb_get_configuration@8
_libusb_get_container_id_descriptor@12
_libusb_get_device@4
_libusb_get_device_address@4
_libusb_get_device_descriptor@8
_libusb_get_device_list@8
_libusb_get_device_speed@4
_libusb_get_max_iso_packet_size@8
_libusb_get_max_packet_size@8
_libusb_get_next_timeout@8
_libusb_get_parent@4
_libusb_get_pollfds@4
_libusb_get_port_number@4
_libusb_get_port_numbers@12
_libusb_get_port_path@16
_libusb_get_ss_endpoint_companion_descriptor@12
_libusb_get_ss_usb_device_capability_descriptor@12
_libusb_get_string_descriptor_ascii@16
_libusb_get_usb_2_0_extension_descriptor@12
_libusb_get_version@0
_libusb_handle_events@4
_libusb_handle_events_completed@8
_libusb_handle_events_locked@8
_libusb_handle_events_timeout@8
_libusb_handle_events_timeout_completed@12
_libusb_has_capability@4
_libusb_hotplug_deregister_callback@8
_libusb_hotplug_get_user_data@8
_libusb_hotplug_register_callback@36
_libusb_init@4
_libusb_interrupt_event_handler@4
_libusb_interrupt_transfer@24
_libusb_kernel_driver_active@8
_libusb_lock_events@4
_libusb_lock_event_waiters@4
_libusb_open@8
_libusb_open_device_with_vid_pid@12
_libusb_pollfds_handle_timeouts@4
_libusb_ref_device@4
_libusb_release_interface@8
_libusb_reset_device@4
_libusb_set_auto_detach_kernel_driver@8
_libusb_set_configuration@8
_libusb_set_debug@8
_libusb_set_interface_alt_setting@12
_libusb_setlocale@4
_libusb_set_log_cb@12
_libusb_set_option
_libusb_set_pollfd_notifiers@16
_libusb_strerror@4
_libusb_submit_transfer@4
_libusb_transfer_get_stream_id@4
_libusb_transfer_set_stream_id@8
_libusb_try_lock_events@4
_libusb_unlock_events@4
_libusb_unlock_event_waiters@4
_libusb_unref_device@4
_libusb_wait_for_event@8
_libusb_wrap_sys_device@12
```

So it seems variadic functions are exported as taking no parameters. Outside of that, I have literally no idea how they are meant to be handled. I would **guess** that despite having no parameter, there's a first param register that's always implicitly set to the number of variadic parameters, and, if non zero, there's a second param register that'll point to an array of params, but, really, that's a question to direct people who develop compilers, which I am not.

mcuee, 2021-09-07:
@pbatard I think that is the best guess for now. 

The export from VS2019 seems to be easy to understand in most case, at a quick glance, I only see one strange point.
```
_libusb_get_port_number@4 --> this does not seem to be correct
_libusb_get_port_numbers@12
```

mcuee, 2021-09-07:
So in this case, the merged https://github.com/libusb/libusb/pull/964 may not be totally correct but it probably may not do harm as well. Let's see if people are encountering the issue or not.

mcuee, 2021-09-07:
With libusb-1.0.24 release, there is a reported issue. 
https://github.com/EttusResearch/uhd/issues/453

mcuee, 2021-09-07:
@pbatard If your guess is correct, how shall we fix the libusb-1.0.def?

```
diff --git a/libusb/libusb-1.0.def b/libusb/libusb-1.0.def
index e357ef5..648ae6a 100644
--- a/libusb/libusb-1.0.def
+++ b/libusb/libusb-1.0.def
@@ -155,7 +155,6 @@ EXPORTS
   libusb_set_interface_alt_setting
   libusb_set_interface_alt_setting@12 = libusb_set_interface_alt_setting
   libusb_set_option
-  libusb_set_option@8 = libusb_set_option
   libusb_set_pollfd_notifiers
   libusb_set_pollfd_notifiers@16 = libusb_set_pollfd_notifiers
   libusb_setlocale
```

pbatard, 2021-09-08:
Outside of trial and error I have very little idea on that topic.

The thing is, from the output I posted above, I see that functions with zero parameters are exported with `@0` (e.g. `_libusb_get_version@0`), but variadics are exported without the any `@something`.

I'm afraid I do not know about the internals of Win32 DLL handling to be able to provide any kind of advice here. Sorry.

mcuee, 2021-09-09:
@pbatard You are being very helpful already. Thanks. Now the question is how to test. 
This issue report comes from https://github.com/EttusResearch/uhd/issues/453 and based on the testing it is a false alarm.

neheb, 2021-12-21:
I'll note that this is a compile time error on clang with MSYS2. Fixed with:

```
--- a/libusb/core.c
+++ b/libusb/core.c
@@ -2181,7 +2181,7 @@ void API_EXPORTED libusb_set_log_cb(libusb_context *ctx, libusb_log_cb cb,
  * on this platform
  * \returns LIBUSB_ERROR_NOT_FOUND if LIBUSB_OPTION_USE_USBDK is valid on this platform but UsbDk is not available
  */
-int API_EXPORTED libusb_set_option(libusb_context *ctx,
+int _cdecl libusb_set_option(libusb_context *ctx,
        enum libusb_option option, ...)
 {
        int arg = 0, r = LIBUSB_SUCCESS;
diff --git a/libusb/libusb.h b/libusb/libusb.h
index 61cacc9..e0ba2b1 100644
--- a/libusb/libusb.h
+++ b/libusb/libusb.h
@@ -2126,7 +2126,7 @@ enum libusb_option {
        LIBUSB_OPTION_MAX = 3
 };

-int LIBUSB_CALL libusb_set_option(libusb_context *ctx, enum libusb_option option, ...);
+int __cdecl libusb_set_option(libusb_context *ctx, enum libusb_option option, ...);

 #if defined(__cplusplus)
 }

```

diabolo38, 2021-12-21:
> -int API_EXPORTED libusb_set_option(libusb_context *ctx,
> +int _cdecl libusb_set_option(libusb_context *ctx,
>         enum libusb_option option, ...)
>  {
>         int arg = 0, r = LIBUSB_SUCCESS;

not correct i think this  is likley to cause the symbol not exported at all,  it's is not enough  for win/mac  to export a symbol  in a dll or dylib 
i suspect it's more the API_EXPORTED that is somehow not correct in case your using the header (not building fro source )

neheb, 2021-12-22:
> not correct i think this is likley to cause the symbol not exported at all, it's is not enough for win/mac to export a symbol in a dll or dylib i suspect it's more the API_EXPORTED that is somehow [...]

Not sure what you mean. The dll has it:

```
strings libusb-1.0.dll | grep set_opt
windows_set_option
_libusb_set_option
libusb_set_option
_libusb_set_option

```

mcuee, 2021-12-23:
> I'll note that this is a compile time error on clang with MSYS2. Fixed with:
>
> ```
> --- a/libusb/core.c
> +++ b/libusb/core.c
> @@ -2181,7 +2181,7 @@ void API_EXPORTED libusb_set_log_cb(libusb_context [...]

Interesting that MSYS2 clang can capture this one. So basically you are saying LIBUSB_CALL (WINAPI or _stdcall under Windows) is not compatible with the varidic function libusb_set_option(), right? This is in line with what Jef Driesen mentioned in the libusb-devel mailing list.

neheb, 2021-12-23:
@mcuee 32-bit. That strings output was with a 32-bit library.

MSYS2 GCC behaves the same. The only difference is that it's a silent failure with GCC and explicit with clang.

mcuee, 2021-12-23:
> @mcuee 32-bit. That strings output was with a 32-bit library.
> 
> MSYS2 GCC behaves the same. The only difference is that it's a silent failure with GCC and explicit with clang.

I see. BTW, how do you install MSYS2 clang32, it seems to me the toolchain is not ready yet.
https://www.msys2.org/docs/environments/
https://issueexplorer.com/issue/msys2/MINGW-packages/9687

neheb, 2021-12-23:
C:\msys2 has a clang32.exe

mcuee, 2021-12-24:
> @mcuee 32-bit. That strings output was with a 32-bit library.
> 
> MSYS2 GCC behaves the same. The only difference is that it's a silent failure with GCC and explicit with clang.

Thanks.

Microsoft Visual Studio C/C++ compiler has the same behavior as MinGW GCC then.

Ref:
https://devblogs.microsoft.com/oldnewthing/20131128-00/?p=2543

>Consider the following function on an x86 system:
>
> void __stdcall something(char *, ...);
> The function declares itself as __stdcall, which is a callee-clean convention. But a variadic function cannot be callee-clean since the callee does not know how many parameters were passed, so it doesn’t know how many it should clean.
>
> The Microsoft Visual Studio C/C++ compiler resolves this conflict by silently converting the calling convention to __cdecl, which is the only supported variadic calling convention for functions that do not take a hidden this parameter.
>
> Why does this conversion take place silently rather than generating a warning or error?
>
> My guess is that it’s to make the compiler options /Gr (set default calling convention to __fastcall) and /Gz (set default calling convention to __stdcall) less annoying.
> 
> Automatic conversion of variadic functions to __cdecl means that you can just add the /Gr or /Gz command line switch to your compiler options, and everything will still compile and run (just with the new calling convention).
> 
> Another way of looking at this is not by thinking of the compiler as converting variadic __stdcall to __cdecl but rather by simply saying “for variadic functions, __stdcall is caller-clean.”

mcuee, 2021-12-24:
> @mcuee 32-bit. That strings output was with a 32-bit library.
> 
> MSYS2 GCC behaves the same. The only difference is that it's a silent failure with GCC and explicit with clang.

Since Microsoft Visual Studio C/C++ compiler has the same behavior as MinGW GCC, can I argue that clang compiler should do the same and this is a MSYS2 clang compiler bug? It seems to me your change is changing the libusb API.

mcuee, 2021-12-24:
@tormodvolden Please take a look at this latest development of this issue. Thanks.

mcuee, 2021-12-24:
> C:\msys64 has a clang32.exe

Sorry but I tried it and it does not work.

```
$ CC=clang ./autogen.sh
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: aclocal --force -I m4
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:23: installing './compile'
configure.ac:26: installing './missing'
examples/Makefile.am: installing './depcomp'
autoreconf: './config.sub' is updated
autoreconf: './config.guess' is updated
autoreconf: Leaving directory '.'
configure: loading site script /etc/config.site
checking for gcc... clang
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether clang accepts -g... yes
checking for clang option to enable C11 features... none needed
checking whether clang understands -c and -o together... yes
checking for g++... g++
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking for inline... inline
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking dependency style of clang... gcc3
checking dependency style of g++... gcc3
checking build system type... i686-w64-mingw32
checking host system type... i686-w64-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by clang... C:/msys64/mingw32/bin/ld
checking if the linker (C:/msys64/mingw32/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /mingw32/bin/nm -B
checking the name lister (/mingw32/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
[... 48 lines cut ...]
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (C:/msys64/mingw32/bin/ld) supports shared libraries... yes
checking dynamic linker characteristics... Win32 ld.exe
checking how to hardcode library paths into programs... immediate
checking for windres... windres
checking whether make supports nested variables... (cached) yes
checking if clang supports -std=gnu11... yes
checking operating system... Windows
checking for struct timespec... yes
checking for sys/time.h... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating libusb-1.0.pc
config.status: creating Makefile
config.status: creating libusb/Makefile
config.status: creating examples/Makefile
config.status: creating tests/Makefile
config.status: creating doc/Makefile
config.status: creating doc/doxygen.cfg
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands
```

neheb, 2021-12-24:
> > MSYS2 GCC behaves the same. The only difference is that it's a silent failure with GCC and explicit with clang.
> 
> Since Microsoft Visual Studio C/C++ compiler has the same behavior as MinGW GCC, can I argue that clang compiler should do the same and this is a MSYS2 clang compiler bug? It seems to me your change is changing the libusb API and we may be reluctant to change that as of now.

I disagree. Maybe the function should be switched to use WINAPIV.

As far as the win32 error, msys2's PKGBUILD has this line:

```
sed -e "s/-mwin32//g" -e "s/-Wl,--add-stdcall-alias//g" -i configure.ac
```

Maybe this will help: https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-libusb/PKGBUILD

mcuee, 2021-12-24:
> I disagree. Maybe the function should be switched to use WINAPIV.

I kind of agree with you but the worry is API change. I will leave this to other admins to comment.

> As far as the win32 error, msys2's PKGBUILD has this line:
> 
> ```
> sed -e "s/-mwin32//g" -e "s/-Wl,--add-stdcall-alias//g" -i configure.ac
> ```
> 
> Maybe this will help: https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-libusb/PKGBUILD

Thanks a lot for your comments.

Is it possible for you to create three pull requests?
1) to address the configure.ac issue for MSYS2 clang. It is probably no problem for other compilers as well.
2) to address clang compiler error issue : use WINAPIV for the variadic function libusb_set_option 
3) The question remains how to address libusb-1.0,def. If you know the answer, that is another pull request.

neheb, 2022-06-26:
@mcuee I just took at the sed line in MSYS2. It gets applied only when compiling with clang. The end result is basically this:

```
--- a/configure.ac
+++ b/configure.ac
@@ -121,14 +121,14 @@ wasm32-**)
        AC_MSG_RESULT([Windows (using Cygwin)])
        backend=windows
        platform=windows
-       EXTRA_CFLAGS="-mwin32"
+       EXTRA_CFLAGS=""
        ;;
 *-mingw* | *msys*)
        AC_MSG_RESULT([Windows])
        backend=windows
        platform=windows
        test "x$enable_shared" = xyes && create_import_lib=yes
-       EXTRA_CFLAGS="-mwin32 -fno-omit-frame-pointer"
+       EXTRA_CFLAGS=" -fno-omit-frame-pointer"
        EXTRA_LDFLAGS="-static-libgcc"
        ;;
 *)
@@ -224,7 +224,7 @@ sunos)
 windows)
        AC_CHECK_TYPES([struct timespec], [], [], [[#include <time.h>]])
        AC_DEFINE([_WIN32_WINNT], [_WIN32_WINNT_VISTA], [Define to the oldest supported Windows version.])
-       LT_LDFLAGS="${LT_LDFLAGS} -avoid-version -Wl,--add-stdcall-alias"
+       LT_LDFLAGS="${LT_LDFLAGS} -avoid-version "
        ;;
 emscripten)
        AC_SUBST(EXEEXT, [.html])
```

The mwin32 line is definitely useless. Not sure about --add-stdcall-alias.

mcuee, 2022-07-09:
@tormodvolden 
It seems to me that pull request #1160 may be a good solution to sort out this messy situation. Please take a look.
* https://github.com/libusb/libusb/pull/1160

tormodvolden, 2022-08-04:
To answer my own question about the ordinal numbers, this seems to explain it accurately: https://docs.microsoft.com/en-us/office/client-developer/excel/developing-dlls

mcuee, 2022-09-05:
Now that #1160 has been merged, I think we can revisit this issue again. The following two lines may not be correct. 
https://github.com/libusb/libusb/blob/master/libusb/libusb-1.0.def#L157_L158

neheb, 2022-09-05:
I'll note that the mingw32 output  in https://github.com/libusb/libusb/pull/1193 , libusb_set_option is missing.

neheb, 2022-09-05:
From testing on mingw32/64 and clang32/64 and looking at warnings, I believe line 158 should be removed.

neheb, 2022-09-05:
clang32:
```
$ objdump.exe -t libusb-1.0.dll | grep libusb_set
[103](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x00001e90 _libusb_set_configuration@8
[106](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x000021a0 _libusb_set_interface_alt_setting@12
[116](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x000029f0 _libusb_set_auto_detach_kernel_driver@8
[117](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x00002a40 _libusb_set_debug@8
[118](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x00002ac0 _libusb_set_log_cb@12
[120](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x00002b20 _libusb_set_option
[275](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x000097a0 _libusb_set_pollfd_notifiers@16
[286](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x0000a520 _libusb_setlocale@4
```

clang64:

```
$ objdump.exe -t libusb-1.0.dll | grep libusb_set
[103](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x00002010 libusb_set_configuration
[106](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x00002330 libusb_set_interface_alt_setting
[116](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x00002c60 libusb_set_auto_detach_kernel_driver
[117](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x00002cb0 libusb_set_debug
[118](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x00002d40 libusb_set_log_cb
[120](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x00002db0 libusb_set_option
[275](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x0000a2a0 libusb_set_pollfd_notifiers
[286](sec  1)(fl 0x00)(ty  20)(scl   2) (nx 0) 0x0000b0a0 libusb_setlocale
```

mingw32:

```
$ objdump.exe -t libusb-1.0.dll | grep libusb_set
[126](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00001afb _libusb_set_configuration@8
[132](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00001d46 _libusb_set_interface_alt_setting@12
[150](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000232b _libusb_set_auto_detach_kernel_driver@8
[151](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00002355 _libusb_set_debug@8
[152](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000239d _libusb_set_log_cb@12
[153](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x000023db _libusb_set_option
[405](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x000081b3 _libusb_set_pollfd_notifiers@16
[448](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 1) 0x000085a0 _libusb_setlocale@4
[3144](sec  1)(fl 0x00)(ty    0)(scl   2) (nx 0) 0x000081b3 _libusb_set_pollfd_notifiers
[3221](sec  1)(fl 0x00)(ty    0)(scl   2) (nx 0) 0x00001d46 _libusb_set_interface_alt_setting
[3224](sec  1)(fl 0x00)(ty    0)(scl   2) (nx 0) 0x000085a0 _libusb_setlocale
[3301](sec  1)(fl 0x00)(ty    0)(scl   2) (nx 0) 0x0000232b _libusb_set_auto_detach_kernel_driver
[3332](sec  1)(fl 0x00)(ty    0)(scl   2) (nx 0) 0x00002355 _libusb_set_debug
[3348](sec  1)(fl 0x00)(ty    0)(scl   2) (nx 0) 0x0000239d _libusb_set_log_cb
[3413](sec  1)(fl 0x00)(ty    0)(scl   2) (nx 0) 0x00001afb _libusb_set_configuration
```

mingw64:

```
$ objdump.exe -t libusb-1.0.dll | grep libusb_set
[147](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000000000001ceb libusb_set_configuration
[153](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000000000001f7c libusb_set_interface_alt_setting
[171](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00000000000026a4 libusb_set_auto_detach_kernel_driver
[172](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00000000000026da libusb_set_debug
[173](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000000000002731 libusb_set_log_cb
[174](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000000000002787 libusb_set_option
[442](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00000000000091c8 libusb_set_pollfd_notifiers
[489](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 1) 0x0000000000009660 libusb_setlocale
```

ucrt64:

```
$ objdump.exe -t libusb-1.0.dll | grep libusb_set
[147](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000000000001ceb libusb_set_configuration
[153](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000000000001f7c libusb_set_interface_alt_setting
[171](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00000000000026a4 libusb_set_auto_detach_kernel_driver
[172](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00000000000026da libusb_set_debug
[173](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000000000002731 libusb_set_log_cb
[174](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000000000002787 libusb_set_option
[442](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00000000000091c8 libusb_set_pollfd_notifiers
[489](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 1) 0x0000000000009660 libusb_setlocale
```

neheb, 2022-09-05:
when getting rid of the def file, output for mingw32 changes:

```
objdump.exe -t libusb-1.0.dll | grep libusb_set
[126](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00001afb _libusb_set_configuration@8
[132](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00001d46 _libusb_set_interface_alt_setting@12
[150](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000232b _libusb_set_auto_detach_kernel_driver@8
[151](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x00002355 _libusb_set_debug@8
[152](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x0000239d _libusb_set_log_cb@12
[153](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x000023db _libusb_set_option
[405](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 0) 0x000081b3 _libusb_set_pollfd_notifiers@16
[448](sec  1)(fl 0x00)(ty   20)(scl   2) (nx 1) 0x000085a0 _libusb_setlocale@4
```

Interesting...

tormodvolden, 2022-09-06:
It is at least clear to me that line 157 must be deleted, since @8 would be only for __stdcall (indicating 8 bytes should be reserved for parameters), while libusb_set_option is __cdecl.

neheb, 2022-09-06:
ignore the clang32 result. it's actually similar to the mingw32 one.

tormodvolden, 2022-09-06:
BTW, in the DLL dumps above, the functions always have a leading underscore on 32-bit but none on 64-bit. Is this explained anywhere?

neheb, 2022-09-06:
from ld's doc page:

```
--leading-underscore
--no-leading-underscore

    For most targets default symbol-prefix is an underscore and is defined in target’s description. By this option it is possible to disable/enable the default underscore symbol-prefix. 
```

I don't know more than that.

https://sourceware.org/binutils/docs-2.38/ld/Options.html

tormodvolden, 2022-09-06:
Here is some discussion about this:
https://stackoverflow.com/questions/45605895/is-the-leading-underscore-for-the-function-names-for-the-stdcall-and-cdecl-c
But I would have preferred a clean definition from Microsoft instead of interpreting people's discussions and reverse engineering...

neheb, 2022-09-06:
sounds like _ prefix and @ suffix is some legacy stuff if completely ignored with 64-bit...

mcuee, 2022-09-12:
Hopefully the following pull request is able to finally fix this issue.
* https://github.com/libusb/libusb/pull/1197

tormodvolden, 2022-09-12:
Can somebody tell me what tool is the consumer of the .def file, like, in which exact process (or linking step) is it used?

mcuee, 2022-09-12:
> Can somebody tell me what tool is the consumer of the .def file, like, in which exact process (or linking step) is it used?

I belive this is the one.
https://github.com/libusb/libusb/blob/master/libusb/Makefile.am#L86

Please refer to my comments here as well..
* https://github.com/libusb/libusb/pull/1193#issuecomment-1242683309

> Last time (before mingw-w64) it is important to make the old mingw.org 32bit compiler DLL to be compatible with MSVC generated 32bit DLL through the libusb-1.0.def file.
> Ref: 
> https://github.com/libusb/libusb/commit/bd623f70d99fad8b975f8d4ee62ea4b1abf1272f
> https://github.com/libusb/libusb/blob/master/configure.ac#L227
>
> Ref: 
> https://github.com/libusb/libusb/commit/6b33cd458921a6023e6e54f3715ead7ed441c648
> https://github.com/libusb/libusb/blob/master/libusb/Makefile.am#L86
>
> Now we do not really support mingw.org 32bit compiler any more, so I am not so sure if this is still required or not.

tormodvolden, 2022-09-12:
So we use it ourselves to build the DLL. That would be consistent with e.g. https://docs.microsoft.com/en-us/cpp/build/exporting-from-a-dll?view=msvc-170
But up till now we also generated the lib file from a built DLL. That is what has been confusing me slightly.

mcuee, 2022-09-12:
> So we use the .def file ourselves to build the DLL. That would be consistent with e.g. https://docs.microsoft.com/en-us/cpp/build/exporting-from-a-dll?view=msvc-170 But up till now we also generated the .def file from a built DLL. That is what has been confusing me slightly.

Actually I am always confused by this myself. And that `libusb_set_option` is really causing troubles here and in other places (eg: Python bindings).

I think the idea is to use the def file to make sure MinGW generated DLL to be compatible with MSVC. The def file is used both for generating the mingw libusb-1.0.dll and MSVC libusb-1.0.dll. But for the details, maybe @pbatard can help to shed some light.

mcuee, 2022-09-12:
@tormodvolden 

From Pete's comment above, it seems to me the only right way is to use `_libusb_set_option`.
https://github.com/libusb/libusb/issues/944#issuecomment-913198521

tormodvolden, 2022-09-12:
In our Wiki (Maintainer's Corner) it is written how to generate the .def file from "the libusb-1.0 DLL generated by MinGW32". I don't know if the old .def file used in the configure.ac snipped above, affects MinGW32 when it builds this DLL, but that would be weird. Is it meant to be only used by MSVC later to generate a compatible DLL?

tormodvolden, 2022-09-12:
So should _libusb_set_option be the only one with leading underscore, or should all functions have it?

neheb, 2022-09-12:
If you want linking failures, sure.

leading underscore is for 32-bit buids. the linker adds them automatically anyway.

mcuee, 2022-09-13:
> In our Wiki (Maintainer's Corner) it is written how to generate the .def file from "the libusb-1.0 DLL generated by MinGW32". I don't know if the old .def file used in the configure.ac snipped [...]

Your understanding was also my understanding before I saw the line in `Makefile.am`. 
I am actually getting more confused now compared to before.
https://github.com/libusb/libusb/blob/master/libusb/Makefile.am#L86

If you ask me, I will say to deprecate `libusb_set_option` and split its function to a few functions...

neheb, 2022-09-13:
If 32-bit DLL compatibility is to be killed, the def file should be removed. Microsoft recommends using declspec(dllX) anyway.

[6 more replies]

closed 2023-01-23 by mcuee