libusb/libusb #945: fxload segfault under macOS

mcuee, 2021-07-03:
I just got my Cypress EZ-USB FX3 SuperSpeed Explorer Kit and the fxload example will cause segfault under macOS.

After FW downloading the FX3 will disconect from USB and the reenumerate to be another device 0x04b4:0x00f1.

```
FX3_SDK_1.3.4_MacOS/cyusb_mac_1.0/fx3_images ❯ ~/build/libusb/libusb/examples/fxload -t fx3 -i cyfxbulksrcsink.img 
found device 'Cypress FX3' [04b4:00f3] (2,19)
microcontroller type: fx3
cyfxbulksrcsink.img: type Cypress IMG format
open firmware image cyfxbulksrcsink.img for RAM upload
normal FW binary executable image with checksum
FX3 bootloader version: 0x000000A9
writing image...
transfer execution to Program Entry at 0x40018ea4
libusb: error [darwin_reenumerate_device] darwin/reenumerate_device: timeout waiting for reenumerate
[1]    64014 segmentation fault  ~/build/libusb/libusb/examples/fxload -t fx3 -i cyfxbulksrcsink.img
```

mcuee, 2021-07-03:
Debug log.

<details>
<summary> click for the full debug log </summary>

```

FX3_SDK_1.3.4_MacOS/cyusb_mac_1.0/fx3_images ❯ ~/build/libusb/libusb/examples/fxload -t fx3 -i cyfxbulksrcsink.img
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000028] [0002ade9] libusb: debug [libusb_init] created default context
[ 0.000051] [0002ade9] libusb: debug [libusb_init] libusb v1.0.24.11630
[ 0.000104] [0002ade9] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000294] [0002ade9] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb4db1e0
[ 0.000326] [0002ade9] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb4db1e0
[ 0.001958] [0002ade9] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.001963] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.001964] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.001966] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.001968] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.001969] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.001970] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.001972] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.001973] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.001974] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.001976] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.001977] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.001978] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.001980] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.001997] [0002ade9] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.001999] [0002ade9] libusb: debug [process_new_device] allocating new device in context 0x15260d5d0 for with session 0xb4db1e0
[ 0.002002] [0002ade9] libusb: debug [process_new_device] found device with address 1 port = 5 parent = 0x0 at 0x1526109ba
[ 0.002050] [0002ade9] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb6d338a
[ 0.002076] [0002ade9] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb6d338a/0x2100000 against cached device with sessionID/locationID 0xb4db1e0/0x2500000
[ 0.002078] [0002ade9] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb6d338a
[ 0.002403] [0002ade9] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002406] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002407] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002409] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002410] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002411] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002413] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002414] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002415] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002416] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002418] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002419] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002420] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002421] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002432] [0002ade9] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002434] [0002ade9] libusb: debug [process_new_device] allocating new device in context 0x15260d5d0 for with session 0xb6d338a
[ 0.002436] [0002ade9] libusb: debug [process_new_device] found device with address 2 port = 1 parent = 0x0 at 0x152610aba
[ 0.002475] [0002ade9] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb7116ed
[ 0.002489] [0002ade9] libusb: debug [darwin_get_cached_device] parent sessionID: 0xb4db1e0
[ 0.002491] [0002ade9] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb7116ed/0x2510000 against cached device with sessionID/locationID 0xb6d338a/0x2100000
[ 0.002493] [0002ade9] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb7116ed/0x2510000 against cached device with sessionID/locationID 0xb4db1e0/0x2500000
[ 0.002494] [0002ade9] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb7116ed
[ 0.002661] [0002ade9] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002663] [0002ade9] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...]
[ 0.242241] [0002adec] libusb: debug [process_new_device] found device with address 24 port = 3 parent = 0x152610780 at 0x152611e8a
[27.181607] [0002ade9] libusb: error [darwin_reenumerate_device] darwin/reenumerate_device: timeout waiting for reenumerate
[27.181640] [0002ade9] libusb: debug [libusb_close]  
[27.181643] [0002ade9] libusb: debug [libusb_release_interface] interface 0
[1]    65495 segmentation fault  ~/build/libusb/libusb/examples/fxload -t fx3 -i cyfxbulksrcsink.img
```

</details>

mcuee, 2021-07-03:
This is more serious than #930 where there is just a error message but no segfault.

LudovicRousseau, 2021-07-03:
Can you run fxload inside lldb and get the backtrace of the crash?

mcuee, 2021-07-04:
Okay, I will try to use lldb.

A bit more info: the bootloader is high speed USB device. After the FW upgrade, it will becomes superspeed USB device.

```
libusb/cyusb3/firmware ❯ lsusb -t
/opt/homebrew/bin/lsusb: line 89: 16#: invalid integer constant (error token is "16#")
  Bus 000.Dev 001: USB 3.1 Bus, 
    |__ Bus 002.Dev 002: 4-Port USB 2.0 Hub, 480Mb/s
        |__ Bus 002.Dev 004: 4-Port USB 2.0 Hub, 480Mb/s
            |__ Bus 002.Dev 006: USB 2.0 Hub [MTT], 480Mb/s
                |__ Bus 002.Dev 010: USB Receiver, 12Mb/s
                |__ Bus 002.Dev 009: USB 2.0 Hub [MTT], 480Mb/s
            |__ Bus 002.Dev 012: WestBridge, 480Mb/s
            |__ Bus 002.Dev 008: PROLiNK PCC3220, 480Mb/s
    |__ Bus 002.Dev 001: 4-Port USB 3.0 Hub, 5Gb/s
        |__ Bus 002.Dev 003: 4-Port USB 3.0 Hub, 5Gb/s
            |__ Bus 002.Dev 005: USB3.0 Card Reader, 5Gb/s
libusb/cyusb3/firmware ❯ ~/build/libusb/libusb/examples/fxload -t fx3 -i BMFW_BulkLoopAuto.img
found device 'Cypress FX3' [04b4:00f3] (2,12)
microcontroller type: fx3
BMFW_BulkLoopAuto.img: type Cypress IMG format
open firmware image BMFW_BulkLoopAuto.img for RAM upload
normal FW binary executable image with checksum
FX3 bootloader version: 0x000000A9
writing image...
transfer execution to Program Entry at 0x4001eca8
libusb: error [darwin_reenumerate_device] darwin/reenumerate_device: timeout waiting for reenumerate
[1]    9676 segmentation fault  ~/build/libusb/libusb/examples/fxload -t fx3 -i BMFW_BulkLoopAuto.img
libusb/cyusb3/firmware took 27s ❯ lsusb -t                                                             
/opt/homebrew/bin/lsusb: line 89: 16#: invalid integer constant (error token is "16#")
  Bus 000.Dev 001: USB 3.1 Bus, 
    |__ Bus 002.Dev 002: 4-Port USB 2.0 Hub, 480Mb/s
        |__ Bus 002.Dev 004: 4-Port USB 2.0 Hub, 480Mb/s
            |__ Bus 002.Dev 006: USB 2.0 Hub [MTT], 480Mb/s
                |__ Bus 002.Dev 010: USB Receiver, 12Mb/s
                |__ Bus 002.Dev 009: USB 2.0 Hub [MTT], 480Mb/s
            |__ Bus 002.Dev 008: PROLiNK PCC3220, 480Mb/s
    |__ Bus 002.Dev 001: 4-Port USB 3.0 Hub, 5Gb/s
        |__ Bus 002.Dev 003: 4-Port USB 3.0 Hub, 5Gb/s
            |__ Bus 002.Dev 005: USB3.0 Card Reader, 5Gb/s
            |__ Bus 002.Dev 013: FX3 BULKLOOP, 5Gb/s
```

mcuee, 2021-07-04:
@LudovicRousseau lldb log here. I am trying to follow lldb guides from the following two links.
https://lldb.llvm.org/use/map.html
https://developer.apple.com/library/archive/documentation/General/Conceptual/lldb-guide/chapters/C5-Examining-The-Call-Stack.html

```
libusb/cyusb3/firmware ❯ lldb -- ~/bin/bin/fxload -t fx3 -i BMFW_BulkLoopAuto.img
(lldb) target create "/Users/mcuee/bin/bin/fxload"
Current executable set to '/Users/mcuee/bin/bin/fxload' (arm64).
(lldb) settings set -- target.run-args  "-t" "fx3" "-i" "BMFW_BulkLoopAuto.img"
(lldb) run
Process 28445 launched: '/Users/mcuee/bin/bin/fxload' (arm64)
found device 'Cypress FX3' [04b4:00f3] (2,15)
microcontroller type: fx3
BMFW_BulkLoopAuto.img: type Cypress IMG format
open firmware image BMFW_BulkLoopAuto.img for RAM upload
normal FW binary executable image with checksum
FX3 bootloader version: 0x000000A9
writing image...
transfer execution to Program Entry at 0x4001eca8
libusb: error [darwin_reenumerate_device] darwin/reenumerate_device: timeout waiting for reenumerate
libusb-1.0.0.dylib was compiled with optimization - stepping may behave oddly; variables may not be available.
Process 28445 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x0000000100124c4c libusb-1.0.0.dylib`darwin_reenumerate_device(dev_handle=0x0000000108008000, capture=false) at darwin_usb.c:1714:6 [opt]
   1711	  cached_configurations = alloca (sizeof (*cached_configurations) * descriptor.bNumConfigurations);
   1712	
   1713	  for (i = 0 ; i < descriptor.bNumConfigurations ; ++i) {
-> 1714	    (*(dpriv->device))->GetConfigurationDescriptorPtr (dpriv->device, i, &cached_configuration);
   1715	    memcpy (cached_configurations + i, cached_configuration, sizeof (cached_configurations[i]));
   1716	  }
   1717	
Target 0: (fxload) stopped.
(lldb) thread backtrace
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
  * frame #0: 0x0000000100124c4c libusb-1.0.0.dylib`darwin_reenumerate_device(dev_handle=0x0000000108008000, capture=false) at darwin_usb.c:1714:6 [opt]
    frame #1: 0x000000010011a760 libusb-1.0.0.dylib`libusb_release_interface(dev_handle=0x0000000108008000, interface_number=0) at core.c:1715:6 [opt]
    frame #2: 0x0000000100120b1c libusb-1.0.0.dylib`darwin_close(dev_handle=0x0000000108008000) at darwin_usb.c:1271:7 [opt]
    frame #3: 0x000000010011a3d4 libusb-1.0.0.dylib`do_close(ctx=<unavailable>, dev_handle=0x0000000108008000) at core.c:1429:2 [opt]
    frame #4: 0x000000010011a1d0 libusb-1.0.0.dylib`libusb_close(dev_handle=0x0000000108008000) at core.c:1483:2 [opt]
    frame #5: 0x0000000100002e50 fxload`main(argc=<unavailable>, argv=<unavailable>) at fxload.c:302:2 [opt]
    frame #6: 0x0000000189a01450 libdyld.dylib`start + 4
(lldb) thread list
Process 28445 stopped
* thread #1: tid = 0x16430, 0x0000000100124c4c libusb-1.0.0.dylib`darwin_reenumerate_device(dev_handle=0x0000000108008000, capture=false) at darwin_usb.c:1714:6, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
  thread #2: tid = 0x16442, 0x00000001899a8dd4 libsystem_kernel.dylib`mach_msg_trap + 8, name = 'org.libusb.device-hotplug'
(lldb) 

```

mcuee, 2021-07-04:
The bootloader device and the new device are very different.

```

libusb/cyusb3/firmware ❯ lsusb1 -vvv -d 04b4:00fe

Bus 002 Device 016: ID 04b4:00fe  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x04b4 
  idProduct          0x00fe 
  bcdDevice            0.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              400mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000641e
      BESL Link Power Management (LPM) Supported
    BESL value     1024 us 
    Deep BESL value    24576 us 
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   3
      Lowest fully-functional device speed is SuperSpeed (5Gbps)
    bU1DevExitLat           0 micro seconds
    bU2DevExitLat           0 micro seconds
can't get debug descriptor: No such file or directory
Device Status:     0x0000
  (Bus Powered)
```

mcuee, 2021-07-04:
This is a regression from 1.0.24 release since 1.0.24 is okay.

```
libusb/cyusb3/firmware ❯ ~/build/libusb/libusb-1.0.24/buildarm64/examples/fxload -t fx3 -i BMFW_BulkLoopAuto.img  
found device 'Cypress FX3' [04b4:00f3] (2,17)
microcontroller type: fx3
BMFW_BulkLoopAuto.img: type Cypress IMG format
open firmware image BMFW_BulkLoopAuto.img for RAM upload
normal FW binary executable image with checksum
FX3 bootloader version: 0x000000A9
writing image...
transfer execution to Program Entry at 0x4001eca8```

mcuee, 2021-07-04:
Trying to narrow down the commit causing this issue.
https://github.com/libusb/libusb/commits/master/libusb/os/darwin_usb.c

Commit d97e85e is okay (darwin: Don't drop partial data on timeout).

```
libusb on  HEAD (d97e85e) [!] ❯ ./examples/fxload -t fx3 -i ../cyusb3/firmware/BMFW_BulkLoopAuto.img
found device 'Cypress FX3' [04b4:00f3] (2,21)
microcontroller type: fx3
../cyusb3/firmware/BMFW_BulkLoopAuto.img: type Cypress IMG format
open firmware image ../cyusb3/firmware/BMFW_BulkLoopAuto.img for RAM upload
normal FW binary executable image with checksum
FX3 bootloader version: 0x000000A9
writing image...
transfer execution to Program Entry at 0x4001eca8
```

mcuee, 2021-07-04:
Looks like pull request #911 introduced this issue.

```
libusb on  HEAD (6a902c3) [!] ❯ ./examples/fxload -t fx3 -i ../cyusb3/firmware/BMFW_BulkLoopAuto.img
found device 'Cypress FX3' [04b4:00f3] (2,23)
microcontroller type: fx3
../cyusb3/firmware/BMFW_BulkLoopAuto.img: type Cypress IMG format
open firmware image ../cyusb3/firmware/BMFW_BulkLoopAuto.img for RAM upload
normal FW binary executable image with checksum
FX3 bootloader version: 0x000000A9
writing image...
transfer execution to Program Entry at 0x4001eca8
libusb: error [darwin_reenumerate_device] darwin/reenumerate_device: timeout waiting for reenumerate
[1]    46117 segmentation fault  ./examples/fxload -t fx3 -i ../cyusb3/firmware/BMFW_BulkLoopAuto.img
```

mcuee, 2021-07-04:
https://github.com/libusb/libusb/commits/master/libusb/os/darwin_usb.c

I can not zoom into the paticular commit within pull request #911. First commit is okay but then the second commit can not be built.

```
libusb on  HEAD (bd097fe) [!] took 3s ❯ make
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in libusb
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_posix.lo
  CC       os/threads_posix.lo
  CC       os/darwin_usb.lo
os/darwin_usb.c:1677:12: error: conflicting types for 'darwin_reenumerate_device'
static int darwin_reenumerate_device (struct libusb_device_handle *dev_handle) {
           ^
os/darwin_usb.c:84:12: note: previous declaration is here
static int darwin_reenumerate_device(struct libusb_device_handle *dev_handle, bool capture);
           ^
os/darwin_usb.c:2301:25: warning: incompatible function pointer types initializing 'int (*)(struct libusb_device_handle *)' with an expression of type 'int (struct libusb_device_handle *, bool)' [-Wincompatible-function-pointer-types]
        .reset_device = darwin_reenumerate_device,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~
1 warning and 1 error generated.
make[2]: *** [os/darwin_usb.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

mcuee, 2021-07-04:
@LudovicRousseau @osy @hjelmn Hopefully now that you have enough info to diagnose this issue. Thanks.

hjelmn, 2021-07-05:
ok. will try to take a look tonight or tomorrow

hjelmn, 2021-07-09:
I see what is happening and I can fix the SEGV but then we are back to the error you saw in the other bug. I know the second commit of #911 doesn't compile but can you try from the third commit. I will see if I can add additional debugging to isolate the issue.

hjelmn, 2021-07-09:
Pushed a change to fix the SEGV and add some debugging. Please run with that and post the debug log if it still fails (either by SEGV or not).

mcuee, 2021-07-09:
Latest git has no SEGV but still have the time out.

```
ibusb/cyusb3/firmware ❯ ~/bin/bin/fxload -t fx3 -i BMFW_BulkLoopAuto.img
found device 'Cypress FX3' [04b4:00f3] (2,8)
microcontroller type: fx3
BMFW_BulkLoopAuto.img: type Cypress IMG format
open firmware image BMFW_BulkLoopAuto.img for RAM upload
normal FW binary executable image with checksum
read bootloader version: LIBUSB_ERROR_TIMEOUT
Could not read bootloader version
libusb: error [darwin_reenumerate_device] darwin/reenumerate_device: timeout waiting for reenumerate
libusb/cyusb3/firmware took 29s ❯ ~/bin/bin/fxload -t fx3 -i BMFW_BulkLoopAuto.img
found device 'Cypress FX3' [04b4:00f3] (2,19)
microcontroller type: fx3
BMFW_BulkLoopAuto.img: type Cypress IMG format
open firmware image BMFW_BulkLoopAuto.img for RAM upload
normal FW binary executable image with checksum
FX3 bootloader version: 0x000000A9
writing image...
transfer execution to Program Entry at 0x4001eca8
libusb: error [darwin_reenumerate_device] darwin/reenumerate_device: timeout waiting for reenumerate
```

<details>
<summary> click to expand for the detailed debug log </summary>

```
libusb/cyusb3/firmware took 27s ❯ export LIBUSB_DEBUG=4
libusb/cyusb3/firmware ❯ ~/bin/bin/fxload -t fx3 -i BMFW_BulkLoopAuto.img
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000040] [0000bf82] libusb: debug [libusb_init] created default context
[ 0.000070] [0000bf82] libusb: debug [libusb_init] libusb v1.0.24.11635
[ 0.000144] [0000bf82] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000399] [0000bf82] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb2d626f
[ 0.000447] [0000bf82] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb2d626f
[ 0.002666] [0000bf82] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002678] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002681] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002684] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002686] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002688] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002690] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002692] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002694] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002696] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002698] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002700] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002702] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002704] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002728] [0000bf82] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002731] [0000bf82] libusb: debug [process_new_device] allocating new device in context 0x14d60d8e0 for with session 0xb2d626f
[ 0.002736] [0000bf82] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x14d70497a
[ 0.002812] [0000bf82] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb542409
[ 0.002850] [0000bf82] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb542409/0x2500000 against cached device with sessionID/locationID 0xb2d626f/0x2100000
[ 0.002854] [0000bf82] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb542409
[ 0.003193] [0000bf82] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003197] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003199] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.003202] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003204] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003206] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.003207] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.003209] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.003211] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.003213] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.003215] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003217] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003219] [0000bf82] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[... 1770 lines cut ...]
[ 0.296121] [0000bfe2] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.296123] [0000bfe2] libusb: debug [process_new_device] allocating new device in context 0x158f06050 for with session 0x144499e3dd
[... cut ...]
[27.148818] [0000bfd9] libusb: debug [libusb_unref_device] destroy device 2.5
[27.148854] [0000bfd9] libusb: debug [libusb_unref_device] destroy device 2.4
[27.148887] [0000bfd9] libusb: debug [libusb_unref_device] destroy device 2.3
[27.148919] [0000bfd9] libusb: debug [libusb_unref_device] destroy device 2.2
[27.148947] [0000bfd9] libusb: debug [libusb_unref_device] destroy device 2.1
[27.149005] [0000bfd9] libusb: debug [usbi_remove_event_source] remove fd 3
libusb/cyusb3/firmware took 27s ❯ 

```

</details>

mcuee, 2021-07-09:
> I see what is happening and I can fix the SEGV but then we are back to the error you saw in the other bug. I know the second commit of #911 doesn't compile but can you try from the third commit. I will see if I can add additional debugging to isolate the issue.

Same for the third commit.
```
libUsb on  HEAD (abd788c) [!] ❯ make
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in libusb
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_posix.lo
  CC       os/threads_posix.lo
  CC       os/darwin_usb.lo
os/darwin_usb.c:2303:54: error: no member named 'service' in 'struct darwin_cached_device'; did you mean 'device'?
  dpriv->device = darwin_device_from_service (dpriv->service);
                                                     ^~~~~~~
                                                     device
os/darwin_usb.h:173:25: note: 'device' declared here
  usb_device_t        **device;
                        ^
os/darwin_usb.c:2319:12: warning: unused variable 'kresult' [-Wunused-variable]
  IOReturn kresult;
           ^
1 warning and 1 error generated.
make[2]: *** [os/darwin_usb.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

hjelmn, 2021-07-09:
What is weird is that the location is changing. That is why it times out. The device was at 0x2113000 but re-attaches at 0x2513000. Can you post a success debug log with 1.0.24?

mcuee, 2021-07-09:
Location will sure change. The boot loader device is a high speed device but after reset, it is a superspeed device.

mcuee, 2021-07-09:
Debug log for 1.0.24.

<details>
<summary> click to expand for the debug log </summary>

```

libusb/cyusb3/firmware ❯ ~/build/libusb/libusb-1.0.24/buildarm64/examples/fxload -t fx3 -i BMFW_BulkLoopAuto.img
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000028] [0001128c] libusb: debug [libusb_init] created default context
[ 0.000049] [0001128c] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000238] [0001128c] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb2d626f
[ 0.000272] [0001128c] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb2d626f
[ 0.002217] [0001128c] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002224] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002225] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002227] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002229] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002230] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002231] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002233] [0001128c] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002234] [0001128c] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002236] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002237] [0001128c] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002238] [0001128c] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002240] [0001128c] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002241] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002258] [0001128c] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002261] [0001128c] libusb: debug [process_new_device] allocating new device in context 0x11e804f00 for with session 0xb2d626f
[ 0.002264] [0001128c] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x11e8077ea
[ 0.002325] [0001128c] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb542409
[ 0.002354] [0001128c] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb542409/0x2500000 against cached device with sessionID/locationID 0xb2d626f/0x2100000
[ 0.002357] [0001128c] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb542409
[ 0.002654] [0001128c] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002659] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002660] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002662] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002663] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002664] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002666] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002667] [0001128c] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002668] [0001128c] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002670] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002671] [0001128c] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002672] [0001128c] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002674] [0001128c] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002675] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002687] [0001128c] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002689] [0001128c] libusb: debug [process_new_device] allocating new device in context 0x11e804f00 for with session 0xb542409
[ 0.002691] [0001128c] libusb: debug [process_new_device] found device with address 2 port = 5 parent = 0x0 at 0x11e805cda
[ 0.002745] [0001128c] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb6e1eef
[ 0.002766] [0001128c] libusb: debug [darwin_get_cached_device] parent sessionID: 0xb2d626f
[ 0.002769] [0001128c] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb6e1eef/0x2110000 against cached device with sessionID/locationID 0xb542409/0x2500000
[ 0.002771] [0001128c] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb6e1eef/0x2110000 against cached device with sessionID/locationID 0xb2d626f/0x2100000
[ 0.002772] [0001128c] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb6e1eef
[ 0.003035] [0001128c] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003039] [0001128c] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...]
[ 0.114929] [0001128c] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.114931] [0001128c] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.114933] [0001128c] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.114934] [0001128c] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.114936] [0001128c] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.114938] [0001128c] libusb: debug [usbi_remove_event_source] remove fd 3
[ 0.114996] [0001128f] libusb: debug [darwin_event_thread_main] darwin event thread exiting
```

</details>

mcuee, 2021-07-09:
Debug log for the last known good commit.

<details>
<summary> Click to expand for the last known good debug log </summary>

```
libUsb on  HEAD (b8b7fd6) [!] ❯ export LIBUSB_DEBUG=4
libUsb on  HEAD (b8b7fd6) [!] ❯ ./examples/fxload -t fx3 -i ../cyusb3/firmware/BMFW_BulkLoopAuto.img
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000029] [0001411b] libusb: debug [libusb_init] created default context
[ 0.000049] [0001411b] libusb: debug [libusb_init] libusb v1.0.24.11610
[ 0.000308] [0001411b] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb2d626f
[ 0.000347] [0001411b] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb2d626f
[ 0.001964] [0001411b] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.001971] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.001972] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.001974] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.001975] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.001977] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.001978] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.001979] [0001411b] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.001981] [0001411b] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.001982] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.001985] [0001411b] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.001986] [0001411b] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.001987] [0001411b] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.001988] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002006] [0001411b] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002008] [0001411b] libusb: debug [process_new_device] allocating new device in context 0x141705630 for with session 0xb2d626f
[ 0.002011] [0001411b] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x141707fca
[ 0.002074] [0001411b] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb542409
[ 0.002102] [0001411b] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb542409/0x2500000 against cached device with sessionID/locationID 0xb2d626f/0x2100000
[ 0.002104] [0001411b] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb542409
[ 0.002352] [0001411b] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002357] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002358] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002360] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002361] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002362] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002364] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002365] [0001411b] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002366] [0001411b] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002368] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002369] [0001411b] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002371] [0001411b] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002372] [0001411b] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002373] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002385] [0001411b] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002387] [0001411b] libusb: debug [process_new_device] allocating new device in context 0x141705630 for with session 0xb542409
[ 0.002389] [0001411b] libusb: debug [process_new_device] found device with address 2 port = 5 parent = 0x0 at 0x1417064fa
[ 0.002443] [0001411b] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb6e1eef
[ 0.002465] [0001411b] libusb: debug [darwin_get_cached_device] parent sessionID: 0xb2d626f
[ 0.002468] [0001411b] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb6e1eef/0x2110000 against cached device with sessionID/locationID 0xb542409/0x2500000
[ 0.002470] [0001411b] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb6e1eef/0x2110000 against cached device with sessionID/locationID 0xb2d626f/0x2100000
[ 0.002472] [0001411b] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb6e1eef
[ 0.002737] [0001411b] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002741] [0001411b] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...]
[ 0.113339] [0001411b] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.113341] [0001411b] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.113342] [0001411b] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.113344] [0001411b] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.113345] [0001411b] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.113347] [0001411b] libusb: debug [usbi_remove_event_source] remove fd 3
[ 0.113403] [0001411e] libusb: debug [darwin_event_thread_main] darwin event thread exiting
```

</details>

mcuee, 2021-07-09:
Maybe we need to check location and VID/PID changes after renumeration, then we can safely assume that device is no longer existing  (totally changed).

hjelmn, 2021-07-09:
Hmm. I don't see a device reset in the log for the last know good version.

hjelmn, 2021-07-09:
I think this is due to ```auto_detach_kernel_driver```. Re-attaching the kernel driver causes the device to be re-enumerated. Probably some issue with error handling in that path. Will add more debugging now so I can determine just what exactly is happening before I attempt to fix it.

hjelmn, 2021-07-09:
Added some logging and fixed some possible issues. Can you give the latest commit a try?

mcuee, 2021-07-10:
Great. Now it is fixed.

```
libusb on  master [!] ❯ ./examples/fxload -t fx3 -i ../cyusb3/firmware/BMFW_BulkLoopAuto.img
found device 'Cypress FX3' [04b4:00f3] (2,12)
microcontroller type: fx3
../cyusb3/firmware/BMFW_BulkLoopAuto.img: type Cypress IMG format
open firmware image ../cyusb3/firmware/BMFW_BulkLoopAuto.img for RAM upload
normal FW binary executable image with checksum
FX3 bootloader version: 0x000000A9
writing image...
transfer execution to Program Entry at 0x4001eca8```

closed 2021-07-10 by mcuee

hjelmn, 2021-07-10:
as suspected the kernel detach failed so the attach failed as well. the error should have been ignored and now is. glad to hear it is fixed.