libusb/libusb #946: libusb_control_transfer returns Error Pipe

Ledz96, 2021-07-06:
I am currently trying to send data to an attached Logitech Unifying Receiver through libusb. I used [this](https://www.usbmadesimple.co.uk/ums_4.htm) resource to figure out the syntax of control transfers, and tried building something that generates random but protocol-compliant transfers.

However, all I am getting for any of the requests that I send is PIPE_ERROR.

I will leave a sample of the code at the bottom. I tried something very simple, such as taking a simple request and modifying the only "free" parameter (as all others are bound), but I still get the same error. 

`libusb_device_handle *dev_handle;
	int r;
	std::vector<uint8_t> data;
	uint32_t timeout = 0xffff;
	data.push_back(1);

	r = libusb_open(dev, &dev_handle);
	if (r != 0) {
		cout<<"ERROR: could not open device"<<endl;
		return false;
	}

	int is_kernel_active;
    is_kernel_active = libusb_kernel_driver_active(dev_handle, 1);
    if (is_kernel_active)
    {
        libusb_detach_kernel_driver(dev_handle, 1);
    }

    auto ret = libusb_claim_interface(dev_handle, 1);
    if (ret != 0)
    {
        throw std::runtime_error("Could not claim interface");
    }
    libusb_error errcode = static_cast<libusb_error>(libusb_control_transfer(dev_handle, 0xa0, 0x8, 0, 0, data.data(), static_cast<uint16_t>(data.size()), timeout));
    ret = libusb_release_interface(dev_handle, 1);
    if (ret != 0)
    {
        throw std::runtime_error("Could not release interface");
    }

    if (is_kernel_active)
    {
        libusb_attach_kernel_driver(dev_handle, 1);
    }`

Am I misunderstanding the way the library is bound to be used, or might there be in issue in the library itself?

mcuee, 2021-07-06:
Please look at hidpp project to see how it should work. You need to know the communication protocol to send the right command. And you may want to use HIDAPI and not libusb.
https://github.com/cvuchener/hidpp

mcuee, 2021-07-06:
I will close this issue for now.

closed 2021-07-06 by mcuee