libusb/libusb #951: Fix regression after commit "core: Refactor initialization and how the default context is handled"

bohdan-tymkiv, 2021-07-14:
Commit 32a22069428cda9d63aa666e92fb8882a83d4515 introduces a regression when `usbi_default_context` is not initialized in case non-NULL context is passed to the `libusb_init()` - [link to error location](https://github.com/libusb/libusb/commit/32a22069428cda9d63aa666e92fb8882a83d4515#diff-c1f9bc250077d41456a3e580fca0ddf5d8c25b741bff6d9b9505990a8b70b254R2277). This causes unwanted performance degradation (at least on Windows platform) because after the commit the `usbi_get_context()` always returns NULL and `log_v()` function is now forced to call `get_env_debug_level()` [for every single debug message](https://github.com/libusb/libusb/blob/339d24c8235476ceee43f7d704415ccab85e2583/libusb/core.c#L2575).
With USB2.0 FS BULK device I am working on performance drop is about 35% (355 KiB/s -> 260 KiB/s).

mcuee, 2021-07-14:
Pull request is welcome.

mcuee, 2021-07-16:
I can reproduce the issue as well with SuperSpeed USB device, from 230MB/sec to 164MB/sec using a pyusb code here.
https://github.com/pyusb/pyusb/issues/235#issuecomment-881274639

mcuee, 2021-07-16:
Test results with 1.0.24 release or the fix #952

```
(py39venv) C:\work\cyusb\cyusb3014-breakout\software [master ≡ +3 ~1 -0 !]> python .\sourcesink_python3.py
ep:       ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x0
ep2:       ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x0
Starting CYUSB3014 python bulk sourcesink test
Write: transfered 231119 kB/s
Read: transfered 232270 kB/s
Write: transfered 231203 kB/s
Read: transfered 233250 kB/s
Write: transfered 230691 kB/s
Read: transfered 232099 kB/s
Write: transfered 231317 kB/s
Read: transfered 232981 kB/s
Write: transfered 232439 kB/s
Read: transfered 233911 kB/s
Write: transfered 231660 kB/s
Read: transfered 233451 kB/s
Write: transfered 233408 kB/s
Read: transfered 234304 kB/s

```

Test results with git head (1.0.24.11636).
```
(py39venv) C:\work\cyusb\cyusb3014-breakout [master ≡ +3 ~1 -0 !]> cd .\software\
(py39venv) C:\work\cyusb\cyusb3014-breakout\software [master ≡ +3 ~1 -0 !]> ls

    Directory: C:\work\cyusb\cyusb3014-breakout\software

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---           7/16/2021  2:41 PM          96984 cyfxbulklpautoenum.img
-a---           7/16/2021  2:41 PM            865 cyfxbulklpautoenum.txt
-a---           7/16/2021  3:03 PM           1529 loopback-test_python3.py
-a---           7/16/2021  3:03 PM           1529 loopback-test.py
-a---           7/16/2021  3:20 PM           1774 sourcesink_python3.py

(py39venv) C:\work\cyusb\cyusb3014-breakout\software [master ≡ +3 ~1 -0 !]> python .\sourcesink_python3.py
ep:       ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x0
ep2:       ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x0
Starting CYUSB3014 python bulk sourcesink test
Read: transfered 166302 kB/s
Write: transfered 165726 kB/s
Read: transfered 164214 kB/s
Write: transfered 163996 kB/s
Read: transfered 166799 kB/s
Write: transfered 166468 kB/s
Read: transfered 166300 kB/s
Write: transfered 166068 kB/s
Read: transfered 166339 kB/s
Write: transfered 165820 kB/s
Read: transfered 165488 kB/s
Write: transfered 164896 kB/s
Read: transfered 167874 kB/s
Write: transfered 167455 kB/s
Read: transfered 165593 kB/s
Write: transfered 165082 kB/s
Read: transfered 167008 kB/s
Write: transfered 166811 kB/s
Read: transfered 166720 kB/s
Write: transfered 166400 kB/s
Read: transfered 167621 kB/s
Write: transfered 166872 kB/s
```

mcuee, 2021-07-16:
https://github.com/libusb/libusb/commit/32a22069428cda9d63aa666e92fb8882a83d4515 is based on the discussion #855. So we may need to revisit #855 to see whether this is a bug or not.

mcuee, 2021-07-16:
Interestingly I do not see the issues under my Mac Mini M1 but the performance is not that good compared to under Windows.

libusb-1.0.24 release.
```

python_usb on  master [?] via 🐍 v3.9.6 (py396venv) took 27s ❯ python cyusb_fx3_sourcesink_python3.py 
ep:       ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x0
ep2:       ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x0
Starting CYUSB3014 python bulk sourcesink test
Read: transfered 166597 kB/s
Write: transfered 166606 kB/s
Read: transfered 163067 kB/s
Write: transfered 163052 kB/s
Read: transfered 161810 kB/s
Write: transfered 161887 kB/s
Read: transfered 161449 kB/s
Write: transfered 161449 kB/s
Read: transfered 161940 kB/s
Write: transfered 161934 kB/s
Write: transfered 162201 kB/s
Read: transfered 162239 kB/s
Read: transfered 159519 kB/s
Write: transfered 159566 kB/s
Read: transfered 164469 kB/s
Write: transfered 164489 kB/s
Read: transfered 161766 kB/s
Write: transfered 161761 kB/s
Read: transfered 158067 kB/s
Write: transfered 158097 kB/s
qRead: transfered 160811 kB/s
Write: transfered 160945 kB/s
Read: transfered 161937 kB/s
Write: transfered 161932 kB/s
```

libusb git head https://github.com/libusb/libusb/commit/339d24c8235476ceee43f7d704415ccab85e2583

```

python_usb on  master [?] via 🐍 v3.9.6 (py396venv) took 10s ❯ python cyusb_fx3_sourcesink_python3.py
ep:       ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x0
ep2:       ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x0
Starting CYUSB3014 python bulk sourcesink test
Read: transfered 169457 kB/s
Write: transfered 169454 kB/s
Read: transfered 160285 kB/s
Write: transfered 160270 kB/s
Read: transfered 162530 kB/s
Write: transfered 162529 kB/s
Read: transfered 158119 kB/s
Write: transfered 158183 kB/s
Read: transfered 161064 kB/s
Write: transfered 161067 kB/s
Write: transfered 157371 kB/s
Read: transfered 157482 kB/s
Write: transfered 160710 kB/s
Read: transfered 160713 kB/s
```

hjelmn, 2021-07-16:
I will take a look today. My FX3 is out for delivery now.

mcuee, 2021-07-16:
BTW, since pyusb does not support async API yet, the pyusb code is using the sync API. So the performance does not match the async API based libusbk or cypress example. That is not related to this issue.
https://github.com/pyusb/pyusb/issues/235

Probably the following example can be modified to test the async API performance.
https://github.com/libusb/libusb/blob/master/examples/sam3u_benchmark.c

mcuee, 2021-07-20:
From #952 discussion.

@hjelmn @LudovicRousseau 
It seems to me there are two potential ways out. Please take a look. Thanks.
1) revert the first part of the changes in https://github.com/libusb/libusb/commit/32a22069428cda9d63aa666e92fb8882a83d4515
usbi_default_context is only set if libusb_init() is called with NULL.

That part of the change is not strictly necessary in the first place as per the discussion https://github.com/libusb/libusb/issues/855.

2) keep the above part of the change, then fix #951 and potential other issues, say change the logging functions in libusb to use the exported/non-default context instead as per Jonas.

hjelmn, 2021-07-20:
I think the right answer might be to make usbi_dbg take the context. Taking a look now.

hjelmn, 2021-07-20:
Can you try #957 and let me know if it helps? I don't have windows so I can't check the changes.

bohdan-tymkiv, 2021-07-20:
> I think the right answer might be to make usbi_dbg take the context.

I agree. I am not an expert in libusb so I may be wrong, but all other logging functions take `ctx` as a parameter and I don't see a reason why `usbi_dbg` should be different. On the other hand, this change will touch almost all source files, it may be difficult to test it on all supported platforms in all configurations etc.

hjelmn, 2021-07-20:
@bohdan-tymkiv Already done. Can you give it a try and let me know if it helps? I will test macOS, linux, and BSD.

bohdan-tymkiv, 2021-07-20:
Tested the #957 on Windows 10 and I can confirm that it resolves the performance issue **but it crashes** [at this location](https://github.com/hjelmn/libusb/blob/update_usbi_dbg/libusb/io.c#L1312). If I comment call to usbi_dbg everything seems to work fine, at least at first glance.
I will run the #957 under Valgrind on Linux but I have no idea how to check for memory access issues on Windows...

hjelmn, 2021-07-20:
See the problem there. Fixing now.

hjelmn, 2021-07-20:
Crash should be fixed now.

bohdan-tymkiv, 2021-07-20:
@hjelmn Running OpenOCD with #957 under Valgrind shows no issues except one described above. Maybe it would be OK to simply drop the call to `usbi_dbg` in `libusb_alloc_transfer`? It seems it is not possible to get the context within `libusb_alloc_transfer` without breaking interface/backward compatibility.

hjelmn, 2021-07-20:
Actually. I agree. Lets just drop that log message.

mcuee, 2021-07-21:
Yes #957 fixed the performance issue.

```

(py39venv) C:\work\libusb\python_usb [master ≡ +2 ~0 -0 !]> python .\cyusb_fx3_sourcesink_python3.py
ep:       ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x0
ep2:       ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x0
Starting CYUSB3014 python bulk sourcesink test
Write: transfered 221208 kB/s
Read: transfered 223511 kB/s
Write: transfered 223465 kB/s
Read: transfered 225129 kB/s
Write: transfered 224153 kB/s
Read: transfered 225879 kB/s
Write: transfered 224248 kB/s
Read: transfered 225720 kB/s
Write: transfered 222408 kB/s
Read: transfered 224135 kB/s
Read: transfered 227725 kB/s
Write: transfered 225934 kB/s
Write: transfered 222334 kB/s
Read: transfered 224444 kB/s
Read: transfered 224443 kB/s
Write: transfered 221948 kB/s
```

mcuee, 2021-07-21:
> Actually. I agree. Lets just drop that log message.

@hjelmn It is still there now, right?

https://github.com/hjelmn/libusb/blob/update_usbi_dbg/libusb/io.c#L1312
```
DEFAULT_VISIBILITY
struct libusb_transfer * LIBUSB_CALL libusb_alloc_transfer(
	int iso_packets)
{
	size_t priv_size;
	size_t alloc_size;
	unsigned char *ptr;
	struct usbi_transfer *itransfer;
	struct libusb_transfer *transfer;

	assert(iso_packets >= 0);
	if (iso_packets < 0)
		return NULL;

	priv_size = PTR_ALIGN(usbi_backend.transfer_priv_size);
	alloc_size = priv_size
		+ sizeof(struct usbi_transfer)
		+ sizeof(struct libusb_transfer)
		+ (sizeof(struct libusb_iso_packet_descriptor) * (size_t)iso_packets);
	ptr = calloc(1, alloc_size);
	if (!ptr)
		return NULL;

	itransfer = (struct usbi_transfer *)(ptr + priv_size);
	itransfer->num_iso_packets = iso_packets;
	itransfer->priv = ptr;
	usbi_mutex_init(&itransfer->lock);
	transfer = USBI_TRANSFER_TO_LIBUSB_TRANSFER(itransfer);
	usbi_dbg(NULL, "transfer %p", transfer);
	return transfer;
}
```

hjelmn, 2021-07-21:
Fixed. Will merge the PR now.

closed 2021-07-21 by hjelmn

jonasmalacofilho, 2021-07-22:
Thanks, @hjelmn!

mcuee, 2021-07-22:
> BTW, since pyusb does not support async API yet, the pyusb code is using the sync API. So the performance does not match the async API based libusbk or cypress example. That is not related to this issue.
> [pyusb/pyusb#235](https://github.com/pyusb/pyusb/issues/235)
> 
> Probably the following example can be modified to test the async API performance.
> https://github.com/libusb/libusb/blob/master/examples/sam3u_benchmark.c

So I go to libusbdotnet which is based on libusb, under macOS the performance is not bad.

```

libusbdotnet_v2 on  v2 took 18s ❯ ./stage/BenchmarkCon/bin/Release/net5.0/BenchmarkCon list notestselect write buffercount=3 buffersize=32768 loop mode=async
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
1. 04B4:00F1 Cypress - FX3
2. 046D:C52B Logitech - USB Receiver
3. 047F:C025 Plantronics C320-M
4. 1A40:0201 
5. 0AC8:3420 Vimicro Corp. - PROLiNK PCC3220
6. 1A40:0201 
7. 0BDA:5411 Generic - 4-Port USB 2.0 Hub
8. 05E3:0749 Generic - USB3.0 Card Reader
9. 0BDA:0411 Generic - 4-Port USB 3.0 Hub
10. 0BDA:0411 Generic - 4-Port USB 3.0 Hub
11. 0BDA:5411 Generic - 4-Port USB 2.0 Hub

Select device (1-11) :1

Benchmark device 04B4:00F1 opened..
Loop Test Information
	Vid / Pid       : 04B4h / 00F1h
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 32768
	Buffer Count    : 3
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
Bulk Write (Ep01h) max packet size: 1024

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 281226203.00 Transfers: 8086 Bytes/s: 281226203.00
Avg. Bytes/s: 288942931.94 Transfers: 17199 Bytes/s: 296153452.65
Avg. Bytes/s: 291884573.12 Transfers: 26336 Bytes/s: 297587420.67
Avg. Bytes/s: 293362093.71 Transfers: 35470 Bytes/s: 297707201.40
Avg. Bytes/s: 294178361.38 Transfers: 44600 Bytes/s: 297393131.68
Avg. Bytes/s: 294741105.78 Transfers: 53733 Bytes/s: 297520426.37
Avg. Bytes/s: 295190625.51 Transfers: 62869 Bytes/s: 297862455.43
Avg. Bytes/s: 295488757.12 Transfers: 72000 Bytes/s: 297557922.46
Avg. Bytes/s: 295693625.44 Transfers: 81130 Bytes/s: 297319243.79
Avg. Bytes/s: 295910920.67 Transfers: 90252 Bytes/s: 297857662.74
Avg. Bytes/s: 296073306.84 Transfers: 99385 Bytes/s: 297687636.52
Avg. Bytes/s: 296204760.63 Transfers: 108519 Bytes/s: 297642661.24
Avg. Bytes/s: 296320007.38 Transfers: 117654 Bytes/s: 297695970.59
Avg. Bytes/s: 296410631.73 Transfers: 126781 Bytes/s: 297583831.34
qAvg. Bytes/s: 296493732.55 Transfers: 135920 Bytes/s: 297651376.26
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
	Vid / Pid       : 04B4h / 00F1h
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 32768
	Buffer Count    : 3
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
	Total Bytes     : 4453859328
	Total Transfers : 135921
	Avg. Bytes/sec  : 296490308.47
	Elapsed Time    : 15.02 seconds

Bulk Write (Ep01h) max packet size: 1024
	Total Bytes     : 4822728704
	Total Transfers : 147178
	Avg. Bytes/sec  : 321044531.44
	Elapsed Time    : 15.02 seconds

Press any key to exit..

```

mcuee, 2021-07-22:
But single direction transfer speed is lower, not so sure if it is because of libusbdotnet or libusb itself.

```
libusbdotnet_v2 on  v2 took 5s ❯ ./stage/BenchmarkCon/bin/Release/net5.0/BenchmarkCon list notestselect read buffercount=4 buffersize=32768 loop mode=async
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
1. 04B4:00F1 Cypress - FX3
2. 046D:C52B Logitech - USB Receiver
3. 047F:C025 Plantronics C320-M
4. 1A40:0201 
5. 0AC8:3420 Vimicro Corp. - PROLiNK PCC3220
6. 1A40:0201 
7. 0BDA:5411 Generic - 4-Port USB 2.0 Hub
8. 05E3:0749 Generic - USB3.0 Card Reader
9. 0BDA:0411 Generic - 4-Port USB 3.0 Hub
10. 0BDA:0411 Generic - 4-Port USB 3.0 Hub
11. 0BDA:5411 Generic - 4-Port USB 2.0 Hub

Select device (1-11) :1

Benchmark device 04B4:00F1 opened..
Loop Test Information
	Vid / Pid       : 04B4h / 00F1h
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 32768
	Buffer Count    : 4
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
Bulk Write (Ep01h) max packet size: 1024

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 273813690.34 Transfers: 7847 Bytes/s: 273813690.34
Avg. Bytes/s: 285690931.32 Transfers: 16955 Bytes/s: 296782138.04
Avg. Bytes/s: 289820435.65 Transfers: 26097 Bytes/s: 297803844.07
Avg. Bytes/s: 291755212.08 Transfers: 35231 Bytes/s: 297428221.92
Avg. Bytes/s: 292893683.41 Transfers: 44357 Bytes/s: 297373395.37
Avg. Bytes/s: 293787759.87 Transfers: 53511 Bytes/s: 298198602.05
Avg. Bytes/s: 294421763.82 Transfers: 62657 Bytes/s: 298186706.89
Avg. Bytes/s: 294871602.07 Transfers: 71802 Bytes/s: 297991034.32
Avg. Bytes/s: 295190648.74 Transfers: 80945 Bytes/s: 297720399.16
Avg. Bytes/s: 295464122.07 Transfers: 90087 Bytes/s: 297907794.94
Avg. Bytes/s: 295719481.83 Transfers: 99238 Bytes/s: 298257134.16
Avg. Bytes/s: 295918548.99 Transfers: 108385 Bytes/s: 298095630.04
Avg. Bytes/s: 296067892.65 Transfers: 117524 Bytes/s: 297850614.57
Avg. Bytes/s: 296207542.31 Transfers: 126665 Bytes/s: 298014796.66
Avg. Bytes/s: 296318649.53 Transfers: 135804 Bytes/s: 297867205.11
Avg. Bytes/s: 296414841.78 Transfers: 144942 Bytes/s: 297851795.26
Avg. Bytes/s: 296487495.72 Transfers: 154077 Bytes/s: 297645056.08
qAvg. Bytes/s: 296564969.26 Transfers: 163232 Bytes/s: 297874936.45
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
	Vid / Pid       : 04B4h / 00F1h
	Interface #     : 00h
	Alt Interface # : 00h
	Priority        : Normal
	Buffer Size     : 32768
	Buffer Count    : 4
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
	Total Bytes     : 5348818944
	Total Transfers : 163233
	Avg. Bytes/sec  : 296562313.60
	Elapsed Time    : 18.04 seconds

Bulk Write (Ep01h) max packet size: 1024
	Total Bytes     : 5807144960
	Total Transfers : 177220
	Avg. Bytes/sec  : 321974803.43
	Elapsed Time    : 18.04 seconds

Press any key to exit..
```

commit 6cae9c6d (2021-07-21): core: update usbi_dbg to take the context as an argument

This commit fixes a performance issue caused by the disconnection of the first context
allocated from the default context. usbi_dbg now takes the explicit context instead of
relying on the default context (which may not exist) in most cases. All call sites have
been updated to pass the context or explicitly pass NULL if the context is not
available. We should actively discourage using NULL as the context in the future and
patch all call sites to always pass the context.

Fixes #951