libusb/libusb #953: Libusb claiming several endpoints from same process return error.

Nithu7, 2021-07-15:
Hi Team,

I am a newbie to LibUSB and programming, please burden me for any silly ask.

I want to communication with my USB device (CDC-ACM) through my C program on android platform.
I am able to do simple read/write through CDC using libusb sync API's.

But the requirement is, I need to communicate with same USB end points parallelly from 2 different places from my program (same process)  from both places I should able to read/write  CDC at any time without facing issues.

**Approach 1:**

Now I am assuming that I need to create 2 thread from my C program and from both I need to create every object required for libusb communication (like for claim interface  and for start communication)
then I should able to communicate to USB CDC from two diffrent places.

1. The above is possible? i.e whether I will face error while claiming libusb interface like  (LIBUSB_ERROR_BUSY) from othe thread  if first thread already communicaiton with CDC.
 
2. Now if i read from USB CDC from both the threads, I will receive the data from both the threads  correclty (copy of data) is that correct?

**Approach 2:**

Create global communication point for libUSB data and share the data 
    - Create everything in one file
    - From global variable shared the data.
    - Provide global API  to write to USB CDC from anywhere of your program.

I am currently using libusb sync API's and I am not sure I need to use Async API for this requirment - This may not relevant to requirment, Please clarify.

Or let me know **if you have better suggestion**.

Thanks.

mcuee, 2021-07-15:
It is easier to discuss question like this using the mailing list. I will suggest you subscribe to the mailing list and ask there.

mcuee, 2021-07-15:
And it is better that you provide the code snippets in order for others to help you better.

Nithu7, 2021-07-15:
Hi mcuee, 

Thank you for the response, I will try to seeking help with mailing list also.

Meanwhile what I want to know  is,  it is possible to get two seperate libusb operation flow for the same usb device (to access and communicate), from my process.  So i can do read/write independenly in both cases.

 Currently  I am facing following issue while trying to create and claim libusb interface 2nd time from my app.

Here 'File 1' and 'File 2' is from same process.

### File 1 (Libusb object 1 flow) :
libusb_init(NULL);- > success
libusb_open_device_with_vid_pid (NULL, 0x12, 0x13) -> success 
**libusb_claim_interface(devh, 6); -> success**
libusb_bulk_transfer() -> accessing for I/O -> success
'
### File 2 (Libusb object 2 flow):
libusb_init(NULL);- > success
libusb_open_device_with_vid_pid (NULL, 0x12, 0x13) -> success 
**libusb_claim_interface(devh, 6)**; -> **FAIL**  _(LIBUSB_ERROR_BUSY)_
libusb_bulk_transfer() -> accessing for I/O -> TBC

**Fail case**
I am facing the issue whenever i am trying to claim the interface again which is already claimed by same process from diffrent part of the program, I think this is expected behavior, Please confirm.

But I dont want to close previosly claimed interface also, I should be able to access same interface from both the places.

1. It is possible avoid the problematic use case by using seperate thread for each libusb (File 1& File 2) flow?
2. OR shoud i use async API's ?
3. OR It is not at all possible? i.e Only one libusb flow is supported for that specific USB device  to access it's interface and endpoint (for Read & Write) ?

Regards
Muthu.

timrprobocom, 2021-07-15:
You can't open the same device from two separate processes.  That's just the way it is, on all of the major operating systems.

You can, of course, create a service to own the device, and have your applications communicate with that service.  If you do it right, that kind of architecture even allows you to make major changes to the device without changing your applications.

diabolo38, 2021-07-15:
simplest not necessary cleanest method : claim interface and  get ep globaly **once**  , provide rd/wr "public" data func to use in several place that kwown about the interface/ep
in that func  use a lock/mutex for a safe "mutual exclusive"  concurrent usage 
the calling threads will be serialized in access to the ep in a random way as  it will occur for concurrent  file rd/wr.

btw cdc/acm is supported via the serial/com os standard usb class driver why not to used that ?

mcuee, 2021-07-16:
It seems to me Android does not support CDC-ACM and other USB serial device natively. I can see Android project like the following using the native Android USB API to access serial port.

https://github.com/mik3y/usb-serial-for-android/tree/master/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver

mcuee, 2021-07-16:
Another Android USB serial project has similar drivers.
https://github.com/felHR85/UsbSerial/tree/master/usbserial/src/main/java/com/felhr/usbserial

Nithu7, 2021-07-16:
Dear Developers,

Thank you so much for the response, now I have an understanding about what i can do and can't do.

**Regarding not using Android USB API:**
We are not using android host JAVA API because we need to do some data security and other such functionalies on CDC data at native level  (few such functionalies supported only at HAL level ) so I don't want to add overhead by receiving CDC in app layer then sending data back them to native layer for processiong then again sending the processed data to app layer.

Regards
Muthu.

closed 2021-07-16 by Nithu7