libusb/libusb #954: Create new libusb release

maxgerhardt, 2021-07-16:
When attempting to use the latest released libusb dll version from this repository with the [Rasberry Pi foundation's version of OpenOCD](https://github.com/raspberrypi/openocd/), I am immediately greeted with a crash-and-burn of the program, here observed more closely by running it GDB (otherwise it just terminates.)

```
[New Thread 19280.0x2f48]

Thread 5 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 19280.0x2f48]
0x00007ff807217734 in libusb_transfer_get_stream_id@4 ()
   from C:\Users\Max\AppData\Local\Programs\Python\Python38\libusb-1.0.dll
(gdb) backtrace
#0  0x00007ff807217734 in libusb_transfer_get_stream_id@4 ()
   from C:\Users\Max\AppData\Local\Programs\Python\Python38\libusb-1.0.dll
#1  0x00007ff80721a809 in libusb_interrupt_transfer@24 ()
   from C:\Users\Max\AppData\Local\Programs\Python\Python38\libusb-1.0.dll
#2  0x00007ff866791bb2 in ucrtbase!_configthreadlocale () from C:\windows\System32\ucrtbase.dll
#3  0x00007ff867447034 in KERNEL32!BaseThreadInitThunk () from C:\windows\System32\kernel32.dll
#4  0x00007ff868da2651 in ntdll!RtlUserThreadStart () from C:\windows\SYSTEM32\ntdll.dll
#5  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

Used version 

[image]

When taking the version before that, v1.0.23, I do not have that problem, Other people, as noted in https://github.com/raspberrypi/picoprobe/issues/3, also run into this problem and say the latest master version does not have this problem.

>v1.0.24 has the Segmentation Fault, latest master does not

Please release a new release of libusb where that bug isn't there.

mcuee, 2021-07-16:
Thanks for the reporting. 1.0.25 release related works are underway as mentioned here: https://github.com/libusb/libusb/issues/777

Please bear with us as we are short of maintainers and the works are all done by volunteers.

mcuee, 2021-07-17:
@maxgerhardt It will be great for the users to git bisect to identify the particular commit which triggers the issue.

mcuee, 2021-07-17:
@maxgerhardt As for https://github.com/earlephilhower/pico-quick-toolchain/issues/4, please use Zadig to install WinUSB driver and try again. libusb-1.0 does not work well with libusb-win32 device driver (libusb0.sys).

maxgerhardt, 2021-07-17:
>@maxgerhardt As for earlephilhower/pico-quick-toolchain#4, please use Zadig to install WinUSB driver and try again. libusb-1.0 does not work well with libusb-win32 device driver (libusb0.sys).

Thanks for the help on that one, but it didn't change anything. On my self-compiled version from the original branch it works, in the openocd version released in the mentioned repo I still get `Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT` the exact same way with WinUSB drivers. The original instructions come from the official Raspberry Pi Foundation's instructions (https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf) that say libusb-win32, though I'll have them know about what you said.

As for the segfault, I think this can be very nicely automated, as the program just crashes on startup, so I can bisect -> recompile libusb -> put it in openocd folder -> attempt running `bin\openocd.exe -f interface/picoprobe.cfg -f target/rp2040.cfg -c "init; reset; exit"`-> observe immediate segfault with or good execution via return value (-large number or 0). I'll try and do this to get you this information.

mcuee, 2021-07-17:
Thanks a lot for the offer to help. 
As for driver selection, please refer to the wiki: https://github.com/libusb/libusb/wiki/Windows#How_to_use_libusb_on_Windows

> Recommended: Use the most recent version of Zadig, an Automated Driver Installer GUI application for WinUSB (recommended), libusb-win32 (not working well, not recommended) and libusbK (only if you hit WinUSB limitations)

mcuee, 2021-07-17:
@maxgerhardt If I venture to guess the issue, it will be around this commit between 1.0.23 and 1.0.24 
https://github.com/libusb/libusb/commit/ba6b8bcb7ea204e65a3deec3be81aacc9f4b6d5a#diff-bd7e9ed3e7d8a4a69d0664da0f6b8dbbff0b2c608d8dd63de02c379c98cf88b4

We have a few issues associated with that commit including one recent (not the same as this issue but related to the above commit).
https://github.com/libusb/libusb/issues/912
https://github.com/libusb/libusb/issues/863
https://github.com/libusb/libusb/issues/844

maxgerhardt, 2021-07-17:
Okay an interesting observation I have is that when I use Zadig to load WinUSB drivers, the 1.0.24 version of libusb-1.0.dll does *not* cause a OpenOCD crash anymore. Only switching back to libusb-win32 drivers makes it fail with 1.0.24. Maybe that's expected though and suggests the Windows driver part of libusb is bad? With [`git bisect run`](https://lwn.net/Articles/317154/) I should be able to find the commit pretty easily now..

maxgerhardt, 2021-07-17:
The results are in. I used the script 

```sh
#!/bin/bash

./autogen.sh

make -C libusb -j16 || exit 125   # an exit code of 125 asks "git bisect"
		   # to "skip" the current commit

cp ./libusb/.libs/libusb-1.0.dll ~/openocd/openocd_installed/mingw64/bin/
~/openocd/openocd_installed/mingw64/bin/openocd.exe -f interface/picoprobe.cfg -f target/rp2040.cfg -c "init; reset; exit"; 
if [ "$?" -eq "0" ]
then
	echo "Commit was ok."
else
	echo "Commit led to crash."
	exit 1
fi
```

with

```
git bisect start v1.0.24 v1.0.23
git bisect run ./my_bisect_test.sh
```

And I encountered one compile error during bisecting (although maybe also two as git says below)..

```
os/windows_winusb.c:470:8: error: no previous prototype for 'windows_open' [-Werror=missing-prototypes]
  470 | HANDLE windows_open(struct libusb_device *dev, const char *path, DWORD access)
      |        ^~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
```

with the final result being

```
Bisecting: 1 revision left to test after this (roughly 1 step)
[d67eb5beaa44c17c09f089a522ff483f4978a978] core: Introduce platform events abstraction
[...]
Error: Can't find a picoprobe device! Please check device connec./my_bisect_test.sh: line 17: 54699 Segmentation fault [...]
Commit led to crash. 
There are only 'skip'ped commits left to test.
The first bad commit could be any of:
ba6b8bcb7ea204e65a3deec3be81aacc9f4b6d5a
d67eb5beaa44c17c09f089a522ff483f4978a978
We cannot bisect more!
```

And the first one of those commit is exactly the commit your mentioned.

mcuee, 2021-07-17:
> Okay an interesting observation I have is that when I use Zadig to load WinUSB drivers, the 1.0.24 version of libusb-1.0.dll does _not_ cause a OpenOCD crash anymore. Only switching back to [...]

Thanks a lot for the fast git bisect.

There is no Windows driver involved in the libusb project. It is purely a user space library. It supports multiple kernel drivers.
1) WinUSB.sys kernel driver from Microsoft, through WinUSB.dll from Microsoft (if libusbk.dll is not present), or through libusbk.dll from libusbk project

libusb-1.0.dll <-- WinUSB.dll <-- WinUSB.sys (if libusbk.dll is not present)
libusb-1.0.dll <-- libusbk.dll <-- WinUSB.dll <-- WinUSB.sys

2) libusbk.sys kernel from libusbk project, through libusbk.dll from libusbk project

libusb-1.0.dll <-- libusbk.dll <-- libusbk.sys

3) libusb0.sys kernel from libusb-win32 project, through libusbk.dll from libusbk project

libusb-1.0.dll <-- libusbk.dll <-- libusb0.sys

4) Windows HID driver 
5) usbdk driver from usbdk project

In terms of support, WinUSB is first class supported driver for libusb project, followed by libusbk.sys, then usbdk/HID/libusb0.sys which may have more issues.

A few more notes.
a) When you use Zadig to install the driver, libusbk.dll will be installed as well.

b) libusb-win32 is a separate project from libusb, libusbk is another separate project from libusb.

c) libusb0.sys is not a bad driver, but libusb support of libusb0.sys is problematic. We are short of resources especially Windows maintainers, it is of low priority to fix the issue with regard to libusb0.sys.
https://github.com/libusb/libusb/issues?q=is%3Aopen+is%3Aissue+label%3Alibusbk_libusb0

mcuee, 2021-07-17:
More info about the relation between libusb project, libusb-win32 project and libusbk project.
https://github.com/libusb/libusb/wiki/FAQ#libusbwin32_libusbK_and_libusb_project

mcuee, 2021-07-17:
@maxgerhardt Please raise an issue with the Raspberry Pi Foundation to update the following document to use WinUSB driver and not libusb-win32 driver. Thanks.
https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf

mcuee, 2021-07-18:
In this case, I will close this issue as it has been fixed in the git head.

closed 2021-07-18 by mcuee

maxgerhardt, 2021-07-18:
Yes, the new release is tracked in #777 anyways.