libusb/libusb #955: xusb warning message under macOS

mcuee, 2021-07-19:
From https://github.com/libusb/libusb/issues/908

Just did a quick hack of xusb example to test and libusb_kernel_driver_active() is working for the Logitech USB Receiver (HID device) under my Mac Mini M1 (Big Sur).

However, there are two questions.
1) Why there is a warning message here?
```
Reading first configuration descriptor:
             nb interfaces: 3
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
```
2) Can we downgrade the following error message to warning?
```
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.
```

Quick hack to xusb example.
```
libusb on  master [!] ❯ git diff
diff --git a/examples/xusb.c b/examples/xusb.c
index bf328fe..e60f7b8 100644
--- a/examples/xusb.c
+++ b/examples/xusb.c
@@ -867,6 +867,30 @@ static int test_device(uint16_t vid, uint16_t pid)
        printf("             nb interfaces: %d\n", nb_ifaces);
        if (nb_ifaces > 0)
                first_iface = conf_desc->usb_interface[0].altsetting[0].bInterfaceNumber;
+
+    r = libusb_kernel_driver_active(handle, 1);
+
+       for (iface = 0; iface < nb_ifaces; iface++)
+       {
+               printf("\nCheck kernel driver active for interface %d... \n", iface);
+               r = libusb_kernel_driver_active(handle, iface);
+               if (r == LIBUSB_SUCCESS) {
+                       perr("   No kernel driver attached.\n");
+               }
+               else if (r == 1) {
+                       perr("   There is a kernel driver attached.\n");
+               }
+               else if (r == LIBUSB_ERROR_NOT_SUPPORTED) {
+                       perr("   Not supported for this platform.\n");
+               }
+               else if (r == LIBUSB_ERROR_NO_DEVICE) {
+                       perr("   Device not present.\n");
+               }
+               else {
+                       perr("   Other errors.\n");
+               }
+       }
+
        for (i=0; i<nb_ifaces; i++) {
                printf("              interface[%d]: id = %d\n", i,
                        conf_desc->usb_interface[i].altsetting[0].bInterfaceNumber);
```

Run log 1:
```

libusb on  master [!] ❯ ./examples/xusb 046d:c52b
Using libusb v1.0.24.11636

Opening device 046D:C52B...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 046D:C52B
         bcdDevice: 2410
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 3
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access

Check kernel driver active for interface 0... 
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   There is a kernel driver attached.

Check kernel driver active for interface 1... 
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   There is a kernel driver attached.

Check kernel driver active for interface 2... 
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   There is a kernel driver attached.
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.01
       endpoint[0].address: 81
           max packet size: 0008
          polling interval: 08
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.02
       endpoint[0].address: 82
           max packet size: 0008
          polling interval: 02
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 83
           max packet size: 0020
          polling interval: 02

Claiming interface 0...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Claiming interface 1...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Claiming interface 2...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Reading string descriptors:
   String (0x01): "Logitech"
   String (0x02): "USB Receiver"

Releasing interface 0...
Releasing interface 1...
Releasing interface 2...
Closing device...
```

mcuee, 2021-07-19:
Run log 2: FTDI USB Serial Device is no longer openning the device in exclusive mode in Big Sur and therefore it shows as no kernel driver attached.

Same two issues here.
1) Why is there a 
Reading first configuration descriptor:
             nb interfaces: 1
libusb: info [darwin_claim_interface] no interface found; setting configuration: 1
libusb: error [darwin_claim_interface] interface not found
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12

```

libusb on  master [!] ❯ ./examples/xusb 0403:6001
Using libusb v1.0.24.11636

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
libusb: info [darwin_claim_interface] no interface found; setting configuration: 1
libusb: error [darwin_claim_interface] interface not found

Check kernel driver active for interface 0... 
   No kernel driver attached.
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...
libusb: warning [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: warning [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Releasing interface 0...
Closing device...
```

mcuee, 2021-07-19:
Full debug log for FTDI USB Serial Port device.

<details>
<summary> click to expand for the full debug log </summary>

```
libusb on  master [!] ❯ ./examples/xusb -d 0403:6001
Using libusb v1.0.24.11636

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000037] [00008c78] libusb: debug [libusb_init] created default context
[ 0.000075] [00008c78] libusb: debug [libusb_init] libusb v1.0.24.11636
[ 0.000139] [00008c78] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000351] [00008c78] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb997dca
[ 0.000394] [00008c78] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb997dca
[ 0.003151] [00008c78] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003156] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003158] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.003159] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003161] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003162] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.003163] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.003164] [00008c78] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.003165] [00008c78] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.003166] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.003167] [00008c78] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003169] [00008c78] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003170] [00008c78] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003171] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003189] [00008c78] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003191] [00008c78] libusb: debug [process_new_device] allocating new device in context 0x159f0afc0 for with session 0xb997dca
[ 0.003193] [00008c78] libusb: debug [process_new_device] found device with address 1 port = 5 parent = 0x0 at 0x159f0e34a
[ 0.003238] [00008c78] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xbba3349
[ 0.003261] [00008c78] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xbba3349/0x2100000 against cached device with sessionID/locationID 0xb997dca/0x2500000
[ 0.003263] [00008c78] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xbba3349
[ 0.003624] [00008c78] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003628] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003629] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.003630] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003631] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003633] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.003634] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.003635] [00008c78] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.003636] [00008c78] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.003637] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.003638] [00008c78] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003640] [00008c78] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003641] [00008c78] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003642] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003653] [00008c78] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003655] [00008c78] libusb: debug [process_new_device] allocating new device in context 0x159f0afc0 for with session 0xbba3349
[ 0.003657] [00008c78] libusb: debug [process_new_device] found device with address 2 port = 1 parent = 0x0 at 0x159f0e3ca
[ 0.003701] [00008c78] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xbbc31eb
[ 0.003717] [00008c78] libusb: debug [darwin_get_cached_device] parent sessionID: 0xb997dca
[ 0.003719] [00008c78] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xbbc31eb/0x2510000 against cached device with sessionID/locationID 0xbba3349/0x2100000
[ 0.003721] [00008c78] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xbbc31eb/0x2510000 against cached device with sessionID/locationID 0xb997dca/0x2500000
[ 0.003722] [00008c78] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xbbc31eb
[ 0.003887] [00008c78] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003891] [00008c78] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...]
[ 0.010004] [00008c78] libusb: debug [libusb_unref_device] destroy device 2.6
[ 0.010060] [00008c78] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.010106] [00008c78] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.010150] [00008c78] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.010183] [00008c78] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.010220] [00008c78] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.010264] [00008c78] libusb: debug [usbi_remove_event_source] remove fd 3
```

</details>

mcuee, 2021-07-19:
Full debug log for Logitech USB Receiver

<details>
<summary> click to expand for the full debug log </summary>

```

libusb on  master [!] ❯ ./examples/xusb -d 046d:c52b
Using libusb v1.0.24.11636

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000038] [00009159] libusb: debug [libusb_init] created default context
[ 0.000077] [00009159] libusb: debug [libusb_init] libusb v1.0.24.11636
[ 0.000145] [00009159] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000326] [00009159] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb997dca
[ 0.000361] [00009159] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb997dca
[ 0.002244] [00009159] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002251] [00009159] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002253] [00009159] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002255] [00009159] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002257] [00009159] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002258] [00009159] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002259] [00009159] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002261] [00009159] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002262] [00009159] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002264] [00009159] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002265] [00009159] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002267] [00009159] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002269] [00009159] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002270] [00009159] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002288] [00009159] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002291] [00009159] libusb: debug [process_new_device] allocating new device in context 0x136e0d4f0 for with session 0xb997dca
[ 0.002294] [00009159] libusb: debug [process_new_device] found device with address 1 port = 5 parent = 0x0 at 0x136f04b7a
[ 0.002347] [00009159] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xbba3349
[ 0.002374] [00009159] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xbba3349/0x2100000 against cached device with sessionID/locationID 0xb997dca/0x2500000
[ 0.002377] [00009159] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xbba3349
[ 0.002716] [00009159] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002720] [00009159] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002721] [00009159] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002723] [00009159] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002724] [00009159] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002725] [00009159] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002727] [00009159] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002728] [00009159] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002730] [00009159] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002731] [00009159] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002733] [00009159] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002734] [00009159] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002735] [00009159] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002737] [00009159] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002749] [00009159] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002751] [00009159] libusb: debug [process_new_device] allocating new device in context 0x136e0d4f0 for with session 0xbba3349
[ 0.002753] [00009159] libusb: debug [process_new_device] found device with address 2 port = 1 parent = 0x0 at 0x136f057da
[ 0.002797] [00009159] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xbbc31eb
[ 0.002813] [00009159] libusb: debug [darwin_get_cached_device] parent sessionID: 0xb997dca
[ 0.002815] [00009159] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xbbc31eb/0x2510000 against cached device with sessionID/locationID 0xbba3349/0x2100000
[ 0.002817] [00009159] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xbbc31eb/0x2510000 against cached device with sessionID/locationID 0xb997dca/0x2500000
[ 0.002818] [00009159] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xbbc31eb
[ 0.002979] [00009159] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002983] [00009159] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...]
[ 0.010322] [00009159] libusb: debug [libusb_unref_device] destroy device 2.6
[ 0.010360] [00009159] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.010417] [00009159] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.010474] [00009159] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.010519] [00009159] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.010567] [00009159] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.010615] [00009159] libusb: debug [usbi_remove_event_source] remove fd 3
```

</details>

hjelmn, 2021-07-20:
Will downgrade to info.

hjelmn, 2021-07-20:
Should be fixed now.

mcuee, 2021-07-20:
Yes the message has been downgraded. However, I am not so sure why the info message is printed out in the xusb example for macOS.

```

libusb on  master ❯ ./examples/xusb 046d:c52b
Using libusb v1.0.24.11640

Opening device 046D:C52B...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 046D:C52B
         bcdDevice: 2410
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 3
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.01
       endpoint[0].address: 81
           max packet size: 0008
          polling interval: 08
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.02
       endpoint[0].address: 82
           max packet size: 0008
          polling interval: 02
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 83
           max packet size: 0020
          polling interval: 02

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: info [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Claiming interface 1...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: info [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Claiming interface 2...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
libusb: info [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-12
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed.

Reading string descriptors:
   String (0x01): "Logitech"
   String (0x02): "USB Receiver"

Releasing interface 0...
Releasing interface 1...
Releasing interface 2...
Closing device...

```

mcuee, 2021-07-21:
Okay, xusb default to info level so this is still acceptable.
https://github.com/libusb/libusb/blob/master/examples/xusb.c#L1112

commit b0fd4d87 (2021-07-20): darwin: downgrade error message on failure to open interface to info

This resolves an issue introduced by the implementation of the kernel detach code. This
code used claim_interface to detect if a kernel driver is attached. The issue is that if
a driver is attached or the interface is not available and an error message was printed
out. Since this is not an error the info level is more appropriate.

Also downgrading a number of warnings to clean up the output.

Closes #955
closed 2021-07-21 by mcuee