libusb/libusb #956: libusb: error [submit_iso_transfer] submiturb failed, errno=12

xiayibing, 2021-07-20:
Any ideas about how to analyze this issue？Use strace tool to get the below info：
ioctl(12, USBDEVFS_SUBMITURB, 0x7ff39017f0d0) = -1 ENOMEM (Cannot allocate memory)

But at the same time the system still have enough memory.  How much memory does  the ioctl function need?
 free -m
               total         used        free      share    buffers/cache  
mem：        1814         399         169          61        1245        
swap：         683         135         548

mcuee, 2021-07-20:
Please post more detail what you are trying to do which cause the error code, like the code snippets, the device, the OS, etc.

mcuee, 2021-07-20:
If you look at libusb codes, there are mulitple places this error can happen, it is not related to your system memory.

xiayibing, 2021-07-20:
> Please post more detail what you are trying to do which cause the error code, like the code snippets, the device, the OS, etc.

This issue is easy to occur.  I also want to know what triggers this issue. 
Now the libusb version is 1.0.24.  
Linux version 5.3.0-28-generic (buildd@lcy01-amd64-009) (gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)) .

The code snippets is below:
            libusb_fill_iso_transfer(priv->transfer,handle,endpoint,buffer,
                                     length,number_of_packets,libusb_callback,(void *)priv,500);
            
            if((ret = usbip_recv_iso(ud, priv)) < 0) {
                err("usbip_recv_iso failed, errno: %d", errno);
                goto EXIT;
            }
            ret = libusb_submit_transfer(priv->transfer);  //this will call ioctl(12, USBDEVFS_SUBMITURB, 0x7ff39017f0d0)

mcuee, 2021-07-20:
Actually isochronous treansfer is pretty difficult to get right. Please post more details of your device (lsusb -vvv) and your codes (minimum codes that can be compiled and can demonstrate the issue).

xiayibing, 2021-07-20:
> Actually isochronous treansfer is pretty difficult to get right. Please post more details of your device (lsusb -vvv) and your codes (minimum but complete codes that can be compiled and can demonstrate the issue).

I think It's nothing to do with the product of this camera: Bus 002 Device 016: ID 18ec:3399 Arkmicro Technologies Inc.
[lsusb.txt](https://github.com/libusb/libusb/files/6848491/lsusb.txt)

mcuee, 2021-07-20:
Codes please. Your codes need to match the device as well.

BTW, why do you want to use libusb with the device? It will not be an easy task.

Does it have a kernel driver (Video4Linux)? If yes you should use that API.
https://www.kernel.org/doc/html/v4.12/media/kapi/v4l2-core.html

If not, does it comply to UVC? If yes you may want to take a look at libuvc here which is using libusb as the base.
https://github.com/libuvc/libuvc

mcuee, 2021-07-20:
BTW, this github is not for technical support but rather for issue reporting. If you consider this to be a libusb issue, please provide all the details in order to diagnose the issue. From what I read this is most likely your code issue and it is not an easy task to talk to the device using libusb.

xiayibing, 2021-07-21:
> BTW, this github is not for technical support but rather for issue reporting. If you consider this to be a libusb issue, please provide all the details in order to diagnose the issue. From what I read this is most likely your code issue and it is not an easy task to talk to the device using libusb.

In submit_iso_transfer there maybe a bug：
The temporary j may have wrong control. Maybe you need to delete it.
for (i = 0, j = 0; i < num_urbs; i++)  //here j =0
for (k = 0; k < num_packets_in_urb; j++, k++) { // here j++
			packet_len = transfer->iso_packet_desc[j].length;// here why not k?

mcuee, 2021-07-21:
You are saying the following code may have a bug right? You need to provide your analysis first. I am pretty sure the codes have not been changed for a long time and I doubt there is an issue with the codes.

https://github.com/libusb/libusb/blob/master/libusb/os/linux_usbfs.c#L2129
```
/* allocate + initialize each URB with the correct number of packets */
	num_packets_remaining = num_packets;
	for (i = 0, j = 0; i < num_urbs; i++) {
		int num_packets_in_urb = MIN(num_packets_remaining, MAX_ISO_PACKETS_PER_URB);
		struct usbfs_urb *urb;
		size_t alloc_size;
		int k;

		alloc_size = sizeof(*urb)
			+ (num_packets_in_urb * sizeof(struct usbfs_iso_packet_desc));
		urb = calloc(1, alloc_size);
		if (!urb) {
			free_iso_urbs(tpriv);
			return LIBUSB_ERROR_NO_MEM;
		}
		urbs[i] = urb;

		/* populate packet lengths */
		for (k = 0; k < num_packets_in_urb; j++, k++) {
			packet_len = transfer->iso_packet_desc[j].length;
			urb->buffer_length += packet_len;
			urb->iso_frame_desc[k].length = packet_len;
		}

		urb->usercontext = itransfer;
		urb->type = USBFS_URB_TYPE_ISO;
		/* FIXME: interface for non-ASAP data? */
		urb->flags = USBFS_URB_ISO_ASAP;
		urb->endpoint = transfer->endpoint;
		urb->number_of_packets = num_packets_in_urb;
		urb->buffer = urb_buffer;

		urb_buffer += urb->buffer_length;
		num_packets_remaining -= num_packets_in_urb;
	}

```

mcuee, 2021-07-21:
Looking at the history, this is the commit which is the last change to the code. I doubt it has the issue.
https://github.com/libusb/libusb/commit/8aad7fd1de08c173ff0fec87abb85bd940526603

xiayibing, 2021-07-21:
> You are saying the following code may have a bug right? You need to provide your analysis first. I am pretty sure the codes have not been changed for a long time and I doubt there is an issue with [...]

At least when num_urbs=2,  j will be start from 129 as  MAX_ISO_PACKETS_PER_URB is 128.
I think MAX_ISO_PACKETS_PER_URB means per urb submit there is 128 for limitation.  It makes me confused. 
Do you think num_iso_packets in transfer structure will have a chance to be larger than 128?

mcuee, 2021-07-21:
I do not know the internals of libusb well enough to comment. Hopefully others can chime in and explain to you.

mcuee, 2021-07-21:
@xiayibing But based on the comments here, the reason you see -ENOMEM is because kmalloc() failed for the large buffer allocation. You may want to reduce the buffersize for the isoc transfer.
https://github.com/libusb/libusb/issues/118

xiayibing, 2021-07-22:
> @xiayibing But based on the comments in ticket #118, the reason you see -ENOMEM is because kmalloc() failed for the large buffer allocation. What is the transfer size you are trying to use here? (that is why to show your codes is important). You may want to reduce the buffersize for the isoc transfer.

This line will cause a lot of  memory fragmentation considering iso-transfering.
urbs = calloc(num_urbs, sizeof(*urbs));

Now the below code seems good and my program has run for 12hours. Please see the attached files.
I'm not sure old code maybe want to handle more urbs which are more than multiple MAX_ISO_PACKETS_PER_URB.
[linux_usbfs.zip](https://github.com/libusb/libusb/files/6859630/linux_usbfs.zip)

mcuee, 2021-07-22:
@xiayibing Can you do a diff of the file you have and the current git? Thanks.

mcuee, 2021-07-22:
Pass discussion: https://github.com/libusb/libusb/issues/118

mcuee, 2021-07-22:
Last commit for the Linux isoc transfer codes: https://github.com/libusb/libusb/commit/8aad7fd1de08c173ff0fec87abb85bd940526603

xiayibing, 2021-07-22:
> @xiayibing Can you do a diff of the file you have and the current git? Thanks.

After 13 hours the issue  occurs again and then it easily occurs again and again.  
[diff.txt](https://github.com/libusb/libusb/files/6859875/diff.txt)

mcuee, 2021-07-22:
@xiayibing You need to show your test codes which you use to reproduce the issue for others to help you. Or at least provide more details of what you are trying to do.

xiayibing, 2021-07-23:
> @xiayibing You need to show your test codes which you use to reproduce the issue for others to help you. Or at least provide more details of what you are trying to do.

So hard for me to show code because camera's host controller is on remote PC running on WIN10.
I notice a issue :  every two operations of interfaces one will cause the urb's submission to local host controller rather than remote host controller by the same call of the ioctl function.  That will cause the kernel's  compaining of the below:
"xhci_hcd 0000:00:14.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 6 comp_code 3".

I used libusb_kernel_driver_active api to confirm that the operated interface is  free from kernel's driver.   The endpoint is the same. Is it possible for the pipe between the endpoint and the host controller changed by the operation of interface?

mcuee, 2021-07-23:
Sorry I can not help you further. Maybe you can send email to libusb-devel mailing list and maybe there are people who can help you better.

mcuee, 2021-07-23:
@xiayibing It is okay that you can not show the code. But at least share what you are trying to do.

It seems to me you are using something like this:
libusb on Linux host -- remote USB solution -- remote Windows 10 PC -- USB Camera

I will venture to say the issue may be related to the "remote USB solution". You need to test your code with the following configuration first.
libusb on Linux host -- USB camera (without the middle man).

mcuee, 2021-07-30:
I will close this issue for now as you are reluctant to share what you are doing. Please reopen when you are ready to provide more information. 

I will also suggest you ask in linux-usb mailing list about the issue you mentioned that the kernel complaining that "xhci_hcd 0000:00:14.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 6 comp_code 3". It may be a Linux kernel usb problem.

mcuee, 2021-07-30:
And if others also facing the issue, please reopen. Thanks.

mcuee, 2021-08-03:
> So hard for me to show code because camera's host controller is on remote PC running on WIN10.
> I notice a issue : every two operations of interfaces one will cause the urb's submission to local host controller rather than remote host controller by the same call of the ioctl function. 

It seems to me your "remote USB solution" is the issue here based on the above. Please test the following configuration first.
libusb on Linux host -- USB camera (without the middle man).

> That will cause the kernel's compaining of the below:
> "xhci_hcd 0000:00:14.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 6 comp_code 3".
> 
As mentioned, I will  suggest you ask in linux-usb mailing list about the issue you mentioned that the kernel complaining that "xhci_hcd 0000:00:14.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 6 comp_code 3". It may be a Linux kernel usb problem.

mcuee, 2021-08-05:
@xiayibing BTW, you can also post the libusb debug info (set LIBUSB_DEBUG=4) if you are reluctant to post your codes and details of your  setup (remote USB solution and your camera).

xiayibing, 2021-08-05:
@mcuee  
I know why ioctl returns ENOMEM but I don't know how to debug it.
When one time of two operations of interface's altnative settings lasts for long time, the use size of buffer/cache is increasing all the time.  I guess the IO operation is not finished really or OS will use it for future. You need  drop caches by  "echo 3 > /proc/sys/vm/drop_caches" to avoid this issue.

The kernel's compaining from xhci_hcd shows there is a interrupt to CPU and finally call xhci_irq.
In fact there is a endpoint of USB camera with interrupt type.  I don't know if this endpoint causes the interruption to CPU.
And with detach from the kernel driver what causes the interruption to local CPU by using libusb. 

I also abide by your advice to ask  in linux-usb mailing list  but nobody response.

xiayibing, 2021-08-09:
This issue is only related to this USB Camera as far as I know.

  idVendor           0x18ec Arkmicro Technologies Inc.
  idProduct          0x3399
  bcdDevice            1.00
  iManufacturer           1 ARKMICRO
  iProduct                2 USB2.0 PC CAMERA

mcuee, 2021-08-09:
@xiayibing Have you tested the following configuration?
libusb on Linux host -- USB camera (without the middle man).

mcuee, 2021-08-09:
@xiayibing Have you tried libuvc as well if the camera is compatible with UVC class? It is based on libusb.
https://github.com/libuvc/libuvc

xiayibing, 2021-08-10:
> @xiayibing Have you tested the following configuration?
> libusb on Linux host -- USB camera (without the middle man).

Yes, I used a tool named xawtv to test and no issue was found.
And also the below debugging can also fix this issue.  I need to make my program sleep for 1 second .
The code is like this:
                    if(transfer->actual_length == 1546 )  { //The first iso actual length 
                        usleep(1000*1000);
                    }

The below infomation shows the procedure.
[pid 1673] recvfrom(13, "\0\0\0\1\0\0\0`\0\1\0\r\0\0\0\1\0\0\0\3\0\0\0\2\0\5\334\0"..., 52, MSG_WAITALL, NULL, NULL) = 52
[pid 1673] mprotect(0x7f0584021000, 266240, PROT_READ|PROT_WRITE) = 0
recvfrom(13, "\0\0\0\0\0\0\v\270\0\0\0\0\0\0\0\0\0\0\v\270\0\0\v\270\0\0\0\0"..., 2048, MSG_WAITALL, NULL, NULL) = 2048
[pid 1673] ioctl(12, USBDEVFS_SUBMITURB, 0x7f05842fbac0) = 0
[pid 1673] epoll_wait(16, [{EPOLLOUT, {u32=10, u64=93969589469194}}, {EPOLLIN, {u32=13, u64=93969589469197}}], 10, 100) = 2
[pid 1673] poll([{fd=10, events=POLLIN}, {fd=11, events=POLLIN}, {fd=12, events=POLLOUT}], 3, 1) = 0 (Timeout)
[pid 1673] recvfrom(13, "\0\0\0\1\0\0\0h\0\1\0\r\0\0\0\1\0\0\0\3\0\0\0\2\0\5\334\0\"..., 52, MSG_WAITALL, NULL, NULL) = 52
[pid 1673] mprotect(0x7f05842fd000, 385024, PROT_READ|PROT_WRITE) = 0
[pid 1673] recvfrom(13, "\0\0\0\0\0\0\v\270\0\0\0\0\0\0\0\0\0\0\v\270\0\0\v\270\0\0\"..., 2048, MSG_WAITALL, NULL, NULL) = 2048
[pid 1673] mprotect(0x7f058435b000, 4096, PROT_READ|PROT_WRITE) = 0

///here the above procedure repeated 8 times of urb submission.
recvfrom(13, "\0\0\0\0\0\0\v\270\0\0\0\0\0\0\0\0\0\0\v\270\0\0\v\270\0\0\0\0\0\0\0\0"..
[pid 1673] ioctl(12, USBDEVFS_SUBMITURB, 0x7f058423d7a0) = 0
poll wait...
///here the end of  8 times urb submission.

[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = 0
[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = -1 EAGAIN (Resource temporarily unavailable)
[pid 1673] nanosleep({tv_sec=1, tv_nsec=0},  <unfinished ...>  ///Here is the sleepping time of 1 s and do once
[pid 1674] <... nanosleep resumed> NULL) = 0  
[pid 1673] recvfrom(13, "\0\0\0\1\0\0\0i\0\1\0\r\0\0\0\1\0\0\0\3\0\0\0\2\0\5\334\0"..., 52, MSG_WAITALL, NULL, NULL) = 52
[pid 1673] recvfrom(13, "\0\0\0\0\0\0\v\270\0\0\0\0\0\0\0\0\0\0\v\270\0\0\v\270\0"..., 2048, MSG_WAITALL, NULL, NULL) = 2048
[pid 1673] ioctl(12, USBDEVFS_SUBMITURB, 0x7f0584062070) = 0
[pid 1673] epoll_wait(16, [{EPOLLOUT, {u32=10, u64=93969589469194}}, {EPOLLIN, {u32=13, u64=93969589469197}}, {EPOLLOUT, {u32=12, u64=93969589469196}}], 10, 100) = 3
[pid 1673] poll([{fd=10, events=POLLIN}, {fd=11, events=POLLIN}, {fd=12, events=POLLOUT}], 3, 1) = 1 ([{fd=12,events=POLLOUT}])
/// Here the reap procedure of the left 8 times of URB submission .
[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = 0
[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = 0
[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = 0
[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = 0
[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = 0
[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = 0
[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = 0
[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = 0
[pid 1673] ioctl(12, USBDEVFS_REAPURBNDELAY, 0x7f058b7fd268) = -1 EAGAIN (Resource temporarily unavailable)

mcuee, 2021-08-10:
In this case, there is no issue with libusb and I will close this issue.

closed 2021-08-10 by mcuee