libusb/libusb #958: Darwin implementation inappropriately masks bits to determine bus

steven807, 2021-07-20:
The line:
https://github.com/libusb/libusb/blob/7ec91f1a538b684ae621cd8fee96559c93a4cf7d/libusb/os/darwin_usb.c#L1123
may be causing problems -- unfortunately on Apple Silicon macs, the location is often of the form `0x00ABCDEF`. This may be the cause of some M1 bugs previously reported.

hjelmn, 2021-07-20:
Not sure I follow. The top byte of the location ID is the bus number. They are numbered from 0. On my M1 Mini I see busses with location IDs:

```
0x00000000
0x01000000
0x02000000
```

Which is what I would expect and is similar to my Intel Macbook Pro. The devices on my M1 are all hooked up to bus 2 and libusb reflects that.

You can see the bus layout by running:

```
ioreg -l -p IOUSB
```

hjelmn, 2021-07-20:
Hmm, interesting. That doesn't match what I am seeing on my M1. What macOS version are you running?

```
  +-o AppleT8103USBXHCI@01000000  <class AppleT8103USBXHCI, id 0x100000299, registered, matched, active, busy 0 (17 ms), retain 39>
  |   {
  |     "IOClass" = "AppleT8103USBXHCI"
  |     "CFBundleIdentifier" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "IOProviderClass" = "AppleARMIODevice"
  |     "kUSBSleepSupported" = Yes
  |     "kUSBSleepPortCurrentLimit" = 3000
  |     "IOPowerManagement" = {"ChildrenPowerState"=2,"DevicePowerState"=0,"CurrentPowerState"=2,"CapabilityFlags"=65536,"MaxPowerState"=3,"DriverPowerState"=0}
  |     "kUSBWakePortCurrentLimit" = 3000
  |     "IOProbeScore" = 10000
  |     "IONameMatch" = "usb-drd,t8103"
  |     "locationID" = 16777216
  |     "IOMatchCategory" = "usb-host"
  |     "UsbHostControllerTierLimit" = 6
  |     "Revision" = <0103>
  |     "controller-statistics" = {"kControllerStatIOCount"=0,"kControllerStatPowerStateTime"={"kPowerStateOff"="3823ms (0%)","kPowerStateSleep"="0ms (0%)","kPowerStateOn"="2982ms (0%)","kPowerStateSuspended"="1842240813ms (99%)"},"kControllerStatSpuriousInterruptCount"=0}
  |     "IONameMatched" = "usb-drd,t8103"
  |     "IOPersonalityPublisher" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "UsbHostControllerDeferRegisterService" = Yes
  |     "CFBundleIdentifierKernel" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "UsbHostControllerStreamPolicy" = 2
  |   }
  |   
  +-o AppleT8103USBXHCI@00000000  <class AppleT8103USBXHCI, id 0x100000295, registered, matched, active, busy 0 (8 ms), retain 40>
  |   {
  |     "IOClass" = "AppleT8103USBXHCI"
  |     "CFBundleIdentifier" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "IOProviderClass" = "AppleARMIODevice"
  |     "kUSBSleepSupported" = Yes
  |     "kUSBSleepPortCurrentLimit" = 3000
  |     "IOPowerManagement" = {"ChildrenPowerState"=0,"DevicePowerState"=0,"CurrentPowerState"=0,"CapabilityFlags"=0,"MaxPowerState"=3,"DriverPowerState"=0}
  |     "kUSBWakePortCurrentLimit" = 3000
  |     "IOProbeScore" = 10000
  |     "IONameMatch" = "usb-drd,t8103"
  |     "locationID" = 0
  |     "IOMatchCategory" = "usb-host"
  |     "UsbHostControllerTierLimit" = 6
  |     "Revision" = <0103>
  |     "controller-statistics" = {"kControllerStatIOCount"=0,"kControllerStatPowerStateTime"={"kPowerStateOff"="1842247639ms (100%)","kPowerStateSleep"="0ms (0%)","kPowerStateOn"="4ms (0%)","kPowerStateSuspended"="1ms (0%)"},"kControllerStatSpuriousInterruptCount"=0}
  |     "IONameMatched" = "usb-drd,t8103"
  |     "IOPersonalityPublisher" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "UsbHostControllerDeferRegisterService" = Yes
  |     "CFBundleIdentifierKernel" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "UsbHostControllerStreamPolicy" = 2
  |   }
  |   
  +-o AppleT8103USBXHCIFL1100@02000000  <class AppleT8103USBXHCIFL1100, id 0x1000006e6, registered, matched, active, busy 0 (234 ms), retain 63>
    | {
    |   "IOClass" = "AppleT8103USBXHCIFL1100"
    |   "kUSBSleepPortCurrentLimit" = 2100
    |   "IOPersonalityPublisher" = "com.apple.driver.usb.AppleEmbeddedUSBXHCIPCI"
    |   "IOPowerManagement" = {"ChildrenPowerState"=3,"DevicePowerState"=2,"CurrentPowerState"=3,"CapabilityFlags"=32768,"MaxPowerState"=3,"DriverPowerState"=0}
    |   "IOProviderClass" = "IOPCIDevice"
    |   "kUSBWakePowerSupply" = 3100
    |   "IOProbeScore" = 2001
    |   "UsbRTD3Supported" = Yes
    |   "locationID" = 33554432
    |   "kUSBWakePortCurrentLimit" = 2100
    |   "IONameMatch" = "pci-xhci-FL1100,t8103"
    |   "CFBundleIdentifierKernel" = "com.apple.driver.usb.AppleEmbeddedUSBXHCIPCI"
    |   "kUSBSleepPowerSupply" = 3100
    |   "IOMatchCategory" = "IODefaultMatchCategory"
    |   "CFBundleIdentifier" = "com.apple.driver.usb.AppleEmbeddedUSBXHCIPCI"
    |   "IOPMResetPowerStateOnWake" = Yes
    |   "IONameMatched" = "pci-xhci-FL1100,t8103"
    |   "Revision" = <0003>
    |   "UsbHostControllerTierLimit" = 6
    |   "UsbBusCurrentPoolID" = 4294969062
    |   "controller-statistics" = {"kControllerStatIOCount"=4499391,"kControllerStatPowerStateTime"={"kPowerStateOff"="0ms (0%)","kPowerStateSleep"="0ms (0%)","kPowerStateOn"="1842247339ms (100%)","kPowerStateSuspended"="0ms (0%)"},"kControllerStatSpuriousInterruptCount"=0}
    |   "kUSBSleepSupported" = Yes
    |   "kUSBHostControllerPropertyPoweredOn" = Yes
    | }

```

steven807, 2021-07-20:
Note, I updated my previous comment to recognize that this may not be a bug, in case you didn't see it.  FWIW I am running Big Sur (11.4) on the M1, whereas I see the other behavior on the Intel MBP on both Catalina (11.15) and Big Sur (tested on 2 different Intel MBPs).

hjelmn, 2021-07-20:
Ah, ok. As you can see in my ioreg there is a bus with location ID == 0 so that seems expected :).

steven807, 2021-07-20:
OK, given that this is expected behavior, I'll close it.  Still, I'm glad to leave this conversation in place -- perhaps it will benefit someone in the future.  I'm not sure why I don't see see sequential bus IDs starting at 0, and I'm still concerned that the line I referenced in the issue has no comment, and I wasn't able to find anything in the Darwin documentation guaranteeing that the high byte is the bus number. But if it ain't broke, don't fix it, I guess.

closed 2021-07-20 by steven807

hjelmn, 2021-07-20:
Ack on the no comment. I will add one to make it clear why that is done.

hjelmn, 2021-07-20:
Done.

mcuee, 2021-07-21:
Just to add the output on my Mac Mini M1.

```

~ ❯ ioreg -l -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 27>
  | {
  |   "IOKitBuildVersion" = "Darwin Kernel Version 20.5.0: Sat May  8 05:10:31 $
  |   "OS Build Version" = "20F71"
  |   "IOBluetoothActive" = Yes
  |   "OSKernelCPUSubtype" = 18446744072635809794
  |   "OSKernelCPUType" = 16777228
  |   "OSPrelinkKextCount" = 5
  |   "IORegistryPlanes" = {"IOPort"="IOPort","IOPower"="IOPower","IOService"="$
  |   "IOConsoleLocked" = No
  |   "IOConsoleUsers" = ({"kCGSSessionOnConsoleKey"=Yes,"kCGSSessionAuditIDKey$
  |   "IOKitDiagnostics" = {"Instance allocation"=9181432,"Container allocation$
  | }
  | 
  +-o AppleT8103USBXHCI@01000000  <class AppleT8103USBXHCI, id 0x100000299, reg$
  |   {
  |     "IOClass" = "AppleT8103USBXHCI"
  |     "CFBundleIdentifier" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "IOProviderClass" = "AppleARMIODevice"
  |     "kUSBSleepSupported" = Yes
  |     "kUSBSleepPortCurrentLimit" = 3000
  |     "IOPowerManagement" = {"ChildrenPowerState"=0,"DevicePowerState"=0,"Cur$
  |     "kUSBWakePortCurrentLimit" = 3000
  |     "IOProbeScore" = 10000
  |     "IONameMatch" = "usb-drd,t8103"
  |     "locationID" = 16777216
  |     "IOMatchCategory" = "usb-host"
  |     "UsbHostControllerTierLimit" = 6
  |     "Revision" = <0103>
  |     "controller-statistics" = {"kControllerStatIOCount"=0,"kControllerStatP$
  |     "IONameMatched" = "usb-drd,t8103"
  |     "IOPersonalityPublisher" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "UsbHostControllerDeferRegisterService" = Yes
  |     "CFBundleIdentifierKernel" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "UsbHostControllerStreamPolicy" = 2
  |   }
  |   
  +-o AppleT8103USBXHCI@00000000  <class AppleT8103USBXHCI, id 0x100000295, reg$
  |   {
  |     "IOClass" = "AppleT8103USBXHCI"
  |     "CFBundleIdentifier" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "IOProviderClass" = "AppleARMIODevice"
  |     "kUSBSleepSupported" = Yes
  |     "kUSBSleepPortCurrentLimit" = 3000
  |     "IOPowerManagement" = {"ChildrenPowerState"=0,"DevicePowerState"=0,"Cur$
  |     "kUSBWakePortCurrentLimit" = 3000
  |     "IOProbeScore" = 10000
  |     "IONameMatch" = "usb-drd,t8103"
  |     "locationID" = 0
  |     "IOMatchCategory" = "usb-host"
  |     "UsbHostControllerTierLimit" = 6
  |     "Revision" = <0103>
  |     "controller-statistics" = {"kControllerStatIOCount"=0,"kControllerStatP$
  |     "IONameMatched" = "usb-drd,t8103"
  |     "IOPersonalityPublisher" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "UsbHostControllerDeferRegisterService" = Yes
  |     "CFBundleIdentifierKernel" = "com.apple.driver.usb.AppleUSBHostT8103"
  |     "UsbHostControllerStreamPolicy" = 2
  |   }
  |   
  +-o AppleT8103USBXHCIFL1100@02000000  <class AppleT8103USBXHCIFL1100, id 0x10$
    | {
    |   "IOClass" = "AppleT8103USBXHCIFL1100"
    |   "kUSBSleepPortCurrentLimit" = 2100
    |   "IOPersonalityPublisher" = "com.apple.driver.usb.AppleEmbeddedUSBXHCIPC$
[... 351 lines cut ...]
              "iProduct" = 4
              "iSerialNumber" = 2
              "bNumConfigurations" = 1
              "USB Product Name" = "USB3.0 Card Reader"
              "USB Address" = 4
              "locationID" = 38866944
              "bDeviceSubClass" = 0
              "bcdUSB" = 784
              "kUSBSerialNumberString" = "000000001536"
              "kUSBCurrentConfiguration" = 1
              "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUS$
              "bDeviceProtocol" = 0
              "USBPortType" = 0
              "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.tr$
              "USB Vendor Name" = "Generic"
              "Device Speed" = 3
              "idVendor" = 1507
              "kUSBProductString" = "USB3.0 Card Reader"
              "USB Serial Number" = "000000001536"
              "IOGeneralInterest" = "IOCommand is not serializable"
              "kUSBAddress" = 4
              "kUSBVendorString" = "Generic"
            }
            
```