libusb/libusb #966: Windows: Possible data race

mcuee, 2021-07-28:
From the mailing list.
https://marc.info/?t=162740011400005&r=1&w=4

On Tue, Jul 27, 2021 at 11:36 PM Bohdan Tymkiv  wrote:
>
> Hello dear libusb maintainers,
>
> I am facing an interesting issue with pyOCD and the DAPLink device. From time to time libusb fails with the following errors:
> (A) libusb: error [composite_copy_transfer_data] program assertion failed - no function to copy transfer data
> (B) libusb: error [windows_submit_transfer] program assertion failed - transfer HANDLE is NULL after transfer was submitted
>
> While issue (B) is non-critical, issue (A) completely breaks the communication with the device.
>
> pyOCD uses pyUSB under the hood (thus synchronous libusb API) and spawns two threads to communicate with the DAPLink:
> Main Thread - writes requests to DAPLink so it submits only OUT transfers.
> New Thread - reads responses from DAPLink so it submits only IN transfers.
> Both threads call libusb_handle_events() and there is a chance that transfer submitted by thread (A) will be handled by a call to libusb_handle_events in thread (B).
>
> It is quite difficult to reproduce the issue but adding a Sleep(10) at the line 750 of windows_common.c
> https://github.com/libusb/libusb/blob/e2edecd3a3f3ff79f58c992862f15b00e4eb3c49/libusb/os/windows_common.c#L750
> causes the issue (B) to appear immediately. That's because during Sleep() the USB transfer completes in the second thread and transfer_priv->handle is now NULL.
>
> To easily reproduce the issue (A), which is critical, Sleep(10) can be added at line 3122 of windows_winusb.c
> https://github.com/libusb/libusb/blob/e2edecd3a3f3ff79f58c992862f15b00e4eb3c49/libusb/os/windows_winusb.c#L3122
>
> DAPLink is a composite device with 5 interfaces.
> I am communication with interface #5 but because transfer_priv->interface_number is not initialized yet when transfer completes, the 'current_interface' here:
> https://github.com/libusb/libusb/blob/e2edecd3a3f3ff79f58c992862f15b00e4eb3c49/libusb/os/windows_winusb.c#L4530
> is not valid (zero).
>
> The rough way of fixing this is to protect the call to priv->backend->submit_transfer() here:
> https://github.com/libusb/libusb/blob/e2edecd3a3f3ff79f58c992862f15b00e4eb3c49/libusb/os/windows_common.c#L738
> by a mutex above, e.g.:
> usbi_mutex_lock(&dev_handle->lock);
> list_add_tail();
> priv->backend->submit_transfer();
> usbi_mutex_unlock(&dev_handle->lock);
>
> I am not sure what impact on performance this will have, probably there is a more elegant way to fix it.
>
> How to reproduce:
> 0. I am using DAPLink on NXP FRDM-K28F board but issue also reproduces e.g. with DAPLink-s on Cypress/Infineon Kits
> 1. Add Sleep(10) at line 3122 of windows_winusb.c
> 2. Install pyOCD 0.31.0 using pip
> 3. Run the following command: pyoct cmd -c "set frequency 16000000" -c " savemem 0x20000000 0x40000 1.bin"
>
> Complete libusb log is attached.
>
> I am not sure but in my understanding calling libusb_submit_transfer from different threads is completely legal from libusb/API point of view.
> If this is not true then I will submit a ticket against pyOCD.
>
> Thanks,
> Bohdan
>

mcuee, 2021-07-28:
libusb debug log
[libusb_log.txt](https://github.com/libusb/libusb/files/6890891/libusb_log.txt)

mcuee, 2021-07-28:
Comments from Tim Roberts.

On Wed, Jul 28, 2021 at 12:52 AM Tim Roberts  wrote:
>
> Bohdan Tymkiv wrote:
> >
> > To easily reproduce the issue (A), which is critical, Sleep(10) can be
> > added at line 3122 of windows_winusb.c
> > https://github.com/libusb/libusb/blob/e2edecd3a3f3ff79f58c992862f15b00e4eb3c49/libusb/os/windows_winusb.c#L3122
> > <https://github.com/libusb/libusb/blob/e2edecd3a3f3ff79f58c992862f15b00e4eb3c49/libusb/os/windows_winusb.c#L3122>
>
> Looking at it from a driver writer's point of view, that code looks
> wrong.  All of the state information for a request needs to be in place
> before the request is submitted, because Murphy ensures that you will
> occasionally be delayed at exactly the wrong moment.  It looks to me
> like line 3123 setting transfer_priv->interface_number needs to be
> scooted up about 30 lines, so it happens before the calls to WinUSBX.
>
>
> > I am not sure but in my understanding calling libusb_submit_transfer
> > from different threads is completely legal from libusb/API point of view.
> > If this is not true then I will submit a ticket against pyOCD.
>
> That is absolutely true.
>
> --
> Tim Roberts, 
> Providenza & Boekelheide, Inc.

mcuee, 2021-07-28:
Comments from

On Wed, Jul 28, 2021 at 1:22 AM Orin Eman <orin.eman@gmail.com> wrote:
>
> On Tue, Jul 27, 2021 at 9:52 AM Tim Roberts <timr@probo.com> wrote:
>>
>> Bohdan Tymkiv wrote:
>> >
>> > To easily reproduce the issue (A), which is critical, Sleep(10) can be
>> > added at line 3122 of windows_winusb.c
>> > https://github.com/libusb/libusb/blob/e2edecd3a3f3ff79f58c992862f15b00e4eb3c49/libusb/os/windows_winusb.c#L3122
>> > <https://github.com/libusb/libusb/blob/e2edecd3a3f3ff79f58c992862f15b00e4eb3c49/libusb/os/windows_winusb.c#L3122>
>>
>> Looking at it from a driver writer's point of view, that code looks
>> wrong.  All of the state information for a request needs to be in place
>> before the request is submitted, because Murphy ensures that you will
>> occasionally be delayed at exactly the wrong moment.  It looks to me
>> like line 3123 setting transfer_priv->interface_number needs to be
>> scooted up about 30 lines, so it happens before the calls to WinUSBX.
>
> I'd agree.  Somewhere between line 3086 and 3092.
>
> There are a few other places where transfer_priv->interface_number is set after asynchronous WinUSB/ReadFile/WriteFile calls that could have the same problem.
>
> Orin.

mcuee, 2021-07-28:
Comments from  Bohdan Tymkiv  

On Wed, Jul 28, 2021 at 1:49 AM Bohdan Tymkiv  wrote:
>
> > It looks to me like line 3123 setting transfer_priv->interface_number needs to be scooted up about 30 lines, so it happens before the calls to WinUSBX.
>
> Yes, I've tried this and it resolves the issue (A). The 'itransfer' structure seems to be protected by the 'itransfer->lock' here
> https://github.com/libusb/libusb/blob/e2edecd3a3f3ff79f58c992862f15b00e4eb3c49/libusb/io.c#L1530
>
> What about this check?
> https://github.com/libusb/libusb/blob/e2edecd3a3f3ff79f58c992862f15b00e4eb3c49/libusb/os/windows_common.c#L753
> This verifies that backend code is obeying the 'contract', but this code is internal to libusb. Can this check be simply removed?
>
> Bohdan

mcuee, 2021-11-11:
FYI, partial fix #967 has been merged. But this issue will remain open since #967 does not totally fix the issue.

mcuee, 2021-11-11:
@bohdan-tymkiv There are some improvement in Windows backend, I am not so sure if will improve for Issue B or not. Is it possible that you try latest git to see how it goes. Thanks.

jacobly0, 2022-02-18:
Here's a best effort [reproducer](https://gist.github.com/jacobly0/f4fb129629abed0b04e256ee8dcfa2bb).  It is able to reproduce the issue for me in under 1000/50000 iterations when run inside/outside a debugger.  To increase the chances of it working you can add one of the following lines of code [before the assertion is checked](https://github.com/libusb/libusb/blob/4622bfcf44db373c53502e3fe873bd611e8332f6/libusb/os/windows_common.c#L750).

Encourage a context switch:
```c
SwitchToThread();
```

Wait for a context switch:
```c
while (!SwitchToThread());
```

Wait a bit:
```c
Sleep(10);
```

Just wait for the bug to occur:
```c
while (transfer_priv->handle != NULL)
    SwitchToThread();
```

mcuee, 2023-01-23:
@jacobly0 

Just wonder if this issue still exist in latest git HEAD. Thanks.

mcuee, 2023-01-31:
Indeed I can reproduce with the file.
```
MINGW64 /c/work/libusb/libusb_tests
$ ./assertion_repro.exe
libusb: error [windows_submit_transfer] program assertion failed - transfer HANDLE is NULL after transfer was submitted
Reproduced after 163337 iterations.
```

[assertion_repro.zip](https://github.com/libusb/libusb/files/10544678/assertion_repro.zip)

mcuee, 2023-01-31:
Tested on another laptop (Windows 11 Acer Swift 3). The first error is not related to this issue -- it is related to access issue with the biometric device (LighTuning Technology EgisTec EH575 fingerprint sensor).

```
PS C:\work\libusb\libusb_test\libusb_test_issue994> .\assertion_issue994.exe
libusb: error [winusbx_open] could not open device \\?\USB#VID_1C7A&PID_0575#077E2F9A#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [5] Access is denied.
libusb: error [windows_submit_transfer] program assertion failed - transfer HANDLE is NULL after transfer was submitted
Reproduced after 68946 iterations.
```

mcuee, 2023-02-01:
The code is only for Windows. I am not so sure if it makes sense to write it in a cross-platform way and adopt it as a test program.
https://gist.github.com/jacobly0/f4fb129629abed0b04e256ee8dcfa2bb#file-assertion_repro-c

mcuee, 2023-04-03:
Another reports of failure here.
* https://github.com/libusb/libusb/issues/1263#issuecomment-1493857765