libusb/libusb #968: Running on bare metal

lonnietc, 2021-07-29:
Hello,

I am interested in a USB lib with minimal clib dependencies so that it might be adaptable to a bare metal project which is working to port over the MUSL Clib standard libraries.

Would libusb be suitable for such an effort or is there another library that might be a better choice to get started with?

Thanks

mcuee, 2021-07-29:
No. libusb needs to run on OS like Linux or Windows, not for bare metal.

closed 2021-07-29 by mcuee

mcuee, 2021-07-30:
For bare metal, usually the silicon chip vendor will provide the USB stack, in your case, I think you want to use the USB Host stack from the silicon chip vendor as the start.

For the embedded side, embedded Linux is the best bet for libusb, and you can see it is based on the existence of usbfs (using usbfs ioctls).
https://github.com/libusb/libusb/blob/master/libusb/os/linux_usbfs.c

mcuee, 2021-07-30:
Please also take note libusb is a host side library, it does not work on the device side.

lonnietc, 2021-08-01:
Thanks again as I am looking for a host side library with minimal dependencies as well as small size so this "might" be the one that I have been seeking.