libusb/libusb #969: Can not get Device string desc when using hotplug on Ubuntu 20.4.

sparkertim, 2021-07-30:
This problem had confused me a long time. I have a FX3 device, **every time I let the hotplug running, and then power up the device,  I can not get the device string descriptions, but the device works normally(open, control read or write).**
I had debugged the program and the libusb dynamic lib,  and I find that libusb_get_string_descriptor_ascii return -6(LIBUSB_ERROR_BUSY), but I can not find any thread that is using this device. I try to use sleep() to wait some threads to unlock, but it did not work. the program will stack in the while(){} loop. I think had initialized libusb properly. 
By the way, I use one thread to process the Libusb events,
{
while (running)
	{
		rec=libusb_handle_events(ctx);
		if(rec != LIBUSB_SUCCESS)
		{
			LOG_MSG(verbose::error, "Libusb Handle events Failed! rec: %d", rec);
		}
	}
}

this thread will start once  the program starts. if I power up my usb device after starting the program, libusb_get_string_descriptor_ascii return -6(LIBUSB_ERROR_BUSY),. Every thing is fine when I power up the device early. 
Can anyone give me some advice?

mcuee, 2021-07-30:
Take note FX3 device will revert back to the default bootloader mode if you use RAM. Did you download the image every time during the power-up? Or do you use I2C EEPROM or SPI Flash?

mcuee, 2021-07-30:
You may want to explain a bit more details of what you are doing.  Maybe you need to show more code snipets? The best is that you can provide a small but complete code so that it is easier for people to try to reproduce the issue.

Does libusb_get_string_descriptor_ascii work with your device without hotplug?

sparkertim, 2021-07-30:
> Take note FX3 device will revert back to the default bootloader mode if you use RAM. Did you download the image every time during the power-up? Or do you use I2C EEPROM or SPI Flash?

I had test the firmware of FX3, the device works fine. I use the I2C EEPROM. libusb_get_string_descriptor_ascii return -6 and nothing had sent to FX3, maybe the lisbusb resource is locked  internal? if the Ubuntu system use  libusb and locked it, it will release it later, but libusb_get_string_descriptor_ascii always return -6;

sparkertim, 2021-07-30:
> You may want to explain a bit more details of what you are doing. Maybe you need to show more code snipets? The best is that you can provide a small but complete code so that it is easier for people [...]

yes, it works. and I can send commands to the device and receive data form it although this func failed.

sparkertim, 2021-07-30:
I'll test some simple code to reproduce it later

mcuee, 2021-07-30:
Thanks for the clarifications.

Please try the latest git head of libusb, which has some fixes for Linux. If that still has the issue with the git head, please post the debug log here with your own program, by setting LIBUSB_DEBUG=4. That may gives some clue.

mcuee, 2021-08-13:
BTW, I seem to have encountered similar issue under Windows. Sometimes the device may not be able to respond to Windows USB descriptor request. It has something to do with sleep. I am not even using libusb. In that case, I have to reset the device and then re-download the firmware image.

mcuee, 2021-08-13:
All in all, I think this may have something to do with the device firmware and nothing to do with libusb. You may want to check usbmon under your Linux system and then ask in linux-usb mailing list. 

I will close this for now. Please reopen wehn you got the time to reproduce with simple codes and post the debug log.

closed 2021-08-13 by mcuee

sparkertim, 2021-09-13:
I had write a test code for this issue
https://github.com/sparkertim/libusb_test.

Output is

Start Initialize USB Driver...
Device monitor: device attached.
Examining [05E3:0612] ([4-2])
Examining [1D6B:0003] ([4])
Examining [05E3:0610] ([3-2])
Examining [1D6B:0002] ([3])
Examining [5744:0102] ([2-6.2])
Examining [05E3:0626] ([2-6])
Examining [1D6B:0003] ([2])
Examining [1532:005C] ([1-9])
Examining [09FB:6010] ([1-6.4])
Found device "unknown" [0x09FB:0x6010] ([1-6.4])
Examining [09FB:6010] ([1-6.3])
Found device "unknown" [0x09FB:0x6010] ([1-6.3])
Examining [05E3:0610] ([1-6])
Examining [8087:0029] ([1-14])
Examining [04D9:A0D1] ([1-10])
Examining [1D6B:0002] ([1])
Open Device Success!
Device Name: USB-BlasterII [1-6.3]
                VID: 0x09fb
                PID: 0x6010
                SpeedLevel: Hight Speed (480Mbps)
                MaxPackageSize: 512
                EndPoint: 0x04 Type: Bulk OUT
                EndPoint: 0x88 Type: Bulk IN

Hotplug capabilites are supported on this platform.
Init HotPlug...
Register hotplug callback function success.
HotPlug initialized.
USB Driver Initialized.
Close Device.
Device monitor: device detached.
Device monitor: device attached.
Examining [09FB:6010] ([1-6.4])
Found device "unknown" [0x09FB:0x6010] ([1-6.4])
Examining [09FB:6010] ([1-6.3])
Found device "unknown" [0x09FB:0x6010] ([1-6.3])
Examining [5744:0102] ([2-6.2])
Examining [05E3:0610] ([1-6])
Examining [05E3:0626] ([2-6])
Examining [05E3:0612] ([4-2])
Examining [1D6B:0003] ([4])
Examining [05E3:0610] ([3-2])
Examining [1D6B:0002] ([3])
Examining [1D6B:0003] ([2])
Examining [1532:005C] ([1-9])
Examining [8087:0029] ([1-14])
Examining [04D9:A0D1] ([1-10])
Examining [1D6B:0002] ([1])
Open Device Success!
Device Name:  [1-6.3]
                VID: 0x09fb
                PID: 0x6010
                SpeedLevel: Hight Speed (480Mbps)
                MaxPackageSize: 512
                EndPoint: 0x04 Type: Bulk OUT
                EndPoint: 0x88 Type: Bulk IN

sparkertim, 2021-09-13:
> All in all, I think this may have something to do with the device firmware and nothing to do with libusb. You may want to check usbmon under your Linux system and then ask in linux-usb mailing list.
> 
> I will close this for now. Please reopen when you got the time to reproduce with simple codes and post the debug log.

I had wrote a test code for this issue
https://github.com/sparkertim/libusb_test.

Connect the dvice (USB Blaster II)
Output is

Start Initialize USB Driver...
Device monitor: device attached.
Examining [05E3:0612] ([4-2])
Examining [1D6B:0003] ([4])
Examining [05E3:0610] ([3-2])
Examining [1D6B:0002] ([3])
Examining [5744:0102] ([2-6.2])
Examining [05E3:0626] ([2-6])
Examining [1D6B:0003] ([2])
Examining [1532:005C] ([1-9])
Examining [09FB:6010] ([1-6.4])
Found device "unknown" [0x09FB:0x6010] ([1-6.4])
Examining [09FB:6010] ([1-6.3])
Found device "unknown" [0x09FB:0x6010] ([1-6.3])
Examining [05E3:0610] ([1-6])
Examining [8087:0029] ([1-14])
Examining [04D9:A0D1] ([1-10])
Examining [1D6B:0002] ([1])
Open Device Success!
Device Name: USB-BlasterII [1-6.3]
VID: 0x09fb
PID: 0x6010
SpeedLevel: Hight Speed (480Mbps)
MaxPackageSize: 512
EndPoint: 0x04 Type: Bulk OUT
EndPoint: 0x88 Type: Bulk IN

Hotplug capabilites are supported on this platform.
Init HotPlug...
Register hotplug callback function success.
HotPlug initialized.
USB Driver Initialized.

Disconnect LOG Mesg:

Close Device.
Device monitor: device detached.
Device monitor: device attached.

Connect again:

Examining [09FB:6010] ([1-6.4])
Found device "unknown" [0x09FB:0x6010] ([1-6.4])
Examining [09FB:6010] ([1-6.3])
Found device "unknown" [0x09FB:0x6010] ([1-6.3])
Examining [5744:0102] ([2-6.2])
Examining [05E3:0610] ([1-6])
Examining [05E3:0626] ([2-6])
Examining [05E3:0612] ([4-2])
Examining [1D6B:0003] ([4])
Examining [05E3:0610] ([3-2])
Examining [1D6B:0002] ([3])
Examining [1D6B:0003] ([2])
Examining [1532:005C] ([1-9])
Examining [8087:0029] ([1-14])
Examining [04D9:A0D1] ([1-10])
Examining [1D6B:0002] ([1])
Open Device Success!
Device Name: [1-6.3]     <<---- this line
VID: 0x09fb
PID: 0x6010
SpeedLevel: Hight Speed (480Mbps)
MaxPackageSize: 512
EndPoint: 0x04 Type: Bulk OUT
EndPoint: 0x88 Type: Bulk IN

when hotplug callback function open the device, the Device Name return "". but the device works fine.

a1ien, 2021-09-13:
If read documentation than you find answer.
>When handling a LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED event it is considered safe to call any libusb function that takes a libusb_device. It also safe to open a device and submit asynchronous transfers. However, most other functions that take a libusb_device_handle are not safe to call. Examples of such functions are any of the synchronous API functions or the blocking functions that retrieve various USB descriptors. These functions must be used outside of the context of the hotplug callback.

sparkertim, 2021-09-13:
> If read documentation than you find answer.

Thanks! I read the hotplug part that I missed before! I tried to get device name outside the callback context, and it works! 
So it means that libusb_get_string_descriptor_ascii use synchronous control transfer? 
Why not use asynchronous control transfer?
How about use control transfer async api to get string descriptor in callback function?

a1ien, 2021-09-13:
You can only submit async transfer in context of the hotplug callback. But for receiving response you must have separate function. You can read about async io https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#details