libusb/libusb #970: libusb iso transfer issue

JunzheFan, 2021-08-02:
Hi, I am using the code snippet listed below to do an ISO write to my USB device, the problem I got was wired: the code either just freeze at that **libusb_submit_transfer** line and doesn't print out any info on the return value or return some **pthread** issue. 
accroding to the API doc, the libusb_submit_transfer is a non-blokcing function, so it should print out the return value right after the function call, so I am confused if the problem is because I didn't manage the thread properly or I didn't provide proper arguements to the fill_iso function before submitting the transfer.
thanks for the help 
`libusb_set_auto_detach_kernel_driver(dev_handle[1], true);
 ibusb_claim_interface(dev_handle[1], 2);
 libusb_fill_iso_transfer(&TransferObject,dev_handle[1],Endpoint_Out,buffer,sizeof(buffer),1,Transfer_callback, nullptr,5000);
 rc= libusb_submit_transfer(&TransferObject);
 qDebug()<<rc;
 ibusb_release_interface(dev_handle[1],2);`

JunzheFan, 2021-08-02:
regarding threadings, my program now has got two threading---one is the main thread for handling the Qt's events, and the other thread is for handling the libusb-specific events(it was created for handling the hotplug USB event in the first place). And the new thread was created using Qthread class provided by Qt

mcuee, 2021-08-03:
Please post more details about your devices (lsusb -vvv), the OS and your codes. Thanks.

JunzheFan, 2021-08-03:
Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00c0  1x 192 bytes
        bInterval               1
        bRefresh                0
        bSynchAddress           0
@mcuee  I am using linux , the text snippet acquired from lsusb -vvv above is the endpoint I would like to write data to for my device

JunzheFan, 2021-08-03:
while(!this->isFinished()){
        rc = libusb_handle_events_completed(nullptr, nullptr);
        if(rc==LIBUSB_SUCCESS){
            if(mEvent==ON_DEVICE_ARRIVED){

                if(!DeviceTree.contains(desc.idProduct)){

                    DeviceTree.insert(desc.idProduct,desc.idVendor);
                    unsigned char arr[50];
                    QString DeviceInfo;
                    if(desc.idProduct==0x0012){

                        libusb_get_string_descriptor_ascii(dev_handle[1],desc.iProduct,arr,sizeof(arr));
                        DeviceInfo.append((const char*)arr);

                        memset(arr, 0, sizeof(arr));
                        libusb_get_string_descriptor_ascii(dev_handle[1],desc.iSerialNumber,arr,sizeof(arr));
                        DeviceInfo.append(":"+QString((const char*)arr));

                        memset(arr, 0, sizeof(arr));
                        libusb_get_string_descriptor_ascii(dev_handle[1],desc.iManufacturer,arr,sizeof(arr));
                        DeviceInfo.append(":"+QString((const char*)arr));
                        emit Send_device_Info(DeviceInfo,desc.bcdDevice);

                       **unsigned char buffer[2]={0x21,0x10};**
                       **libusb_set_auto_detach_kernel_driver(dev_handle[1], true);**
                       **libusb_claim_interface(dev_handle[1], 2);**
                       **libusb_fill_iso_transfer(&TransferObject,dev_handle[1],Endpoint_Out,buffer,sizeof(buffer),1,Transfer_callback, nullptr,5000);**
                       **rc= libusb_submit_transfer(&TransferObject);**
                       **qDebug()<<rc;**
                       **libusb_release_interface(dev_handle[1],2);**
                    }
                    if(desc.idProduct==0x0001){

                        libusb_get_string_descriptor_ascii(dev_handle[0],desc.iProduct,arr,sizeof(arr));
                        DeviceInfo.append((const char*)arr);

                        memset(arr, 0, sizeof(arr));
                        libusb_get_string_descriptor_ascii(dev_handle[0],desc.iSerialNumber,arr,sizeof(arr));
                        DeviceInfo.append(":"+QString((const char*)arr));

                        memset(arr, 0, sizeof(arr));
                        libusb_get_string_descriptor_ascii(dev_handle[0],desc.iManufacturer,arr,sizeof(arr));
                        DeviceInfo.append(":"+QString((const char*)arr));
                        emit Send_device_Info(DeviceInfo,desc.bcdDevice);

                    }

                }

            }
            if(mEvent==ON_DEVICE_LEFT){

                if(desc.idProduct==0x0012){
                    emit Remove_BT100();
                }
                if(desc.idProduct==0x0001){
                    emit Remove_UC100();
                }
            }
        }
        sleep(3);
    }
Here is the loop I created to let libusb keep handling the events, I guess I did something wrong here that led to the thread crash? it works fine on handling the hotplug events, and I can see the claim interface return success as well

JunzheFan, 2021-08-03:
@mcuee I just saw these works in the asyc API section----- This step involves allocating memory for a USB transfer. This is the generic transfer object mentioned above. At this stage, the transfer is "blank" with no details about what type of I/O it will be used for.Allocation is done with the libusb_alloc_transfer() function. You must use this function rather than allocating your own transfers.

So we are not allowed to allocate this transfer object like we normally instantiate an object in c, and must use this function to allocate memory for that struct transfer object?  I might assume this might be the reason for the thread crash, I am gonna give it a go anyway

mcuee, 2021-08-03:
isoc is more difficult to get working. Please post the full details of "lsusb -vvv" as well. Thanks. 

There is another thread related to Linux isoc recently. 
https://github.com/libusb/libusb/issues/956

mcuee, 2021-08-03:
Another thing, it could also be related to device FW. How do you know the device is okay? Do you have vendor provided software (eg: under Windows) to verify that it is working?

mcuee, 2021-08-03:
BTW, I am not an expert in isoc transfer and threading issues. You may want to post in libusb-devel mailing list for more inputs.

JunzheFan, 2021-08-03:
@mcuee ok no worry, and I am pretty confused about how to properly use this lib_usb_handle_events function, so do I need to submit the events prior to this function call, or this function can pick up any event in the middle of the function call?

JunzheFan, 2021-08-04:
void USBworker::Libusb_Init() {

    int rc;
    //init the usb session(context)
    libusb_init(nullptr);

    //register hotplug_callback
    rc = libusb_hotplug_register_callback(nullptr, LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED,
                                          static_cast<libusb_hotplug_flag>(0), 0x3368, 0x0001,
                                          LIBUSB_HOTPLUG_MATCH_ANY, hotplug_callback, nullptr,
                                          &hp[0]);
    if (LIBUSB_SUCCESS != rc) {
        printf("Error registering a hotplug callback\n");
        libusb_exit(nullptr);
        return;
    }

    //register hotplug_callback_detach
    rc = libusb_hotplug_register_callback (nullptr, LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT, static_cast<libusb_hotplug_flag>(0), 0x3368,
                                           0x0001, LIBUSB_HOTPLUG_MATCH_ANY,
                                           hotplug_callbackhotplug_callback_detach, nullptr, &hp[1]);
    if (LIBUSB_SUCCESS != rc) {
        fprintf (stderr, "Error registering callback 1\n");
        libusb_exit (nullptr);
        return;
    }

    //register hotplug_callback
    rc = libusb_hotplug_register_callback(nullptr, LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED,
                                          static_cast<libusb_hotplug_flag>(0), 0x3015, 0x0012,
                                          LIBUSB_HOTPLUG_MATCH_ANY, hotplug_callback, nullptr,
                                          &hp[2]);
    if (LIBUSB_SUCCESS != rc) {
        printf("Error registering a hotplug callback\n");
        libusb_exit(nullptr);
        return;
    }

    //register hotplug_callback_detach
    rc = libusb_hotplug_register_callback (nullptr, LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT, static_cast<libusb_hotplug_flag>(0), 0x3015,
                                           0x0012, LIBUSB_HOTPLUG_MATCH_ANY,
                                           hotplug_callbackhotplug_callback_detach, nullptr, &hp[3]);
    if (LIBUSB_SUCCESS != rc) {
        fprintf (stderr, "Error registering callback 1\n");
        libusb_exit (nullptr);
        return;
    }

    while (1){

       rc=libusb_handle_events_timeout_completed(nullptr, &timeout, nullptr);
      // qDebug()<<"1";
       if(rc==LIBUSB_SUCCESS){

           if(mEvent == Event::ON_DEVICE_ARRIVED){
               unsigned char arr[50];
               QString DeviceInfo;
               if(desc.idProduct==0x0012){

                   libusb_get_string_descriptor_ascii(dev_handle[1],desc.iProduct,arr,sizeof(arr));
                   DeviceInfo.append((const char*)arr);

                   memset(arr, 0, sizeof(arr));
                   libusb_get_string_descriptor_ascii(dev_handle[1],desc.iSerialNumber,arr,sizeof(arr));
                   DeviceInfo.append(":"+QString((const char*)arr));

                   memset(arr, 0, sizeof(arr));
                   libusb_get_string_descriptor_ascii(dev_handle[1],desc.iManufacturer,arr,sizeof(arr));
                   DeviceInfo.append(":"+QString((const char*)arr));
                   emit Render_BT100_UI(DeviceInfo,desc.bcdDevice);

               }
               mEvent=Event::IDLE;
           }
           if(mEvent == Event::ON_DEVICE_LEFT){

           }
           if(mEvent == Event::ON_DEVICE_WRITE){

               auto TransferObject = libusb_alloc_transfer(0);
               unsigned char buffer[2]={0x21,0x10};
               libusb_set_auto_detach_kernel_driver(dev_handle[1], true);
               libusb_claim_interface(dev_handle[1], 3);
               libusb_fill_interrupt_transfer(TransferObject,dev_handle[1],EP_OUT,buffer,sizeof(buffer),Transfer_Callback, nullptr,0);
               rc=libusb_submit_transfer(TransferObject);
               if(rc==LIBUSB_SUCCESS){
                   qDebug()<<"good";
                   mEvent=Event::IDLE;
               }

           }
           if(mEvent == Event::ON_DEVICE_READ){

           }
       }
       else break;
    }

}

void USBworker::Transfer_Callback(struct libusb_transfer *transfer){

     qDebug()<<transfer->status;

}

@mcuee like I am using interrupt transfer this time, I can see the submit_tranfer method this time returns success, but that Callback function never get fired,

mcuee, 2021-08-04:
@JunzheFan I am not an expert on async transfer (not a programmer myself), please use libusb-devel mailing list for help. Thanks.

JunzheFan, 2021-08-05:
@mcuee okay no problem, and one more question, I am trying to get into the DFU(firmware upgrade mode), so I suppose the lib_usb_control_transfer is the desired function to use ?  But I have a series of instructions that the vendor says can be used to get into DFU mode, but I am quite not sure how to properly fill out the fileds of that function, do I just send these instructions to Endpoint0? I will also message the dev team for help, thanks

mcuee, 2021-08-05:
For DFU, you can use dfu-util, it uses libusb. And yes DFU spec uses control transfer. But you do not need to do the hard work by yourself, as dfu-util has already done it for you.
http://dfu-util.sourceforge.net/

closed 2021-08-06 by JunzheFan