libusb/libusb #972: MacOS Monterey 12.0 Beta, libuvc, libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access

hl0521, 2021-08-04:
When using the function "uvc_open" in libuvc on MacOS Monterey 12.0, the following exception will be thrown: 

libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access.

Does anyone know how to solve this problem？

The complete log is as follows：

[device.c:322] begin uvc_open
[ 0.706899] [00018c33] libusb: debug [libusb_open] open 20.6
[ 0.707071] [00018c33] libusb: debug [darwin_open] device open for access
[device.c:325/uvc_open] libusb_open() = 0
[device.c:345] begin uvc_open_internal
[device.c:892] begin uvc_ref_device
[device.c:897] end uvc_ref_device
[device.c:430] begin uvc_get_device_info
[ 0.707100] [00018c33] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.707105] [00018c33] libusb: debug [parse_configuration] skipping descriptor 0xb
[ 0.707109] [00018c33] libusb: debug [parse_endpoint] skipping descriptor 0x25
[ 0.707112] [00018c33] libusb: debug [parse_endpoint] skipping descriptor 0xb
[ 0.707115] [00018c33] libusb: debug [parse_endpoint] skipping descriptor 0x25
[device.c:1009] begin uvc_scan_control
[device.c:552] begin uvc_get_device_descriptor
[ 0.707124] [00018c33] libusb: debug [libusb_get_device_descriptor]  
[ 0.707126] [00018c33] libusb: debug [libusb_open] open 20.6
[ 0.707160] [00018c33] libusb: debug [darwin_open] device open for access
[ 0.707165] [00018c33] libusb: debug [libusb_submit_transfer] transfer 0x7fb688c3dbf8
[ 0.707361] [00018c33] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.707368] [00018c33] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.707371] [00018c33] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.707376] [00018c33] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 0.707522] [00018c32] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.707546] [00018c33] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.707549] [00018c33] libusb: debug [handle_event_trigger] event triggered
[ 0.707551] [00018c33] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 0.707554] [00018c33] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb688c3dbf8 has callback 0x100dfafb0
[ 0.707556] [00018c33] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.707559] [00018c33] libusb: debug [libusb_free_transfer] transfer 0x7fb688c3dbf8
[ 0.707564] [00018c33] libusb: debug [libusb_submit_transfer] transfer 0x7fb688e42a08
[ 0.707586] [00018c33] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.707589] [00018c33] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.707591] [00018c33] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 0.707602] [00018c32] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.707613] [00018c33] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.707615] [00018c33] libusb: debug [handle_event_trigger] event triggered
[ 0.707617] [00018c33] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 0.707619] [00018c33] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb688e42a08 has callback 0x100dfafb0
[ 0.707621] [00018c33] libusb: debug [sync_transfer_cb] actual_length=40
[ 0.707624] [00018c33] libusb: debug [libusb_free_transfer] transfer 0x7fb688e42a08
[ 0.707628] [00018c33] libusb: debug [libusb_submit_transfer] transfer 0x7fb688e42a08
[ 0.707660] [00018c33] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.707663] [00018c33] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.707665] [00018c33] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 0.707799] [00018c32] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.707821] [00018c33] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.707825] [00018c33] libusb: debug [handle_event_trigger] event triggered
[ 0.707826] [00018c33] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 0.707829] [00018c33] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb688e42a08 has callback 0x100dfafb0
[ 0.707831] [00018c33] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.707836] [00018c33] libusb: debug [libusb_free_transfer] transfer 0x7fb688e42a08
[ 0.707840] [00018c33] libusb: debug [libusb_submit_transfer] transfer 0x7fb68a882298
[ 0.707867] [00018c33] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.707870] [00018c33] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.707872] [00018c33] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 0.707883] [00018c32] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.707894] [00018c33] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.707896] [00018c33] libusb: debug [handle_event_trigger] event triggered
[ 0.707898] [00018c33] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 0.707900] [00018c33] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb68a882298 has callback 0x100dfafb0
[ 0.707902] [00018c33] libusb: debug [sync_transfer_cb] actual_length=24
[... 96 lines cut ...]
[device.c:1061] end uvc_scan_control (0)
[device.c:455] end uvc_get_device_info (0)
[device.c:358/uvc_open_internal] claiming control interface 0
[device.c:929] begin uvc_claim_if
[ 0.708100] [00018c33] libusb: debug [libusb_detach_kernel_driver] interface 0
[ 0.708103] [00018c33] libusb: debug [darwin_detach_kernel_driver] attempting to detach kernel driver from device
[ 0.708239] [00018c33] libusb: info [darwin_detach_kernel_driver] no capture entitlements. can not detach the kernel driver for this device
[device.c:942/uvc_claim_if] claiming interface 0
[ 0.708245] [00018c33] libusb: debug [libusb_claim_interface] interface 0
[ 0.708766] [00018c33] libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
[device.c:951] end uvc_claim_if (-3)
[device.c:965] begin uvc_release_if
[device.c:966/uvc_release_if] releasing interface 0
[device.c:968/uvc_release_if] attempt to release unclaimed interface 0

[device.c:969] end uvc_release_if (0)
[ 0.708784] [00018c33] libusb: debug [libusb_close]  
[device.c:908] begin uvc_unref_device
[device.c:916] end uvc_unref_device
[device.c:1653] begin uvc_free_devh
[device.c:477] begin uvc_free_device_info
[device.c:530] end uvc_free_device_info
[device.c:1663] end uvc_free_devh
[device.c:411] end uvc_open_internal (-3)
[device.c:333] end uvc_open (-3)

mcuee, 2021-08-05:
That is just a info message and not a problem.
[ 0.708766] [00018c33] libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access

mcuee, 2021-08-05:
Do you have real problem here other than the message?

mcuee, 2021-08-05:
Reference: we have downgraded the message from warning to info as it is normal.
https://github.com/libusb/libusb/issues/955

hl0521, 2021-08-05:
> 
> 
> Do you have real problem here other than the message?

My program use the libuvc (https://github.com/libuvc/libuvc) to control the webcam. The program can work on macOS 10.x and macOS 11.x. It doesn't work on macOS 12.0 that is Apple's new system released recently. The cause is that the function "uvc_open" failed. Function "uvc_open" works as follows:

uvc_open
->uvc_open_internal
   ->uvc_claim_if
       ->libusb_claim_interface
	  ->usbi_backend.claim_interface ==> darwin_capture_claim_interface
	      ->darwin_claim_interface
		  ->USBInterfaceOpen  ---------->return "USBInterfaceOpen: another process has device opened for exclusive access"

Maybe this problem is caused by some changes in Apple’s new system, i'm not sure. And I don't know how to solve this problem, so come here for help.

mcuee, 2021-08-05:
Can you try the latest libusb git? And post the debug log again.

hl0521, 2021-08-05:
> 
> 
> Can you try the latest libusb git? And post the debug log again.

I updated the latest code from the libusb yesterday and got the above log.

The log starts with [device.c:322] begin uvc_open
            ends with [device.c:333] end uvc_open (-3)

mcuee, 2021-08-05:
I see. Thanks. Maybe there is a change on the Apple side. Anyway, this is a beta release. You may want to try again when there is a formal release.

Right now please post the debug log for macOS 11.x as well since you say it works. By comparing the logs, you may be able to identify the differences. It could be that there is a kernel driver in the 12.x beta which opens the device for exclusive access.

One thing you can try is to see if you can use the latest addition to the macOS side, the kernel driver detaching function, but you need to have the right entitlement. I have not tried this myself. More details are in pull request https://github.com/libusb/libusb/pull/911

hl0521, 2021-08-05:
> I see. Thanks. Maybe there is a change on the Apple side. Anyway, this is a beta release. You may want to try again when there is a formal release.
>
> Right now please post the debug log for macOS [...]

On macOS 11.4, the log as follows:

[device.c:322] begin uvc_open
[ 7.424639] [0014a84b] libusb: debug [libusb_open] open 20.10
[ 7.424768] [0014a84b] libusb: debug [darwin_open] device open for access
[device.c:325/uvc_open] libusb_open() = 0
[device.c:345] begin uvc_open_internal
[device.c:892] begin uvc_ref_device
[device.c:897] end uvc_ref_device
[device.c:430] begin uvc_get_device_info
[ 7.424779] [0014a84b] libusb: debug [libusb_get_config_descriptor] index 0
[ 7.424783] [0014a84b] libusb: debug [parse_configuration] skipping descriptor 0xb
[ 7.424785] [0014a84b] libusb: debug [parse_endpoint] skipping descriptor 0x25
[ 7.424788] [0014a84b] libusb: debug [parse_endpoint] skipping descriptor 0xb
[ 7.424791] [0014a84b] libusb: debug [parse_endpoint] skipping descriptor 0x25
[device.c:1009] begin uvc_scan_control
[device.c:552] begin uvc_get_device_descriptor
[ 7.425724] [0014a84b] libusb: debug [libusb_get_device_descriptor]  
[ 7.425735] [0014a84b] libusb: debug [libusb_open] open 20.10
[ 7.425789] [0014a84b] libusb: debug [darwin_open] device open for access
[ 7.425829] [0014a84b] libusb: debug [libusb_submit_transfer] transfer 0x7fcd2a56fd68
[ 7.425948] [0014a84b] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 7.425973] [0014a84b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 7.425977] [0014a84b] libusb: debug [handle_events] event sources modified, reallocating event data
[ 7.425992] [0014a84b] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 7.426368] [0014a84b] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 7.426375] [0014a84b] libusb: debug [handle_event_trigger] event triggered
[ 7.426378] [0014a84b] libusb: debug [handle_event_trigger] hotplug message received
[ 7.426414] [0014a84b] libusb: debug [libusb_unref_device] destroy device 20.6
[ 7.426423] [0014a84b] libusb: debug [libusb_unref_device] destroy device 20.5
[ 7.426446] [0014a84b] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 7.426457] [0014a84b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 7.426464] [0014a84b] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 7.427444] [0014a840] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 7.428181] [0014a84b] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 7.428194] [0014a84b] libusb: debug [handle_event_trigger] event triggered
[ 7.428564] [0014a84b] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 7.428570] [0014a84b] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fcd2a56fd68 has callback 0x1088976e0
[ 7.428574] [0014a84b] libusb: debug [sync_transfer_cb] actual_length=4
[ 7.428581] [0014a84b] libusb: debug [libusb_free_transfer] transfer 0x7fcd2a56fd68
[ 7.428588] [0014a84b] libusb: debug [libusb_submit_transfer] transfer 0x7fcd2aca5b88
[ 7.428648] [0014a84b] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 7.428653] [0014a84b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 7.428656] [0014a84b] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 7.428932] [0014a840] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 7.428962] [0014a84b] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 7.428969] [0014a84b] libusb: debug [handle_event_trigger] event triggered
[ 7.428971] [0014a84b] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 7.428974] [0014a84b] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fcd2aca5b88 has callback 0x1088976e0
[ 7.428977] [0014a84b] libusb: debug [sync_transfer_cb] actual_length=40
[ 7.428981] [0014a84b] libusb: debug [libusb_free_transfer] transfer 0x7fcd2aca5b88
[ 7.429000] [0014a84b] libusb: debug [libusb_submit_transfer] transfer 0x7fcd277bbdd8
[ 7.429078] [0014a84b] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 7.429083] [0014a84b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 7.429087] [0014a84b] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 7.429369] [0014a840] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 7.429568] [0014a84b] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 7.429715] [0014a84b] libusb: debug [handle_event_trigger] event triggered
[ 7.429720] [0014a84b] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 7.429724] [0014a84b] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fcd277bbdd8 has callback 0x1088976e0
[ 7.429727] [0014a84b] libusb: debug [sync_transfer_cb] actual_length=4
[ 7.429731] [0014a84b] libusb: debug [libusb_free_transfer] transfer 0x7fcd277bbdd8
[ 7.429738] [0014a84b] libusb: debug [libusb_submit_transfer] transfer 0x7fcd2b91e5f8
[... cut ...]
[device.c:951] end uvc_claim_if (0)
[ 7.433260] [0014a84b] libusb: debug [libusb_get_device_descriptor]  
[ 7.433264] [0014a84b] libusb: debug [libusb_submit_transfer] transfer 0x7fcd2aca2978
[ 7.433269] [0014a84b] libusb: debug [ep_to_pipeRef] converting ep address 0x84 to pipeRef and interface
[ 7.433272] [0014a84b] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[device.c:382/uvc_open_internal] libusb_submit_transfer() = 0
[device.c:399] end uvc_open_internal (0)
[device.c:333] end uvc_open (0)

mcuee, 2021-08-05:
Main difference:
macOS 11.x
```
[device.c:942/uvc_claim_if] claiming interface 0
[ 7.430490] [0014a84b] libusb: debug [libusb_claim_interface] interface 0
[ 7.432406] [0014a84b] libusb: debug [get_endpoints] building table of endpoints.
[ 7.433185] [0014a84b] libusb: debug [get_endpoints] interface: 0 pipe 1: dir: 1 number: 4
[ 7.433247] [0014a84b] libusb: debug [darwin_claim_interface] interface opened
```

macOS 12 beta: claiming interface failed. So there is a kernel driver (or maybe system application) which opened the device for exclusive access. You may have to contact Apple for help.

```
[device.c:942/uvc_claim_if] claiming interface 0
[ 0.708245] [00018c33] libusb: debug [libusb_claim_interface] interface 0
[ 0.708766] [00018c33] libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
[device.c:951] end uvc_claim_if (-3)```

hjelmn, 2021-08-08:
Think I know how I broke it. Fixing now....

hjelmn, 2021-08-08:
Opps. Wrong bug :).

mcuee, 2021-08-08:
@hl0521 You may want to give latest git a try again (to run with sudo) under macOS 12 beta. Nathan just fixed an issue and it might help you under macOS 12 beta.

Reference discussion: https://github.com/libusb/libusb/pull/911

hl0521, 2021-08-08:
> 
> 
> @hl0521 You may want to give latest git a try again (to run with sudo) under macOS 12 beta. Nathan just fixed an issue and it might help you under macOS 12 beta.
> 
> Reference discussion: #911

It works when running with sudo. Thanks.

I also found another issue: when calling "uvc_close" in libuvc, the following line of code in function "libusb_free_transfer" oftern lead to program crash. 
    usbi_dbg(TRANSFER_CTX(transfer), "transfer %p", transfer);
The cause is null pointer in TRANSFER_CTX(transfer), the object may have been released in other place. When I comment this line of code, the program can work.

mcuee, 2021-08-08:
Thanks for the updates. Then there is no issue with libusb already. I will close this issue.

The proper way (without using sudo) is for the developer to get the right entitlement as per the following discussion: libusb/libusb#911

> kUSBReEnumerateCaptureDeviceMask requires either the application to be running as root OR the application has the entitlement com.apple.vm.device-access AND the user has accepted the authorization for the capture with IOServiceAuthorize.

mcuee, 2021-08-08:
> I also found another issue: when calling "uvc_close" in libuvc, the following line of code in function "libusb_free_transfer" oftern lead to program crash.
> `usbi_dbg(TRANSFER_CTX(transfer), [...]

You may want to report the issue to libuvc, looks like a bug in libuvc. But if you think it is an issue with libusb, please create a new issue and post the debug log. Thanks.

scchn, 2021-08-12:
Non-VM apps on the App Store have no way to get the 'com.apple.vm.device-access' or run as root by default.
Contacted apple and they suggested that I file a bug about this problem since macOS 12 is still in beta.

mcuee, 2021-08-12:
> Non-VM apps on the App Store have no way to get the 'com.apple.vm.device-access' or run as root by default.
> Contacted apple and they suggested that I file a bug about this problem since macOS 12 is still in beta.

Thanks for the updates. As for the first sentence, I am not so sure if it is correct or not. I think it is not correct to say "non-VM apps".

scchn, 2021-08-12:
Ha, me either.

Quote from the mail:
```
That entitlement is intended for virtual machine (VM) apps that need to ship in the Mac App Store
(hence the `vm` in `com.apple.vm.device-access`).
```

mcuee, 2021-08-12:
Do not worry too much about the vm in the name.

https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_vm_device-access
```The entitlement is required to use the IOUSBHost APIs for USB device capture.```

mcuee, 2021-10-30:
https://github.com/libusb/libusb/wiki/FAQ#how-can-i-run-libusb-applications-under-mac-os-x-if-there-is-already-a-kernel-extension-installed-for-the-device-and-claim-exclusive-access

Updates on Oct 2021: with the merging of pull request #911, support for detaching kernel drivers with authorization has been added. This will be included in the upcoming 1.0.25 release. Take note the underlying macOS capture APIs only work on the whole device and not on individual interfaces. So this will force all the kernel extensions (drivers) bound to all the interfaces of a USB Composite devices to be released (Issue #920). You will need to get the entitlement from Apple Developer support, create a provisioning profile with that entitlement, and build your app with that profile. Please take note that command line apps cannot use provisioning profiles and therefore cannot hold this entitlement.

mcuee, 2021-11-09:
Please refer to the following discussion as well for further info. Apple support seems to say that the 'com.apple.vm.device-access' is not the right entitlement for this purpose.
- #1014

mcuee, 2021-11-12:
Ref: https://github.com/libuvc/libuvc/issues/188

If you are just testing, then you can use latest libusb git and sudo. 

```
mcuee@mcuees-Mac-mini build % sudo ./example 
UVC initialized
Device found
Device opened
DEVICE CONFIGURATION (0ac8:3420/[none]) ---
Status: idle
VideoControl:
	bcdUVC: 0x0100
VideoStreaming(1):
	bEndpointAddress: 130
	Formats:
	UncompressedFormat(1)
		  bits per pixel: 16
		  GUID: 5955593200001000800000aa00389b71 (YUY2)
		  default frame: 1
		  aspect ratio: 0x0
		  interlace flags: 00
		  copy protect: 00
			FrameDescriptor(1)
			  capabilities: 00
			  size: 640x480
			  bit rate: 768000-196608000
			  max frame size: 614400
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(2)
			  capabilities: 00
			  size: 352x288
			  bit rate: 768000-196608000
			  max frame size: 202752
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(3)
			  capabilities: 00
			  size: 320x240
			  bit rate: 768000-196608000
			  max frame size: 153600
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(4)
			  capabilities: 00
			  size: 176x144
			  bit rate: 768000-196608000
			  max frame size: 50688
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(5)
			  capabilities: 00
			  size: 160x120
			  bit rate: 768000-196608000
			  max frame size: 38400
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			StillFrameDescriptor
			  bEndPointAddress: 00
			  wWidth(1) = 640
			  wHeight(1) = 480
			  wWidth(2) = 352
			  wHeight(2) = 288
			  wWidth(3) = 320
			  wHeight(3) = 240
			  wWidth(4) = 176
			  wHeight(4) = 144
			  wWidth(5) = 160
			  wHeight(5) = 120
END DEVICE CONFIGURATION

First format: (YUY2) 640x480 30fps
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 614400
dwMaxPayloadTransferSize: 2688
bInterfaceNumber: 1
Streaming...
Enabling auto exposure ...
 ... full AE not supported, trying aperture priority mode
 ... enabled aperture priority auto exposure mode
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
...

```

mcuee, 2021-11-12:
ANother libuvc issue is here: https://github.com/libuvc/libuvc/issues/194

mmppeegg, 2021-12-20:
> com.apple.vm.device-access

So this entitlement is not required for USBInterfaceOpen, right? Can I say this might be an issue of macOS 12 itself?

mcuee, 2021-12-24:
> > com.apple.vm.device-access
> 
> So this entitlement is not required for USBInterfaceOpen, right? Can I say this might be an issue of macOS 12 itself? My app must not be used with root.

No, it is required. Please go to the other discussion to check out potential workaround -- that is to say, do not use libusb/libuvc.
https://github.com/libusb/libusb/issues/1014

mcuee, 2022-02-06:
A good summary here:
https://github.com/joelpurra/node-uvc/issues/2

mcuee, 2022-02-06:
I have updated the wiki here.
https://github.com/libusb/libusb/wiki/FAQ#How_can_I_run_libusb_applications_under_Mac_OS_X_if_there_is_already_a_kernel_extension_installed_for_the_device_and_claim_exclusive_access

And I will close this issue for now. Please continue in #1014 
* #1014

mcuee, 2022-06-10:
Re-open this one so that future issues can be linked here.

mcuee, 2022-06-10:
Related issue from libuvc github.
* https://github.com/libuvc/libuvc/issues/194
* https://github.com/libuvc/libuvc/issues/188

Hucent, 2022-09-05:
I asked this question to the Apple , and they said:

_Based on your request Apple Developer Technical Support believes that your question is answered by the Core Media I/O documentation page linked below: 

Overriding the default USB video class extension:
https://developer.apple.com/documentation/coremediaio/overriding_the_default_usb_video_class_extension?language=objc_

Has someone try it?

Hucent, 2022-09-05:
I asked this problem to the Apple, and they said:

Based on your request Apple Developer Technical Support believes that your question is answered by the Core Media I/O documentation page linked below: 

Overriding the default USB video class extension:
https://developer.apple.com/documentation/coremediaio/overriding_the_default_usb_video_class_extension?language=objc

Did someone try this?

mcuee, 2022-09-05:
> I asked this problem to the Apple, and they said:
>
> Based on your request Apple Developer Technical Support believes that your question is answered by the Core Media I/O documentation page linked [...]

This should work. Basically it is the same as previously used method -- codeless kext, now it is codeless DriverKit extension. You do need to have the valid developer signature in order to load the extension.

Ref: https://developer.apple.com/documentation/systemextensions/installing_system_extensions_and_drivers?language=objc
>During the activation process, the system verifies:
> Your extension is installed in the Contents/Library/SystemExtensions folder of your app.
> Your app is installed in an appropriate Applications directory of the system.
> The code signature of your extension.
> The entitlements in the code signature match the entitlements granted to your development team.
> The identifer in your activation request matches the one in your system extension bundle.
> The identifier isn’t already in use by another system extension.

The main drawback compared to previous situation is that it will no longer function as a UVC camera and you can only use libuvc. It may be okay for certain use cases though.

If possible, I think it is better not to use libusb/libuvc in this case. It was the same situation last time with HID device. Later HIDAPI project was launched to use native HID API under Mac OS X (now macOS). Maybe the application should use native macOS API now.

Macschrauber, 2022-12-16:
It's quite a while but found it via google. For me it was "libusb.kext" installed by G-Flash.

kextunload it solved it

mcuee, 2023-01-24:
> It's quite a while but found it via google. For me it was "libusb.kext" installed by G-Flash.
> 
> kextunload it solved it

Hmm, that is strange to know. I do not think this is the cause for majority of the users. I believe it is the Apple `com.apple.UVCService`.

mcuee, 2023-01-24:
With Apple macOS Ventura 13.2, I need to use sudo in order to find the USB camera (PROLiNK PCC3220 USB web campera as the test device)

Using git HEAD (same for older release version).
```
mcuee@mcuees-Mac-mini libusb % ./examples/testlibusb 
Dev (bus 2, device 10): 0BDA - 5411 speed: 480M
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
Dev (bus 2, device 9): 0BDA - 5411 speed: 480M
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
Dev (bus 2, device 3): 05E3 - 0749 speed: 5G
  Manufacturer:              Generic
  Product:                   USB3.0 Card Reader
Dev (bus 2, device 2): 0BDA - 0411 speed: 5G
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub
Dev (bus 2, device 1): 0BDA - 0411 speed: 5G
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub

mcuee@mcuees-Mac-mini libusb % sudo ./examples/testlibusb
Password:
Dev (bus 2, device 12): 0AC8 - 3420 speed: 480M
  Manufacturer:              Vimicro Corp.
  Product:                   PROLiNK PCC3220
Dev (bus 2, device 10): 0BDA - 5411 speed: 480M
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
Dev (bus 2, device 9): 0BDA - 5411 speed: 480M
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
Dev (bus 2, device 3): 05E3 - 0749 speed: 5G
  Manufacturer:              Generic
  Product:                   USB3.0 Card Reader
Dev (bus 2, device 2): 0BDA - 0411 speed: 5G
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub
Dev (bus 2, device 1): 0BDA - 0411 speed: 5G
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub

```

Youw, 2023-02-01:
> Overriding the default USB video class extension:
https://developer.apple.com/documentation/coremediaio/overriding_the_default_usb_video_class_extension?language=objc
> 
> Did someone try this?

My team has successfully used it - that is the way to go.

> Basically it is the same as previously used method -- codeless kext, now it is codeless DriverKit extension.

Not exactly: (even codeless) kext is loaded by the kernel and running in the kernel space. It is deprecated since macOS 12.<something> and it is practically impossible to make/release a product with thiks kind of kext - Apple will not allow it.
They specifically suggested to use USB/UVC SystemExtension which is running in user-space, and allows you either get exclusive access to to UVC device (so no other application will see it) or even develope a proxy/driver for that UVC device.
But you not only need to have a valid Apple Developer Program membership - you also need to be a vendor of that UVC device. They do not allow making extensions for just any camera.

mcuee, 2023-02-16:
So there is a solution but it is not easy for most users. 

Anyway, this is not a libusb issue now. I will close this one.

closed 2023-02-16 by mcuee

eagle8625, 2025-06-17:
> > Overriding the default USB video class extension:
> > https://developer.apple.com/documentation/coremediaio/overriding_the_default_usb_video_class_extension?language=objc
> > Did someone try this?
> 
> My team has successfully used it - that is the way to go.
> 
> > Basically it is the same as previously used method -- codeless kext, now it is codeless DriverKit extension.
> 
> Not exactly: (even codeless) kext is loaded by the kernel and running in the kernel space. It is deprecated since macOS 12. and it is practically impossible to make/release a product with thiks kind of kext - Apple will not allow it. They specifically suggested to use USB/UVC SystemExtension which is running in user-space, and allows you either get exclusive access to to UVC device (so no other application will see it) or even develope a proxy/driver for that UVC device. But you not only need to have a valid Apple Developer Program membership - you also need to be a vendor of that UVC device. They do not allow making extensions for just any camera.

Hello Youw,
Could you share your successful example for https://developer.apple.com/documentation/coremediaio/overriding_the_default_usb_video_class_extension?language=objc?
which is really appreciated.