libusb/libusb #974: Better cleanup upon libusb_close(0 and libusb_exit()

mcuee, 2021-08-09:
Ref: https://github.com/vpelletier/python-libusb1/issues/72

It seems to me the issue may be more with python-libusb1, however, it may be possible to improve on libusb side as well to reduce the chance of  libusb warning of device still being referenced.

macOS seems to be better now with the fixes after fixing the warnings mentioned in issue #924. 

Windows seems to be still more problematic. 

And the debug log shows device unref after libusb_exit for the Linux backend. I am not so sure if that is good enough already.

mcuee, 2021-08-09:
As mentioned in #924, dfu-util has to do extra clean-ups to remove the warning. It may not be a libusb issue but I am not so sure if libusb can help here.
https://sourceforge.net/p/dfu-util/dfu-util/ci/63c6388603057fca8b84a3b900d53ee41673a7e1/

mcuee, 2021-08-18:
It could be libusb Windows issue as well.
https://github.com/vpelletier/python-libusb1/issues/72#issuecomment-900168898
From python-libusb1 developer:
> Thanks for the extensive testing and debugging.
> 
> > The following debug log seems to be a bit suspicious. Is it possible that only the default context was detroyed and not the second second context?
> 
> One particularity of python-libusb1 is that my code never uses the default context, because this seemed the sane thing to do, and it was mapping very well in the object world. I suspect this may differ from most uses, and could cause unusual codepaths to be exercised. Could the windows backend accidentally rely on this ?
> 
> The default context destruction triggering warnings about ~still-referenced devices~ leaked `libusb_devices` looks to me as another hint that there may be some confusion about the context.

mcuee, 2021-08-23:
BTW, I could not reproduce the issue using mod C examples from libusb to test non-default libusb context. So I am not so sure if we have a real issue here. 
https://github.com/vpelletier/python-libusb1/issues/72

mcuee, 2021-08-30:
Still whenever I use libusb-compat-0.1 under Windows, I always see "device still referenced" warnings in the end, even though I do not see such warnings with libusb-1.0.

Example codes

libusb-compat-0.1
https://github.com/mcuee/picusb/blob/master/lvr_winusb/libusb_winusb2.c

libusb1
https://github.com/mcuee/picusb/blob/master/lvr_winusb/libusb1_winusb2.c

```
C:\work\picusb\picusb\lvr_winusb [master ≡ +25 ~1 -0 !]> .\libusb0_winusb2.exe
lvr_winusb with Vendor Id: 925 and Product Id: 150c found.
Testing control transfer:
20, 20; 21, 21;
Testing interrupt transfer:
40, 40; 41, 41;
Testing bulk transfer:
00, 00; 01, 01; 02, 02; 03, 03; 04, 04; 05, 05; 06, 06; 07, 07;
08, 08; 09, 09; 0a, 0a; 0b, 0b; 0c, 0c; 0d, 0d; 0e, 0e; 0f, 0f;
10, 10; 11, 11; 12, 12; 13, 13; 14, 14; 15, 15; 16, 16; 17, 17;
18, 18; 19, 19; 1a, 1a; 1b, 1b; 1c, 1c; 1d, 1d; 1e, 1e; 1f, 1f;
20, 20; 21, 21; 22, 22; 23, 23; 24, 24; 25, 25; 26, 26; 27, 27;
28, 28; 29, 29; 2a, 2a; 2b, 2b; 2c, 2c; 2d, 2d; 2e, 2e; 2f, 2f;
30, 30; 31, 31; 32, 32; 33, 33; 34, 34; 35, 35; 36, 36; 37, 37;
38, 38; 39, 39; 3a, 3a; 3b, 3b; 3c, 3c; 3d, 3d; 3e, 3e; 3f, 3f;
libusb: warning [libusb_exit] device 1.44 still referenced
libusb: warning [libusb_exit] device 1.29 still referenced
libusb: warning [libusb_exit] device 1.4 still referenced
libusb: warning [libusb_exit] device 1.5 still referenced
libusb: warning [libusb_exit] device 1.3 still referenced
libusb: warning [libusb_exit] device 1.6 still referenced
libusb: warning [libusb_exit] device 2.0 still referenced
libusb: warning [libusb_exit] device 1.0 still referenced

C:\work\picusb\picusb\lvr_winusb [master ≡ +25 ~1 -0 !]> .\libusb1_winusb2.exe
Successfully find the LVR WINUSB device
Successfully set usb configuration 1
Successfully claimed interface
Testing control transfer:
14, 14; 15, 15;
Testing interrupt transfer:
00, 00; 01, 01;
Testing bulk transfer:
00, 00; 01, 01; 02, 02; 03, 03; 04, 04; 05, 05; 06, 06; 07, 07;
08, 08; 09, 09; 0a, 0a; 0b, 0b; 0c, 0c; 0d, 0d; 0e, 0e; 0f, 0f;
10, 10; 11, 11; 12, 12; 13, 13; 14, 14; 15, 15; 16, 16; 17, 17;
18, 18; 19, 19; 1a, 1a; 1b, 1b; 1c, 1c; 1d, 1d; 1e, 1e; 1f, 1f;
20, 20; 21, 21; 22, 22; 23, 23; 24, 24; 25, 25; 26, 26; 27, 27;
28, 28; 29, 29; 2a, 2a; 2b, 2b; 2c, 2c; 2d, 2d; 2e, 2e; 2f, 2f;
30, 30; 31, 31; 32, 32; 33, 33; 34, 34; 35, 35; 36, 36; 37, 37;
38, 38; 39, 39; 3a, 3a; 3b, 3b; 3c, 3c; 3d, 3d; 3e, 3e; 3f, 3f;
```

mcuee, 2021-08-30:
Debug log for the above code using libusb-compat-0.1 under Windows.

<details>
<summary> click to expand for the full debug logs </summary>

```

C:\work\picusb\picusb\lvr_winusb [master ≡ +25 ~1 -0 !]> $Env:LIBUSB_DEBUG=4
C:\work\picusb\picusb\lvr_winusb [master ≡ +25 ~1 -0 !]> .\libusb0_winusb2.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000337] [0000242c] libusb: debug [libusb_init] libusb v1.0.24.11650
[ 0.000511] [0000242c] libusb: debug [usbi_add_event_source] add HANDLE 000000000000008C events 0
[ 0.000664] [0000242c] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000802] [0000242c] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000088 events 0
[ 0.000972] [0000242c] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.001121] [0000242c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.005835] [0000242c] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.011518] [0000242c] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.018654] [0000242c] libusb: debug [windows_init] UsbDk backend is available
[ 0.018994] [0000242c] libusb: debug [libusb_get_device_list]
[ 0.020108] [00006a98] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.039401] [0000242c] libusb: debug [winusb_get_device_list] allocating new device for session [94]
[ 0.039650] [0000242c] libusb: debug [winusb_get_device_list] allocating new device for session [95]
[ 0.040091] [0000242c] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.040342] [0000242c] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.040524] [0000242c] libusb: debug [winusb_get_device_list] allocating new device for session [10]
[ 0.041168] [0000242c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.041388] [0000242c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.041508] [0000242c] libusb: debug [winusb_get_device_list] allocating new device for session [70]
[ 0.041915] [0000242c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.042106] [0000242c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.042251] [0000242c] libusb: debug [winusb_get_device_list] allocating new device for session [71]
[ 0.042610] [0000242c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.042795] [0000242c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.043046] [0000242c] libusb: debug [winusb_get_device_list] allocating new device for session [72]
[ 0.043465] [0000242c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.043647] [0000242c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.043794] [0000242c] libusb: debug [winusb_get_device_list] allocating new device for session [73]
[ 0.044146] [0000242c] libusb: debug [get_api_type] driver(s): WINUSB
[ 0.044318] [0000242c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.044441] [0000242c] libusb: debug [winusb_get_device_list] allocating new device for session [74]
[ 0.044792] [0000242c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A36D&SUBSYS_091A1028&REV_10\3&11583659&0&A0' bus number 1
[ 0.044992] [0000242c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15DB&SUBSYS_091A1028&REV_02\71D9F51854B3020000' bus number 2
[ 0.046703] [0000242c] libusb: debug [winusb_get_device_list] extra GUID: {865C4DDE-10B6-6373-FAEE-1506B321A619}
[ 0.046840] [0000242c] libusb: debug [winusb_get_device_list] found existing device for session [70]
[ 0.047058] [0000242c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.047182] [0000242c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.047317] [0000242c] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 3): 'USB\VID_046D&PID_C534\5&E9F3E45&0&3'
[ 0.047560] [0000242c] libusb: debug [winusb_get_device_list] found existing device for session [10]
[ 0.047752] [0000242c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.047875] [0000242c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.048006] [0000242c] libusb: debug [init_device] (bus: 1, addr: 6, depth: 1, port: 14): 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
[ 0.048486] [0000242c] libusb: debug [winusb_get_device_list] extra GUID: {ecceff35-146c-4ff3-acd9-8f992d09acdd}
[ 0.048619] [0000242c] libusb: debug [winusb_get_device_list] found existing device for session [74]
[ 0.048827] [0000242c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.048943] [0000242c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 46 bytes)
[ 0.049082] [0000242c] libusb: debug [init_device] (bus: 1, addr: 48, depth: 1, port: 1): 'USB\VID_0925&PID_150C\5&E9F3E45&0&1'
[ 0.049323] [0000242c] libusb: debug [winusb_get_device_list] found existing device for session [73]
[ 0.049523] [0000242c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.049638] [0000242c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 288 bytes)
[ 0.049776] [0000242c] libusb: debug [init_device] (bus: 1, addr: 29, depth: 1, port: 2): 'USB\VID_047F&PID_C056\D1CEC32927974D5F9BD6B2AEBF2EA8E3'
[... cut ...]
[ 0.134427] [0000242c] libusb: warning [libusb_exit] device 1.4 still referenced
[ 0.134472] [0000242c] libusb: warning [libusb_exit] device 1.5 still referenced
[ 0.134517] [0000242c] libusb: warning [libusb_exit] device 1.3 still referenced
[ 0.134562] [0000242c] libusb: warning [libusb_exit] device 1.6 still referenced
[ 0.134606] [0000242c] libusb: warning [libusb_exit] device 2.0 still referenced
[ 0.134652] [0000242c] libusb: warning [libusb_exit] device 1.0 still referenced
```

</details>

mcuee, 2021-09-01:
Adding libusb_close() call will make the issue going away. However, the question is if the libusb_close() call is needed or not.

```
#include <stdio.h>
#include <libusb.h>

int main(int argc, char* argv[]){
  libusb_context* libusbContext = nullptr;
  if(libusb_init(&libusbContext) != LIBUSB_SUCCESS)
    return -1;

  libusb_device** devList = nullptr;
  const int devCount = libusb_get_device_list(libusbContext, &devList);
  if(devCount <= 0){
    libusb_exit(libusbContext);
    return 0;
  }

  for(int i = 0; i < devCount; ++i){
    libusb_device_descriptor desc;
    libusb_device_handle* handle = nullptr;
    if(libusb_open(devList[i], &handle) != LIBUSB_SUCCESS) continue;
    libusb_get_device_descriptor(devList[i], &desc);
    libusb_close(handle);
  }
  libusb_free_device_list(devList, 1);
  libusb_exit(libusbContext);
  return 0;
}
```

Debug log no issues.
```
(py39venv) C:\work\libusb\tests> .\libusb1_win_leak1.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000324] [00003e34] libusb: debug [libusb_init] libusb v1.0.24.11650
[ 0.000492] [00003e34] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000094 events 0
[ 0.000642] [00003e34] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000790] [00003e34] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000098 events 0
[ 0.000940] [00003e34] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.001078] [00003e34] libusb: debug [htab_create] using 1021 entries hash table
[ 0.005862] [00003e34] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.012283] [00003e34] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.020234] [00003e34] libusb: debug [windows_init] UsbDk backend is available
[ 0.020597] [00003e34] libusb: debug [libusb_get_device_list]
[ 0.021726] [00003328] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.042208] [00003e34] libusb: debug [winusb_get_device_list] allocating new device for session [93]
[ 0.042455] [00003e34] libusb: debug [winusb_get_device_list] allocating new device for session [94]
[ 0.043019] [00003e34] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.043270] [00003e34] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.043438] [00003e34] libusb: debug [winusb_get_device_list] allocating new device for session [10]
[ 0.043914] [00003e34] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.044136] [00003e34] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.044281] [00003e34] libusb: debug [winusb_get_device_list] allocating new device for session [1E]
[ 0.044912] [00003e34] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.045111] [00003e34] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.045287] [00003e34] libusb: debug [winusb_get_device_list] allocating new device for session [71]
[ 0.045705] [00003e34] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.045946] [00003e34] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.046098] [00003e34] libusb: debug [winusb_get_device_list] allocating new device for session [72]
[ 0.046494] [00003e34] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.046719] [00003e34] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.046838] [00003e34] libusb: debug [winusb_get_device_list] allocating new device for session [73]
[ 0.047302] [00003e34] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.047525] [00003e34] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.047713] [00003e34] libusb: debug [winusb_get_device_list] allocating new device for session [74]
[ 0.048090] [00003e34] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A36D&SUBSYS_091A1028&REV_10\3&11583659&0&A0' bus number 1
[ 0.048255] [00003e34] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15DB&SUBSYS_091A1028&REV_02\71D9F51854B3020000' bus number 2
[ 0.049990] [00003e34] libusb: debug [winusb_get_device_list] extra GUID: {865C4DDE-10B6-6373-FAEE-1506B321A619}
[... 90 lines cut ...]
[ 0.090207] [00003e34] libusb: debug [libusb_open] open 1.0 returns -12
[ 0.090351] [00003e34] libusb: debug [libusb_open] open 1.2
[ 0.090493] [00003e34] libusb: debug [libusb_open] open 1.2 returns -5
[ 0.090641] [00003e34] libusb: debug [libusb_open] open 2.0
[ 0.090724] [00003e34] libusb: debug [winusb_open] unsupported API call for 'open' (unrecognized device driver)
[ 0.090831] [00003e34] libusb: debug [libusb_open] open 2.0 returns -12
[ 0.090908] [00003e34] libusb: debug [libusb_open] open 1.27
[ 0.091030] [00003e34] libusb: debug [libusb_get_device_descriptor]
[ 0.091118] [00003e34] libusb: debug [libusb_close]
[ 0.091266] [00003e34] libusb: debug [libusb_open] open 1.4
[ 0.091332] [00003e34] libusb: debug [libusb_open] open 1.4 returns -5
[ 0.091397] [00003e34] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.091471] [00003e34] libusb: debug [libusb_unref_device] destroy device 1.5
[ 0.091541] [00003e34] libusb: debug [libusb_unref_device] destroy device 1.9
[ 0.091606] [00003e34] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.091670] [00003e34] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.091733] [00003e34] libusb: debug [libusb_unref_device] destroy device 1.27
[ 0.091817] [00003e34] libusb: debug [libusb_unref_device] destroy device 1.4
[ 0.091880] [00003e34] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.091943] [00003e34] libusb: debug [libusb_exit]
[ 0.092054] [00003328] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.094280] [00003e34] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000098
[ 0.094449] [00003e34] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000094

```

mcuee, 2021-09-01:
> Adding libusb_close() call will make the issue going away. However, the question is if the libusb_close() call is needed or not.

Technically libusb_close() is needed as per the API documentation.
https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#ga779bc4f1316bdb0ac383bddbd538620e

> void libusb_close	(	libusb_device_handle * 	dev_handle	)	
> Close a device handle. Should be called on all open handles before your application exits.
> Internally, this function destroys the reference that was added by libusb_open() on the given device.
> This is a non-blocking function; no requests are sent over the bus.

mcuee, 2021-09-01:
On the other hand,   libusb_free_device_list() is not strictly needed and it is still okay if I comment out the code.

mcuee, 2021-09-08:
https://github.com/vpelletier/python-libusb1/issues/77#issuecomment-914846587

Indeed this is good enough to reproduce the issue.
```
C:\work\libusb\libusb [master ≡ +5 ~3 -0 !]> $Env:LIBUSB_DEBUG=3
C:\work\libusb\libusb [master ≡ +5 ~3 -0 !]> .\x64\Release\examples\listdevs.exe
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
0424:5744 (bus 3, device 5) path: 4.3.2
0bda:5413 (bus 3, device 6) path: 2.3
0bda:0413 (bus 3, device 3) path: 4.3
0424:2740 (bus 3, device 14) path: 2.3.2.5
413c:b06e (bus 3, device 7) path: 2.5
046d:c534 (bus 1, device 1) path: 1
8087:0aaa (bus 1, device 4) path: 14
0424:2744 (bus 3, device 8) path: 2.3.2
0bda:5487 (bus 3, device 2) path: 2
0bda:0487 (bus 3, device 1) path: 4
047f:c056 (bus 3, device 11) path: 2.3.2.2
0bda:402e (bus 3, device 9) path: 2.3.4
8086:a36d (bus 1, device 0)
413c:2107 (bus 3, device 12) path: 2.3.2.3
0a5c:5842 (bus 1, device 2) path: 10
8086:15f0 (bus 3, device 0)
8086:15db (bus 2, device 0)
0bda:8153 (bus 3, device 4) path: 4.4
413c:b06f (bus 3, device 10) path: 2.3.5
046d:c077 (bus 3, device 13) path: 2.3.2.4
0bda:58fd (bus 1, device 3) path: 11
0424:5744 (bus 3, device 5) path: 4.3.2
0bda:5413 (bus 3, device 6) path: 2.3
0bda:0413 (bus 3, device 3) path: 4.3
0424:2740 (bus 3, device 14) path: 2.3.2.5
413c:b06e (bus 3, device 7) path: 2.5
046d:c534 (bus 1, device 1) path: 1
8087:0aaa (bus 1, device 4) path: 14
0424:2744 (bus 3, device 8) path: 2.3.2
0bda:5487 (bus 3, device 2) path: 2
0bda:0487 (bus 3, device 1) path: 4
047f:c056 (bus 3, device 11) path: 2.3.2.2
0bda:402e (bus 3, device 9) path: 2.3.4
8086:a36d (bus 1, device 0)
413c:2107 (bus 3, device 12) path: 2.3.2.3
0a5c:5842 (bus 1, device 2) path: 10
8086:15f0 (bus 3, device 0)
8086:15db (bus 2, device 0)
0bda:8153 (bus 3, device 4) path: 4.4
413c:b06f (bus 3, device 10) path: 2.3.5
046d:c077 (bus 3, device 13) path: 2.3.2.4
0bda:58fd (bus 1, device 3) path: 11
libusb: warning [libusb_exit] device 3.1 still referenced
libusb: warning [libusb_exit] device 3.3 still referenced
libusb: warning [libusb_exit] device 3.0 still referenced
```

Code for easy reference:
```
/*
 * libusb example program to list devices on the bus
 * Copyright © 2007 Daniel Drake <dsd@gentoo.org>
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General Public
 * License along with this library; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA
 */

#include <stdio.h>

#include "libusb.h"

static void print_devs(libusb_device **devs)
{
	libusb_device *dev;
	int i = 0, j = 0;
	uint8_t path[8]; 

	while ((dev = devs[i++]) != NULL) {
		struct libusb_device_descriptor desc;
		int r = libusb_get_device_descriptor(dev, &desc);
		if (r < 0) {
			fprintf(stderr, "failed to get device descriptor");
			return;
		}

		printf("%04x:%04x (bus %d, device %d)",
			desc.idVendor, desc.idProduct,
			libusb_get_bus_number(dev), libusb_get_device_address(dev));

		r = libusb_get_port_numbers(dev, path, sizeof(path));
		if (r > 0) {
			printf(" path: %d", path[0]);
			for (j = 1; j < r; j++)
				printf(".%d", path[j]);
		}
		printf("\n");
	}
}

int main(void)
{
	libusb_device **devs;
	libusb_device* refdev1, * refdev2;
	int r;
	ssize_t cnt;

	r = libusb_init(NULL);
	if (r < 0)
		return r;

	cnt = libusb_get_device_list(NULL, &devs);
    if (cnt < 1) {
		libusb_exit(NULL);
		return (int)cnt;
		
	}
	
	print_devs(devs);
	refdev1 = devs[0];
	libusb_ref_device(refdev1);
	libusb_free_device_list(devs, 1);

	cnt = libusb_get_device_list(NULL, &devs);
	if (cnt < 0){
		libusb_exit(NULL);
		return (int) cnt;
	}

	print_devs(devs);
	
	refdev2 = devs[0];
	libusb_ref_device(refdev2);
	libusb_free_device_list(devs, 1);

	libusb_unref_device(refdev1);
	libusb_unref_device(refdev2);

	libusb_exit(NULL);
	return 0;
}
```

mcuee, 2021-09-08:
Looks like there is a bug in libusb Windows backend.
https://github.com/vpelletier/python-libusb1/issues/77

lerdmann, 2022-01-02:
I think this problem is the same as in #1038

mcuee, 2022-02-05:
> I think this problem is the same as in #1038

I tend to think #1038 may not be the same since my issues are more with Windows.
* 1038

mcuee, 2022-04-01:
No issues under Linux (Ubuntu 20.04 x64).
```
mcuee@UbuntuSwift3:~/build/libusb-1.0.26-rc1$ ./examples/listdevs 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000102] [0000133e] libusb: debug [libusb_init] created default context
[ 0.000127] [0000133e] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.000145] [0000133e] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000156] [0000133e] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000161] [0000133e] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000178] [0000133e] libusb: debug [get_kernel_version] reported kernel version is 5.15.10-051510-generic
[ 0.000203] [0000133e] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000208] [0000133e] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000217] [0000133e] libusb: debug [op_init] sysfs is available
[ 0.000478] [0000134c] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.003503] [0000133e] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.003516] [0000133e] libusb: debug [linux_get_device_address] scan usb1
[ 0.003547] [0000133e] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.003551] [0000133e] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.003555] [0000133e] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.003648] [0000133e] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.003655] [0000133e] libusb: debug [linux_get_device_address] scan usb2
[ 0.003681] [0000133e] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.003688] [0000133e] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.003691] [0000133e] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.003777] [0000133e] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.003784] [0000133e] libusb: debug [linux_get_device_address] scan usb3
[ 0.003808] [0000133e] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.003814] [0000133e] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.003818] [0000133e] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.003905] [0000133e] libusb: debug [linux_get_device_address] getting address for device: 3-10 detached: 0
[ 0.003914] [0000133e] libusb: debug [linux_get_device_address] scan 3-10
[ 0.003941] [0000133e] libusb: debug [linux_get_device_address] bus=3 dev=5
[ 0.003946] [0000133e] libusb: debug [linux_enumerate_device] busnum 3 devaddr 5 session_id 773
[ 0.003951] [0000133e] libusb: debug [linux_enumerate_device] allocating new device for 3/5 (session 773)
[ 0.003982] [0000133e] libusb: debug [linux_get_parent_info] dev 0x55f634737a20 (3-10) has parent 0x55f634755d50 (usb3) port 10
[ 0.004052] [0000133e] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.004059] [0000133e] libusb: debug [linux_get_device_address] scan 3-3
[ 0.004083] [0000133e] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.004088] [0000133e] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.004094] [0000133e] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.004120] [0000133e] libusb: debug [linux_get_parent_info] dev 0x55f634757440 (3-3) has parent 0x55f634755d50 (usb3) port 3
[ 0.004192] [0000133e] libusb: debug [linux_get_device_address] getting address for device: 3-5 detached: 0
[ 0.004197] [0000133e] libusb: debug [linux_get_device_address] scan 3-5
[ 0.004223] [0000133e] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.004228] [0000133e] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.004234] [0000133e] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.004261] [0000133e] libusb: debug [linux_get_parent_info] dev 0x55f634757360 (3-5) has parent 0x55f634755d50 (usb3) port 5
[ 0.004333] [0000133e] libusb: debug [linux_get_device_address] getting address for device: 3-7 detached: 0
[ 0.004339] [0000133e] libusb: debug [linux_get_device_address] scan 3-7
[ 0.004364] [0000133e] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.004369] [0000133e] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.004375] [0000133e] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.004399] [0000133e] libusb: debug [linux_get_parent_info] dev 0x55f634738c30 (3-7) has parent 0x55f634755d50 (usb3) port 7
[ 0.004479] [0000133e] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.004488] [0000133e] libusb: debug [linux_get_device_address] scan usb4
[ 0.004512] [0000133e] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.004518] [0000133e] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[ 0.004523] [0000133e] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
[ 0.004618] [0000133e] libusb: debug [libusb_get_device_list]  
[ 0.004634] [0000133e] libusb: debug [libusb_get_device_descriptor]  
1d6b:0003 (bus 4, device 1)
[... cut ...]
[ 0.004913] [0000133e] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.004916] [0000133e] libusb: debug [libusb_unref_device] destroy device 3.5
[ 0.004919] [0000133e] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.004921] [0000133e] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.004924] [0000133e] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.004929] [0000133e] libusb: debug [usbi_remove_event_source] remove fd 4
[ 0.004936] [0000133e] libusb: debug [usbi_remove_event_source] remove fd 3

```

mcuee, 2022-04-01:
No issues either under Windows with libusb-1.0.26rc1.

```

PS C:\work\libusb\libusb_test\libusb-1.0.26-rc1> $Env:LIBUSB_DEBUG=4
PS C:\work\libusb\libusb_test\libusb-1.0.26-rc1> .\x64\Release\examples\listdevs.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000334] [00002f38] libusb: debug [libusb_init] created default context
[ 0.000521] [00002f38] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.000697] [00002f38] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000094 events 0
[ 0.000887] [00002f38] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001039] [00002f38] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000098 events 0
[ 0.001217] [00002f38] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.001382] [00002f38] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003123] [00002f38] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.004773] [00002f38] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.006486] [00002f38] libusb: info [windows_init] UsbDk backend is not available
[ 0.006606] [00002f38] libusb: debug [libusb_get_device_list]
[ 0.006955] [00003434] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.012940] [00002f38] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.013057] [00002f38] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.013467] [00002f38] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.013540] [00002f38] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.013568] [00002f38] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.013595] [00002f38] libusb: debug [winusb_get_device_list] allocating new device for session [1F]
[ 0.013732] [00002f38] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.013780] [00002f38] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.013810] [00002f38] libusb: debug [winusb_get_device_list] allocating new device for session [34]
[ 0.013989] [00002f38] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.014042] [00002f38] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.014077] [00002f38] libusb: debug [winusb_get_device_list] allocating new device for session [62]
[ 0.014172] [00002f38] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.014217] [00002f38] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.014246] [00002f38] libusb: debug [winusb_get_device_list] allocating new device for session [63]
[ 0.014422] [00002f38] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.014494] [00002f38] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.015048] [00002f38] libusb: debug [winusb_get_device_list] found existing device for session [34]
[ 0.015160] [00002f38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.015211] [00002f38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.015256] [00002f38] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&586B51A&0&10'
[ 0.015339] [00002f38] libusb: debug [winusb_get_device_list] extra GUID: {16126A05-3179-4589-9DB8-952AFFD297D5}
[ 0.015373] [00002f38] libusb: debug [winusb_get_device_list] found existing device for session [62]
[ 0.015428] [00002f38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.015475] [00002f38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 512 bytes)
[ 0.015512] [00002f38] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 5): 'USB\VID_04F2&PID_B6DD\0001'
[ 0.015581] [00002f38] libusb: debug [winusb_get_device_list] extra GUID: {20348E47-A897-468E-9D0E-56307802FC2A}
[ 0.015611] [00002f38] libusb: debug [winusb_get_device_list] found existing device for session [1F]
[ 0.015664] [00002f38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.015696] [00002f38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 46 bytes)
[ 0.015734] [00002f38] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 7): 'USB\VID_1C7A&PID_0575\077E2F9A'
[ 0.015961] [00002f38] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&12C539F&0&0' reports 16 ports
[ 0.016022] [00002f38] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&12C539F&0&0'
[ 0.016136] [00002f38] libusb: debug [winusb_get_device_list] found existing device for session [63]
[ 0.016190] [00002f38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.016225] [00002f38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 84 bytes)
[ 0.016261] [00002f38] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 3): 'USB\VID_046D&PID_C52B\5&586B51A&0&3'
[ 0.016438] [00002f38] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&1A63DC7C&0&0' reports 5 ports
[ 0.016477] [00002f38] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&1A63DC7C&0&0'
[... cut ...]
[ 0.019175] [00002f38] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.019204] [00002f38] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.019230] [00002f38] libusb: debug [libusb_exit] destroying default context
[ 0.019263] [00003434] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.019555] [00002f38] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000098
[ 0.019592] [00002f38] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000094

```

mcuee, 2022-04-01:
So I think I will close this one.

closed 2022-04-01 by mcuee