libusb/libusb #975: libusb_interrupt_transfer always timeout

fayfive, 2021-08-09:
libusb version 1.0.24
kernel linux-3.11
I write a mouse driver. I  can receive the endpoint data and it's correct. But libusb_interrupt_transfer is always timeout even I set the time 10s. The data sent to me at timeout and the actual length is not zero.  I don't know if this is normal phenomenon that lib_interrupt_transfer accept data periodically and controlled by myself?
Thanks!

mcuee, 2021-08-10:
Most likely you are not writing the codes correctly. Please post your codes and debug log (LIBUSB_DEBUG=4). Do you know the communication protocol of your mouse? And why do you want to use libusb for a mouse?

mcuee, 2021-08-13:
By the way, Github issues are more for issues reporting and not technical support. You may want to provide more details and ask in the libusb-devel mailing list.

closed 2021-08-13 by mcuee