libusb/libusb #976: libusb_bulk_transfer returns -7 on win10,64bit

Lelouchccc, 2021-08-11:
I use libusb_bulk_transfer in Visual studio2019, and compile the project to an test.exe, then send data by this to stm32.

Using of function libusb_bulk_transfer:
libusb_bulk_transfer(handle, endpoint_out, data, data_length, &size, 5000);
When i set the variable data_length to 128, libusb_bulk_transfer returns correct, and test.exe send data to mcu normally ；
When i set the variable data_length to 256 or 512(more than 150 bytes around), libusb_bulk_transfer returns -7, "Operation timed out".

My usb device info displayed in BUS HOUND:
Bus Type:        USB 2.0 mode
Bus Speed:       480 megabit (hi-speed)
Power Drawn:     100 milliamps @ 5.0 volts
Endpoint 0:      Type=CTL  Class=08 SubClass=06 Protocol=50 MaxPacket=8
Endpoint 1 OUT:  Type=BULK Class=08 SubClass=06 Protocol=50 MaxPacket=200
Endpoint 1 IN:   Type=BULK Class=08 SubClass=06 Protocol=50 MaxPacket=200

Can anyone give me some suggestions about the problem?

mcuee, 2021-08-12:
Fix your FW if you develop the FW by yourself.

mcuee, 2021-08-12:
You can also post the debug log by setting environment variable LIBUSB_DEBUG=4. 

And if this is a real USB mass storage device, maybe you want to try out xusb first.
https://github.com/libusb/libusb/blob/master/examples/xusb.c#L316

BTW, it is not good to start with a USB mass storage device with libusb. Use a proper custom device class if you want to learn to use libusb.

mcuee, 2021-08-12:
libusb FAQ:
https://github.com/libusb/libusb/wiki/FAQ#Can_I_use_libusb_to_open_a_file_on_an_USB_storage_device

Lelouchccc, 2021-08-12:
Thanks for your help.

I did use the example code from xusb.c, and after send readcap cmd, mcu returned result shows a block size is 512 bytes, so it should be okay to send 512 bytes data.

I am adding log in dll to see what's wrong, and i found that when executing **WaitForMultipleObjects** in function usbi_wait_for_events, result is 1 but not 0.Then in **handle_timer_trigger** function, status is set to TIMED_OUT.

Does this mean my usb device didn't response normally, so WaitForMultipleObjects got no signal for handle0?

(PS:I didn't find how to print usbi_dbg log, so i added print log to debug, print log and source code are attached.)
[out.txt](https://github.com/libusb/libusb/files/6974076/out.txt)
[libusb-1.0.24_debug_modi_part.zip](https://github.com/libusb/libusb/files/6974102/libusb-1.0.24_debug_modi_part.zip)

mcuee, 2021-08-12:
You set environment variable LIBUSB_DEBUG=4 and then run your application to get the debug log.

mcuee, 2021-08-12:
In any case, github is only for issue reporting and not for technical support. You may want to post the questions to libusb-devel mailing list for support.

closed 2021-08-12 by mcuee