libusb/libusb #977: Libusb missing Byte on reception

TH3R3ch1, 2021-08-16:
Hi, so I'm trying to Send/receive Data to an HID Device. This uses an vendor specific request as I'm (to some extend) in
control of the FW of the device. This is running on Win10 using VSCode. 
Below you can find the function calls, the debug Logs, the Data on the Bus (using HW sniffer) as well as the return value 
and Buffer contents after the command. the first Call (transmitting the Read Command) works fine, but on the following 
2 (receiving the requested Data) somewhere the last Byte is lost. I don't have that much experience using libusb, so maybe
some of you can spot that misstake ?

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
libusb_control_transfer(workinghandle, 0x21, 0x09, 0x0200, 0x0000, bufIn, 32, 10000);
[20.034603] [00000ecc] libusb: debug [libusb_alloc_transfer] transfer 0101C130
[20.034629] [00000ecc] libusb: debug [libusb_submit_transfer] transfer 0101C130
[20.034642] [00000ecc] libusb: debug [add_to_flying_list] arm timer for timeout in 10000ms (first in line)
[20.034668] [00000ecc] libusb: debug [libusb_claim_interface] interface 0
[20.034686] [00000ecc] libusb: debug [hid_claim_interface] claimed interface 0
[20.034712] [00000ecc] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[20.034725] [00000ecc] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 02 to interface 0
[20.034742] [00000ecc] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[20.034753] [00000ecc] libusb: debug [hid_submit_control_transfer] will use interface 0
[20.034769] [00000ecc] libusb: debug [_hid_set_report] report ID: 0x00
[20.035104] [00000ecc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[20.035135] [00000ecc] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[20.035266] [000006b0] libusb: debug [windows_iocp_thread] transfer 0101C130 completed, length 32
[20.035303] [00000ecc] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[20.035322] [00000ecc] libusb: debug [handle_event_trigger] event triggered
[20.035362] [00000ecc] libusb: debug [windows_handle_transfer_completion] handling transfer 0101C130 completion with errcode 0, length 32
[20.035388] [00000ecc] libusb: debug [libusb_release_interface] interface 0
[20.035401] [00000ecc] libusb: debug [auto_release] auto-released interface 0
[20.035413] [00000ecc] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[20.035439] [00000ecc] libusb: debug [usbi_handle_transfer_completion] transfer 0101C130 has callback 7B487910
[20.035455] [00000ecc] libusb: debug [sync_transfer_cb] actual_length=32
[20.035477] [00000ecc] libusb: debug [libusb_free_transfer] transfer 0101C130
BUS:
	Setup:	21 09 00 02 00 00 20 00
	Write:	00 05 D0 84 00 00 10 CC CC CC CC CC CC CC CC CC
			CC CC CC CC CC CC CC CC CC CC CC CC CC CC CC CC
Software:
	Length reported: 32
	Buffer: 00 05 D0 84 00 00 10 CC CC CC CC CC CC CC CC CC
			CC CC CC CC CC CC CC CC CC CC CC CC CC CC CC CC

libusb_control_transfer(workinghandle, 0xA1, 0x01, 0x0100, 0x0000, bufIn, 32, 10000);
[20.640819] [00000ecc] libusb: debug [libusb_alloc_transfer] transfer 01023D88
[20.640890] [00000ecc] libusb: debug [libusb_submit_transfer] transfer 01023D88
[20.640919] [00000ecc] libusb: debug [add_to_flying_list] arm timer for timeout in 10000ms (first in line)
[20.640977] [00000ecc] libusb: debug [libusb_claim_interface] interface 0
[20.641025] [00000ecc] libusb: debug [hid_claim_interface] claimed interface 0
[20.641086] [00000ecc] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[20.641117] [00000ecc] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 02 to interface 0
[20.641149] [00000ecc] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[20.641194] [00000ecc] libusb: debug [hid_submit_control_transfer] will use interface 0
[20.641225] [00000ecc] libusb: debug [_hid_get_report] report ID: 0x00
[20.641619] [00000ecc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[20.641678] [00000ecc] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[20.641626] [000006b0] libusb: debug [windows_iocp_thread] transfer 01023D88 completed, length 32
[20.641808] [00000ecc] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[20.641862] [00000ecc] libusb: debug [handle_event_trigger] event triggered
[20.641887] [00000ecc] libusb: debug [windows_handle_transfer_completion] handling transfer 01023D88 completion with errcode 0, length 32
[20.641928] [00000ecc] libusb: debug [libusb_release_interface] interface 0
[20.641955] [00000ecc] libusb: debug [auto_release] auto-released interface 0
[20.641976] [00000ecc] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[20.642046] [00000ecc] libusb: debug [usbi_handle_transfer_completion] transfer 01023D88 has callback 7B487910
[20.642078] [00000ecc] libusb: debug [sync_transfer_cb] actual_length=31
[20.642143] [00000ecc] libusb: debug [libusb_free_transfer] transfer 01023D88
BUS:
	Setup:	A1 01 00 01 00 00 20 00
	Read:	00 22 05 05 03 04 07 07 03 04 05 05 03 04 07 07
			03 04 07 07 03 04 05 05 03 04 07 07 03 04 05 05
Software:			
	Length reported: 31
	Buffer:	00 22 05 05 03 04 07 07 03 04 05 05 03 04 07 07
			03 04 07 07 03 04 05 05 03 04 07 07 03 04 05 CC

libusb_control_transfer(workinghandle, 0xA1, 0x01, 0x0100, 0x0000, bufIn, 32, 10000);
[21.258102] [00000ecc] libusb: debug [libusb_alloc_transfer] transfer 01023D88
[21.258130] [00000ecc] libusb: debug [libusb_submit_transfer] transfer 01023D88
[21.258141] [00000ecc] libusb: debug [add_to_flying_list] arm timer for timeout in 10000ms (first in line)
[21.258155] [00000ecc] libusb: debug [libusb_claim_interface] interface 0
[21.258164] [00000ecc] libusb: debug [hid_claim_interface] claimed interface 0
[21.258178] [00000ecc] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[21.258184] [00000ecc] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 02 to interface 0
[21.258192] [00000ecc] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[21.258197] [00000ecc] libusb: debug [hid_submit_control_transfer] will use interface 0
[21.258203] [00000ecc] libusb: debug [_hid_get_report] report ID: 0x00
[21.258550] [00000ecc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[21.258621] [00000ecc] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[21.258650] [000006b0] libusb: debug [windows_iocp_thread] transfer 01023D88 completed, length 32
[21.258674] [00000ecc] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[21.258695] [00000ecc] libusb: debug [handle_event_trigger] event triggered
[21.258767] [00000ecc] libusb: debug [windows_handle_transfer_completion] handling transfer 01023D88 completion with errcode 0, length 32
[21.258794] [00000ecc] libusb: debug [libusb_release_interface] interface 0
[21.258813] [00000ecc] libusb: debug [auto_release] auto-released interface 0
[21.258829] [00000ecc] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[21.258909] [00000ecc] libusb: debug [usbi_handle_transfer_completion] transfer 01023D88 has callback 7B487910
[21.258928] [00000ecc] libusb: debug [sync_transfer_cb] actual_length=31
[21.258949] [00000ecc] libusb: debug [libusb_free_transfer] transfer 01023D88
BUS:
	Setup:	A1 01 00 01 00 00 20 00
	Read:	00 22 03 04 90 00 00 00 00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00		
Software:
	Length reported: 31
	Buffer:	00 22 03 04 90 00 00 00 00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 CC

mcuee, 2021-08-16:
Is this the same as https://github.com/libusb/libusb/issues/902?

mcuee, 2021-08-16:
Recommendation is to use HIDAPI and not libusb for HID device.
You need to use latest HIDAPI git head. It has the fixes for https://github.com/libusb/hidapi/pull/286 and https://github.com/libusb/hidapi/pull/232.

TH3R3ch1, 2021-08-17:
Ok, so I tried to use HIDAPI, but that seems to be not possible, as I didn't find a way to send vendor requests. I also tried implementing my own sync_control_transfer (with some modifications), thou without any big gains. What I find interesting is:
[windows_handle_transfer_completion] reports the correct amount of 32 Bytes, independent of the length given by me. I think that at some point there are 32 Bytes received, but something discards one Byte ?

mcuee, 2021-08-17:
For HID device, usually you use Feature Report for vendor specific requests.

As mentioned in #902, there may be a bug in libusb Windows HID backend. Can you try the suggestion mentioned there?
https://github.com/libusb/libusb/issues/902#issuecomment-810726897

And I can tell you the HID backend will not be fixed any time soon unless someone comes out with a fix.

TH3R3ch1, 2021-08-17:
So I got it to work using HIDAPI (was using it the wrong way). Regarding (https://github.com/libusb/libusb/issues/902#issuecomment-810726897) :
I used Report ID 0 so it should use the given function, I haven't checked it, but it seems to be correct (at least in my case).

mcuee, 2021-08-18:
> So I got it to work using HIDAPI (was using it the wrong way). Regarding ([#902 (comment)](https://github.com/libusb/libusb/issues/902#issuecomment-810726897)) :
> I used Report ID 0 so it should use the given function, I haven't checked it, but it seems to be correct (at least in my case).

Glad to hear that you have got HIDAPI working.

As for libusb HID backend, what if you change the following lines as per the comment?
https://github.com/libusb/libusb/blob/c6a35c56016ea2ab2f19115d2ea1e85e0edae155/libusb/os/windows_winusb.c#L4111-L4117

```
				if (transfer_priv->hid_buffer[0] == 0) {
					// Discard the 1 byte report ID prefix
					// length--; //remove this line to see if that helps
					memcpy(transfer_priv->hid_dest, transfer_priv->hid_buffer + 1, length);
				} else {
					memcpy(transfer_priv->hid_dest, transfer_priv->hid_buffer, length);
				}
```

mcuee, 2021-08-20:
Before the issue is fixed, one potential workaround is to use WinUSB driver to replace the HID driver, as mentioned in #902.

mcuee, 2021-08-23:
@TH3R3ch1 Just wondering if you can try out the above changes for the libusb HID backend. 
If not, that is perfectly fine as well. I will label this as duplicate of #902 in that case.

TH3R3ch1, 2021-08-23:
@mcuee In the next few weeks sadly not, I might come back later and investigate this further (in about 4 weeks ), I currently just can't invest time here.

commit f7624bf0 (2021-09-23): Fix Windows HID backend missing byte

When the report id is zero, we should indeed skip the first byte of
transfer_priv->hid_buffer, since it is not part of the actual report.
But the report size should not change.

Thanks to Jonas Malaco who has done the investigation in this issue.
Reference: Jonas Malaco's comments
https://github.com/libusb/libusb/issues/902#issuecomment-810726897

Fixes #902
Fixes #977
Closes #986
closed 2021-09-23 by tormodvolden