libusb/libusb #982: libusb_set_interface_alt_setting(...) crashes in 1.0.24 in Win10.

carmangary, 2021-08-22:
The same call in version 1.0.23 does not crash.  Here is a code snippet of the crashing code.

    r = libusb_open(this->usbdevice, &this->devh_ctl);
    if (r != 0) return;

    r = libusb_open(this->usbdevice, &this->devh_data);
    if (r != 0) return;

    r = libusb_claim_interface(this->devh_ctl, 0);
    if (r != 0) return;

    r = libusb_claim_interface(this->devh_data, 1);
    if (r != 0) return;

    r = libusb_set_interface_alt_setting(this->devh_data, 1, 1);  // CRASHES HERE
    if (r != 0) return;

mcuee, 2021-08-22:
Which driver are you using? Please use the lates git head and try again. I believe it shuld work. If not, post the debug log by setting the env variable LIBUSB_DEBUG=4.

carmangary, 2021-08-22:
I use vcpkg and compile from the source it provides.  I will check to see if they are current or not.  Does someone from this team update the vcpkg ports?  Or is that done by some other volunteer?  

It will take me a little time to provide more information because switching versions with vcpkg is a huge pain.

mcuee, 2021-08-23:
We do not maintain vcpkg or any other third party distribution on any OS (eg: MSYS2 MinGW-w64, Homebrew for macOS, various Linux distros). 

Even the Windows binary provided by libusb project is for convenience and may not guarantee to work because of different compiler versions.

mcuee, 2021-08-23:
vcpkg seems to only support release version from the formula, unlike Homebrew on macOS.
https://github.com/microsoft/vcpkg/tree/master/ports/libusb

mcuee, 2021-08-23:
But you should be able to use plain VS project here to add libusb (dll or static lib) into your project.
https://github.com/libusb/libusb/tree/master/msvc

mcuee, 2021-08-23:
@carmangary In the mean time, even with 1.0.24, can you tell which driver you are using? And then post the debug log. There are a few issues with libusbk.sys and libusb0.sys driver which have been fixed in git head.

Ref: 
https://github.com/libusb/libusb/issues/954 (libusb0.sys)
https://github.com/libusb/libusb/issues/912 (libusbk.sys)
https://github.com/libusb/libusb/issues/844 (libusbk.sys)

carmangary, 2021-08-23:
I don't actually know what driver you are referring to.  I am building a device controller library that uses libusb.  I don't know the details of the device controller library other than how you use its API.  If you can provide more info about what you are looking for I will do my best to get what you ask for.

mcuee, 2021-08-23:
You have to know the driver used to use libusb under Windows.
https://github.com/libusb/libusb/wiki/Windows#Driver_Installation

mcuee, 2021-08-23:
The device descriptor information will be good info as well. You can use this.
https://www.uwe-sieber.de/usbtreeview_e.html

mcuee, 2021-08-23:
BTW, is the follwoing the same device, just two interfaces? Then you should open it only once.
```
r = libusb_open(this->usbdevice, &this->devh_ctl);
if (r != 0) return;

r = libusb_open(this->usbdevice, &this->devh_data);
if (r != 0) return;
```

tormodvolden, 2021-08-29:
What does the stack trace say?

mcuee, 2022-04-01:
I am going to close this as there is no update from the OP for long time. Please reopen if you have more info.

closed 2022-04-01 by mcuee