libusb/libusb #983: libusb_get_string_descriptor_ascii(...) crashes in 1.0.24 in Win10 for custom ftdi device

SamsonBox, 2021-08-23:
I have a ftdi device with customized vendor- & product-Id and manufacturer string. When I run the testlibusb.c the call 
`ret = libusb_get_string_descriptor_ascii(handle, desc.iManufacturer, string, sizeof(string));`
crashes when the ftdi device is hit. I got the following crash:
`Unhandled exception at 0x00007FF65CE78BD4 in test_libusb.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.
`
When I look in the disassembly I see the following line (the crash occurs in the last line):
usbi_signal_transfer_completion:
00007FF65CE78BB0  push        rbx  
00007FF65CE78BB2  sub         rsp,30h  
00007FF65CE78BB6  mov         rax,qword ptr [rcx+78h]  
00007FF65CE78BBA  mov         rbx,rcx  
00007FF65CE78BBD  test        rax,rax  
00007FF65CE78BC0  je          usbi_signal_transfer_completion+0B7h (07FF65CE78C67h)  
00007FF65CE78BC6  mov         rax,qword ptr [rax+40h]  
00007FF65CE78BCA  mov         qword ptr [rsp+40h],rsi  
00007FF65CE78BCF  mov         qword ptr [rsp+48h],rdi  
00007FF65CE78BD4  mov         rdi,qword ptr [rax+30h]  

The same code in version 1.0.23 does not crash. Are there some limitations on the manufacturer string?

Thank you in advance

mcuee, 2021-08-25:
Please provide more details.
1) which driver are you using?
2) please provide the debug log by running your program with environmental variable LIBUSB_DEBUG=4
3) please try the latest git head. It has quite some issues fixed with regard to Windows.

mcuee, 2021-08-25:
Here is my quick test for my FT2232H device.
git head is okay with WinUSB and libusbk driver
1.024 is okay with WinUSB but not libusbk driver
1.0.23 is okay with WinUSB and libusbk driver.

Do not use libusb-win32 driver. It is not working with libusb-1.0.24 and git. It is a known issue: https://github.com/libusb/libusb/issues/761

SamsonBox, 2021-08-25:
@mcuee 
I use libusbk. We can not use WinUSB as we need to open a device twice (with different intefaces). I will check if the git HEAD is working for me.

SamsonBox, 2021-08-25:
@mcuee 
You are right. The latest git HEAD is working. Thank you! I will close the issue.

closed 2021-08-25 by SamsonBox

mcuee, 2021-11-02:
#1016 should be able to fix this issue for libusb0.sys.