libusb/libusb #987: usbdk backend usage -- how to improve

mcuee, 2021-08-31:
Right now existing applications need to be modified in order to use Windows usbdk backend.
https://github.com/libusb/libusb/wiki/Windows#driver-installation
```
libusb_context * ctx = NULL;
libusb_init(&ctx);
libusb_set_option(ctx, LIBUSB_OPTION_USE_USBDK);
```

Back in 2018 when I was asking in the openocd-devel to see if it is possible to add the support of usbdk and the following was the answer.
On Tue, Jun 12, 2018 at 12:15 AM, Paul Fertser wrote:
> Hello Xiaofan,
>
> Thank you for the information.
>
> On Mon, Jun 11, 2018 at 09:47:24PM +0800, Xiaofan Chen wrote:
>> https://github.com/libusb/libusb/wiki/Windows#Driver_Installation
>
> IMHO it would be so much nicer if libusb included autodetection for
> that... It seems like a layering violation to offload the decision
> regarding selected backend to the application. In what case can it be
> useful?
>

mcuee, 2021-08-31:
On the other hand, sometimes it may not be so simple and  more code changes are necessary.
Example in pyusb.
https://github.com/pyusb/pyusb/issues/209

mcuee, 2021-08-31:
Take note document has been updated.
https://github.com/libusb/libusb/commit/c6a68829828a69b44dc0646defbf440bc7b0c5d1

mcuee, 2021-08-31:
So far I can see the major projects are not really adopting usbdk backend (eg: libftdi and openocd). On one hand, I think this is probably a good thing as usbdk may create some issues for some Windows systems. On he other hand, it does mean that the users have to modify the source codes and build the projects when they want to use usbdk.

tormodvolden, 2021-09-08:
So it used to be a compile-time option and now is a run-time option. The joke is that unless the application has some mechanism for getting the preference from the user, users still have to recompile the application if they want to change it.

And if the application is going to offer some mechanism, then it should ideally be some uniform way to do it, and then maybe libusb should offer this mechanism instead*. Which basically means an environment variable, loved by many, hated by some.

*) if no other application changes are needed, this I don't know.

mcuee, 2021-09-09:
Actualy the idea of environment variable is not bad. You can see we use both LIBUSB_DEBUG env varibale and libusb_set_debug (before using libusb_set_option).

Maybe it is a good idea to offer environmental variable for the options inside libusb_option.
1) LIBUSB_OPTION_LOG_LEVEL -- already have, LIBUSB_DEBUG
2) LIBUSB_OPTION_USE_USBDK -- new env variable, something like LIBUSB_USBDK: =1 (use usbdk), =0 (default, not using usbdk)
3)LIBUSB_OPTION_WEAK_AUTHORITY (deprecated) -- new env variable, something like LIBUSB_WEAK_AUTHORITY : =1 (use weak authority), =0 (default, not using the option)

4)LIBUSB_OPTION_NO_ENUMERATION -- new env variable, something like LIBUSB_NO_DEVICE_DISCOVERY: =1 (with no enumeration), =0 (default, with enumeration)

mcuee, 2021-09-09:
By right, no application change is needed, if the device is supported by usbdk (not all devices are supported by usbdk).

tormodvolden, 2021-09-09:
Since LIBUSB_OPTION_WEAK_AUTHORITY is deprecated (it is basically renamed to LIBUSB_OPTION_NO_ENUMERATION) we wouldn't need to add an environment variable for it.

mcuee, 2023-02-17:
Comments about usbdk backend here.
* https://github.com/libusb/libusb/issues/250#issuecomment-1434568980

> Dolphin has been using 1.0.26 since April 2022 but I think the majority of users are not using the usbdk backend. Sadly we haven't been able to recommend it for a while because of general instability issues and even reports of blue screens iirc.

mcuee, 2023-02-17:
Quick testing of the usbdk backend in git HEAD.

```
PS C:\work\libusb\libusb_usbdk> git diff
diff --git a/examples/listdevs.c b/examples/listdevs.c
index b5a9494c..6c70006d 100644
--- a/examples/listdevs.c
+++ b/examples/listdevs.c
@@ -58,7 +58,7 @@ int main(void)
        r = libusb_init_context(/*ctx=*/NULL, /*options=*/NULL, /*num_options=*/0);
        if (r < 0)
                return r;
-
+       libusb_set_option(NULL, LIBUSB_OPTION_USE_USBDK);
        cnt = libusb_get_device_list(NULL, &devs);
        if (cnt < 0){
                libusb_exit(NULL);
diff --git a/examples/testlibusb.c b/examples/testlibusb.c
index 394cec52..58f53898 100644
--- a/examples/testlibusb.c
+++ b/examples/testlibusb.c
@@ -290,7 +290,7 @@ int main(int argc, char *argv[])
        r = libusb_init_context(/*ctx=*/NULL, /*options=*/NULL, /*num_options=*/0);
        if (r < 0)
                return r;
-
+       libusb_set_option(NULL, LIBUSB_OPTION_USE_USBDK);
        if (device_name) {
                r = test_wrapped_device(device_name);
        } else {
diff --git a/examples/xusb.c b/examples/xusb.c
index 441af017..e2a9a52b 100644
--- a/examples/xusb.c
+++ b/examples/xusb.c
@@ -1172,7 +1172,7 @@ int main(int argc, char** argv)
        r = libusb_init_context(/*ctx=*/NULL, /*options=*/NULL, /*num_options=*/0);
        if (r < 0)
                return r;
-
+       libusb_set_option(NULL, LIBUSB_OPTION_USE_USBDK);
        // If not set externally, and no debug option was given, use info log level
        if ((old_dbg_str == NULL) && (!debug_mode))
                libusb_set_option(NULL, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_INFO);

PS C:\work\libusb\libusb_usbdk> .\examples\xusb.exe 0403:6001
Using libusb v1.0.26.11784

Opening device 0403:6001...
  Failed.
PS C:\work\libusb\libusb_usbdk> .\examples\xusb.exe 0403:6001
Using libusb v1.0.26.11784

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "US232R"
   String (0x03): "FTDCRPIE"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...

PS C:\work\libusb\libusb_usbdk> .\examples\listdevs.exe
046d:c52b (bus 2, device 4) path: 3
04f2:b6dd (bus 2, device 6) path: 5
1c7a:0575 (bus 2, device 8) path: 7
8087:0026 (bus 2, device 11) path: 10
0403:6001 (bus 2, device 2) path: 1

```

mcuee, 2023-02-17:
And running `testlibusb.exe` caused BSOD on my Windows 11 laptop. 

So I guess the instability of usbdk is real and we may not be able to recommend usbdk backend as of now.

mcuee, 2023-02-17:
Output from Blue Screen Viewer from the mini-dump.
https://www.nirsoft.net/utils/blue_screen_view.html

```
==================================================
Dump File         : 021723-14218-01.dmp
Crash Time        : 17/2/2023 9:02:55 pm
Bug Check String  : 
Bug Check Code    : 0x0000010d
Parameter 1       : 00000000`0000000d
Parameter 2       : ffffc381`c3782d30
Parameter 3       : ffffc381`c59a4010
Parameter 4       : ffffc381`c2cc4670
Caused By Driver  : Wdf01000.sys
Caused By Address : Wdf01000.sys+45b50
File Description  : Kernel Mode Driver Framework Runtime
Product Name      : Microsoft® Windows® Operating System
Company           : Microsoft Corporation
File Version      : 1.33.22621.1 (WinBuild.160101.0800)
Processor         : x64
Crash Address     : ntoskrnl.exe+428c50
Stack Address 1   : 
Stack Address 2   : 
Stack Address 3   : 
Computer Name     : 
Full Path         : C:\WINDOWS\Minidump\021723-14218-01.dmp
Processors Count  : 8
Major Version     : 15
Minor Version     : 22621
Dump File Size    : 2,104,844
Dump File Time    : 17/2/2023 9:03:25 pm
==================================================
```

tormodvolden, 2023-02-17:
Wow. And since we haven't gotten any complaints, I guess it means very few people have been trying it.

Are you in your tests using usbdk as a filter driver or device driver (if that question makes sense)?

mcuee, 2023-02-17:
> Wow. And since we haven't gotten any complaints, I guess it means very few people have been trying it.
> 
> Are you in your tests using usbdk as a filter driver or device driver (if that question makes sense)?

There is no such thing as "device driver mode" for usbdk. It works like a filter driver only. You can see it is attached to 5 USB devices in my system.

```
PS C:\Program Files\UsbDk Runtime Library> .\UsbDkController.exe -n
Enumerate USB devices
Found 5 USB devices:
0. FilterID: 2, Port: 3, ID: 046d:c52b, Configs: 1, Speed: 2
        USB\VID_046D&PID_C52B 3
Descriptor for configuration #0: size 84
1. FilterID: 2, Port: 5, ID: 04f2:b6dd, Configs: 1, Speed: 3
        USB\VID_04F2&PID_B6DD 0001
Descriptor for configuration #0: size 512
2. FilterID: 2, Port: 1, ID: 0403:6001, Configs: 1, Speed: 2
        USB\VID_0403&PID_6001 FTDCRPIE
Descriptor for configuration #0: size 32
3. FilterID: 2, Port: 7, ID: 1c7a:0575, Configs: 1, Speed: 3
        USB\VID_1C7A&PID_0575 077E2F9A
Descriptor for configuration #0: size 46
4. FilterID: 2, Port: 10, ID: 8087:0026, Configs: 1, Speed: 2
        USB\VID_8087&PID_0026 10
Descriptor for configuration #0: size 200

```

mcuee, 2023-02-19:
Interestingly if I build libusb-compat-0.1 with the following patch to use libusb usbdk backend, the example testlibusb application does not cause BSOD.

I have renamed the examples from libusb-compat-0.1 as `lsusb0` and `testlibusb0` in the below run log.

```
PS C:\work\libusb\libusb-compat_usbdk> git diff
diff --git a/libusb/core.c b/libusb/core.c
index 3bc9c25..80ec01f 100644
--- a/libusb/core.c
+++ b/libusb/core.c
@@ -175,12 +175,12 @@ API_EXPORTED void usb_init(void)
        usbi_dbg("");

        if (!ctx) {
-               r = libusb_init(&ctx);
+               r = libusb_init(&ctx, NULL, NULL);
                if (r < 0) {
                        usbi_err("initialization failed!");
                        return;
                }
-
+       libusb_set_option(&ctx, LIBUSB_OPTION_USE_USBDK);
                /* usb_set_debug can be called before usb_init */
                if (usb_debug)
                        libusb_set_debug(ctx, 3);

PS C:\work\libusb\libusb_bin\libusb_bin_usbdk> .\lsusb0.exe
libusb: debug [windows_set_option] switching context 00007ff8fc0de030 to use UsbDk backend
05e3:0612
8086:9a13
0403:6001
8087:0026
04f2:b6dd
1c7a:0575
8086:a0ed
05e3:0610
046d:c52b

PS C:\work\libusb\libusb_bin\libusb_bin_usbdk> .\testlibusb0.exe
libusb: debug [windows_set_option] switching context 00007ff8fc0de030 to use UsbDk backend
Dev #1: 05E3 - 0612
Dev #0: 8086 - 9A13
Dev #7: 0403 - 6001
Dev #5: 8087 - 0026
Dev #3: 04F2 - B6DD
Dev #4: 1C7A - 0575
Dev #0: 8086 - A0ED
Dev #1: 05E3 - 0610
Dev #2: Logitech - USB Receiver
```

The above seems to show more devices than libusb-1.0 `listdev` application. And libusb-1.0 `testlibusb` application will 100% cause BSOD in my system.

```
PS C:\Program Files\UsbDk Runtime Library> .\UsbDkController.exe -n
Enumerate USB devices
Found 7 USB devices:
0. FilterID: 1, Port: 2, ID: 05e3:0612, Configs: 1, Speed: 4
        USB\VID_05E3&PID_0612 2
Descriptor for configuration #0: size 31
1. FilterID: 2, Port: 2, ID: 05e3:0610, Configs: 1, Speed: 3
        USB\VID_05E3&PID_0610 2
Descriptor for configuration #0: size 41
2. FilterID: 2, Port: 3, ID: 046d:c52b, Configs: 1, Speed: 2
        USB\VID_046D&PID_C52B 3
Descriptor for configuration #0: size 84
3. FilterID: 2, Port: 5, ID: 04f2:b6dd, Configs: 1, Speed: 3
        USB\VID_04F2&PID_B6DD 0001
Descriptor for configuration #0: size 512
4. FilterID: 2, Port: 7, ID: 1c7a:0575, Configs: 1, Speed: 3
        USB\VID_1C7A&PID_0575 077E2F9A
Descriptor for configuration #0: size 46
5. FilterID: 2, Port: 10, ID: 8087:0026, Configs: 1, Speed: 2
        USB\VID_8087&PID_0026 10
Descriptor for configuration #0: size 200
6. FilterID: 4, Port: 2, ID: 0403:6001, Configs: 1, Speed: 2
        USB\VID_0403&PID_6001 A50285BI
Descriptor for configuration #0: size 32

PS C:\work\libusb\libusb_bin\libusb_bin_usbdk> .\listdevs.exe
05e3:0612 (bus 1, device 3) path: 2
05e3:0610 (bus 2, device 3) path: 2
046d:c52b (bus 2, device 4) path: 3
04f2:b6dd (bus 2, device 6) path: 5
1c7a:0575 (bus 2, device 8) path: 7
8087:0026 (bus 2, device 11) path: 10
0403:6001 (bus 4, device 3) path: 2

```

mcuee, 2024-04-30:
usbdk seems to casue quite some stability issues and it has not been updated for quite a while. I have updated libusb Wiki to discourage the use of usbdk.
https://github.com/libusb/libusb/wiki/Windows#user-content-Driver_Installation
https://github.com/libusb/libusb/wiki/FAQ#user-content-What_about_usbdk

mcuee, 2026-04-17:
My proposal is to deprecate usbdk backend after 1.0.30 or 1.0.31 release and remove usbdk backend completely.
* https://github.com/libusb/libusb/issues/1804