libusb/libusb #988: libusb-compat-0.1 devices still referenced upon exit under Windows

mcuee, 2021-09-01:
From here:
https://github.com/libusb/libusb/issues/974

Even the example lsusb.c and testlibusb.c are having the same behavior.

Super simple test application (subset of lsusb.c)

```
#include <usb.h>
#include <stdio.h>

int main( void )
{
	struct usb_bus *busses;
	struct usb_bus *bus;
    usb_set_debug(255);
    usb_init();
    usb_find_busses();
    usb_find_devices();

	return 0;
}

```

Debug log:
```
(py39venv) C:\work\libusb\tests> .\libusb_compat_leak1.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000349] [00005fe8] libusb: debug [libusb_init] libusb v1.0.24.11650
[ 0.000513] [00005fe8] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000094 events 0
[ 0.000673] [00005fe8] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000810] [00005fe8] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000098 events 0
[ 0.000965] [00005fe8] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.001110] [00005fe8] libusb: debug [htab_create] using 1021 entries hash table
[ 0.005843] [00005fe8] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.011122] [00005fe8] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.019561] [00005fe8] libusb: debug [windows_init] UsbDk backend is available
[ 0.019931] [00005fe8] libusb: debug [libusb_get_device_list]
[ 0.021071] [00000a2c] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.041610] [00005fe8] libusb: debug [winusb_get_device_list] allocating new device for session [93]
[ 0.041865] [00005fe8] libusb: debug [winusb_get_device_list] allocating new device for session [94]
[ 0.042322] [00005fe8] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.042565] [00005fe8] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.042730] [00005fe8] libusb: debug [winusb_get_device_list] allocating new device for session [10]
[ 0.043134] [00005fe8] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.043338] [00005fe8] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.043493] [00005fe8] libusb: debug [winusb_get_device_list] allocating new device for session [1E]
[ 0.044204] [00005fe8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.044406] [00005fe8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.044571] [00005fe8] libusb: debug [winusb_get_device_list] allocating new device for session [71]
[ 0.044904] [00005fe8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.045104] [00005fe8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.045261] [00005fe8] libusb: debug [winusb_get_device_list] allocating new device for session [72]
[ 0.045648] [00005fe8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.045840] [00005fe8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.045977] [00005fe8] libusb: debug [winusb_get_device_list] allocating new device for session [73]
[ 0.046504] [00005fe8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.046686] [00005fe8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.046814] [00005fe8] libusb: debug [winusb_get_device_list] allocating new device for session [74]
[ 0.047187] [00005fe8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A36D&SUBSYS_091A1028&REV_10\3&11583659&0&A0' bus number 1
[ 0.047356] [00005fe8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15DB&SUBSYS_091A1028&REV_02\71D9F51854B3020000' bus number 2
[ 0.049085] [00005fe8] libusb: debug [winusb_get_device_list] extra GUID: {865C4DDE-10B6-6373-FAEE-1506B321A619}
[ 0.049236] [00005fe8] libusb: debug [winusb_get_device_list] found existing device for session [71]
[ 0.049505] [00005fe8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.049662] [00005fe8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.049821] [00005fe8] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 3): 'USB\VID_046D&PID_C534\5&E9F3E45&0&3'
[ 0.050132] [00005fe8] libusb: debug [winusb_get_device_list] found existing device for session [10]
[ 0.050408] [00005fe8] libusb: debug [init_device] found 1 configurations (current config: 1)
[... 162 lines cut ...]
[ 0.104452] [00005fe8] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.104501] [00005fe8] libusb: debug [parse_endpoint] skipping descriptor 0x30
[ 0.104550] [00005fe8] libusb: debug [libusb_get_device_descriptor]
[ 0.104599] [00005fe8] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.104650] [00005fe8] libusb: debug [parse_endpoint] skipping descriptor 0x25
[ 0.104698] [00005fe8] libusb: debug [parse_endpoint] skipping descriptor 0x25
[ 0.104757] [00005fe8] libusb: debug [libusb_get_device_descriptor]
[ 0.104807] [00005fe8] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.104864] [00005fe8] libusb: debug [libusb_get_device_descriptor]
[ 0.104916] [00005fe8] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.104970] [00005fe8] libusb: debug [libusb_get_device_descriptor]
[ 0.105019] [00005fe8] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.105068] [00005fe8] libusb: debug [parse_endpoint] skipping descriptor 0x30
[ 0.105551] [00005fe8] libusb: debug [libusb_exit]
[ 0.105627] [00005fe8] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000098
[ 0.105684] [00005fe8] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000094
[ 0.105738] [00005fe8] libusb: warning [libusb_exit] device 1.9 still referenced
[ 0.105787] [00005fe8] libusb: warning [libusb_exit] device 1.2 still referenced
[ 0.105836] [00005fe8] libusb: warning [libusb_exit] device 1.4 still referenced
[ 0.105886] [00005fe8] libusb: warning [libusb_exit] device 1.3 still referenced
[ 0.105956] [00005fe8] libusb: warning [libusb_exit] device 1.27 still referenced
[ 0.106005] [00005fe8] libusb: warning [libusb_exit] device 1.5 still referenced
[ 0.106055] [00005fe8] libusb: warning [libusb_exit] device 2.0 still referenced
[ 0.106104] [00005fe8] libusb: warning [libusb_exit] device 1.0 still referenced
```

mcuee, 2021-09-01:
The following is okay.
```
#include <usb.h>
#include <stdio.h>

int main( void )
{
    struct usb_bus *busses;
    struct usb_bus *bus;
    usb_set_debug(255);
    usb_init();
    usb_find_busses();
    //usb_find_devices();

    return 0;
}

```

Debug log:
```
(py39venv) C:\work\libusb\tests> .\libusb_compat_leak3.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000333] [00002b38] libusb: debug [libusb_init] libusb v1.0.24.11650
[ 0.000512] [00002b38] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000094 events 0
[ 0.000674] [00002b38] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000781] [00002b38] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000098 events 0
[ 0.000914] [00002b38] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.001056] [00002b38] libusb: debug [htab_create] using 1021 entries hash table
[ 0.006228] [00002b38] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.012230] [00002b38] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.019718] [00002b38] libusb: debug [windows_init] UsbDk backend is available
[ 0.020164] [00002b38] libusb: debug [libusb_get_device_list]
[ 0.021410] [00002f04] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.041463] [00002b38] libusb: debug [winusb_get_device_list] allocating new device for session [93]
[ 0.041738] [00002b38] libusb: debug [winusb_get_device_list] allocating new device for session [94]
[ 0.042176] [00002b38] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.042412] [00002b38] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.042572] [00002b38] libusb: debug [winusb_get_device_list] allocating new device for session [10]
[ 0.042966] [00002b38] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.043159] [00002b38] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.043318] [00002b38] libusb: debug [winusb_get_device_list] allocating new device for session [1E]
[ 0.043925] [00002b38] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.044123] [00002b38] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.044279] [00002b38] libusb: debug [winusb_get_device_list] allocating new device for session [71]
[ 0.044654] [00002b38] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.044864] [00002b38] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.045024] [00002b38] libusb: debug [winusb_get_device_list] allocating new device for session [72]
[ 0.045392] [00002b38] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.045591] [00002b38] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.045737] [00002b38] libusb: debug [winusb_get_device_list] allocating new device for session [73]
[ 0.046106] [00002b38] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.046310] [00002b38] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.046472] [00002b38] libusb: debug [winusb_get_device_list] allocating new device for session [74]
[ 0.046860] [00002b38] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A36D&SUBSYS_091A1028&REV_10\3&11583659&0&A0' bus number 1
[ 0.047065] [00002b38] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15DB&SUBSYS_091A1028&REV_02\71D9F51854B3020000' bus number 2
[ 0.048739] [00002b38] libusb: debug [winusb_get_device_list] extra GUID: {865C4DDE-10B6-6373-FAEE-1506B321A619}
[ 0.048886] [00002b38] libusb: debug [winusb_get_device_list] found existing device for session [71]
[ 0.049173] [00002b38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.049343] [00002b38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.049511] [00002b38] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 3): 'USB\VID_046D&PID_C534\5&E9F3E45&0&3'
[ 0.049796] [00002b38] libusb: debug [winusb_get_device_list] found existing device for session [10]
[ 0.050052] [00002b38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.050210] [00002b38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.050339] [00002b38] libusb: debug [init_device] (bus: 1, addr: 5, depth: 1, port: 14): 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
[ 0.050823] [00002b38] libusb: debug [winusb_get_device_list] found existing device for session [74]
[ 0.051050] [00002b38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.051169] [00002b38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 288 bytes)
[ 0.051281] [00002b38] libusb: debug [init_device] (bus: 1, addr: 9, depth: 1, port: 2): 'USB\VID_047F&PID_C056\D1CEC32927974D5F9BD6B2AEBF2EA8E3'
[... 36 lines cut ...]
[ 0.057966] [00002b38] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\CONVERTEDDEVICE&COL02\5&379854AA&0&0001' (non USB HID, newly connected, etc.) - ignoring
[ 0.059624] [00002b38] libusb: debug [winusb_get_device_list] setting composite interface for [74]:
[ 0.059711] [00002b38] libusb: debug [set_composite_interface] interface[3] already set - ignoring HID collection: HID\VID_047F&PID_C056&MI_03&COL02\F&39E6F119&0&0001
[... cut ...]
[ 0.063494] [00002b38] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.063587] [00002b38] libusb: debug [libusb_unref_device] destroy device 1.27
[ 0.063677] [00002b38] libusb: debug [libusb_unref_device] destroy device 1.4
[ 0.063765] [00002b38] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.064287] [00002b38] libusb: debug [libusb_exit]
[ 0.064396] [00002b38] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000098
[ 0.064491] [00002b38] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000094
```

mcuee, 2021-09-01:
For the release version of libusb-1.0.21, 1.0.22, 1.0.23 and 1.0.24, the warning message is like the following.
```
[ 0.108524] [00000ebc] libusb: debug [libusb_exit]
[ 0.108613] [00000ebc] libusb: debug [libusb_exit] destroying default context
[ 0.108682] [00000ebc] libusb: warning [libusb_exit] some libusb_devices were leaked
```

mcuee, 2021-09-08:
Looks like there is a bug in libusb Windows backend.
https://github.com/vpelletier/python-libusb1/issues/77

mcuee, 2022-04-01:
This is still there. Probably this is a bug of libusb-compat.

Here is the debug log under my Mac Mini M1 running macOS 12.3.

```

mcuee@mcuees-Mac-mini libusb-compat-0.1 % ./examples/testlibusb
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000022] [0000affd] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.000038] [0000affd] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000252] [0000affd] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x141a4a7bd4
[ 0.000287] [0000affd] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x141a4a7bd4
[ 0.001765] [0000affd] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.001771] [0000affd] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.001773] [0000affd] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.001775] [0000affd] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0xff
[ 0.001777] [0000affd] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0xff
[ 0.001778] [0000affd] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0xff
[ 0.001779] [0000affd] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.001781] [0000affd] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x04b4
[ 0.001782] [0000affd] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8613
[ 0.001784] [0000affd] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0xa001
[ 0.001786] [0000affd] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.001787] [0000affd] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.001789] [0000affd] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.001790] [0000affd] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.001806] [0000affd] libusb: debug [darwin_check_configuration] active config: 0, first config: 1
[ 0.001809] [0000affd] libusb: debug [process_new_device] allocating new device in context 0x12ce04ad0 for with session 0x141a4a7bd4
[ 0.001813] [0000affd] libusb: debug [process_new_device] found device with address 12 port = 1 parent = 0x0 at 0x6000013a4dba
[ 0.001857] [0000affe] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.002009] [0000affe] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.002026] [0000affd] libusb: warning [libusb_init] installing new context as implicit default
[ 0.002032] [0000affd] libusb: debug [libusb_get_device_list]  
[ 0.002059] [0000affd] libusb: debug [libusb_get_device_list]  
[ 0.002070] [0000affd] libusb: debug [libusb_get_device_descriptor]  
[ 0.002073] [0000affd] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.002088] [0000affd] libusb: debug [libusb_open] open 2.12
[ 0.002140] [0000affd] libusb: debug [darwin_open] device open for access
Dev #12: 04B4 - 8613
[ 0.002151] [0000affd] libusb: debug [libusb_close]  
[ 0.002169] [0000affd] libusb: debug [libusb_exit]  
[ 0.002192] [0000affe] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 0.002304] [0000affd] libusb: debug [usbi_remove_event_source] remove fd 3
[ 0.002312] [0000affd] libusb: warning [libusb_exit] device 2.12 still referenced

```

mcuee, 2022-04-01:
Same under Linux (Ubuntu Linux 20.04 x64 version).

```
mcuee@UbuntuSwift3:~/build/libusb-compat-0.1$ ./examples/testlibusb 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000102] [00001212] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.000137] [00001212] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000146] [00001212] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000150] [00001212] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000167] [00001212] libusb: debug [get_kernel_version] reported kernel version is 5.15.10-051510-generic
[ 0.000190] [00001212] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000200] [00001212] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000209] [00001212] libusb: debug [op_init] sysfs is available
[ 0.001337] [00001220] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.003910] [00001212] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.003914] [00001212] libusb: debug [linux_get_device_address] scan usb1
[ 0.003927] [00001212] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.003928] [00001212] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.003931] [00001212] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.003967] [00001212] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.003969] [00001212] libusb: debug [linux_get_device_address] scan usb2
[ 0.003979] [00001212] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.003980] [00001212] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.003983] [00001212] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.004030] [00001212] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.004032] [00001212] libusb: debug [linux_get_device_address] scan usb3
[ 0.004042] [00001212] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.004044] [00001212] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.004047] [00001212] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.004081] [00001212] libusb: debug [linux_get_device_address] getting address for device: 3-10 detached: 0
[ 0.004083] [00001212] libusb: debug [linux_get_device_address] scan 3-10
[ 0.004093] [00001212] libusb: debug [linux_get_device_address] bus=3 dev=5
[ 0.004095] [00001212] libusb: debug [linux_enumerate_device] busnum 3 devaddr 5 session_id 773
[ 0.004097] [00001212] libusb: debug [linux_enumerate_device] allocating new device for 3/5 (session 773)
[ 0.004111] [00001212] libusb: debug [linux_get_parent_info] dev 0x55f913ed7a20 (3-10) has parent 0x55f913ef5d50 (usb3) port 10
[ 0.004138] [00001212] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.004140] [00001212] libusb: debug [linux_get_device_address] scan 3-3
[ 0.004150] [00001212] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.004152] [00001212] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.004155] [00001212] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.004166] [00001212] libusb: debug [linux_get_parent_info] dev 0x55f913ef7440 (3-3) has parent 0x55f913ef5d50 (usb3) port 3
[ 0.004192] [00001212] libusb: debug [linux_get_device_address] getting address for device: 3-5 detached: 0
[ 0.004194] [00001212] libusb: debug [linux_get_device_address] scan 3-5
[ 0.004205] [00001212] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.004207] [00001212] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.004209] [00001212] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.004221] [00001212] libusb: debug [linux_get_parent_info] dev 0x55f913ef7360 (3-5) has parent 0x55f913ef5d50 (usb3) port 5
[ 0.004248] [00001212] libusb: debug [linux_get_device_address] getting address for device: 3-7 detached: 0
[ 0.004250] [00001212] libusb: debug [linux_get_device_address] scan 3-7
[ 0.004261] [00001212] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.004262] [00001212] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.004265] [00001212] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.004276] [00001212] libusb: debug [linux_get_parent_info] dev 0x55f913ed8c30 (3-7) has parent 0x55f913ef5d50 (usb3) port 7
[ 0.004301] [00001212] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.004303] [00001212] libusb: debug [linux_get_device_address] scan usb4
[ 0.004313] [00001212] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.004316] [00001212] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[ 0.004322] [00001212] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
[ 0.004358] [00001212] libusb: warning [libusb_init] installing new context as implicit default
[ 0.004362] [00001212] libusb: debug [libusb_get_device_list]  
[ 0.004383] [00001212] libusb: debug [libusb_get_device_list]  
[... cut ...]
[ 0.128144] [00001212] libusb: warning [libusb_exit] device 3.4 still referenced
[ 0.128147] [00001212] libusb: warning [libusb_exit] device 3.3 still referenced
[ 0.128153] [00001212] libusb: warning [libusb_exit] device 3.2 still referenced
[ 0.128160] [00001212] libusb: warning [libusb_exit] device 3.5 still referenced
[ 0.128166] [00001212] libusb: warning [libusb_exit] device 3.1 still referenced
[ 0.128173] [00001212] libusb: warning [libusb_exit] device 2.1 still referenced
[ 0.128180] [00001212] libusb: warning [libusb_exit] device 1.1 still referenced

```

mcuee, 2022-04-01:
I have closed #974 as it seems to be okay. So I tend to think there is an issue with libusb-compat-0.1 or at least its usage of libusb-1.0 APIs.

mcuee, 2022-04-01:
Same under my WIndows 11 machine.
```
$ ./examples/lsusb.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000957] [0000327c] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.001257] [0000327c] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000a0 events 0
[ 0.001555] [0000327c] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001852] [0000327c] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000a4 events 0
[ 0.002151] [0000327c] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.002294] [0000327c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003945] [0000327c] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.005638] [0000327c] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.007307] [0000327c] libusb: info [windows_init] UsbDk backend is not available
[ 0.007412] [0000327c] libusb: warning [libusb_init] installing new context as implicit default
[ 0.007477] [0000327c] libusb: debug [libusb_get_device_list]
[ 0.007505] [00001c14] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.013539] [0000327c] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.013782] [0000327c] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.014267] [0000327c] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.014462] [0000327c] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.014582] [0000327c] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.014676] [0000327c] libusb: debug [winusb_get_device_list] allocating new device for session [1F]
[ 0.014960] [0000327c] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.015055] [0000327c] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.015183] [0000327c] libusb: debug [winusb_get_device_list] allocating new device for session [34]
[ 0.015484] [0000327c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.015557] [0000327c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.015601] [0000327c] libusb: debug [winusb_get_device_list] allocating new device for session [62]
[ 0.015717] [0000327c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.015775] [0000327c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.015817] [0000327c] libusb: debug [winusb_get_device_list] allocating new device for session [63]
[ 0.016006] [0000327c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.016081] [0000327c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.016761] [0000327c] libusb: debug [winusb_get_device_list] found existing device for session [34]
[ 0.016895] [0000327c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.016953] [0000327c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.017028] [0000327c] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&586B51A&0&10'
[ 0.017133] [0000327c] libusb: debug [winusb_get_device_list] extra GUID: {16126A05-3179-4589-9DB8-952AFFD297D5}
[ 0.017195] [0000327c] libusb: debug [winusb_get_device_list] found existing device for session [62]
[ 0.017284] [0000327c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.017347] [0000327c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 512 bytes)
[ 0.017409] [0000327c] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 5): 'USB\VID_04F2&PID_B6DD\0001'
[ 0.017509] [0000327c] libusb: debug [winusb_get_device_list] extra GUID: {20348E47-A897-468E-9D0E-56307802FC2A}
[ 0.017557] [0000327c] libusb: debug [winusb_get_device_list] found existing device for session [1F]
[ 0.017633] [0000327c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.017680] [0000327c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 46 bytes)
[ 0.017729] [0000327c] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 7): 'USB\VID_1C7A&PID_0575\077E2F9A'
[ 0.017956] [0000327c] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&12C539F&0&0' reports 16 ports
[ 0.018036] [0000327c] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&12C539F&0&0'
[ 0.018158] [0000327c] libusb: debug [winusb_get_device_list] found existing device for session [63]
[ 0.018227] [0000327c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.018274] [0000327c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 84 bytes)
[ 0.018322] [0000327c] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 3): 'USB\VID_046D&PID_C52B\5&586B51A&0&3'
[ 0.018503] [0000327c] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&1A63DC7C&0&0' reports 5 ports
[ 0.018572] [0000327c] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&1A63DC7C&0&0'
[... cut ...]
[ 0.037722] [0000327c] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000a0
[ 0.037783] [0000327c] libusb: warning [libusb_exit] device 1.1 still referenced
[ 0.037836] [0000327c] libusb: warning [libusb_exit] device 1.2 still referenced
[ 0.037915] [0000327c] libusb: warning [libusb_exit] device 1.4 still referenced
[ 0.037967] [0000327c] libusb: warning [libusb_exit] device 1.3 still referenced
[ 0.038030] [0000327c] libusb: warning [libusb_exit] device 2.0 still referenced
[ 0.038081] [0000327c] libusb: warning [libusb_exit] device 1.0 still referenced
```

mcuee, 2023-01-24:
Just want to mention that the issue is still present with latest version of libusb git HEAD and libusb-compat-0.1.

tormodvolden, 2023-03-31:
Does libusb-compat-0.1 use libusb_free_device_list() with unref_devices = true?

I guess it references all devices by running libusb_get_device_list() and then it must unref them all before calling libusb_exit().

mcuee, 2023-03-31:
The following simple addition to libusb debug log seems to help a bit.

```
diff --git a/libusb/core.c b/libusb/core.c
index 17e61172..5a7680e8 100644
--- a/libusb/core.c
+++ b/libusb/core.c
@@ -1272,6 +1272,7 @@ libusb_device * LIBUSB_CALL libusb_ref_device(libusb_device *dev)
        long refcnt;

        refcnt = usbi_atomic_inc(&dev->refcnt);
+       usbi_dbg(DEVICE_CTX(dev), "Device %d.%d reference count increased to %ld", dev->bus_number, dev->device_address, refcnt);
        assert(refcnt >= 2);

        return dev;
@@ -1290,6 +1291,7 @@ void API_EXPORTED libusb_unref_device(libusb_device *dev)
                return;

        refcnt = usbi_atomic_dec(&dev->refcnt);
+       usbi_dbg(DEVICE_CTX(dev), "Device %d.%d reference count decreased to %ld", dev->bus_number, dev->device_address, refcnt);
        assert(refcnt >= 0);

        if (refcnt == 0) {
diff --git a/libusb/version_nano.h b/libusb/version_nano.h
index 34620df0..08af623b 100644
--- a/libusb/version_nano.h
+++ b/libusb/version_nano.h
@@ -1 +1 @@
-#define LIBUSB_NANO 11791
+#define LIBUSB_NANO 22791

MINGW64 /c/work/libusb/libusb_test/libusb_issue1165
$ ./examples/listdevs.exe
8087:0026 (bus 1, device 2) path: 10
04f2:b6dd (bus 1, device 5) path: 5
1c7a:0575 (bus 1, device 3) path: 7
8086:a0ed (bus 1, device 0)
046d:c52b (bus 1, device 1) path: 3
8086:9a13 (bus 2, device 0)

MINGW64 /c/work/libusb/libusb-compat
$ export LIBUSB_DEBUG=4
MINGW64 /c/work/libusb/libusb-compat
$ ./examples/lsusb.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001253] [000042c4] libusb: debug [libusb_init_context] libusb v1.0.26.22791
[ 0.001575] [000042c4] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000058 events 0
[ 0.001786] [000042c4] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001979] [000042c4] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000b4 events 0
[ 0.002320] [000042c4] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.002544] [000042c4] libusb: debug [htab_create] using 1021 entries hash table
[ 0.004222] [000042c4] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.006071] [000042c4] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.008052] [000042c4] libusb: info [windows_init] UsbDk backend is not available
[ 0.008179] [000042c4] libusb: debug [libusb_init_context] installing new context as implicit default
[ 0.008248] [000042c4] libusb: debug [libusb_get_device_list]
[ 0.008277] [00001d70] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.016356] [000042c4] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.016500] [000042c4] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.016960] [000042c4] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.017055] [000042c4] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.017122] [000042c4] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.017178] [000042c4] libusb: debug [winusb_get_device_list] allocating new device for session [1C]
[ 0.017368] [000042c4] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.017445] [000042c4] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.017497] [000042c4] libusb: debug [winusb_get_device_list] allocating new device for session [3C]
[ 0.017711] [000042c4] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.017778] [000042c4] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.017827] [000042c4] libusb: debug [winusb_get_device_list] allocating new device for session [6C]
[... 312 lines cut ...]
[ 0.048891] [000042c4] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.048939] [000042c4] libusb: debug [parse_endpoint] skipping descriptor 0x30
[ 0.048988] [000042c4] libusb: debug [libusb_ref_device] Device 2.0 reference count increased to 2
[ 0.049036] [000042c4] libusb: debug [libusb_unref_device] Device 1.2 reference count decreased to 1
[ 0.049083] [000042c4] libusb: debug [libusb_unref_device] Device 1.5 reference count decreased to 1
[ 0.049130] [000042c4] libusb: debug [libusb_unref_device] Device 1.3 reference count decreased to 1
[ 0.049177] [000042c4] libusb: debug [libusb_unref_device] Device 1.0 reference count decreased to 5
[ 0.049225] [000042c4] libusb: debug [libusb_unref_device] Device 1.1 reference count decreased to 1
[ 0.049272] [000042c4] libusb: debug [libusb_unref_device] Device 2.0 reference count decreased to 1
8087:0026
04f2:b6dd
1c7a:0575
8086:a0ed
046d:c52b
8086:9a13
[ 0.051910] [000042c4] libusb: debug [libusb_exit]
[ 0.052005] [000042c4] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000b4
[ 0.052069] [000042c4] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000058
[ 0.052123] [000042c4] libusb: warning [libusb_exit] device 1.1 still referenced
[ 0.052174] [000042c4] libusb: warning [libusb_exit] device 1.5 still referenced
[ 0.052223] [000042c4] libusb: warning [libusb_exit] device 1.2 still referenced
[ 0.052277] [000042c4] libusb: warning [libusb_exit] device 1.3 still referenced
[ 0.052339] [000042c4] libusb: warning [libusb_exit] device 2.0 still referenced
[ 0.052398] [000042c4] libusb: warning [libusb_exit] device 1.0 still referenced
```

mcuee, 2023-03-31:
As a comparison, the debug log for libusb's listdevs exmple is here.
```
MINGW64 /c/work/libusb/libusb_test/libusb_issue1165
$ ./examples/listdevs.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001054] [0000284c] libusb: debug [libusb_init_context] created default context
[ 0.001283] [0000284c] libusb: debug [libusb_init_context] libusb v1.0.26.22791
[ 0.001526] [0000284c] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000b0 events 0
[ 0.001745] [0000284c] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001881] [0000284c] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000b4 events 0
[ 0.001931] [0000284c] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.001967] [0000284c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003504] [0000284c] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.005170] [0000284c] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.006927] [0000284c] libusb: info [windows_init] UsbDk backend is not available
[ 0.007030] [0000284c] libusb: debug [libusb_get_device_list]
[ 0.007107] [00001df8] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.014435] [0000284c] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.014712] [0000284c] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.014855] [0000284c] libusb: debug [winusb_get_device_list] allocating new device for session [7]
[ 0.014997] [0000284c] libusb: debug [winusb_get_device_list] allocating new device for session [8]
[ 0.015497] [0000284c] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.015762] [0000284c] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.015902] [0000284c] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.016039] [0000284c] libusb: debug [winusb_get_device_list] allocating new device for session [1E]
[ 0.016324] [0000284c] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.016488] [0000284c] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.016628] [0000284c] libusb: debug [winusb_get_device_list] allocating new device for session [3E]
[ 0.016931] [0000284c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.017083] [0000284c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.017202] [0000284c] libusb: debug [winusb_get_device_list] allocating new device for session [70]
[ 0.017428] [0000284c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.017503] [0000284c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.017552] [0000284c] libusb: debug [winusb_get_device_list] allocating new device for session [71]
[ 0.017728] [0000284c] libusb: debug [get_api_type] driver(s): USBSTOR
[ 0.017807] [0000284c] libusb: debug [winusb_get_device_list] allocating new device for session [72]
[ 0.018043] [0000284c] libusb: debug [libusb_ref_device] Device 0.0 reference count increased to 2
[ 0.018088] [0000284c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.018162] [0000284c] libusb: debug [libusb_unref_device] Device 1.0 reference count decreased to 1
[ 0.018225] [0000284c] libusb: debug [libusb_ref_device] Device 0.0 reference count increased to 2
[ 0.018276] [0000284c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.018328] [0000284c] libusb: debug [libusb_unref_device] Device 2.0 reference count decreased to 1
[ 0.018998] [0000284c] libusb: debug [libusb_ref_device] Device 2.0 reference count increased to 2
[ 0.019080] [0000284c] libusb: debug [libusb_ref_device] Device 0.0 reference count increased to 2
[ 0.019127] [0000284c] libusb: debug [winusb_get_device_list] found existing device for session [5]
[ 0.019253] [0000284c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.019309] [0000284c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 31 bytes)
[ 0.019368] [0000284c] libusb: debug [init_device] (bus: 2, addr: 2, depth: 1, port: 2): 'USB\VID_05E3&PID_0612\5&1A00B1C&0&2'
[ 0.019418] [0000284c] libusb: debug [libusb_ref_device] Device 2.2 reference count increased to 3
[ 0.019522] [0000284c] libusb: debug [libusb_ref_device] Device 2.2 reference count increased to 4
[ 0.019572] [0000284c] libusb: debug [libusb_ref_device] Device 0.0 reference count increased to 2
[ 0.019620] [0000284c] libusb: debug [winusb_get_device_list] found existing device for session [72]
[ 0.019709] [0000284c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.019765] [0000284c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 44 bytes)
[ 0.019820] [0000284c] libusb: debug [init_device] (bus: 2, addr: 3, depth: 2, port: 2): 'USB\VID_0781&PID_5591\0401D5032AFA44996ADA6E7FB56B51153C7D26AC8C49311D73C10CF96B01E67412E700000000000000000000DE646461000810189155810714AA1A5B'
[ 0.019883] [0000284c] libusb: debug [libusb_ref_device] Device 2.3 reference count increased to 3
[ 0.019974] [0000284c] libusb: debug [libusb_ref_device] Device 1.0 reference count increased to 2
[... cut ...]
[ 0.043346] [0000284c] libusb: debug [libusb_unref_device] Device 2.0 reference count decreased to 0
[ 0.043391] [0000284c] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.043437] [0000284c] libusb: debug [libusb_exit] destroying default context
[ 0.043489] [00001df8] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.043876] [0000284c] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000b4
[ 0.043949] [0000284c] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000b0
```

mcuee, 2023-04-01:
@tormodvolden and @hjelmn 

Is this because libusb-compat-0.1 only does the clean up when libusb-1.0 dynamic library is unloaded as there is no `usb_exit` in libusb-0.1 API?

mcuee, 2023-04-01:
Debug log for libusb-compat-0.1 `lsusb` example.

<details>
<summary> click for debug log </summary>

```
mcuee@mcuees-Mac-mini test_issue1165 % ./lsusb
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000029] [000131cd] libusb: debug [libusb_init_context] libusb v1.0.26.11791
[ 0.000052] [000131cd] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000112] [000131d0] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000381] [000131d0] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.000535] [000131cd] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xf326c5572
[ 0.000582] [000131cd] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xf326c5572
[ 0.003010] [000131cd] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003019] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003022] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.003025] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003028] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003030] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.003033] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.003035] [000131cd] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.003038] [000131cd] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.003040] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.003043] [000131cd] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003045] [000131cd] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003048] [000131cd] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003050] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003076] [000131cd] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003080] [000131cd] libusb: debug [process_new_device] allocating new device in context 0x14ef04420 for with session 0xf326c5572
[ 0.003085] [000131cd] libusb: debug [process_new_device] found device with address 11 port = 6 parent = 0x0 at 0x6000020c48ba
[ 0.003168] [000131cd] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xf326d622f
[ 0.003201] [000131cd] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xf326d622f/0x2200000 against cached device with sessionID/locationID 0xf326c5572/0x2600000
[ 0.003206] [000131cd] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xf326d622f
[ 0.003457] [000131cd] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003463] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003466] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.003469] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003471] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003473] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.003476] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.003478] [000131cd] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.003481] [000131cd] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.003484] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.003486] [000131cd] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003489] [000131cd] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003491] [000131cd] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003494] [000131cd] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003510] [000131cd] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003514] [000131cd] libusb: debug [process_new_device] allocating new device in context 0x14ef04420 for with session 0xf326d622f
[ 0.003517] [000131cd] libusb: debug [process_new_device] found device with address 10 port = 2 parent = 0x0 at 0x6000020c4aba
[ 0.003595] [000131cd] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xf328a0add
[ 0.003620] [000131cd] libusb: debug [darwin_get_cached_device] parent sessionID: 0xf326c5572
[ 0.003624] [000131cd] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xf328a0add/0x2610000 against cached device with sessionID/locationID 0xf326d622f/0x2200000
[ 0.003627] [000131cd] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xf328a0add/0x2610000 against cached device with sessionID/locationID 0xf326c5572/0x2600000
[ 0.003630] [000131cd] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xf328a0add
[... cut ...]
[ 0.005088] [00013f29] libusb: debug [libusb_unref_device] Device 2.10 reference count decreased to 0
[ 0.005091] [00013f29] libusb: debug [libusb_unref_device] destroy device 2.10
[ 0.005093] [00013f29] libusb: debug [libusb_unref_device] Device 2.11 reference count decreased to 0
[ 0.005096] [00013f29] libusb: debug [libusb_unref_device] destroy device 2.11
[ 0.005117] [00013f2b] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 0.005849] [00013f29] libusb: debug [usbi_remove_event_source] remove fd 3
```

</details>

mcuee, 2023-04-01:
The following patch to libusb-compat-0.1 seems to fix the issue. My thought is that libusb will handle the device ref and unref so there should not be handling required by libusb-compat-0.1. But I am not so sure if this is the right fix. 

I have only tested under Linux as of now.

```
mcuee@UbuntuSwift3:~/build/libusb-compat-0.1$ git diff
diff --git a/libusb/core.c b/libusb/core.c
index 3bc9c25..09068f3 100644
--- a/libusb/core.c
+++ b/libusb/core.c
@@ -584,14 +584,12 @@ static int initialize_device(struct usb_device *dev)
        dev->num_children = 0;
        dev->children = NULL;
 
-       libusb_ref_device(newlib_dev);
        return 0;
 }
 
 static void free_device(struct usb_device *dev)
 {
        clear_device(dev);
-       libusb_unref_device(dev->dev);
        free(dev);
 }
```

Debug log.

<details>
<summary> click for the debug log </summary>

```
mcuee@UbuntuSwift3:~/build/libusb-compat-0.1$ export LIBUSB_DEBUG=4
mcuee@UbuntuSwift3:~/build/libusb-compat-0.1$ ./examples/lsusb 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000028] [0000820a] libusb: debug [libusb_init_context] libusb v1.0.26.11791
[ 0.000056] [0000820a] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000064] [0000820a] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000068] [0000820a] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000090] [0000820a] libusb: debug [get_kernel_version] reported kernel version is 5.15.10-051510-generic
[ 0.000123] [0000820a] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000129] [0000820a] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000138] [0000820a] libusb: debug [op_init] sysfs is available
[ 0.000585] [00008218] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.004205] [0000820a] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004209] [0000820a] libusb: debug [linux_get_device_address] scan usb1
[ 0.004222] [0000820a] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.004224] [0000820a] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.004226] [0000820a] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.004290] [0000820a] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.004292] [0000820a] libusb: debug [linux_get_device_address] scan usb2
[ 0.004300] [0000820a] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.004301] [0000820a] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.004303] [0000820a] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.004343] [0000820a] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.004345] [0000820a] libusb: debug [linux_get_device_address] scan 2-1
[ 0.004354] [0000820a] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.004356] [0000820a] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.004357] [0000820a] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.004367] [0000820a] libusb: debug [libusb_ref_device] Device 2.1 reference count increased to 2
[ 0.004368] [0000820a] libusb: debug [linux_get_parent_info] dev 0x561311e23ab0 (2-1) has parent 0x561311e235a0 (usb2) port 1
[ 0.004408] [0000820a] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.004409] [0000820a] libusb: debug [linux_get_device_address] scan usb3
[ 0.004432] [0000820a] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.004433] [0000820a] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.004434] [0000820a] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.004475] [0000820a] libusb: debug [linux_get_device_address] getting address for device: 3-10 detached: 0
[ 0.004478] [0000820a] libusb: debug [linux_get_device_address] scan 3-10
[ 0.004488] [0000820a] libusb: debug [linux_get_device_address] bus=3 dev=6
[... 163 lines cut ...]
[ 0.005152] [0000820a] libusb: debug [libusb_unref_device] Device 3.1 reference count decreased to 4
[ 0.005154] [0000820a] libusb: debug [libusb_unref_device] Device 3.3 reference count decreased to 0
[ 0.005156] [0000820a] libusb: debug [libusb_unref_device] destroy device 3.3
[ 0.005158] [0000820a] libusb: debug [libusb_unref_device] Device 3.1 reference count decreased to 3
[ 0.005160] [0000820a] libusb: debug [libusb_unref_device] Device 3.2 reference count decreased to 0
[ 0.005162] [0000820a] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.005164] [0000820a] libusb: debug [libusb_unref_device] Device 3.1 reference count decreased to 2
[ 0.005166] [0000820a] libusb: debug [libusb_unref_device] Device 3.6 reference count decreased to 0
[ 0.005168] [0000820a] libusb: debug [libusb_unref_device] destroy device 3.6
[ 0.005170] [0000820a] libusb: debug [libusb_unref_device] Device 3.1 reference count decreased to 1
[ 0.005173] [0000820a] libusb: debug [libusb_unref_device] Device 3.1 reference count decreased to 0
[ 0.005175] [0000820a] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.005177] [0000820a] libusb: debug [libusb_unref_device] Device 2.2 reference count decreased to 0
[... cut ...]
[ 0.005185] [0000820a] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.005187] [0000820a] libusb: debug [libusb_unref_device] Device 1.1 reference count decreased to 0
[ 0.005189] [0000820a] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.005196] [00008218] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 0.005228] [0000820a] libusb: debug [usbi_remove_event_source] remove fd 4
[ 0.005232] [0000820a] libusb: debug [usbi_remove_event_source] remove fd 3
```

</details>

mcuee, 2023-04-02:
Same for macOS, once I have the patched version of libusb-compat-0.1, it seems to work as expected. I have also tested libusb-compat-0.1 `testlibusb` example and it is also okay.

Modified version of libusb (just to add the debug print for the reference count) and modified version of libusb-compat-0.1 are installed to `~/bin`. Please also take note that I am using static build for the example `lsusb` to avoid linking to Homebrew installed libusb and libusb-compat-0.1.

<details>
<summary> click for debug log under macOS </summary>

```
mcuee@mcuees-Mac-mini test_issue1165 % gcc -o lsusb lsusb.c ~/bin/lib/libusb.a ~/bin/lib/libusb-1.0.a -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -Wl,-framework,Security

mcuee@mcuees-Mac-mini test_issue1165 % ./lsusb 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000051] [00006efc] libusb: debug [libusb_init_context] libusb v1.0.26.11791
[ 0.000081] [00006efc] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000184] [00006f0a] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000594] [00006f0a] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.000829] [00006efc] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x8c3b602
[ 0.000888] [00006efc] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x8c3b602
[ 0.005893] [00006efc] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.005907] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.005911] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.005914] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.005917] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.005919] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.005921] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.005924] [00006efc] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.005927] [00006efc] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.005930] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.005933] [00006efc] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.005935] [00006efc] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.005938] [00006efc] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.005940] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.005965] [00006efc] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.005969] [00006efc] libusb: debug [process_new_device] allocating new device in context 0x124f043b0 for with session 0x8c3b602
[ 0.005974] [00006efc] libusb: debug [process_new_device] found device with address 1 port = 2 parent = 0x0 at 0x60000225c0ba
[ 0.006054] [00006efc] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x8c93ad8
[ 0.006092] [00006efc] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x8c93ad8/0x2600000 against cached device with sessionID/locationID 0x8c3b602/0x2200000
[ 0.006096] [00006efc] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x8c93ad8
[ 0.006342] [00006efc] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.006350] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.006352] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.006355] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.006358] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.006360] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.006363] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.006365] [00006efc] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.006368] [00006efc] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.006370] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.006373] [00006efc] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.006376] [00006efc] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.006378] [00006efc] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.006381] [00006efc] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.006397] [00006efc] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.006401] [00006efc] libusb: debug [process_new_device] allocating new device in context 0x124f043b0 for with session 0x8c93ad8
[ 0.006405] [00006efc] libusb: debug [process_new_device] found device with address 2 port = 6 parent = 0x0 at 0x60000225c2ba
[ 0.006469] [00006efc] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x8e092a9
[... cut ...]
[ 0.010346] [00006efc] libusb: debug [libusb_unref_device] Device 2.2 reference count decreased to 0
[ 0.010349] [00006efc] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.010352] [00006efc] libusb: debug [libusb_unref_device] Device 2.1 reference count decreased to 0
[ 0.010355] [00006efc] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.010383] [00006f0a] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 0.011480] [00006efc] libusb: debug [usbi_remove_event_source] remove fd 3
```

</details>