libusb/libusb #990: To keep libusb_set_debug() and not deprecate it

mcuee, 2021-09-05:
https://github.com/vpelletier/python-libusb1/issues/75#issuecomment-913113419

Comments from https://github.com/vpelletier/python-libusb1/issues/75 by Tormod Volden.
>The simplest for now and foreseeable future would be if libusb could keep libusb_set_debug() and not deprecate it. Other than that, libusb has only two other options and they are simple flags without option argument:
LIBUSB_OPTION_USE_USBDK and LIBUSB_OPTION_NO_ENUMERATION (previously named LIBUSB_OPTION_WEAK_AUTHORITY). Any new options in the pipeline that I am aware of are also just flags. Then the language bindings for libusb_set_option() wouldn't need to deal with arguments.
>
>It is nice for a library to be future-proof and get everything right from the beginning, and the new libusb_set_option() is enormously scalable and elegant with variadic macros, however it is wonderful over-engineering for a problem that libusb doesn't have, and rather should try to avoid (lots of options with various arguments).

tormodvolden, 2021-09-07:
I do have another suggestion though, make it `libusb_set_option(enum option, int value)`

This will be backwards compatible for LIBUSB_OPTION_DEBUG so no established code will need changes. Flag options can use 1 for on and 0 for off. Only any existing users of LIBUSB_OPTION_USE_USBDK and LIBUSB_OPTION_WEAK_AUTHORITY (LIBUSB_OPTION_NO_ENUMERATION) will need updating, but they are experimental and not in wide use anyway.

tormodvolden, 2021-09-22:
The pull request Better Android Support #874 suggests a couple of new options with JAVA pointers as arguments. That is very platform-specific though, not sure if libusb_set_option() is the best solution there.

xloem, 2021-11-02:
Regarding 874, if intptr_t is used the pointers could be passed as ints.   It would be kind if implementors of such changes could also contribute a change to the PR.

I would personally propose an interface that was fully backward compatible to existing released interfaces.

commit fcf0c710 (2023-03-12): Do not deprecate libusb_set_debug and libusb_init

These functions are used in a lot of existing downstream code
and the deprecation makes -Werror builds fail.

It seems reasonable to keep these functions supported at least until a
major API overhaul.

Closes #990
Closes #1236
closed 2023-03-12 by tormodvolden