libusb/libusb #993: Android: libusb: error [submit_iso_transfer] submiturb failed, errno=12

zhouconglong, 2021-09-09:
Hi, I use this he libuvc in android, when I start_streaming 
retrun:
libusb: error [submit_iso_transfer] submiturb failed, errno=12

mcuee, 2021-09-09:
Again you need to provide more details. What is the Android system? What is the uvc devices? What is the libusb version used? Do you see issues under normal Linux host (not Android)?

mcuee, 2021-09-09:
https://github.com/libusb/libusb/wiki/Android
Currently there are multiple issues of libusb under Android. But your issue may not may not be related. So you need to make sure your codes work under normal Linux first before trying Android.

closed 2021-09-12 by zhouconglong

mcuee, 2021-09-12:
@zhouconglong It will be good to tell us whether you have sorted out the issue or not. If you have sorted out the issue, it will be great that you can share what you have done to sort out the issue.