libusb/libusb #994: 1.0.24 - windows_exit hangs sometimes when exiting on with WinUSB

cnlohr, 2021-09-22:
Frequently, when setting up several pipes, opening and closing a device the following lines of code hang; even if I unplug the device, `WaitForSingleObject` never completes.

```c
	if (WaitForSingleObject(priv->completion_port_thread, INFINITE) == WAIT_FAILED)
		usbi_err(ctx, "failed to wait for I/O completion port thread: %s", windows_error_str(0));
```

I've even verified that that the completion_port_thread completes, and it does properly.  So I don't know why the `WaitForSingleObjectCall()` hangs.  Has anyone else ever seen anything like this?

I can step through in the debugger and see the thread returning.

mcuee, 2021-09-23:
Have you tested latest git to see if it performs the same as 1.0.24 release?

BTW, I am not so sure about this has anything to do with - #974.

mcuee, 2021-09-23:
@cnlohr BTW, would you help to post your test code and debug log (by setting LIBUS_DEBUG=4)? Thanks. That will help me to recreate the issue. It will also help others to identify the issue and come out with the fixes or workarounds.

cnlohr, 2021-10-14:
This is probably easier to read since it doesn't double-up the log messages.

Additionally, looking through the log, I don't see the error, but I am definitely getting a message that the transfer timed out on my side.

[libusblog_after_failthrough2.txt](https://github.com/libusb/libusb/files/7349797/libusblog_after_failthrough2.txt)

mcuee, 2021-10-19:
> If I extend the timeout to be indefinite, then I am unable to repro the problem. It seems the issue revolves around timing out and/or canceling transfers.

Can you post the debug in this case?

mcuee, 2022-06-18:
@cnlohr Please help to test libusb-1.0.26 to see if the issues have been resolved. Thanks.

cnlohr, 2022-11-09:
I was not able to reproduce reliably the issue on 1.0.24, but I am not seeing it on 1.0.26.  It appears this may have fixed it.

tormodvolden, 2023-01-22:
Is it still hanging on WaitForSingleObject( <thread>, INFINITE ) ?

cnlohr, 2023-01-27:
I don't know for sure, I didn't take those notes.  We will be migrating away from this system before public release, so it is unlikely I will spend more time trying to chase this down.  If I do, for whatever reason, I will post back here.  But we still do have an issue where `windows_exit` freezes.

mcuee, 2023-01-31:
@cnlohr 

Thanks for the updates. Just wondering if you see any similarity of your issue with the following reported issue.
* https://github.com/libusb/libusb/issues/966

cnlohr, 2023-02-02:
No similarity.  

At this point it probably makes sense to close this issue.  If we end up not moving on or I get some time I will test.

closed 2023-02-02 by cnlohr